mcguirepr89/BirdNET-Pi #238: Birdnet-Pi accessible as hostname.local but blank screen if I enter the IP adress.

mistaoh, 2022-05-03:
Hi and thanks for a wonderful product. I really enjoy going through what it has heard during the day.

I installed it successfully on a rpi3b+ (I think) and it works like a charm if I enter `hostname.local`, or `birdpi.local` in my case, in firefox. But if I try to connect by using only the IP address, `192.168.2.224`, on my internal network I only get a blank screen in my web browser and nothing more happens. I'm not sure why this happens and It could be some issue on my side but I never noticed it on any other similar device I use.

The reason I'm unable to use the mDNS name is because I connect through VPN wich is set up to use a specific DNS.

nilspupils, 2022-05-03:
Hi @mistaoh and welcome to BridNET-Pi! 
Could you please provide the output of "ifconfig" in the terminal for now? I am sure the others are going to chime in soon!

bdoner, 2022-05-03:
I think it eventually comes down to [this line](https://github.com/mcguirepr89/BirdNET-Pi/blob/58f61b7a34e68b272331b2969bfddbc8759c653f/scripts/install_services.sh#L206=) listening to localhost, hostname.local and the configured birdnet-pi URL. 
I don't know if any kind of wildcards are supported in a Caddyfile, since I don't know Caddy :)

mistaoh, 2022-05-03:
@nilspupils 
Thank you very much.
My ifconfig output: (I replaced mac address and ipv6 address with xx) 
```birdpi:~/BirdNET-Pi$ ifconfig
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether xx  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 10552850  bytes 7874242771 (7.3 GiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 10552850  bytes 7874242771 (7.3 GiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.2.224  netmask 255.255.255.0  broadcast 192.168.2.255
        inet6 xx  prefixlen 64  scopeid 0x20<link>
        ether xx  txqueuelen 1000  (Ethernet)
        RX packets 193914  bytes 26487019 (25.2 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 241804  bytes 345181025 (329.1 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

birdpi:~/BirdNET-Pi$
```

It does also respond to ping `192.168.2.224`

@bdoner 

Just read your reply while writing my answer, tried it and it works now. Huge thanks!

ehpersonal38, 2022-05-03:
This is actually intended behavior. Quoting Patrick for the fix for anyone in the future who sees this issue:

"1. SSH into the BirdNET-Pi
2. Get its IPv4 address ```hostname -I``` or ```ip a```
3. Set your `BIRDNETPI_URL` in `birdnet.conf` to those results, example:
   ```
   BIRDNETPI_URL=http://192.168.1.120
   ```
4. Then run ```sudo update_caddyfile.sh```

Then your installation is accessible at the IP address."

Glad you got it working!

closed 2022-05-03 by ehpersonal38

mistaoh, 2022-05-03:
@ehpersonal38 
Ok, I think it could be something to reference in the installation instructions. I could not find any information about it when I was trying to find what the issue could be.
After adding the preferred URL in the web interface the Caddyfile line was modified to reflect the changes as well. 

Thank you to everyone that chimed in to help!

ehpersonal38, 2022-05-03:
> @ehpersonal38 Ok, I think it could be something to reference in the installation instructions. I could not find any information about it when I was trying to find what the issue could be. After [...]

It's now in the installation wiki! Thanks for making BirdNET-Pi better. Happy birding!!

kinteki, 2022-06-20:
Hi! 
birdnet.conf overwritten to my IP address, but now 403 Forbidden error. What can I do? 
Sinc, Zoltan

bdoner, 2022-06-22:
> 

Hi @kinteki 

What is your currently set URL under the advanced settings for the field " BirdNET-Pi URL:"? That setting will overwrite your `birdnet.conf`.

kinteki, 2022-07-01:
Hi @bdoner 

birdnet.conf : BIRDNETPI_URL=HTTP://192.168.1.157/birdnet

after reboot: 404 :(

mcguirepr89, 2022-07-01:
Hi, @kinteki -- the `BIRDNETPI_URL` cannot contain a path -- you will have to remove the `/birdnet` and then reboot

kinteki, 2022-07-01:
sure not capitalized. when i try just http://192.168.1.157 this is the answer:
[image]