mcguirepr89/BirdNET-Pi #239: File Manager not working after new installation

Svardsten53, 2022-05-03:
Today I decided to do a new and clean installation on my "production" system. Everything seem to work except File Manager. When I try to start it I only get an empty page. In the last weeks I have worked a lot in File Manager and need it now when we have left the desktop version of the OS. I have not done any translations or other modifications yet, so it's just a standard installation.

By the way, http://birdnetpi.local/ is no longer reachable after I made the first settings. I have done the update of Caddyfile in order to have the connection to birdnetpi.com and that works.

mcguirepr89, 2022-05-03:
Hi, @Svardsten53 -- I was not able to replicate the problem on a clean installation just now -- File-Manager is working for me.

There is the `installation-$(date +%F).log` which you're welcome to look through for the error, or post it here for others to aid in troubleshooting. Likely, though, this was a networking issue during installation (if I were to guess). But the logs always tell us what is going on, so please use them to provide more information.

Thanks!

netdisciple, 2022-05-03:
i ran into a similar problem yesterday and decided to rebuild. i will report back about my success but want to contribute to this and look forward to working with you.

Svardsten53, 2022-05-04:
Now I have experimented a lot to find what is causing the error. I have managed to get File Manager to work, but only when I run from birdnetpi.local. Should it not work from the internet as well or have I misunderstood?

I have had some indications that it is due to my internal network, but wait to search further until I have replaced my NAS. The disks are full and the NAS needs to be upgraded to be able to mount larger.

ehpersonal38, 2022-05-04:
> Now I have experimented a lot to find what is causing the error. I have managed to get File Manager to work, but only when I run from birdnetpi.local. Should it not work from the internet as well or [...]

FWIW my file manager does work from the internet. Try accessing the direct link for file manager, http://{birdnetdomain}/scripts/filemanager/filemanager.php?p=

Svardsten53, 2022-05-05:
@ehpersonal38: Thank you! The extension to the URL worked fine and I'm able to use File Manager also from the Internet. I also tried to change the error reporting to true, but can't see any messages or are they hidden somewhere in the Linux file system?  ;-)

ehpersonal38, 2022-05-05:
That's good to hear! You're welcome! 

>can't see any messages or are they hidden somewhere in the Linux file system

The errors would've shown directly on the page, and the fact that they didn't indicates to me that this is an iFrame issue. But seeing as you've got it working now, unless you want to debug further then I will close this issue soon.

closed 2022-05-05 by Svardsten53