mcguirepr89/BirdNET-Pi #241: empty / corrupted files still causing analysis to break after midnight

kaeltis, 2022-05-05:
According to #214, #218 and #223 the bug that causes the analysis to stop working over midnight should've been fixed, however I still experienced the same problem running fc6cf7bd84975d35502f70ece65e134a458a7a53 (fresh install from yesterday).

The logs complained about a file:
```May 05 00:00:39 birdnetpi spectrogram.sh[234868]: sox FAIL formats: can't open input file `/home/pi/BirdSongs/May-2022/04-Wednesday/2022-05-04-birdnet-23:59:53.wav': No such file or director```

However the actual 0 byte file was a completely different one:
```
pi@birdnetpi:~/BirdSongs/May-2022/04-Wednesday$ ls -al
total 8
drwxr-xr-x 2 pi pi 4096 May  5 00:00 .
drwxr-xr-x 5 pi pi 4096 May  5 00:00 ..
-rw-r--r-- 1 pi pi    0 May  4 18:42 2022-05-04-birdnet-18:42:38.wav
```

After deleting this file, the analysis started working again.

Full log:
```
May 04 23:59:00 birdnetpi birdnet_analysis.sh[233385]: RECORDING_LENGTH set to 15
May 04 23:59:07 birdnetpi systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 1647.
May 04 23:59:07 birdnetpi systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
May 04 23:59:07 birdnetpi systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
May 04 23:59:07 birdnetpi spectrogram.sh[233538]: sox WARN wav: Premature EOF on .wav input file
May 04 23:59:07 birdnetpi systemd[1]: spectrogram_viewer.service: Succeeded.
May 04 23:59:08 birdnetpi birdnet_analysis.sh[233385]: analyze.py --i /home/pi/BirdSongs/May-2022/04-Wednesday/2022-05-04-birdnet-23:58:53.wav --o /home/pi/BirdSongs/May-2022/04-Wednesday/2022-05-04-birdnet-23:58:53.wav.csv --lat XXXX --lon XXXX --week 17 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
May 04 23:59:09 birdnetpi server.py[399]: ANALYZING AUDIO... DONE! Time 1.6 SECONDS
May 04 23:59:09 birdnetpi server.py[399]: WRITING RESULTS TO /home/pi/BirdSongs/May-2022/04-Wednesday/2022-05-04-birdnet-23:58:38.wav.csv ... DONE! WROTE 0 RESULTS.
May 04 23:59:09 birdnetpi server.py[399]: [NEW CONNECTION] ('127.0.0.1', 34444) connected.
May 04 23:59:09 birdnetpi server.py[399]: [ACTIVE CONNECTIONS] 1
May 04 23:59:09 birdnetpi server.py[399]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
May 04 23:59:09 birdnetpi server.py[399]: FULL FILENAME: -/home/pi/BirdSongs/May-2022/04-Wednesday/2022-05-04-birdnet-23:58:53.wav-
May 04 23:59:09 birdnetpi server.py[399]: Date-time: 2022-05-04 23:58:53
May 04 23:59:10 birdnetpi birdnet_analysis.sh[233566]: 0.0;3.0-('Strix aluco_Waldkauz', 0.07849456)
May 04 23:59:10 birdnetpi birdnet_analysis.sh[233566]: 3.0;6.0-('Luscinia megarhynchos_Nachtigall', 0.2396897)
May 04 23:59:10 birdnetpi birdnet_analysis.sh[233566]: 6.0;9.0-('Luscinia megarhynchos_Nachtigall', 0.05754697)
May 04 23:59:10 birdnetpi birdnet_analysis.sh[233566]: 9.0;12.0-('Bubo bubo_Uhu', 0.066098325)
May 04 23:59:10 birdnetpi birdnet_analysis.sh[233566]: 12.0;15.0-('Dryocopus martius_Schwarzspecht', 0.0496973)
May 04 23:59:10 birdnetpi systemd[1]: birdnet_analysis.service: Succeeded.
May 04 23:59:10 birdnetpi systemd[1]: birdnet_analysis.service: Consumed 3.021s CPU time.
May 04 23:59:12 birdnetpi systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 1127.
May 04 23:59:12 birdnetpi systemd[1]: Stopped BirdNET Analysis.
May 04 23:59:12 birdnetpi systemd[1]: birdnet_analysis.service: Consumed 3.021s CPU time.
May 04 23:59:12 birdnetpi systemd[1]: Started BirdNET Analysis.
May 04 23:59:13 birdnetpi birdnet_analysis.sh[233593]: Files loaded
May 04 23:59:16 birdnetpi birdnet_analysis.sh[233593]: RECORDING_LENGTH set to 15
May 04 23:59:17 birdnetpi systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 1648.
May 04 23:59:17 birdnetpi systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
May 04 23:59:17 birdnetpi systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
May 04 23:59:17 birdnetpi spectrogram.sh[233680]: sox WARN wav: Premature EOF on .wav input file
May 04 23:59:17 birdnetpi systemd[1]: spectrogram_viewer.service: Succeeded.
May 04 23:59:24 birdnetpi birdnet_analysis.sh[233593]: analyze.py --i /home/pi/BirdSongs/May-2022/04-Wednesday/2022-05-04-birdnet-23:59:08.wav --o /home/pi/BirdSongs/May-2022/04-Wednesday/2022-05-04-birdnet-23:59:08.wav.csv --lat XXXX --lon XXXX --week 17 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
May 04 23:59:24 birdnetpi server.py[399]: ANALYZING AUDIO... DONE! Time 1.6 SECONDS
May 04 23:59:24 birdnetpi server.py[399]: WRITING RESULTS TO /home/pi/BirdSongs/May-2022/04-Wednesday/2022-05-04-birdnet-23:58:53.wav.csv ... DONE! WROTE 0 RESULTS.
May 04 23:59:24 birdnetpi server.py[399]: [NEW CONNECTION] ('127.0.0.1', 34446) connected.
May 04 23:59:24 birdnetpi server.py[399]: [ACTIVE CONNECTIONS] 1
May 04 23:59:24 birdnetpi server.py[399]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
May 04 23:59:24 birdnetpi server.py[399]: FULL FILENAME: -/home/pi/BirdSongs/May-2022/04-Wednesday/2022-05-04-birdnet-23:59:08.wav-
May 04 23:59:24 birdnetpi server.py[399]: Date-time: 2022-05-04 23:59:08
May 04 23:59:25 birdnetpi birdnet_analysis.sh[233762]: 0.0;3.0-('Oriolus oriolus_Pirol', 0.07349175)
May 04 23:59:25 birdnetpi birdnet_analysis.sh[233762]: 3.0;6.0-('Puffinus puffinus_Atlantiksturmtaucher', 0.055073142)
May 04 23:59:25 birdnetpi birdnet_analysis.sh[233762]: 6.0;9.0-('Luscinia megarhynchos_Nachtigall', 0.08708503)
May 04 23:59:25 birdnetpi birdnet_analysis.sh[233762]: 9.0;12.0-('Bubo bubo_Uhu', 0.08471422)
May 04 23:59:25 birdnetpi birdnet_analysis.sh[233762]: 12.0;15.0-('Luscinia megarhynchos_Nachtigall', 0.057424873)
May 04 23:59:25 birdnetpi systemd[1]: birdnet_analysis.service: Succeeded.
May 04 23:59:25 birdnetpi systemd[1]: birdnet_analysis.service: Consumed 2.993s CPU time.
May 04 23:59:27 birdnetpi systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 1649.
May 04 23:59:27 birdnetpi systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
May 04 23:59:27 birdnetpi systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
May 04 23:59:27 birdnetpi systemd[1]: spectrogram_viewer.service: Succeeded.
May 04 23:59:28 birdnetpi systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 1128.
[... 167 lines cut ...]
May 05 00:00:49 birdnetpi spectrogram.sh[235055]: sox FAIL formats: can't open input file `/home/pi/BirdSongs/May-2022/04-Wednesday/2022-05-04-birdnet-23:59:53.wav': No such file or directory
May 05 00:00:49 birdnetpi systemd[1]: spectrogram_viewer.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
May 05 00:00:49 birdnetpi systemd[1]: spectrogram_viewer.service: Failed with result 'exit-code'.
May 05 00:00:49 birdnetpi systemd[1]: birdnet_analysis.service: Succeeded.
May 05 00:00:51 birdnetpi systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 1141.
May 05 00:00:51 birdnetpi systemd[1]: Stopped BirdNET Analysis.
May 05 00:00:51 birdnetpi systemd[1]: Started BirdNET Analysis.
May 05 00:00:53 birdnetpi systemd[1]: birdnet_analysis.service: Succeeded.
May 05 00:00:54 birdnetpi dhcpcd[896]: wlan0: Router Advertisement from fe80::d221:f9ff:fe5b:cf73
May 05 00:00:54 birdnetpi dhcpcd[896]: wlan0: Router Advertisement from fe80::d221:f9ff:fe5b:cf73
May 05 00:00:55 birdnetpi systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 1142.
May 05 00:00:55 birdnetpi systemd[1]: Stopped BirdNET Analysis.
May 05 00:00:55 birdnetpi systemd[1]: Started BirdNET Analysis.
May 05 00:00:55 birdnetpi dhcpcd[896]: wlan0: Router Advertisement from fe80::d221:f9ff:fe5b:cf73
May 05 00:00:56 birdnetpi systemd[1]: birdnet_analysis.service: Succeeded.
May 05 00:00:58 birdnetpi systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 1143.
May 05 00:00:58 birdnetpi systemd[1]: Stopped BirdNET Analysis.
May 05 00:00:58 birdnetpi systemd[1]: Started BirdNET Analysis.
May 05 00:00:59 birdnetpi systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 1658.
May 05 00:00:59 birdnetpi systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
May 05 00:00:59 birdnetpi systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
May 05 00:00:59 birdnetpi spectrogram.sh[235246]: sox FAIL formats: can't open input file `/home/pi/BirdSongs/May-2022/04-Wednesday/2022-05-04-birdnet-23:59:53.wav': No such file or directory
May 05 00:00:59 birdnetpi systemd[1]: spectrogram_viewer.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
May 05 00:00:59 birdnetpi systemd[1]: spectrogram_viewer.service: Failed with result 'exit-code'.
```

nilspupils, 2022-05-05:
It's about time we call it the 'haunted' midnight bug!

mzakharo, 2022-05-05:
I confirm installation from yesterday (`fc6cf7bd8`) still has this issue -> detection stops working overnight.

ehpersonal38, 2022-05-05:
Quoting Patrick:
> `find` is where I've decided to solve this. I'll just let 0 byte corrupted files live peacefully on the system and let `find` be the one to ignore them :)

However I say down with the 0 byte files! Diplomacy is not an option anymore, we must erase them! :)

mcguirepr89, 2022-05-05:
Hi, @kaeltis and @mzakharo -- thanks for letting me know.

I see what you mean about these pull requests, but I also have no idea how they were meant to address these corrupted files, since the changes on those PRs simply don't affect any specific files.

I will put something together shortly and will kindly ask that you update and then report back if you are seeing the issue any longer.

I will let you know here when to perform the update via "Tools" > "System Controls" > "Update"

My best regards,
Patrick

mcguirepr89, 2022-05-05:
Hello again -- please update via "Tools" > "System Controls" > "Update"

The planned fix is here https://github.com/mcguirepr89/BirdNET-Pi/commit/3705334570646aa000598ce2e51e05f2068b6ed4

Please do let me know if you encounter the issue again and I will kindly ask that you send me your corrupted audio files so that I may replicate your system errors while troubleshooting the solution.

My best regards to you all,
Patrick

cdkl, 2022-05-05:
I ran into this a couple of times this week as well. Had a temporary hack in place to fix this, but I'm happy to see a proper fix land. Thanks Patrick!

kaeltis, 2022-05-06:
Worked for me as well, thanks!

closed 2022-05-06 by kaeltis

malone-patpat, 2022-05-06:
I did an update to try and fix this as well and it introduced a new issue.  This error shows up  "birdpi birdnet_analysis.sh[26557]: /usr/local/bin/birdnet_analysis.sh: line 99: lsof: command not found", and the processing started failing.  The lsof command is not in the lite os version.  When I installed lsof and rebooted, it started working again.

mcguirepr89, 2022-05-06:
Perform another update -- this is a big change that required two updates. Going forward, there won't ever be another need for two updates back to back like this.
Sorry for the scare, but that error will clear up after the next update you perform (which will take a little longer than usual as it will install the missing `lsof` command)

mcguirepr89, 2022-05-06:
I got ahead of myself and didn't even read the last line wherein you mention manually installing `lsof` to fix the issue -- my bad!!! But I anticipate this update will make some folks worried because it breaks their installation until they do the next update -- all in the name of progress, though