mcguirepr89/BirdNET-Pi #250: gotty errors on x86_64

Phaeton, 2022-05-07:
Hi:

Attempted to build this out on a x86_64 Debian VM while waiting for a Docker build and initially was pretty excited with how far I was able to get. I modified the install_birdnet.sh script to not check for ARM, then updated requirements.txt to use "tflite_runtime" versus the ARM compiled version. Everything installed without error and after a reboot, I was able to get to the web interface. Initially noticed that the spectrograms were displaying as broken images. Then noticed Species Stats threw a bunch of errors. 

[image]

Moved on and noticed that View Log did not work. Went ahead and added a RTSP stream and watched the clips start piling up. They never seemed to process so I tailed journalctl and saw "Failed to execute" and "Failed to step EXEC spawning" errors for /usr/local/bin/gotty. I initially thought there was a missing shebang or /bin/bash in the service, but I think where I have landed tonight is that there must be something in the "custom" gotty build; is it specific to ARM? Errors were occuring for both birdnet_log.service and web_terminal.service.

I am also getting "birdnet spectrogram.sh[#####]: sox FAIL formats: can't open input file `': No such file or directory". Not sure if that is related to above or another separate item. "analyzing_now.txt" is not being created, but I imagine that's related to one of the earlier issues. 

Really excited to get this working, appreciate all the hard work that has gone into this so far!

mcguirepr89, 2022-05-07:
Hi, @Phaeton -- awesome that you're porting things to x86 -- we've got several users also interested in doing that, so I'm sure that any info you share about your accomplishments will be greatly appreciated!

And you are correct, the included `gotty` was compiled for ARM64 and also was compiled to support a path argument. I will try to find the exact repo from which I built the binary and I'm sure you'll be able to put one together for x86 to replace the included binary.

Regarding the spectrogram error, we'll need to get the logging working first (or at least lets just do one thing at a time). Great work so far!

mcguirepr89, 2022-05-07:
Good news!! I found the repo and there is already an x86 binary for `gotty` that is compiled the same way as the included `gotty` binary!

You can get it [here](https://github.com/sorenisanerd/gotty/releases)

mcguirepr89, 2022-05-07:
Also, looking at the original repo (https://github.com/kahst/BirdNET-Lite), you'll probably need to replace `tflite_runtime` in `requirements.txt` with the full `tensorflow` to get things working. There is a flex delegate that I don't think comes standard in the `tflite_runtime` `pip` site-package.

MrksHfmn, 2022-05-07:
@Phaeton Really great! Please share your successes! I am very interested in an x86 release :)

Phaeton, 2022-05-07:
@mcguirepr89 I replaced `gotty` from the scripts folder with the 1.3.0 AMD version from your earlier link. 

>You can get it [here](https://github.com/sorenisanerd/gotty/releases)

Specifically, this [one](https://github.com/sorenisanerd/gotty/releases/download/v1.3.0/gotty_v1.3.0_linux_amd64.tar.gz). This eliminated the error messages in journalctl, but View Log and Web Terminal still do not pull up. Any thoughts as to what to try next? I also replaced `tflite_runtime` with `tensorflow` .

Phaeton, 2022-05-07:
So there are definitely two things going on. One is a permissions issue, the second is potentially differences in the builds. Here is the code from GitHub for the web_terminal.service unadjusted. `ExecStart=/usr/local/bin/gotty --address localhost -w -p 8888 -P terminal --title-format "BirdNET-Pi Terminal" bash
`

If I try running that standalone, I get an error stating that the "-P" flag is invalid. If I remove `-P terminal,` I can get it to create the terminal in a browser, but not via the BirdNet webpage. Here is what I used. `ExecStart=sudo -E /usr/local/bin/gotty --address 0.0.0.0 -w -p 8888 --title-format "BirdNET-Pi Terminal" bash
`

`sudo -E` is needed as I must have some type of `$PATH` issue. I updated `--address 0.0.0.0` to access from a remote computer.

Phaeton, 2022-05-08:
Was able to resolve the permissions issues, and I can view the logs via gotty in a separate browser window, see above. 

From journalctl, current error messages are 
`May 07 23:17:34 birdnet daily_plot.py[8989]: Traceback (most recent call last):
May 07 23:17:34 birdnet daily_plot.py[8989]:   File "/usr/local/bin/daily_plot.py", line 59, in <module>
May 07 23:17:34 birdnet daily_plot.py[8989]:     norm = plt.Normalize(confmax.values.min(), confmax.values.max())
May 07 23:17:34 birdnet daily_plot.py[8989]:   File "/home/redacted/BirdNET-Pi/birdnet/lib/python3.9/site-packages/numpy/core/_methods.py", line 44, in _amin
May 07 23:17:34 birdnet daily_plot.py[8989]:     return umr_minimum(a, axis, None, out, keepdims, initial, where)
May 07 23:17:34 birdnet daily_plot.py[8989]: ValueError: zero-size array to reduction operation minimum which has no identity
May 07 23:17:39 birdnet spectrogram.sh[9080]: cat: /home/redacted/BirdNET-Pi/analyzing_now.txt: No such file or directory
May 07 23:17:39 birdnet spectrogram.sh[9081]: sox FAIL formats: can't open input file ': No such file or directory`

mcguirepr89, 2022-05-09:
Those errors seem to imply that the core services are still not functioning properly (they will populate the `analyzing_now.txt` file that is currently empty and causing the error. Likewise, plotly is reading an empty database (at least that is my supposition based on those log excerpts). 

As this effort is a good bit different than simply supporting BirdNET-Pi, I'd like to propose a remote desktop session between the two of us if that would be something you might be okay with? It would help a lot in being able to see how your system is failing. I'm not sure what your timezone is, but I'm GMT -4 right now, so US Eastern Daylight Time. I won't get my feelings hurt if this isn't something you're into, but if you are, I think we'll make a lot of progress rather quickly. If interested, [email me](mailto:mcguirepr89@gmail.com) and we'll arrange a workflow so to speak.

My best,
Patrick

Phaeton, 2022-05-13:
I think we can close this one. The main issue raised is definitely related to arm vs amd versions of `gotty`.  There is also something different with the `-P` flag, but that may be a result of me using version 1.3 versus I'm not confident what is used in the main repo. Regarding the VM build, it seems there are a lot of issues with the approach I am taking. I would love to see this get Docker support as that would allow for much easier adoption. For now I think I'll wait for Docker or hope to get lucky and find a Pi4.

closed 2022-05-13 by Phaeton