mcguirepr89/BirdNET-Pi #256: Exclusion list is ignored

oliverk71, 2022-05-08:
I noticed that BirdNET-Pi seems to ignore the exclusion list on my system. I have three species in the exclusion list, but these nevertheless show up in the recognition chart and there als also recordings of those species.

mcguirepr89, 2022-05-08:
Hi, Oliver!!
I try ink I fixed that this morning -- I noticed it too, from a recent update. 

https://github.com/mcguirepr89/BirdNET-Pi/commit/c3908a0435153de4076a0e93bb68a299bdb23906

That's what I spotted and fixed -- try another update and let me know if it continues. 

I'm keeping an eye on mine as well

oliverk71, 2022-05-08:
Hi Patrick, 
nice, then just close the issue.
I have another one, but this is possibly only in my installation: When I click on "Today detections" nothing is shown. Maybe since the last major update, I cannot tell exactly.

mcguirepr89, 2022-05-09:
I'm still seeing species from my exclusion list showing up in my data, so this issue is still open, unfortunately

mcguirepr89, 2022-05-09:
I'm hopeful that [this fix](https://github.com/mcguirepr89/BirdNET-Pi/commit/aa4908559c6c4737fa6407f0408b85d9f45167d2) will address this :crossed_fingers:

oliverk71, 2022-05-12:
Again, species from the exclusion list are put in the recognition list.

mcguirepr89, 2022-05-13:
Hi, @oliverk71 

I haven't been able to reproduce this issue since the fix. Would you mind running an update and pasting its output here so that I can ensure there are no issue with updates on your end?

oliverk71, 2022-05-16:
Sorry for the delay. Here it is: 
```
fatal: pathspec 'privacy_server.py' did not match any files
Saved working directory and index state WIP on main: 2739d59 Merge pull request #286 from mcguirepr89/revert-274-localhost-everywhere
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Already up to date.
On branch main
Your branch is up to date with 'origin/main'.

Changes not staged for commit:
  (use "git add ..." to update what will be committed)
  (use "git restore ..." to discard changes in working directory)
	modified:   ../model/labels.txt
	modified:   analyze.py
	modified:   daily_plot.py
	modified:   plotly_streamlit.py
	modified:   server.py

Untracked files:
  (use "git add ..." to include in what will be committed)
	../BirdDB.txt
	../HUMAN.txt
	../analyzing_now.txt
	../apprise.txt
	../birdnet/
	../exclude_species_list.txt
	../firstrun.ini
	../include_species_list.txt
	../lastrun.txt
	BirdDB.txt
	birds.db
	disk_check_exclude.txt
	exclude_species_list.txt
	labels.txt
	lastrun.txt
	thisrun.txt
	../templates/avahi-alias@.service
	../templates/birdnet_analysis.service
	../templates/birdnet_log.service
	../templates/birdnet_recording.service
	../templates/birdnet_server.service
	../templates/birdnet_stats.service
	../templates/chart_viewer.service
	../templates/custom_recording.service
	../templates/extraction.service
	../templates/livestream.service
	../templates/pushed_notifications.service
	../templates/spectrogram_viewer.service
	../templates/web_terminal.service
	../thisrun.txt

no changes added to commit (use "git add" and/or "git commit -a")
Dropped refs/stash@{0} (287fc1bb503ea7e2c233349d5f843e98df12f23b)
```

oliverk71, 2022-05-20:
@mcguirepr89 I think I am going to reinstall the whole thing