mcguirepr89/BirdNET-Pi #263: List index out of range after update

Svardsten53, 2022-05-09:
After an update analysis stopped and I'm getting the following:
```
22:50:14 systemd[1]: Started BirdNET Analysis Server.
22:50:17 server.py[32864]: Traceback (most recent call last):
22:50:17 server.py[32864]:   File "/usr/local/bin/server.py", line 51, in <module>
22:50:17 server.py[32864]:     priv_thresh = float("." + str(str(str([i for i in this_run if i.startswith('PRIVACY_THRESHOLD')]).split('=')[1]).split('\\')[0]))/10
22:50:17 server.py[32864]: IndexError: list index out of range
22:50:17 systemd[1]: birdnet_server.service: Main process exited, code=exited, status=1/FAILURE
22:50:17 systemd[1]: birdnet_server.service: Failed with result 'exit-code'.
22:50:17 systemd[1]: birdnet_server.service: Consumed 3.806s CPU time.
22:50:19 systemd[1]: Stopped BirdNET Analysis Server.
22:50:19 systemd[1]: birdnet_server.service: Consumed 3.806s CPU time.
22:50:19 systemd[1]: Started BirdNET Analysis Server.
22:50:20 systemd[1]: Stopping BirdNET Analysis Server...
22:50:20 systemd[1]: birdnet_server.service: Succeeded.
22:50:20 systemd[1]: Stopped BirdNET Analysis Server.
22:50:20 systemd[1]: birdnet_server.service: Consumed 2.311s CPU time.
22:50:20 systemd[1]: Started BirdNET Analysis Server.
22:50:24 server.py[32984]: Traceback (most recent call last):

```

I have tried restarting all services and did a reboot, but still the same problem. Is there a way to fix this?

I'm tinkering with a routine to do translations and some modifications of pages and before an update I do a restore of the original php and py files in the scripts directory in order to not getting unmerged files etc, so it might be possible I have caused this issue myself.

mcguirepr89, 2022-05-09:
Hi, Lars --

Can I ask you what method you used to update?

Svardsten53, 2022-05-10:
I used the usual one in the menu.

Svardsten53, 2022-05-10:
I thought I solved the problem by doing the following:

- Backup of birds.db
- Uninstallation (which preserves the BirdSongs folder)
- New installation
- Restoring the database, which was not easy from a plain file, NOT a GZ backup, difficult to replace the running db.
- Adding my own translations

It looks fine in the log, but the confidence values are very low. Even if I lower  the confidence threshold to 0,3 and get some detections they are not written to the database but they are in /BirdSongs/Extracted/By-Date folder.
```
11:45:15 server.py[16257]: WRITING RESULTS TO /home/tyreso/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-11:44:25.wav.csv ... DONE! WROTE 1 RESULTS.
```
There seems to be some problem with the database. I guess the best thing is to do a clean installation, but it should have been nice to have the observations from the latest weeks saved. I'll see if I can reproduce the problem on my test system.

mcguirepr89, 2022-05-10:
Hi, Lars,

During this step:
>Restoring the database, which was not easy from a plain file, NOT a GZ backup, difficult to replace the running db.

Did you [use `scp` or WinSCP` as detailed in the wiki](https://github.com/mcguirepr89/BirdNET-Pi/wiki/Backup-and-Restore-the-Database#restoring-the-database)?

ehpersonal38, 2022-05-10:
> Even if I lower the confidence threshold to 0,3 and get some detections they are not written to the database but they are in /BirdSongs/Extracted/By-Date folder.

I had this issue the other day when I was messing with birds.db over FTP. No detections got logged. I had to `chmod 777` birds.db, then reboot the system, then it all started working again.

mcguirepr89, 2022-05-10:
@ehpersonal38 Nice avatar!

Svardsten53, 2022-05-10:
Thank you both for your help!

I first used File Manager to transfer the file from Windows, didn't konw about the bug or WinSCP. After using WinSCP everything works much better and I have detections in the database again.

closed 2022-05-10 by mcguirepr89