mcguirepr89/BirdNET-Pi #266: Failures on New Install

mcguirepr89, 2022-05-10:
### Discussed in https://github.com/mcguirepr89/BirdNET-Pi/discussions/265

<div type='discussions-op-text'>

<sup>Originally posted by **jacobabehm** May 10, 2022</sup>
I have an install that's been running for a few days.  Every few hours, it hangs on an error.  The following from log is a successful connection followed by a failure:

```
12:49:02 server.py[28213]: [ACTIVE CONNECTIONS] 1
12:49:02 server.py[28213]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
12:49:02 server.py[28213]: FULL FILENAME: -/home/dnkn/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-12:48:47.wav-
12:49:02 server.py[28213]: Date-time: 2022-05-10 12:48:47
12:49:04 birdnet_analysis.sh[52834]: 0.0;3.0-('Cardinalis cardinalis_Northern Cardinal', 0.20219456)
12:49:04 birdnet_analysis.sh[52834]: 3.0;6.0-('Spizella pusilla_Field Sparrow', 0.051027566)
12:49:04 birdnet_analysis.sh[52834]: 6.0;9.0-('Vireo gilvus_Warbling Vireo', 0.10197324)
12:49:04 birdnet_analysis.sh[52834]: 9.0;12.0-('Branta canadensis_Canada Goose', 0.24285354)
12:49:04 birdnet_analysis.sh[52834]: 12.0;15.0-('Scolopax minor_American Woodcock', 0.042452153)
12:49:04 systemd[1]: birdnet_analysis.service: Succeeded.
12:49:06 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 135.
12:49:06 systemd[1]: Stopped BirdNET Analysis.
12:49:06 systemd[1]: Started BirdNET Analysis.
12:49:07 birdnet_analysis.sh[52860]: Files loaded
12:49:07 birdnet_analysis.sh[52860]: RECORDING_LENGTH set to 15
12:49:07 env[52894]: Line = 2022-05-10-birdnet-12:48:47.wav.csv
12:49:27 birdnet_analysis.sh[52860]: analyze.py --i /home/dnkn/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-12:49:02.wav --o /home/dnkn/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-12:49:02.wav.csv --lat 44.9863 --lon -92.7278 --week 18 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
12:49:27 server.py[28213]: ANALYZING AUDIO... DONE! Time 1.5 SECONDS
12:49:27 server.py[28213]: WRITING RESULTS TO /home/dnkn/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-12:48:47.wav.csv ... DONE! WROTE 0 RESULTS.
12:49:27 server.py[28213]: [NEW CONNECTION] ('127.0.0.1', 49272) connected.
12:49:27 server.py[28213]: [ACTIVE CONNECTIONS] 1
12:49:28 server.py[28213]: READING AUDIO DATA... /home/dnkn/BirdNET-Pi/birdnet/lib/python3.9/site-packages/librosa/util/decorators.py:88: UserWarning: PySoundFile failed. Trying audioread instead.
12:49:28 server.py[28213]:   return f(*args, **kwargs)
12:49:28 server.py[28213]: Exception in thread Thread-174:
12:49:28 server.py[28213]: Traceback (most recent call last):
12:49:28 server.py[28213]:   File "/home/dnkn/BirdNET-Pi/birdnet/lib/python3.9/site-packages/librosa/core/audio.py", line 155, in load
12:49:28 server.py[28213]:     context = sf.SoundFile(path)
12:49:28 server.py[28213]:   File "/home/dnkn/BirdNET-Pi/birdnet/lib/python3.9/site-packages/soundfile.py", line 629, in __init__
12:49:28 server.py[28213]:     self._file = self._open(file, mode_int, closefd)
12:49:28 server.py[28213]:   File "/home/dnkn/BirdNET-Pi/birdnet/lib/python3.9/site-packages/soundfile.py", line 1183, in _open
12:49:28 server.py[28213]:     _error_check(_snd.sf_error(file_ptr),
12:49:28 server.py[28213]:   File "/home/dnkn/BirdNET-Pi/birdnet/lib/python3.9/site-packages/soundfile.py", line 1357, in _error_check
12:49:28 server.py[28213]:     raise RuntimeError(prefix + _ffi.string(err_str).decode('utf-8', 'replace'))
12:49:28 server.py[28213]: RuntimeError: Error opening '/home/dnkn/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-12:49:02.wav': System error.
12:49:28 server.py[28213]: During handling of the above exception, another exception occurred:
12:49:28 server.py[28213]: Traceback (most recent call last):
12:49:28 server.py[28213]:   File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
12:49:28 server.py[28213]:     self.run()
12:49:28 server.py[28213]:   File "/usr/lib/python3.9/threading.py", line 892, in run
12:49:28 server.py[28213]:     self._target(*self._args, **self._kwargs)
12:49:28 server.py[28213]:   File "/usr/local/bin/server.py", line 334, in handle_client
12:49:28 server.py[28213]:     audioData = readAudioData(args.i, args.overlap)
12:49:28 server.py[28213]:   File "/usr/local/bin/server.py", line 125, in readAudioData
12:49:28 server.py[28213]:     sig, rate = librosa.load(path, sr=sample_rate, mono=True, res_type='kaiser_fast')
12:49:28 server.py[28213]:   File "/home/dnkn/BirdNET-Pi/birdnet/lib/python3.9/site-packages/librosa/util/decorators.py", line 88, in inner_f
12:49:28 server.py[28213]:     return f(*args, **kwargs)
12:49:28 server.py[28213]:   File "/home/dnkn/BirdNET-Pi/birdnet/lib/python3.9/site-packages/librosa/core/audio.py", line 174, in load
12:49:28 server.py[28213]:     y, sr_native = __audioread_load(path, offset, duration, dtype)
12:49:28 server.py[28213]:   File "/home/dnkn/BirdNET-Pi/birdnet/lib/python3.9/site-packages/librosa/core/audio.py", line 198, in __audioread_load
12:49:28 server.py[28213]:     with audioread.audio_open(path) as input_file:
12:49:28 server.py[28213]:   File "/home/dnkn/BirdNET-Pi/birdnet/lib/python3.9/site-packages/audioread/__init__.py", line 111, in audio_open
12:49:28 server.py[28213]:     return BackendClass(path)
12:49:28 server.py[28213]:   File "/home/dnkn/BirdNET-Pi/birdnet/lib/python3.9/site-packages/audioread/rawread.py", line 62, in __init__
12:49:28 server.py[28213]:     self._fh = open(filename, 'rb')
12:49:28 server.py[28213]: FileNotFoundError: [Errno 2] No such file or directory: '/home/dnkn/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-12:49:02.wav'
```

Also, I was getting a different failure after the most recent upgrade:

```
12:05:34 server.py[17047]: [NEW CONNECTION] ('127.0.0.1', 48902) connected.
12:05:34 server.py[17047]: [ACTIVE CONNECTIONS] 1
12:05:34 server.py[17047]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
12:05:34 server.py[17047]: FULL FILENAME: -/home/dnkn/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-12:05:19.wav-
12:05:34 server.py[17047]: Date-time: 2022-05-10 12:05:19
12:05:36 server.py[17047]: ANALYZING AUDIO... Exception in thread Thread-45:
12:05:36 server.py[17047]: Traceback (most recent call last):
12:05:36 server.py[17047]:   File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
12:05:36 server.py[17047]:     self.run()
12:05:36 server.py[17047]:   File "/usr/lib/python3.9/threading.py", line 892, in run
12:05:36 server.py[17047]:     self._target(*self._args, **self._kwargs)
12:05:36 server.py[17047]:   File "/usr/local/bin/server.py", line 363, in handle_client
12:05:36 server.py[17047]:     writeResultsToFile(detections, min_conf, args.o)
12:05:36 server.py[17047]:   File "/usr/local/bin/server.py", line 256, in writeResultsToFile
12:05:36 server.py[17047]:     sendAppriseNotifications(str(entry[0]),str(entry[1]));
12:05:36 server.py[17047]:   File "/usr/local/bin/server.py", line 229, in sendAppriseNotifications
12:05:36 server.py[17047]:     if os.path.getsize(userDir + '/BirdNET-Pi/apprise.txt') > 0:
12:05:36 server.py[17047]:   File "/usr/lib/python3.9/genericpath.py", line 50, in getsize
12:05:36 server.py[17047]:     return os.stat(filename).st_size
12:05:36 server.py[17047]: FileNotFoundError: [Errno 2] No such file or directory: '/home/dnkn/BirdNET-Pi/apprise.txt'
```

I think this failure only occurred when there was a new results to write.  If the job complete with "DONE! WROTE 0 RESULTS", no failure.  There was no "apprise.txt" file, so I created one.  The jobs now seem to complete fine with the above exception.  Nothing is ever written to "apprise.txt".

Any idea what's going on?</div>

mcguirepr89, 2022-05-10:
https://github.com/mcguirepr89/BirdNET-Pi/discussions/265#discussioncomment-2725324

ehpersonal38, 2022-05-10:
I'll be interested to see the cause of this bug -- it's reminding me of the Great Midnight Bug of 2022.

mcguirepr89, 2022-05-11:
Ok I found the bug and am now working on a fix -- stay tuned! (but don't hold your breath, could take a little while)

mcguirepr89, 2022-05-11:
The cause of the bug:
lines 98, 101, and 102 of `birdnet_analysis.sh`

Why it used to work:
The old code checked audio file length while incrementing the variable `a`. If `a`'s value surpassed the `RECORDING_LENGTH` value, the audio clip is considered to be of a different (former) `RECORDING_LENGTH` and is discarded.

The new code checks whether the loaded audio files are currently open and in use by another process. If so, it waits 2 seconds and checks again until the file is closed.

Unfortunately, the new code, which doesn't utilize the `a` variable for file length comparison still kept the variable incrementing in the loop, so it has allowed occasions wherein the `a` value is able to increment past the `RECORDING_LENGTH` value, and then discards the file. You can see this process here:

```
16:51:02 birdnet_analysis.sh[198798]: + '[' 14 -ge 15 ']'
16:51:02 birdnet_analysis.sh[198798]: + a=16
16:51:02 birdnet_analysis.sh[198956]: ++ lsof -t /home/dnkn/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-16:50:33.wav
16:51:02 birdnet_analysis.sh[198798]: + '[' -z '198249
16:51:02 birdnet_analysis.sh[198798]: 198928' ']'
16:51:02 birdnet_analysis.sh[198798]: + sleep 2
16:51:02 birdnet_analysis.sh[198798]: + '[' 16 -ge 15 ']'
16:51:02 birdnet_analysis.sh[198798]: + rm -f /home/dnkn/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-16:50:33.wav
16:51:02 birdnet_analysis.sh[198798]: + break
16:51:02 birdnet_analysis.sh[198798]: + grep 5050 /dev/fd/63
16:51:02 birdnet_analysis.sh[198976]: ++ netstat -tulpn
16:51:02 birdnet_analysis.sh[198798]: + '[' -f null ']'
16:51:02 birdnet_analysis.sh[198798]: + INCLUDEPARAM=
16:51:02 birdnet_analysis.sh[198798]: + '[' -f null ']'
16:51:02 birdnet_analysis.sh[198798]: + EXCLUDEPARAM=
16:51:02 birdnet_analysis.sh[198798]: + '[' '!' -z ']'
16:51:02 birdnet_analysis.sh[198798]: + BIRDWEATHER_ID_PARAM=
16:51:02 birdnet_analysis.sh[198798]: + BIRDWEATHER_ID_LOG=
16:51:02 birdnet_analysis.sh[198798]: + echo analyze.py --i /home/dnkn/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-16:50:33.wav --o /home/dnkn/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-16:50:33.wav.csv --lat 44.9863 --lon -92.7278 --week 18 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
16:51:02 birdnet_analysis.sh[198798]: analyze.py --i /home/dnkn/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-16:50:33.wav --o /home/dnkn/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-16:50:33.wav.csv --lat 44.9863 --lon -92.7278 --week 18 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
16:51:02 birdnet_analysis.sh[198798]: + analyze.py --i /home/dnkn/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-16:50:33.wav --o /home/dnkn/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-16:50:33.wav.csv --lat 44.9863 --lon -92.7278 --week 18 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
16:51:03 server.py[115570]: Exception in thread Thread-574:
16:51:03 server.py[115570]: Traceback (most recent call last):
16:51:03 server.py[115570]:   File "/home/dnkn/BirdNET-Pi/birdnet/lib/python3.9/site-packages/librosa/core/audio.py", line 155, in load
16:51:03 server.py[115570]:     context = sf.SoundFile(path)
16:51:03 server.py[115570]:   File "/home/dnkn/BirdNET-Pi/birdnet/lib/python3.9/site-packages/soundfile.py", line 629, in __init__
16:51:03 server.py[115570]:     self._file = self._open(file, mode_int, closefd)
16:51:03 server.py[115570]:   File "/home/dnkn/BirdNET-Pi/birdnet/lib/python3.9/site-packages/soundfile.py", line 1183, in _open
16:51:03 server.py[115570]:     _error_check(_snd.sf_error(file_ptr),
16:51:03 server.py[115570]:   File "/home/dnkn/BirdNET-Pi/birdnet/lib/python3.9/site-packages/soundfile.py", line 1357, in _error_check
16:51:03 server.py[115570]:     raise RuntimeError(prefix + _ffi.string(err_str).decode('utf-8', 'replace'))
16:51:03 server.py[115570]: RuntimeError: Error opening '/home/dnkn/BirdSongs/May-2022/10-Tuesday/2022-05-10-birdnet-16:50:33.wav': System error.
```

The fix that should address this simply removes the `a` iterator and its rule to remove files when it is greater than the `RECORDING_LENGTH` value.

You can [take a look at it here ](https://github.com/mcguirepr89/BirdNET-Pi/commit/6cc2e0b0fd92219fbdde7d1ce707ef8bb4f69f95 )and are invited to update to resolve this issue.

Let's all keep an eye on things and we'll see whether this resolution is adequate. 

My best regards as always,
Patrick

closed 2022-05-13 by mcguirepr89