mcguirepr89/BirdNET-Pi #276: Setting a password

filikun, 2022-05-14:
So this is the second time I'm trying to set a password for the tools menu and not being able to login. Is there any restrictions to length or letters/numbers/special characters  being used here? First time I had 16 chars with letters, numbers and special characters and now second time I have the same mix but only 8 letters. 

Or does the username change or something?

mcguirepr89, 2022-05-14:
Hi, @filikun 

I'm sorry that you're having trouble setting your web page password. I've not heard of this issue yet, so we'll need to see what is going on.

>Is there any restrictions to length or letters/numbers/special characters being used here?

No

>Or does the username change or something?

Nope

>Can I change password with ssh access to the host?

Yep!

To see exactly what password is configured in `birdnet.conf` file, run this command from SSH:
```
grep -e '^CADDY_PWD' ~/BirdNET-Pi/birdnet.conf
```

To ensure that the password configured in `birdnet.conf` is the password set in the `Caddyfile`, run:
```
sudo update_caddyfile.sh
```

If you encounter any issues with those steps, please let me know by pasting the erroneous output here for further review.

(If your password contains an `=` I could see that causing problems -- likewise if it were to contain quotations `"` or backticks ``` ` ```, I could see that causing problems)

filikun, 2022-05-14:
Thanks for the fast reply as always @mcguirepr89! 

I run both of them, the password do match. I'm using ´%j73D!zK´. So the username is still `birdnet`? That does not work for me.

filikun, 2022-05-14:
```
CADDY_PWD=%j73D!zK
filikun@birdnetpi:~/BirdNET-Pi$ sudo update_caddyfile.sh
+ '[' -d /etc/caddy ']'
+ '[' -f /etc/caddy/Caddyfile ']'
+ cp /etc/caddy/Caddyfile /etc/caddy/Caddyfile.original
+ '[' -z '%j73D!zK' ']'
++ caddy hash-password -plaintext '%j73D!zK'
+ HASHWORD=JDJhJDE0JHk0ZDNhRTdOODVVR29rdFNyYTlYL09neUJVRy8wSWFPai5GWlBFWXBZcGtkT                                                            zZBV2pkL3F1
+ cat
++ hostname
+ sudo systemctl reload caddy

```

mcguirepr89, 2022-05-14:
To confirm, you are unable to login using 
Username: `birdnet`
Password: `%j73D!zK`

Could you try changing the password in `birdnet.conf` to `test` and then rerun `sudo update_caddyfile.sh` to determine whether the password content is playing a role?

mcguirepr89, 2022-05-14:
Also, [this reference](https://www.ncsc.gov.uk/blog-post/the-logic-behind-three-random-words#:~:text='Three%20random%20words%20will%20generate,password%20as%20it's%20quite%20guessable.) may be relevant here.

filikun, 2022-05-14:
Changing it to `test` worked fine :) So there might be a problem with the special signs then?

mcguirepr89, 2022-05-14:
Yes it seems that the special characters are the culprit here -- I will look into what may be needed to allow them in this context. Thank you for your help getting to the bottom of this issue. I will update you on this as I think of a way to allow these characters. I'm certainly you're not the only user interested in using special characters to protect their installation.

My best regards,
Patrick

I will rename this issue to describe the issue we found and will keep it open for others

filikun, 2022-05-14:
Thanks a bunch! 

I tried taking away `%` to `j73D!zK` did not work, then removed `!` as well to 'j73DzK' and then it worked so yes, special character is the problem.

I use Bitwarden to autogenerate passwords for me, so it's not like I'm trying to remember all my random 16 character passwords :D haha. Il just add a 16 character without special ones for no.

filikun, 2022-05-15:
Today I sometimes get the login promt and it would not accept my password. The thing is, sometimes it does work and sometimes it does not. Not sure if my setup is just broken.

Running the `update_caddyfile.sh` I get the following _after_ it have shown me the password:

```
+ cat
++ hostname
+ sudo systemctl reload caddy
Job for caddy.service failed.
See "systemctl status caddy.service" and "journalctl -xe" for details.
```

Is that normal?

ehpersonal38, 2022-05-15:
> 

Could you post the output of `systemctl status caddy.service` too?

mcguirepr89, 2022-05-15:
This is likely due to [the bug that was just found and fixed](https://github.com/mcguirepr89/BirdNET-Pi/pull/281). 

This can likely be fixed with "Tools" > "System Controls" > "Update" (if the updater will work)

Then:
```
sudo update_caddyfile.sh
```
you'll likely get the same complaint about not being able to reload (since it's not loaded), so you'll do this:
```
sudo systemctl restart caddy
```

Please let me know if that helps resolve the issue you're describing.
(Sorry to jump in @ehpersonal38)

filikun, 2022-05-15:
> Could you post the output of `systemctl status caddy.service` too?

```
May 15 18:56:30 birdnetpi systemd[1]: Reloading Caddy.
May 15 18:56:30 birdnetpi caddy[20061]: {"level":"info","ts":1652633790.3815663,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","c>
May 15 18:56:30 birdnetpi caddy[20061]: reload: adapting config using caddyfile: parsing caddyfile tokens for 'basicauth': /etc/caddy/Caddyfile:10 - Error d>
May 15 18:56:30 birdnetpi systemd[1]: caddy.service: Control process exited, code=exited, status=1/FAILURE
May 15 18:56:30 birdnetpi systemd[1]: Reload failed for Caddy.
May 15 18:58:03 birdnetpi systemd[1]: Reloading Caddy.
May 15 18:58:03 birdnetpi caddy[22122]: {"level":"info","ts":1652633883.6477597,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","c>
May 15 18:58:03 birdnetpi caddy[22122]: reload: adapting config using caddyfile: parsing caddyfile tokens for 'basicauth': /etc/caddy/Caddyfile:10 - Error d>
May 15 18:58:03 birdnetpi systemd[1]: caddy.service: Control process exited, code=exited, status=1/FAILURE
May 15 18:58:03 birdnetpi systemd[1]: Reload failed for Caddy.
```

Ill try the fixes

jmherbst, 2022-05-15:
I've also opened https://github.com/mcguirepr89/BirdNET-Pi/pull/283 to hopefully protect a bit more around passwords with special characters

filikun, 2022-05-15:
No errors after running `sudo update_caddyfile.sh`

mcguirepr89, 2022-05-15:
I merged @jmherbst's fix to allow special characters in Basic Auth passwords. Thanks again, @jmherbst! :handshake:

bdoner, 2022-05-17:
> ```
> May 15 18:56:30 birdnetpi systemd[1]: Reloading Caddy.
> May 15 18:56:30 birdnetpi caddy[20061]: {"level":"info","ts":1652633790.3815663,"msg":"using provided [...]

I got this error as well yesterday. Seems like a new rule have been added in the Caddyfile on line 10, which is invalid syntax. Had to manually remove some of it (on mobile, can't remember the exact string) before Caddy could run again.
it's a different issue from this.

mcguirepr89, 2022-05-24:
I want to revisit this since `birdnet.conf` cannot be properly sourced when the password contains certain characters. I'm thinking the simplest approach is changing the quotations around the password to single quotes to disallow shell expansion, and then setting something up that checks the new password that is input to be sure it doesn't contain and single quotes that would break `birdnet.conf`

Anyone have any thoughts?

bdoner, 2022-05-24:
My initial thought, to avoid breaking characters in `birdnet.conf`, would simply be to base64 encode the password as long as it resides in the config file.

ehpersonal38, 2022-05-24:
And while we're at it, why not sha256 hash the passwords?

jmherbst, 2022-05-24:
Good point on the single-quotes to prevent shell expansion...

What if we took the input directly from the POST and used the caddy `hash-password` https://caddyserver.com/docs/command-line#caddy-hash-password function?

mcguirepr89, 2022-05-24:
There is certainly tremendous value in hashing the passwords before storing them in files -- it would prevent any password recovery, but it would be easy enough to reset from the command line.

So you guys know, when you update the password or the custom URL in "Advanced Settings," the existing Caddyfile is recreated for the new settings. The password input in "Advanced Settings" get hashed by `caddy hash-password` and is input into the Caddyfile.

If it would be best, this hashed password can also be the password that is updated in `birdnet.conf`. Then, the "hashwords" will match in `birdnet.conf` and the `Caddyfile`

Does that seem reasonable?

bdoner, 2022-05-24:
> And while we're at it, why not sha256 hash the passwords?

Because the caddy `hash-password` is not a `sha256`. So the password is needed at some point in order to update the caddyfile. What jmherbst suggest is pretty much the same as you suggest, just using a different algorithm.

> it would prevent any password recovery

Which inherently isn't a bad this imo.

ehpersonal38, 2022-05-24:
In update_caddyfile.sh (I assume this is what gets called after a user changes the pw on the web gui?), couldn't you just change:
`HASHWORD=$(caddy hash-password -plaintext ${CADDY_PWD})`
to
`HASHWORD=${CADDY_PWD}`

And then in advanced.php where it writes the user PW to birdnet.conf:
Change from:
`$caddy_pwd = $_GET['caddy_pwd']`
To
`$caddy_pwd = shell_exec("caddy hash-password $_GET['caddy_pwd']")`

So that way PHP hashes the user submitted password, that hash gets put in birdnet.conf AND into /etc/caddy/Caddyfile. And then when a user logs in with Basic Auth, you check like 
`if(shell_exec("caddy hash-password $_SERVER['PHP_AUTH_PW']") != $config['CADDY_PWD']) || $submitteduser !== 'birdnet'){`

Sorry, that might be missing the mark completely. Even reading back through it I'm not sure I understand the problem completely. Maybe it'll give you some ideas though 😄

ehpersonal38, 2022-05-24:
Oh right, I'm from the days of md5 where that was the case!

mcguirepr89, 2022-05-24:
Yeah, it would work for encryption, but I'm not sure it is the way to go:
```
birdnetpi1:~/BirdNET-Pi$ sha256sum <(test)
e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855  /dev/fd/63
birdnetpi1:~/BirdNET-Pi$ sha256sum <(test)
e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855  /dev/fd/63
birdnetpi1:~/BirdNET-Pi$ sha256sum <(test)
e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855  /dev/fd/63
birdnetpi1:~/BirdNET-Pi$ sha256sum <(test)
e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855  /dev/fd/63
birdnetpi1:~/BirdNET-Pi$ sha256sum <(test)
e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855  /dev/fd/63
birdnetpi1:~/BirdNET-Pi$ md5sum <(test)
d41d8cd98f00b204e9800998ecf8427e  /dev/fd/63
birdnetpi1:~/BirdNET-Pi$ md5sum <(test)
d41d8cd98f00b204e9800998ecf8427e  /dev/fd/63
birdnetpi1:~/BirdNET-Pi$ md5sum <(test)
d41d8cd98f00b204e9800998ecf8427e  /dev/fd/63
```

ehpersonal38, 2022-05-24:
md5 and sha256 are both hashing actually. Would scrypt work instead?
```
pi@birdnetpi:~/BirdNET-Pi/scripts$ caddy hash-password --algorithm scrypt --plaintext TEST
ldYtRCQlR/0fJUmv9Q0uwB3lyiFPsHNj/sTq4vm1+y4=
pi@birdnetpi:~/BirdNET-Pi/scripts$ caddy hash-password --algorithm scrypt --plaintext TEST
ldYtRCQlR/0fJUmv9Q0uwB3lyiFPsHNj/sTq4vm1+y4=
pi@birdnetpi:~/BirdNET-Pi/scripts$ caddy hash-password --algorithm scrypt --plaintext TEST
ldYtRCQlR/0fJUmv9Q0uwB3lyiFPsHNj/sTq4vm1+y4=
```

bdoner, 2022-05-24:
`scrypt` is a better alternative than `sha` fwiw.

ehpersonal38, 2022-05-25:
Also if we wanted to use bcrypt we could just use:
```php
$hash = base64_decode(shell_exec("caddy hash-password --plaintext teST123"));
echo $hash."<br>";

if (password_verify('teST123', $hash)) {
    echo 'Password is valid!';
} else {
    echo 'Invalid password.';
}
```
Output:
```
$2a$14$XXrJJQF8SG90UGJdq3LgUOM6FQ3UxRQIdhzX8D7r4CCx/IhqTcVii
Password is valid!
```

mcguirepr89, 2022-05-25:
I'll try to futz with this soon -- for now, just don't use special characters -- or if concerned with security (like you need special characters allllll up in the password, or even just want one), you can always edit the Caddyfile manually and set whatever username and password you'd like. Remember to hash it `caddy hash-password`!

mcguirepr89, 2022-08-22:
I just wanted to note here that the new Laravel framework will address this AND add more security AND be easier to adjust in the future AND will rate limit AND won't actually be basic Auth so no old-school dialogs.

My best,
Patrick

closed 2023-02-10 by ehpersonal38