mcguirepr89/BirdNET-Pi #278: Not Broadcasting on Port 443

marshalmiller, 2022-05-15:
I have set up everything and installed an SSL cert via DNS txt record, but I cannot get BirdNET-Pi to broadcast on port 443.  I am not super familiar with Caddy. Is there an easy way to do this? I didn't see anything on Caddy's page.

mcguirepr89, 2022-05-15:
Hi, @marshalmiller 
I do hope you will read the Caddy documentation more, as I know it has a great deal of information on how to enable the HTTPS protocol (indeed, it is its main unique feature and is very heavily referenced, even among the bold lettering on Caddy's main page 
[image])

Please do try searching their documentation for "HTTPS" and you will find all the information needed to enable the built-in HTTPS handling for which Caddy is best known.
[image]

marshalmiller, 2022-05-15:
@mcguirepr89  Due to my certificate management solution, I cannot use the zeroSSL project to issue certificates. I was hoping there was an alternative way to activate port 443, outside of caddy. If there isn't, that is fine. I was just wondering. I didn't see any other way to manually set the port.

mcguirepr89, 2022-05-15:
I think I understand, though unfortunately, utilizing port 443 would happen within Caddy since it is the primary web server. The only alternative I can think of which would allow you to control TCP packet flow for port 443 is if you wanted to install a different web server (Nginx or Apache), then you could handle binding to 443 "outside of caddy."

Since this isn't an issue with Caddy or BirdNET-Pi, I'm going to close it, though please feel free to continue to respond if you have additional questions regarding your particular setup.

My best,
Patrick

closed 2022-05-15 by mcguirepr89

marshalmiller, 2022-05-15:
Thank you for your time and help.

jmherbst, 2022-05-15:
FWIW @marshalmiller - you may look into https://caddyserver.com/docs/caddyfile/directives/tls and the related https://caddy.community/t/how-to-use-dns-provider-modules-in-caddy-2/8148