mcguirepr89/BirdNET-Pi #279: No module named 'apprise'

filikun, 2022-05-15:
I needed to remove the USB mic from the pi for a couple of hours but can't get it to work again. Logs looking like this:

```
16:49:01 systemd[1]: birdnet_server.service: Failed with result 'exit-code'.
16:49:01 systemd[1]: birdnet_server.service: Consumed 10.232s CPU time.
16:49:11 systemd[1]: birdnet_server.service: Scheduled restart job, restart counter is at 1772.
16:49:11 systemd[1]: Stopping BirdNET Analysis...
16:49:11 systemd[1]: birdnet_analysis.service: Succeeded.
16:49:11 systemd[1]: Stopped BirdNET Analysis.
16:49:11 systemd[1]: Stopped BirdNET Analysis Server.
16:49:11 systemd[1]: birdnet_server.service: Consumed 10.232s CPU time.
16:49:11 systemd[1]: Started BirdNET Analysis Server.
16:49:11 systemd[1]: Started BirdNET Analysis.
16:49:20 server.py[1177424]: Traceback (most recent call last):
16:49:20 server.py[1177424]:   File "/usr/local/bin/server.py", line 28, in <module>
16:49:20 server.py[1177424]:     import apprise
16:49:20 server.py[1177424]: ModuleNotFoundError: No module named 'apprise'
16:49:21 systemd[1]: birdnet_server.service: Main process exited, code=exited, status=1/FAILURE
16:49:21 systemd[1]: birdnet_server.service: Failed with result 'exit-code'.
16:49:21 systemd[1]: birdnet_server.service: Consumed 8.962s CPU time.
16:49:31 systemd[1]: birdnet_server.service: Scheduled restart job, restart counter is at 1773.
16:49:31 systemd[1]: Stopping BirdNET Analysis...
16:49:31 systemd[1]: birdnet_analysis.service: Succeeded.
16:49:31 systemd[1]: Stopped BirdNET Analysis.
16:49:31 systemd[1]: Stopped BirdNET Analysis Server.
16:49:31 systemd[1]: birdnet_server.service: Consumed 8.962s CPU time.
16:49:31 systemd[1]: Started BirdNET Analysis Server.
16:49:31 systemd[1]: Started BirdNET Analysis.
```

mcguirepr89, 2022-05-15:
Hi, @filikun 
Those things are unrelated so far as I can tell. The complaint in the log is clear that the `apprise` module is not installed. An update via "Tools" > "System Controls" > "Update" should have done this for you, but obviously failed.

I recommend you run the following to ensure your system is up to date.

```
git -C ~/BirdNET-Pi checkout -f
git -C ~/BirdNET-Pi pull
update_birdnet.sh
```

Since `apprise` should have been installed during the update, but did not, run these commands _if_ the commands above don't address your issue:
```
~/BirdNET-Pi/birdnet/bin/pip3 install apprise
```

If the steps above to not work to solve the issue, please let me know and include any relevant log excepts or on-screen errors you encounter.

My best,
Patrick

filikun, 2022-05-15:
The first step seems to have worked well, thanks! It does not seem however that my instance have started listening on the mic. If I check the log it looks like it's analysing things from yesterday (14th) and I can't see any new entries to "todays detections". Is it so that it has some kind of backlog it works on?

The log updates all the time but here is a clip:

```
18:49:14 server.py[367]: [NEW CONNECTION] ('127.0.0.1', 46596) connected.
18:49:14 server.py[367]: [ACTIVE CONNECTIONS] 1
18:49:14 server.py[367]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
18:49:14 server.py[367]: FULL FILENAME: -BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-16:05:13.wav-
18:49:14 server.py[367]: Date-time: 2022-05-14 16:05:13
18:49:19 birdnet_analysis.sh[10161]: 0.0;3.0-('Abeillia abeillei_Smaragdstrupe', nan)
18:49:19 birdnet_analysis.sh[10161]: 3.0;6.0-('Abeillia abeillei_Smaragdstrupe', nan)
18:49:19 birdnet_analysis.sh[10161]: 6.0;9.0-('Abeillia abeillei_Smaragdstrupe', nan)
18:49:19 birdnet_analysis.sh[10161]: 9.0;12.0-('Abeillia abeillei_Smaragdstrupe', nan)
18:49:19 birdnet_analysis.sh[10161]: 12.0;15.0-('Abeillia abeillei_Smaragdstrupe', nan)
18:49:19 birdnet_analysis.sh[368]: RECORDING_LENGTH set to 15
18:49:19 birdnet_analysis.sh[368]: analyze.py --i BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-16:05:28.wav --o BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-16:05:28.wav.csv --lat 56.1411 --lon 12.7864 --week 19 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
18:49:20 server.py[367]: ANALYZING AUDIO... DONE! Time 4.4 SECONDS
18:49:20 server.py[367]: WRITING RESULTS TO BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-16:05:13.wav.csv ... DONE! WROTE 0 RESULTS.
18:49:20 server.py[367]: [NEW CONNECTION] ('127.0.0.1', 46598) connected.
18:49:20 server.py[367]: [ACTIVE CONNECTIONS] 1
18:49:22 server.py[367]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
18:49:22 server.py[367]: FULL FILENAME: -BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-16:05:28.wav-
18:49:22 server.py[367]: Date-time: 2022-05-14 16:05:28
18:49:26 birdnet_analysis.sh[10213]: 0.0;3.0-('Abeillia abeillei_Smaragdstrupe', nan)
18:49:26 birdnet_analysis.sh[10213]: 3.0;6.0-('Abeillia abeillei_Smaragdstrupe', nan)
18:49:26 birdnet_analysis.sh[10213]: 6.0;9.0-('Abeillia abeillei_Smaragdstrupe', nan)
18:49:26 birdnet_analysis.sh[10213]: 9.0;12.0-('Abeillia abeillei_Smaragdstrupe', nan)
18:49:26 birdnet_analysis.sh[10213]: 12.0;15.0-('Abeillia abeillei_Smaragdstrupe', nan)
18:49:26 birdnet_analysis.sh[368]: RECORDING_LENGTH set to 15
18:49:27 birdnet_analysis.sh[368]: analyze.py --i BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-16:05:42.wav --o BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-16:05:42.wav.csv --lat 56.1411 --lon 12.7864 --week 19 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
18:49:27 server.py[367]: ANALYZING AUDIO... DONE! Time 3.8 SECONDS
18:49:27 server.py[367]: WRITING RESULTS TO BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-16:05:28.wav.csv ... DONE! WROTE 0 RESULTS.
18:49:27 server.py[367]: [NEW CONNECTION] ('127.0.0.1', 46600) connected.
18:49:27 server.py[367]: [ACTIVE CONNECTIONS] 1
18:49:27 server.py[367]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
18:49:27 server.py[367]: FULL FILENAME: -BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-16:05:42.wav-
18:49:27 server.py[367]: Date-time: 2022-05-14 16:05:42
18:49:31 birdnet_analysis.sh[10353]: 0.0;3.0-('Abeillia abeillei_Smaragdstrupe', nan)
18:49:31 birdnet_analysis.sh[10353]: 3.0;6.0-('Abeillia abeillei_Smaragdstrupe', nan)
18:49:31 birdnet_analysis.sh[10353]: 6.0;9.0-('Abeillia abeillei_Smaragdstrupe', nan)
18:49:31 birdnet_analysis.sh[10353]: 9.0;12.0-('Abeillia abeillei_Smaragdstrupe', nan)
18:49:31 birdnet_analysis.sh[10353]: 12.0;15.0-('Abeillia abeillei_Smaragdstrupe', nan)
18:49:31 birdnet_analysis.sh[368]: RECORDING_LENGTH set to 15
18:49:31 birdnet_analysis.sh[368]: analyze.py --i BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-16:05:57.wav --o BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-16:05:57.wav.csv --lat 56.1411 --lon 12.7864 --week 19 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
18:49:31 server.py[367]: ANALYZING AUDIO... DONE! Time 3.9 SECONDS
18:49:31 server.py[367]: WRITING RESULTS TO BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-16:05:42.wav.csv ... DONE! WROTE 0 RESULTS.
18:49:31 server.py[367]: [NEW CONNECTION] ('127.0.0.1', 46604) connected.
18:49:31 server.py[367]: [ACTIVE CONNECTIONS] 1
18:49:32 server.py[367]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
18:49:32 server.py[367]: FULL FILENAME: -BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-16:05:57.wav-
18:49:32 server.py[367]: Date-time: 2022-05-14 16:05:57
```

mcguirepr89, 2022-05-15:
It is definitely working on a large queue of files that haven't yet been processed yet, which were not analyzed due to the missing `apprise` module.
It will take a while for that to catch up (a long while), but if I were you I would just keep an eye on the logs, checking every few hours to see how things are coming along, taking note of how quickly it is catching up.

I'll try to update you, too, on improvements to the update routine since a failure in the update tool was ultimately what introduced the bug on your system.

filikun, 2022-05-15:
> It is definitely working on a large queue of files that haven't yet been processed yet, which were not analyzed due to the missing `apprise` module. It will take a while for that to catch up (a long [...]

Thanks a lot! Would the spectrogram be black until it caches up? And non of the analysed results is updating in the statistics.

filikun, 2022-05-15:
[image]

```
20:03:09 server.py[367]: WRITING RESULTS TO BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-19:41:28.wav.csv ... DONE! WROTE 0 RESULTS.
20:03:09 server.py[367]: [NEW CONNECTION] ('127.0.0.1', 48446) connected.
20:03:09 server.py[367]: [ACTIVE CONNECTIONS] 1
20:03:10 server.py[367]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
20:03:10 server.py[367]: FULL FILENAME: -BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-19:41:43.wav-
20:03:10 server.py[367]: Date-time: 2022-05-14 19:41:43
20:03:14 birdnet_analysis.sh[110972]: 0.0;3.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:14 birdnet_analysis.sh[110972]: 3.0;6.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:14 birdnet_analysis.sh[110972]: 6.0;9.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:14 birdnet_analysis.sh[110972]: 9.0;12.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:14 birdnet_analysis.sh[110972]: 12.0;15.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:15 birdnet_analysis.sh[368]: RECORDING_LENGTH set to 15
20:03:15 birdnet_analysis.sh[368]: analyze.py --i BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-19:41:58.wav --o BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-19:41:58.wav.csv --lat 56.1411 --lon 12.7864 --week 19 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
20:03:15 server.py[367]: ANALYZING AUDIO... DONE! Time 4.8 SECONDS
20:03:15 server.py[367]: WRITING RESULTS TO BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-19:41:43.wav.csv ... DONE! WROTE 0 RESULTS.
20:03:15 server.py[367]: [NEW CONNECTION] ('127.0.0.1', 48448) connected.
20:03:15 server.py[367]: [ACTIVE CONNECTIONS] 1
20:03:16 server.py[367]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
20:03:16 server.py[367]: FULL FILENAME: -BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-19:41:58.wav-
20:03:16 server.py[367]: Date-time: 2022-05-14 19:41:58
20:03:20 birdnet_analysis.sh[111115]: 0.0;3.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:20 birdnet_analysis.sh[111115]: 3.0;6.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:20 birdnet_analysis.sh[111115]: 6.0;9.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:20 birdnet_analysis.sh[111115]: 9.0;12.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:20 birdnet_analysis.sh[111115]: 12.0;15.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:20 birdnet_analysis.sh[368]: RECORDING_LENGTH set to 15
20:03:20 birdnet_analysis.sh[368]: analyze.py --i BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-19:42:13.wav --o BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-19:42:13.wav.csv --lat 56.1411 --lon 12.7864 --week 19 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
20:03:21 server.py[367]: ANALYZING AUDIO... DONE! Time 4.4 SECONDS
20:03:21 server.py[367]: WRITING RESULTS TO BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-19:41:58.wav.csv ... DONE! WROTE 0 RESULTS.
20:03:21 server.py[367]: [NEW CONNECTION] ('127.0.0.1', 48450) connected.
20:03:21 server.py[367]: [ACTIVE CONNECTIONS] 1
20:03:21 server.py[367]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
20:03:21 server.py[367]: FULL FILENAME: -BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-19:42:13.wav-
20:03:21 server.py[367]: Date-time: 2022-05-14 19:42:13
20:03:25 birdnet_analysis.sh[111255]: 0.0;3.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:25 birdnet_analysis.sh[111255]: 3.0;6.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:25 birdnet_analysis.sh[111255]: 6.0;9.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:25 birdnet_analysis.sh[111255]: 9.0;12.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:25 birdnet_analysis.sh[111255]: 12.0;15.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:25 birdnet_analysis.sh[368]: RECORDING_LENGTH set to 15
20:03:25 birdnet_analysis.sh[368]: analyze.py --i BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-19:42:28.wav --o BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-19:42:28.wav.csv --lat 56.1411 --lon 12.7864 --week 19 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
20:03:25 server.py[367]: ANALYZING AUDIO... DONE! Time 3.9 SECONDS
20:03:25 server.py[367]: WRITING RESULTS TO BirdSongs/May-2022/14-Saturday/2022-05-14-birdnet-19:42:13.wav.csv ... DONE! WROTE 0 RESULTS.
20:03:25 server.py[367]: [NEW CONNECTION] ('127.0.0.1', 48452) connected.
20:03:25 server.py[367]: [ACTIVE CONNECTIONS] 1
20:03:26 server.py[367]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
```

mcguirepr89, 2022-05-15:
Those indicate a time where there was no sound in the audio files (perhaps from when you needed to unplug it?)

But if you are continuing to see 
```
20:03:25 birdnet_analysis.sh[111255]: 0.0;3.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:25 birdnet_analysis.sh[111255]: 3.0;6.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:25 birdnet_analysis.sh[111255]: 6.0;9.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:25 birdnet_analysis.sh[111255]: 9.0;12.0-('Abeillia abeillei_Smaragdstrupe', nan)
20:03:25 birdnet_analysis.sh[111255]: 12.0;15.0-('Abeillia abeillei_Smaragdstrupe', nan)
```
as your results, then the analysis is mostly a waste of time, as those files are empty. This is also indicated by black (empty) spectrograms. This all means that files are being made (so no issues with the services, they all think things are going smoothly), but your input device is not providing any data. Not sure what could be going on there.

If you want to remove the backlog and try to start over, I would do it in the terminal via:
```
stop_core_services.sh
rm -drfv ~/BirdSongs/May-2022/14-Saturday
rm -drfv ~/BirdSongs/May-2022/15-Sunday
sudo reboot
```
then check when the system comes back up that the "Currently Analyzing" spectrogram on the "Overview" page is showing audio data.

To do the same things on the web interface:
- "Tools" > "Services" > "Stop Core Services"
- "Tools" > "File Manager"
- Locate in `$HOME/BirdSongs/May-2022` the `14-Saturday` and `15-Sunday` directories and delete them.
- "Tools" > "System Controls" > "Reboot"

filikun, 2022-05-15:
> Those indicate a time where there was no sound in the audio files (perhaps from when you needed to unplug it?)
>
> But if you are continuing to see
>
> ```
> 20:03:25 birdnet_analysis.sh[111255]: [...]

Thanks a lot, it's now up and running as normal!

mcguirepr89, 2022-05-15:
Thanks -- I'm going to keep this open for others since there seems to be something buggy going on somewhere. At the very least I promised you updates about changes to the Update routine -- [I'm soliciting help from the community](https://github.com/mcguirepr89/BirdNET-Pi/discussions/282) since I'm just making this all up as I go and bet there is someone out there who knows more than I.

My best,
Patrick