mcguirepr89/BirdNET-Pi #280: Sox warnings - Of any concern?

jmherbst, 2022-05-15:
Hello, 

I often see a lot of `sox WARN...` logs in my `/var/log/syslog` - Should these be of any concern?

Here is a full analysis snipped ( I think? ) 
```
May 15 12:37:27 birdnetpi systemd[1]: Started BirdNET Analysis.
May 15 12:37:28 birdnetpi birdnet_analysis.sh[689908]: Files loaded
May 15 12:37:29 birdnetpi birdnet_analysis.sh[689908]: RECORDING_LENGTH set to 15
May 15 12:37:31 birdnetpi env[689971]: Line = 2022-05-15-birdnet-12:37:04.wav.csv
May 15 12:37:31 birdnetpi env[689971]: Line = 9.0;12.0;Pipilo erythrophthalmus;Eastern Towhee;0.8791912
May 15 12:37:31 birdnetpi env[689971]: NEWFILE=Eastern_Towhee-88-2022-05-15-birdnet-12:37:04.mp3
May 15 12:37:31 birdnetpi env[690038]: sox WARN trim: End position is after expected end of audio.
May 15 12:37:31 birdnetpi env[690038]: sox WARN trim: Last 1 position(s) not reached.
May 15 12:37:32 birdnetpi systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 6096.
May 15 12:37:32 birdnetpi systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
May 15 12:37:32 birdnetpi systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
May 15 12:37:32 birdnetpi spectrogram.sh[690042]: sox WARN wav: Premature EOF on .wav input file
May 15 12:37:32 birdnetpi systemd[1]: spectrogram_viewer.service: Succeeded.
May 15 12:37:35 birdnetpi birdnet_analysis.sh[689908]: analyze.py --i /home/birder/BirdSongs/May-2022/15-Sunday/2022-05-15-birdnet-12:37:19.wav --o /home/birder/BirdSongs/May-2022/15-Sunday/2022-05-15-birdnet-12:37:19.wav.csv --lat 33.9721 --lon -84.4406 --week 19 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7 --birdweather_id "IN_USE"
May 15 12:37:35 birdnetpi server.py[388]: ANALYZING AUDIO... DONE! Time 1.6 SECONDS
May 15 12:37:35 birdnetpi server.py[388]: WRITING RESULTS TO /home/birder/BirdSongs/May-2022/15-Sunday/2022-05-15-birdnet-12:37:04.wav.csv ... DONE! WROTE 1 RESULTS.
May 15 12:37:35 birdnetpi server.py[388]: 2022-05-15;12:37:04;Pipilo erythrophthalmus;Eastern Towhee;0.8791912;33.9721;-84.4406;0.7;19;1.25;0.0Eastern_Towhee-0.8791912-2022-05-15-birdnet-12:37:04.mp3
May 15 12:37:35 birdnetpi server.py[388]: Soundscape POST Response Status -  201
May 15 12:37:35 birdnetpi server.py[388]: { "timestamp": "2022-05-15T12:37:13-04:00","lat": 33.9721,"lon": -84.4406,"soundscapeId": 9056338,"soundscapeStartTime": 9.0,"soundscapeEndTime": 12.0,"commonName": "Eastern Towhee","scientificName": "Pipilo erythrophthalmus","algorithm": "alpha","confidence": 0.8791912 }
May 15 12:37:35 birdnetpi server.py[388]: Detection POST Response Status -  201
May 15 12:37:35 birdnetpi server.py[388]: [NEW CONNECTION] ('127.0.0.1', 47790) connected.
May 15 12:37:35 birdnetpi server.py[388]: [ACTIVE CONNECTIONS] 1
May 15 12:37:35 birdnetpi server.py[388]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
May 15 12:37:35 birdnetpi server.py[388]: FULL FILENAME: -/home/birder/BirdSongs/May-2022/15-Sunday/2022-05-15-birdnet-12:37:19.wav-
May 15 12:37:35 birdnetpi server.py[388]: Date-time: 2022-05-15 12:37:19
May 15 12:37:37 birdnetpi birdnet_analysis.sh[690062]: 0.0;3.0-('Piranga olivacea_Scarlet Tanager', 0.075196825)
May 15 12:37:37 birdnetpi birdnet_analysis.sh[690062]: 3.0;6.0-('Icterus galbula_Baltimore Oriole', 0.08173101)
May 15 12:37:37 birdnetpi birdnet_analysis.sh[690062]: 6.0;9.0-('Pipilo erythrophthalmus_Eastern Towhee', 0.49324965)
May 15 12:37:37 birdnetpi birdnet_analysis.sh[690062]: 9.0;12.0-('Pipilo erythrophthalmus_Eastern Towhee', 0.21082132)
May 15 12:37:37 birdnetpi birdnet_analysis.sh[690062]: 12.0;15.0-('Buteo platypterus_Broad-winged Hawk', 0.07050251)
May 15 12:37:37 birdnetpi systemd[1]: birdnet_analysis.service: Succeeded.
May 15 12:37:39 birdnetpi systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 4180.
May 15 12:37:39 birdnetpi systemd[1]: Stopped BirdNET Analysis.
```

mcguirepr89, 2022-05-15:
Those can be safely ignored. They're just stating the extraction's END (and sometimes START) point is impossible, so it uses the end of the file (or beginning) instead. But I'm going to look into them further since `sox` shouldn't need to do that, but obviously it is, so that sucks :/

mcguirepr89, 2022-05-15:
I think I see now why that warning is allowed to get generated -- the default `position` syntax for `trim` is `+`, so I think `=` needs to be explicitly specified on line 120 of `extract_new_birdsounds.sh`

I'm testing that now and will update you as to whether that makes those warnings cease.

Thanks for pointing this out. I've noticed it before, but have not looked very closely at the manual for `sox` on those specific calls. Hopefully this will have identified the origin of those warnings.

mcguirepr89, 2022-05-15:
So far so good on #284:

You can see when I made the change on my system:
[image]

jmherbst, 2022-05-15:
I've just updated, and I'm still seeing the `Premature EOF on .wav` warning.

```
May 15 16:27:43 birdnetpi systemd[1]: Started BirdNET Analysis.
May 15 16:27:44 birdnetpi birdnet_analysis.sh[851136]: Files loaded
May 15 16:27:44 birdnetpi birdnet_analysis.sh[851136]: RECORDING_LENGTH set to 15
May 15 16:27:44 birdnetpi env[851172]: Line = 2022-05-15-birdnet-16:27:20.wav.csv
May 15 16:27:47 birdnetpi systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 7439.
May 15 16:27:47 birdnetpi systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
May 15 16:27:47 birdnetpi systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
May 15 16:27:47 birdnetpi spectrogram.sh[851218]: sox WARN wav: Premature EOF on .wav input file
May 15 16:27:47 birdnetpi systemd[1]: spectrogram_viewer.service: Succeeded.
May 15 16:27:51 birdnetpi birdnet_analysis.sh[851136]: analyze.py --i /home/birder/BirdSongs/May-2022/15-Sunday/2022-05-15-birdnet-16:27:35.wav --o /home/birder/BirdSongs/May-2022/15-Sunday/2022-05-15-birdnet-16:27:35.wav.csv --lat 33.9721 --lon -84.4406 --week 19 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7 --birdweather_id "IN_USE"
May 15 16:27:51 birdnetpi server.py[388]: ANALYZING AUDIO... DONE! Time 1.6 SECONDS
May 15 16:27:51 birdnetpi server.py[388]: WRITING RESULTS TO /home/birder/BirdSongs/May-2022/15-Sunday/2022-05-15-birdnet-16:27:20.wav.csv ... DONE! WROTE 0 RESULTS.
May 15 16:27:51 birdnetpi server.py[388]: [NEW CONNECTION] ('127.0.0.1', 49746) connected.
May 15 16:27:51 birdnetpi server.py[388]: [ACTIVE CONNECTIONS] 1
May 15 16:27:51 birdnetpi server.py[388]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
May 15 16:27:51 birdnetpi server.py[388]: FULL FILENAME: -/home/birder/BirdSongs/May-2022/15-Sunday/2022-05-15-birdnet-16:27:35.wav-
May 15 16:27:51 birdnetpi server.py[388]: Date-time: 2022-05-15 16:27:35
May 15 16:27:53 birdnetpi birdnet_analysis.sh[851248]: 0.0;3.0-('Human_Human', 0.0)
May 15 16:27:53 birdnetpi birdnet_analysis.sh[851248]: 3.0;6.0-('Corvus ossifragus_Fish Crow', 0.1088708)
May 15 16:27:53 birdnetpi birdnet_analysis.sh[851248]: 6.0;9.0-("Limnothlypis swainsonii_Swainson's Warbler", 0.10364607)
May 15 16:27:53 birdnetpi birdnet_analysis.sh[851248]: 9.0;12.0-('Melanerpes carolinus_Red-bellied Woodpecker', 0.031417135)
May 15 16:27:53 birdnetpi birdnet_analysis.sh[851248]: 12.0;15.0-('Turdus migratorius_American Robin', 0.19701256)
May 15 16:27:53 birdnetpi systemd[1]: birdnet_analysis.service: Succeeded.
May 15 16:27:55 birdnetpi systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 5103.
May 15 16:27:55 birdnetpi systemd[1]: Stopped BirdNET Analysis.
```

Any concern for that one?

mcguirepr89, 2022-05-15:
Still not concerned about that one, but concerned as to whether the update did what it should have.

Can you verify that line 120 has the `=` signs to the left of START and END like this `=${START} and this `=${END}`

```
nl -ba scripts/extract_new_birdsounds.sh  | grep 120
```

jmherbst, 2022-05-15:
Yep, the update brought those in:
```
birdnetpi:~/BirdNET-Pi$ nl -ba scripts/extract_new_birdsounds.sh  | grep 120
   120	      trim ="${START}" ="${END}"
```

jmherbst, 2022-05-16:
Nope! Haven't seen either of those since the `extract_new_birdsounds.sh` change

jmherbst, 2022-05-16:
great to hear! 🙏

mcguirepr89, 2022-05-16:
[I went ahead and did that](https://github.com/mcguirepr89/BirdNET-Pi/commit/57778d405f668cf25c58a038bf743e83ab37afd4) so you should no longer need to ignore the warnings, they will simply no longer be reported.