mcguirepr89/BirdNET-Pi #289: current version of Installer does not work

DD4WH, 2022-05-18:
Serveral errors during the installation:

[image]

[image]

[image]

mgzob, 2022-05-18:
I hit this wall yesterday. I -believe- this is the full set of commands required to unblock the install.

> sudo apt install -y debian-keyring debian-archive-keyring apt-transport-https
> curl -1sLf 'https://dl.cloudsmith.io/public/caddy/stable/gpg.key' | sudo gpg --dearmor -o /usr/share/keyrings/caddy-stable-archive-keyring.gpg
> curl -1sLf 'https://dl.cloudsmith.io/public/caddy/stable/debian.deb.txt' | sudo tee /etc/apt/sources.list.d/caddy-stable.list
> sudo apt update
> sudo apt install caddy
> sudo apt-get install python3-venv python3-dev sqlite3

Note: Some are not required; I believe debian-keyring / others may already be sorted? I also don't remember if I added `icecast2` or not. If you do include icecast2, you don't need to do any configuration during install.

mcguirepr89, 2022-05-18:
Hello!

There is an issue upstream with Caddy's newest GPG key signature -- it will need to get sorted out upstream. It likely works sometimes and not others due to DNS caching of the old key signature location since the URL is the same. I suspect this glitch will not last long. It's hard to say when it started, but likely in the last few days shortly after the version update from caddy. I was able to install caddy today using the installation excerpt @mgzob mentions, so it _seems_ that is has been fixed. Please let me know if you still see the error on your next installation attempt -- if so, I may look into mimicking their setup script within the BirdNET-Pi installer and then include the `caddy` binary to have one less thing "upstream" on which we must depend for stability.

DD4WH, 2022-05-18:
Perfect!
Thank you, @mgzob and @mcguirepr89 

* I issued all the commands in this comment https://github.com/mcguirepr89/BirdNET-Pi/issues/289#issuecomment-1130273053
* and then started the usual one-line-installer 
* It took two attempts (the first attempt gave a network error after some minutes), not sure why . . . maybe a network failure here at home.
* now it works!

mcguirepr89, 2022-05-18:
Caddy updated the documentation for Debian installations, so [this](https://github.com/mcguirepr89/BirdNET-Pi/commit/354fce1aaf69b1f6f5a6d4a55689ce10e5f812db) possibly fixed the installer. Let me know!

DD4WH, 2022-05-19:
Thanks a lot, Patrick !!! With the latest installer it works perfectly as usual with the one-line-installer and without the additional commands!

Thanks also, @mgzob for your help and the commands, which also worked when I installed the former version yesterday.

closed 2022-05-19 by DD4WH