mcguirepr89/BirdNET-Pi #293: Feature Request - Extra notification info

lloydbayley, 2022-05-19:
Here's a thought. What about being able to get the Total, Today, Last Hour, Species Detected Today and Total Number of Species as notifications. For example, to be able to bung them into MQTT to use.

ehpersonal38, 2022-05-19:
Great solution Patrick!

lloydbayley, 2022-05-19:
You mean like the one I finished about an hour ago? Hehe

var_detect_last_hour=$(sqlite3 $HOME/BirdNET-Pi/scripts/birds.db "SELECT COUNT(*) FROM detections WHERE date='""$(date '+%Y-%m-%d')'"" AND Time >=datetime('now', '-1 Hour')>
var_detect_today=$(sqlite3 $HOME/BirdNET-Pi/scripts/birds.db "SELECT COUNT(*) FROM detections WHERE date='""$(date '+%Y-%m-%d')'"";")
var_unique_today=$(sqlite3 $HOME/BirdNET-Pi/scripts/birds.db "SELECT COUNT(DISTINCT Com_Name) FROM detections WHERE date='""$(date '+%Y-%m-%d')'"";")
var_total_detections=$(sqlite3 $HOME/BirdNET-Pi/scripts/birds.db "SELECT COUNT(*) FROM detections;")
var_total_species=$(sqlite3 $HOME/BirdNET-Pi/scripts/birds.db "SELECT COUNT(DISTINCT Com_Name) FROM detections;")

$HOME/BirdNET-Pi/birdnet/bin/apprise -vv -b "${var_detect_last_hour}" mqtt://10.61.2.101/MyBirdNetPi/Detections_Last_Hour
$HOME/BirdNET-Pi/birdnet/bin/apprise -vv -b "${var_detect_today}" mqtt://10.61.2.101/MyBirdNetPi/Total_Detections_Today
$HOME/BirdNET-Pi/birdnet/bin/apprise -vv -b "${var_unique_today}" mqtt://10.61.2.101/MyBirdNetPi/Unique_Detections_Today
$HOME/BirdNET-Pi/birdnet/bin/apprise -vv -b "${var_total_detections}" mqtt://10.61.2.101/MyBirdNetPi/Total_Detections
$HOME/BirdNET-Pi/birdnet/bin/apprise -vv -b "${var_total_species}" mqtt://10.61.2.101/MyBirdNetPi/Total_Species

It's not as tiddly-tidy as yours but it does the trick.

A agree that that sort of thing to allow topics with depth would be nice in API fashion and also perhaps native JSON for added "datapullability" (not really a word but it does the job)!

lloydbayley, 2022-05-20:
Turns out I cocked the last_hour one up by forgetting the other half of the syntax. Your line worked perfectly (as I expected). Thanks!

closed 2022-05-24 by mcguirepr89