mcguirepr89/BirdNET-Pi #302: sound recording via pulse breaks every few hours

hoover67, 2022-05-22:
Hi folks, 

first of all thanks for creating this fantastic project! 

I've installed via curl on my new pi400, everything works nicely but every few hours, the sound recording will break and detections will become impossible as a result. I'm seeing these messages in daemon.log: 

3:59:02 birdnetpi [livestream.sh](http://livestream.sh/)[663927]: default: Input/output error
May 21 00:00:01 birdnetpi birdnet_recording.sh[218986]: Failed to create secure directory (/run/user/1000/pulse): No such file or directory

Sometimes the system starts working again by itself, and a reboot usually fixes the issue as well. 

Any idea what could be causing this? Thanks! 

Uwe 

PS: USB recording device is a "gaming / podcast microphone": 

birdnetpi:~/BirdNET-Pi$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 04d9:0007 Holtek Semiconductor, Inc. Raspberry Pi Internal Keyboard
Bus 001 Device 003: ID 4c4a:4155 Jieli Technology UACDemoV1.0
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

mcguirepr89, 2022-05-22:
Hi, @hoover67 and thank you so much for your kind praise of the system. Thank you also for including the relevant log excerpts from what you've observed -- it's always very helpful and appreciated. Since I haven't seen this error on other systems, I would like to begin by troubleshooting the hardware.

By any chance, do you have another microphone that can be used to test? If so, switch them out and let it run for a day while monitoring for error.

hoover67, 2022-05-22:
Hi Patrick, 

I may be "seeing ghosts" but it looks like the sound recording breaks when I log out of the system (connected via ssh). I don't know if some form of power saving may be active on the pi400 which shuts down the USB port or something similar... Very weird. 

I don't have a 2nd USB microphone at the moment but I'll ask around and try another device. Two buddies of mine also run birdnet pi using the same microphone and on their end there are no issues. 

As a workaround I've now written a little shell script that'll reboot the pi400 whenever exessive audio Input/output errors are found in /var/log/daemon.log. Also, I've connected the microphone to  a different USB port on the pi now, let's hope it helps. 

Thanks again for your work on this project, it's much appreciated! 

All the best, Uwe

mcguirepr89, 2022-05-22:
>I may be "seeing ghosts" but it looks like the sound recording breaks when I log out of the system (connected via ssh). I don't know if some form of power saving may be active on the pi400 which shuts down the USB port or something similar... Very weird.

Super helpful observation!! There is a chance that your installation (for reasons still unknown) may not be configured for autologin on tty1. There does need to be a user logged in for `pulse` to work, though the RaspiOS enables an autologin for the default user on tty1.

To be _totally_ sure that the autologin is properly configured, you can run through the steps below and we'll cross our fingers that you see an improvement:
```
sudo raspi-config
```
[image]

[image]

[image]

hoover67, 2022-05-22:
Thanks for the hint! I've now configured the auto-login using raspi-config, although on my end it's listed as the "pi" user, not "birdie" as in your example. All of the birdnet stuff runs as the "pi" user so I guess that's ok. 

Fingers crossed this'll work now :-) 

All the best, Uwe

mcguirepr89, 2022-05-22:
>PS: I should have noted I installed the pi400 "headless" using the ssh touch file on the boot partition. I had to jump through quite a few hoops because the default user pi / raspberry would not work for some weird reason, that may also have something to do with it.

This is interesting :thinking: The current [installation guide](https://github.com/mcguirepr89/BirdNET-Pi/wiki/Installation-Guide) will configure a totally headless system and can be configured with a non-`pi` user (which is a recommended security precaution) -- let me know any other places that your installation procedure differed from the guide and we might learn a bit more about what is going on (or not).

hoover67, 2022-05-22:
I downloaded the recommended distro, dd'ed it to an sd card (32GB), created the "ssh" touch file and tried to log into the system using the IP provided by my Fritzbox.

This would not work (pi / raspberry) as the /etc/shadow entry for "pi" had an "*" in the password column. 

I created a known hash on my laptop (running mint 20) and pasted that to the /etc/shadow file on the sd card, this way I was able to log in using a known password. 

I then ran the installer for birdnet (which apparently was broken for a few weeks) and had to fiddle with all sorts of settings in order to get the components to work. The last uninstall / reinstall I did was this morning. (I received the pi400 last Thursday which also was the day of the first birdnet installation attempt). 

Is there an easy way to verify the "auto console" login config has succeeded? 

Sorry for bothering you with my pi400 install troubles, I do realize it's not exactly a birdnet-pi related issue :)

All the best, 

Uwe

hoover67, 2022-05-22:
Closing the issue as it's apparently an installation problem with my pi400. Thanks for the support & all the best, 

Uwe

closed 2022-05-22 by hoover67

cathcartd, 2022-07-02:
Note i had a similar issue. The main difference in my setup was i didn't create a user with the rpi imager, and i didn't set login via password over ssh. What i saw was that pulseaudio was only running when i was ssh'd in. The console autologin as the pi user didn't seem to work (not sure if that's a good idea security wise - though i guess if you have physical access to the console you probably have physical access to the sd card). Pulseaudio would exit cleanly when the ssh session was dropped. Then when not ssh'd in i would see log entries indicating difficulty autospawning pulseaudio

```
Jul  2 15:01:31 birdnetpi birdnet_recording.sh[1950158]: + pulseaudio --start
Jul  2 15:01:31 birdnetpi birdnet_recording.sh[1950163]: E: [autospawn] core-util.c: Failed to create secure directory (/run/user/1000/pulse): No such file or directory
Jul  2 15:01:31 birdnetpi birdnet_recording.sh[1950163]: W: [autospawn] lock-autospawn.c: Cannot access autospawn lock.
Jul  2 15:01:31 birdnetpi birdnet_recording.sh[1950163]: E: [pulseaudio] main.c: Failed to acquire autospawn lock
...
Jul  2 15:01:31 birdnetpi livestream.sh[1950161]: Failed to create secure directory (/run/user/1000/pulse): No such file or directory
Jul  2 15:01:31 birdnetpi livestream.sh[1950161]: [alsa @ 0x5592cad050] cannot open audio device default (No such file or directory)
...
Jul  2 15:01:37 birdnetpi birdnet_recording.sh[1950348]: + arecord -f S16_LE -c2 -r48000 -t wav --max-file-time 15 -D default --use-strftime /home/pi/BirdSongs/%B-%Y/%d-%A/%F-birdnet-%H:%M:%S.wav
Jul  2 15:01:37 birdnetpi birdnet_recording.sh[1950356]: Failed to create secure directory (/run/user/1000/pulse): No such file or directory
Jul  2 15:01:37 birdnetpi birdnet_recording.sh[1950356]: arecord: main:830: audio open error: No such file or directo
```
I was going to try moving pulse to [system wide](https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/SystemWide/) - but actually just disabling the per-user service (`sudo systemctl --global disable pulseaudio.service pulseaudio.socket`) seems to have allowed the autostart client to work without issue.