mcguirepr89/BirdNET-Pi #303: No website login possible after changing password

oliverk71, 2022-05-22:
Is there maybe a maximum password length for the website? I changed the password to another one and cannot login since then. I also tried the old password. My (new) password is about 20 characters long.

oliverk71, 2022-05-22:
Don't know what happened, but it seems I killed the caddy server. But I can still login via SSH.

ehpersonal38, 2022-05-22:
You can always check `$HOME/BirdNET-Pi/birdnet.conf` to see if CADDY_PWD is what you expect.

oliverk71, 2022-05-22:
Interesting! My password contained a special character ('$'), but this and the following two characters were not recognized! 
However, I cannot reach the website. I edited the Caddyfile as I did so many times, I guess this is still possible?

oliverk71, 2022-05-22:
May it be possible that my special password killed caddy?

oliverk71, 2022-05-22:
I tried to `sudo update_caddyfile.sh` and received the following output (I edited password to multiple X): 

```
+ '[' -d /etc/caddy ']'
+ '[' -f /etc/caddy/Caddyfile ']'
+ cp /etc/caddy/Caddyfile /etc/caddy/Caddyfile.original
+ '[' -z KHLpmc892pnbewR47 ']'
++ caddy hash-password -plaintext XXXXXXXXXXXXXXXX
+ HASHWORD=JDJhJDE0JDRtclRNR1RvTnp0blVvaFJLOGEzYnVvcGp2dmdzWTNja0V0b2VyeUNDL3JVdkRRVi9ET25h
+ cat
++ hostname
+ sudo systemctl reload caddy
caddy.service is not active, cannot reload.
```

Not sure if this can have to do something with the special character in the password I entered and the calculated hash, which maybe does not match, anymore?

ehpersonal38, 2022-05-22:
#276 should've fixed this issue. Can you verify you've updated in the last 7 days, or if not, do so and try again.

oliverk71, 2022-05-22:
I did a clean install, today. 

Okay, I did `update_birdnet.sh` and received the same error which made me consider a clean install before...

```
**fatal: pathspec 'privacy_server.py' did not match any files**
Saved working directory and index state WIP on main: c1cfc98 Merge pull request #301 from lamkin/drl/crontab-redirection
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Already up to date.
On branch main
Your branch is up to date with 'origin/main'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   ../model/labels.txt
	modified:   analyze.py
	modified:   daily_plot.py
	modified:   plotly_streamlit.py
	modified:   server.py

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	../BirdDB.txt
	../HUMAN.txt
	../analyzing_now.txt
	../apprise.txt
	../birdnet/
	../exclude_species_list.txt
	../firstrun.ini
	../include_species_list.txt
	birds.db
	exclude_species_list.txt
	include_species_list.txt
	labels.txt
	lastrun.txt
	thisrun.txt
	../templates/avahi-alias@.service
	../templates/birdnet_analysis.service
	../templates/birdnet_log.service
	../templates/birdnet_recording.service
	../templates/birdnet_server.service
	../templates/birdnet_stats.service
	../templates/chart_viewer.service
	../templates/custom_recording.service
	../templates/extraction.service
	../templates/livestream.service
	../templates/pushed_notifications.service
	../templates/spectrogram_viewer.service
	../templates/web_terminal.service

no changes added to commit (use "git add" and/or "git commit -a")
Dropped refs/stash@{0} (f14e7d36731238cf377ea31ae059620c3fad2d22)
```
`sudo service caddy restart` either takes long time or just does not work in my case.

oliverk71, 2022-05-22:
Output of `sudo service caddy restart`:
```
Job for caddy.service failed because a timeout was exceeded.
See "systemctl status caddy.service" and "journalctl -xe" for details.
```
Output of `systemclt status caddy.service`:
● caddy.service - Caddy
     Loaded: loaded (/lib/systemd/system/caddy.service; enabled; vendor preset>
     Active: failed (Result: timeout) since Sun 2022-05-22 22:37:48 CEST; 3min>
       Docs: https://caddyserver.com/docs/
    Process: 31866 ExecStart=/usr/bin/caddy run --environ --config /etc/caddy/>
   Main PID: 31866 (code=exited, status=0/SUCCESS)
        CPU: 1min 30.573s

May 22 22:37:48 birdnetpi-elsen caddy[31866]: {"level":"info","ts":1653251868.>
May 22 22:37:48 birdnetpi-elsen caddy[31866]: {"level":"info","ts":1653251868.>
May 22 22:37:48 birdnetpi-elsen caddy[31866]: {"level":"info","ts":1653251868.>
May 22 22:37:48 birdnetpi-elsen caddy[31866]: {"level":"info","ts":1653251868.>
May 22 22:37:48 birdnetpi-elsen caddy[31866]: {"level":"info","ts":1653251868.>
May 22 22:37:48 birdnetpi-elsen caddy[31866]: {"level":"info","ts":1653251868.>
May 22 22:37:48 birdnetpi-elsen caddy[31866]: {"level":"info","ts":1653251868.>
May 22 22:37:48 birdnetpi-elsen systemd[1]: caddy.service: Failed with result >
May 22 22:37:48 birdnetpi-elsen systemd[1]: Failed to start Caddy.
May 22 22:37:48 birdnetpi-elsen systemd[1]: caddy.service: Consumed 1min 30.57>

Output of `journalctl -xe`:

```
May 22 22:43:06 birdnetpi-elsen server.py[394]: WRITING RESULTS TO /home/olive>
May 22 22:43:06 birdnetpi-elsen server.py[394]: [NEW CONNECTION] ('127.0.0.1',>
May 22 22:43:06 birdnetpi-elsen server.py[394]: [ACTIVE CONNECTIONS] 1
May 22 22:43:06 birdnetpi-elsen server.py[394]: READING AUDIO DATA... DONE! RE>
May 22 22:43:06 birdnetpi-elsen server.py[394]: FULL FILENAME: -/home/oliver/B>
May 22 22:43:06 birdnetpi-elsen server.py[394]: Date-time: 2022-05-22 22:42:47
May 22 22:43:11 birdnetpi-elsen birdnet_analysis.sh[35339]: 0.0;3.0-('Coturnix>
May 22 22:43:11 birdnetpi-elsen birdnet_analysis.sh[35339]: 3.0;6.0-('Coturnix>
May 22 22:43:11 birdnetpi-elsen birdnet_analysis.sh[35339]: 6.0;9.0-('Coturnix>
May 22 22:43:11 birdnetpi-elsen birdnet_analysis.sh[35339]: 9.0;12.0-('Coturni>
May 22 22:43:11 birdnetpi-elsen birdnet_analysis.sh[35339]: 12.0;15.0-('Coturn>
May 22 22:43:11 birdnetpi-elsen systemd[1]: birdnet_analysis.service: Succeede>
░░ Subject: Unit succeeded
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit birdnet_analysis.service has successfully entered the 'dead' state.
May 22 22:43:11 birdnetpi-elsen systemd[1]: birdnet_analysis.service: Consumed>
░░ Subject: Resources consumed by unit runtime
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit birdnet_analysis.service completed and consumed the indicated reso>
lines 3467-3489/3489 (END)
May 22 22:43:06 birdnetpi-elsen server.py[394]: WRITING RESULTS TO /home/oliver/BirdSongs/May-2022/22-Sunday/2022-05-22-birdnet-22:42:32.wav.csv ... DONE! WROTE 0 RESULTS.
May 22 22:43:06 birdnetpi-elsen server.py[394]: [NEW CONNECTION] ('127.0.0.1', 47794) connected.
May 22 22:43:06 birdnetpi-elsen server.py[394]: [ACTIVE CONNECTIONS] 1
May 22 22:43:06 birdnetpi-elsen server.py[394]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
May 22 22:43:06 birdnetpi-elsen server.py[394]: FULL FILENAME: -/home/oliver/BirdSongs/May-2022/22-Sunday/2022-05-22-birdnet-22:42:47.wav-
May 22 22:43:06 birdnetpi-elsen server.py[394]: Date-time: 2022-05-22 22:42:47
May 22 22:43:11 birdnetpi-elsen birdnet_analysis.sh[35339]: 0.0;3.0-('Coturnix coturnix_Wachtel', 0.10662905)
May 22 22:43:11 birdnetpi-elsen birdnet_analysis.sh[35339]: 3.0;6.0-('Coturnix coturnix_Wachtel', 0.124791406)
May 22 22:43:11 birdnetpi-elsen birdnet_analysis.sh[35339]: 6.0;9.0-('Coturnix coturnix_Wachtel', 0.12570535)
May 22 22:43:11 birdnetpi-elsen birdnet_analysis.sh[35339]: 9.0;12.0-('Coturnix coturnix_Wachtel', 0.12613085)
May 22 22:43:11 birdnetpi-elsen birdnet_analysis.sh[35339]: 12.0;15.0-('Coturnix coturnix_Wachtel', 0.13719805)
May 22 22:43:11 birdnetpi-elsen systemd[1]: birdnet_analysis.service: Succeeded.
░░ Subject: Unit succeeded
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit birdnet_analysis.service has successfully entered the 'dead' state.
May 22 22:43:11 birdnetpi-elsen systemd[1]: birdnet_analysis.service: Consumed 1.404s CPU time.
░░ Subject: Resources consumed by unit runtime
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit birdnet_analysis.service completed and consumed the indicated resources.
```

ehpersonal38, 2022-05-22:
> Okay, I did update_birdnet.sh and received the same error which made me consider a clean install before...

Actually there's no error here - the first line that says "fatal" is just in reference to the fact that a file that it tried to delete was already deleted, harmless. @mcguirepr89 we should probably prevent this from showing, since it confused me too! 

As for the rest, hopefully Patrick can help soon.

mcguirepr89, 2022-05-23:
To perform the test without the web interface, do the following and please end me any output:
1. Connect via SSH
2. Edit the `birdnet.conf` file's `BIRDNETPI_URL` so that it is empty
3. Set the `CADDY_PWD` to `CADDY_PWD=test`
4. Stop the `caddy service`
5. Update the caddyfile using the script, but this time we will instruct it to `restart` caddy instead of `reload`ing it:
   ```
   sed 's/reload/restart/' ~/BirdNET-Pi/scripts/update_caddyfile.sh | sudo bash
   ```

Please relay you output here and after step 6, please test reaching the local web interface via http://_HostnameOfTheBirdNET-Pi._local

oliverk71, 2022-05-23:
Finally at home. I just did the above steps and it looks very good. Here is the output:
```

+ '[' -d /etc/caddy ']'
+ '[' -f /etc/caddy/Caddyfile ']'
+ cp /etc/caddy/Caddyfile /etc/caddy/Caddyfile.original
+ '[' -z test ']'
++ caddy hash-password -plaintext test
+ HASHWORD=JDJhJDE0JGc0cG03WVBPdGdaUkhJOWxoVkVNRXVJLnoudXlRcDdCQmx5LmZUMmR2V2V4b2huRDY5cHVt
+ cat
++ hostname
+ sudo systemctl restart caddy
```
I guess I can move on and add domains and password, now.

oliverk71, 2022-05-23:
So I tried again ... and ended up with the same result. My new password has only a '-' as special character, but I think, the issue is caused by entering the domain name, which I just entered in the usual format: https://domain.birdnetpi.com 

I deleted the domain from the birdnet.conf without editing the password and then used this `sed 's/reload/restart/' ~/BirdNET-Pi/scripts/update_caddyfile.sh | sudo bash` to update the caddy file and get caddy running again and it worked. Then I rebooted, just to be sure and it still works. By now, I did not enter a domain because I feel it won't work. 

Any output or log I shall provide you @mcguirepr89 ?

oliverk71, 2022-05-23:
I added a domain in the settings and it works by now. Let's see if it still works after reboot. But, no, it does not. Output of `systemctl status caddy.service`

```
● caddy.service - Caddy
     Loaded: loaded (/lib/systemd/system/caddy.service; enabled; vendor preset: enabled)
     Active: failed (Result: timeout) since Mon 2022-05-23 20:01:21 CEST; 2s ago
       Docs: https://caddyserver.com/docs/
    Process: 864 ExecStart=/usr/bin/caddy run --environ --config /etc/caddy/Caddyfile (code=killed, signal=KILL)
   Main PID: 864 (code=killed, signal=KILL)
        CPU: 1min 25.358s

May 23 20:01:21 birdnetpi-elsen systemd[1]: caddy.service: State 'stop-sigterm' timed out. Killing.
May 23 20:01:21 birdnetpi-elsen systemd[1]: caddy.service: Killing process 864 (caddy) with signal SIGKILL.
May 23 20:01:21 birdnetpi-elsen systemd[1]: caddy.service: Killing process 943 (caddy) with signal SIGKILL.
May 23 20:01:21 birdnetpi-elsen systemd[1]: caddy.service: Killing process 964 (n/a) with signal SIGKILL.
May 23 20:01:21 birdnetpi-elsen systemd[1]: caddy.service: Killing process 965 (caddy) with signal SIGKILL.
May 23 20:01:21 birdnetpi-elsen systemd[1]: caddy.service: Killing process 978 (n/a) with signal SIGKILL.
May 23 20:01:21 birdnetpi-elsen systemd[1]: caddy.service: Main process exited, code=killed, status=9/KILL
May 23 20:01:21 birdnetpi-elsen systemd[1]: caddy.service: Failed with result 'timeout'.
May 23 20:01:21 birdnetpi-elsen systemd[1]: Failed to start Caddy.
May 23 20:01:21 birdnetpi-elsen systemd[1]: caddy.service: Consumed 1min 25.358s CPU time.
```

oliverk71, 2022-05-23:
This time, I added the domains manually by editing the Caddfile and rebooted, then. Does not work. Caddy won't start. Output:

```
● caddy.service - Caddy
     Loaded: loaded (/lib/systemd/system/caddy.service; enabled; vendor preset: enabled)
     Active: failed (Result: timeout) since Mon 2022-05-23 20:09:01 CEST; 16s ago
       Docs: https://caddyserver.com/docs/
    Process: 859 ExecStart=/usr/bin/caddy run --environ --config /etc/caddy/Caddyfile (code=killed, signal=KILL)
   Main PID: 859 (code=killed, signal=KILL)
        CPU: 1min 25.814s

May 23 20:09:01 birdnetpi-elsen systemd[1]: caddy.service: State 'stop-sigterm' timed out. Killing.
May 23 20:09:01 birdnetpi-elsen systemd[1]: caddy.service: Killing process 859 (caddy) with signal SIGKILL.
May 23 20:09:01 birdnetpi-elsen systemd[1]: caddy.service: Killing process 935 (caddy) with signal SIGKILL.
May 23 20:09:01 birdnetpi-elsen systemd[1]: caddy.service: Killing process 936 (caddy) with signal SIGKILL.
May 23 20:09:01 birdnetpi-elsen systemd[1]: caddy.service: Killing process 965 (caddy) with signal SIGKILL.
May 23 20:09:01 birdnetpi-elsen systemd[1]: caddy.service: Killing process 976 (caddy) with signal SIGKILL.
May 23 20:09:01 birdnetpi-elsen systemd[1]: caddy.service: Main process exited, code=killed, status=9/KILL
May 23 20:09:01 birdnetpi-elsen systemd[1]: caddy.service: Failed with result 'timeout'.
May 23 20:09:01 birdnetpi-elsen systemd[1]: Failed to start Caddy.
May 23 20:09:01 birdnetpi-elsen systemd[1]: caddy.service: Consumed 1min 25.814s CPU time.
```

mcguirepr89, 2022-05-23:
I'm putting together a test for you. I should have it ready soon

oliverk71, 2022-05-23:
I am pretty sure, now, that caddy does not like any domains other than local in the Caddyfile, at least no https-domains.

mcguirepr89, 2022-05-23:
Let's please try increasing the systemd timeout for the certificate maintenance:
```
sudo systemctl edit caddy
```
Within the blank space provided at the top of the document, type or paste the following:
```
[Service]
TimeoutSec=300s
```
Then reload and restart the service, watching the output as it runs:
```
sudo systemctl daemon-reload && sudo systemctl restart caddy & journalctl -fu caddy
```

That will follow the log output while the service is restarting. You'll have to press "Ctrl + C" to stop following the log.

Please let me know how that goes for you and what kind of output you encounter --thank you!

oliverk71, 2022-05-23:
Here is the output (just a second)

```
-- Journal begins at Mon 2022-04-04 16:41:41 CEST. --
May 23 20:16:45 birdnetpi-elsen caddy[809]: {"level":"warn","ts":1653329805.809754,"msg":"Caddyfile input is not formatted; run the 'caddy fmt' command to fix inconsistencies","adapter":"caddyfile","file":"/etc/caddy/Caddyfile","line":2}
May 23 20:16:45 birdnetpi-elsen caddy[809]: {"level":"info","ts":1653329805.8665295,"logger":"admin","msg":"admin endpoint started","address":"tcp/localhost:2019","enforce_origin":false,"origins":["//127.0.0.1:2019","//localhost:2019","//[::1]:2019"]}
May 23 20:16:45 birdnetpi-elsen caddy[809]: {"level":"warn","ts":1653329805.867622,"logger":"http","msg":"server is listening only on the HTTP port, so no automatic HTTPS will be applied to this server","server_name":"srv0","http_port":80}
May 23 20:16:45 birdnetpi-elsen caddy[809]: {"level":"info","ts":1653329805.8678176,"logger":"tls.cache.maintenance","msg":"started background certificate maintenance","cache":"0x4000523500"}
May 23 20:17:38 birdnetpi-elsen caddy[809]: {"level":"info","ts":1653329858.4540906,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}
May 23 20:17:38 birdnetpi-elsen caddy[809]: {"level":"info","ts":1653329858.4546592,"msg":"serving initial configuration"}
May 23 20:17:38 birdnetpi-elsen systemd[1]: Started Caddy.
May 23 20:17:38 birdnetpi-elsen caddy[809]: {"level":"info","ts":1653329858.4292834,"logger":"tls","msg":"cleaning storage unit","description":"FileStorage:/var/lib/caddy/.local/share/caddy"}
May 23 20:17:38 birdnetpi-elsen caddy[809]: {"level":"info","ts":1653329858.7524002,"logger":"tls","msg":"finished cleaning storage units"}
May 23 21:16:26 birdnetpi-elsen caddy[809]: {"level":"error","ts":1653333386.0116692,"logger":"http.handlers.reverse_proxy","msg":"aborting with incomplete response","error":"write tcp 192.168.1.5:80->192.168.1.26:47036: write: broken pipe"}
May 23 21:56:46 birdnetpi-elsen systemd[1]: Stopping Caddy...
May 23 21:56:46 birdnetpi-elsen caddy[809]: {"level":"info","ts":1653335806.13255,"msg":"shutting down apps, then terminating","signal":"SIGTERM"}
May 23 21:56:46 birdnetpi-elsen caddy[809]: {"level":"warn","ts":1653335806.1399534,"msg":"exiting; byeee!! 👋","signal":"SIGTERM"}
May 23 21:56:46 birdnetpi-elsen caddy[809]: {"level":"info","ts":1653335806.203225,"logger":"tls.cache.maintenance","msg":"stopped background certificate maintenance","cache":"0x4000523500"}
May 23 21:56:46 birdnetpi-elsen caddy[809]: {"level":"info","ts":1653335806.2434175,"logger":"admin","msg":"stopped previous server","address":"tcp/localhost:2019"}
May 23 21:56:46 birdnetpi-elsen caddy[809]: {"level":"info","ts":1653335806.2435753,"msg":"shutdown complete","signal":"SIGTERM","exit_code":0}
May 23 21:56:46 birdnetpi-elsen systemd[1]: caddy.service: Succeeded.
May 23 21:56:46 birdnetpi-elsen systemd[1]: Stopped Caddy.
May 23 21:56:46 birdnetpi-elsen systemd[1]: caddy.service: Consumed 1min 6.918s CPU time.
May 23 21:56:46 birdnetpi-elsen systemd[1]: Starting Caddy...
May 23 21:56:47 birdnetpi-elsen caddy[36863]: caddy.HomeDir=/var/lib/caddy
May 23 21:56:47 birdnetpi-elsen caddy[36863]: caddy.AppDataDir=/var/lib/caddy/.local/share/caddy
May 23 21:56:47 birdnetpi-elsen caddy[36863]: caddy.AppConfigDir=/var/lib/caddy/.config/caddy
May 23 21:56:47 birdnetpi-elsen caddy[36863]: caddy.ConfigAutosavePath=/var/lib/caddy/.config/caddy/autosave.json
May 23 21:56:47 birdnetpi-elsen caddy[36863]: caddy.Version=v2.5.1 h1:bAWwslD1jNeCzDa+jDCNwb8M3UJ2tPa8UZFFzPVmGKs=
May 23 21:56:47 birdnetpi-elsen caddy[36863]: runtime.GOOS=linux
May 23 21:56:47 birdnetpi-elsen caddy[36863]: runtime.GOARCH=arm64
May 23 21:56:47 birdnetpi-elsen caddy[36863]: runtime.Compiler=gc
May 23 21:56:47 birdnetpi-elsen caddy[36863]: runtime.NumCPU=4
May 23 21:56:47 birdnetpi-elsen caddy[36863]: runtime.GOMAXPROCS=4
May 23 21:56:47 birdnetpi-elsen caddy[36863]: runtime.Version=go1.18.1
May 23 21:56:47 birdnetpi-elsen caddy[36863]: os.Getwd=/
May 23 21:56:47 birdnetpi-elsen caddy[36863]: LANG=en_GB.UTF-8
May 23 21:56:47 birdnetpi-elsen caddy[36863]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
May 23 21:56:47 birdnetpi-elsen caddy[36863]: NOTIFY_SOCKET=/run/systemd/notify
May 23 21:56:47 birdnetpi-elsen caddy[36863]: HOME=/var/lib/caddy
May 23 21:56:47 birdnetpi-elsen caddy[36863]: LOGNAME=caddy
May 23 21:56:47 birdnetpi-elsen caddy[36863]: USER=caddy
May 23 21:56:47 birdnetpi-elsen caddy[36863]: INVOCATION_ID=5cdebefd848b4cbca6d66ac6890b1d6a
May 23 21:56:47 birdnetpi-elsen caddy[36863]: JOURNAL_STREAM=8:137410
May 23 21:56:47 birdnetpi-elsen caddy[36863]: {"level":"info","ts":1653335807.4312065,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","config_adapter":""}
May 23 21:56:47 birdnetpi-elsen caddy[36863]: {"level":"warn","ts":1653335807.445908,"msg":"Caddyfile input is not formatted; run the 'caddy fmt' command to fix inconsistencies","adapter":"caddyfile","file":"/etc/caddy/Caddyfile","line":2}
May 23 21:56:47 birdnetpi-elsen caddy[36863]: {"level":"info","ts":1653335807.4609063,"logger":"admin","msg":"admin endpoint started","address":"tcp/localhost:2019","enforce_origin":false,"origins":["//localhost:2019","//[::1]:2019","//127.0.0.1:2019"]}
May 23 21:56:47 birdnetpi-elsen caddy[36863]: {"level":"info","ts":1653335807.462232,"logger":"tls.cache.maintenance","msg":"started background certificate maintenance","cache":"0x40002fecb0"}
May 23 21:56:47 birdnetpi-elsen caddy[36863]: {"level":"warn","ts":1653335807.4626143,"logger":"http","msg":"server is listening only on the HTTP port, so no automatic HTTPS will be applied to this server","server_name":"srv0","http_port":80}

```

oliverk71, 2022-05-23:
Should I add a https domain, first? I think so.

oliverk71, 2022-05-23:
And here is the output of `sudo systemctl restart caddy & journalctl -fu caddy`

```
[3] 37017
-- Journal begins at Mon 2022-04-04 16:41:41 CEST. --
May 23 22:01:49 birdnetpi-elsen caddy[36993]: {"level":"info","ts":1653336109.8470447,"logger":"http","msg":"server is listening only on the HTTPS port but has no TLS connection policies; adding one to enable TLS","server_name":"srv0","https_port":443}
May 23 22:01:49 birdnetpi-elsen caddy[36993]: {"level":"info","ts":1653336109.8471928,"logger":"http","msg":"enabling automatic HTTP->HTTPS redirects","server_name":"srv0"}
May 23 22:01:49 birdnetpi-elsen caddy[36993]: {"level":"warn","ts":1653336109.8472788,"logger":"http","msg":"server is listening only on the HTTP port, so no automatic HTTPS will be applied to this server","server_name":"srv1","http_port":80}
May 23 22:01:49 birdnetpi-elsen caddy[36993]: {"level":"info","ts":1653336109.8493488,"logger":"tls.cache.maintenance","msg":"started background certificate maintenance","cache":"0x40000d4bd0"}
May 23 22:03:17 birdnetpi-elsen caddy[36993]: {"level":"info","ts":1653336197.452583,"logger":"tls","msg":"cleaning storage unit","description":"FileStorage:/var/lib/caddy/.local/share/caddy"}
May 23 22:03:17 birdnetpi-elsen caddy[36993]: {"level":"info","ts":1653336197.4533231,"logger":"http","msg":"enabling automatic TLS certificate management","domains":["grevenbroich-elsen.birdnetpi.com"]}
May 23 22:03:17 birdnetpi-elsen caddy[36993]: {"level":"info","ts":1653336197.460856,"logger":"tls","msg":"finished cleaning storage units"}
May 23 22:03:17 birdnetpi-elsen caddy[36993]: {"level":"info","ts":1653336197.4740732,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}
May 23 22:03:17 birdnetpi-elsen caddy[36993]: {"level":"info","ts":1653336197.4754,"msg":"serving initial configuration"}
May 23 22:03:17 birdnetpi-elsen systemd[1]: Started Caddy.
May 23 22:03:44 birdnetpi-elsen caddy[36993]: {"level":"info","ts":1653336224.4697242,"msg":"shutting down apps, then terminating","signal":"SIGTERM"}
May 23 22:03:44 birdnetpi-elsen caddy[36993]: {"level":"warn","ts":1653336224.4698982,"msg":"exiting; byeee!! 👋","signal":"SIGTERM"}
May 23 22:03:44 birdnetpi-elsen systemd[1]: Stopping Caddy...
May 23 22:03:44 birdnetpi-elsen caddy[36993]: {"level":"info","ts":1653336224.507862,"logger":"tls.cache.maintenance","msg":"stopped background certificate maintenance","cache":"0x40000d4bd0"}
May 23 22:03:44 birdnetpi-elsen caddy[36993]: {"level":"info","ts":1653336224.509935,"logger":"admin","msg":"stopped previous server","address":"tcp/localhost:2019"}
May 23 22:03:44 birdnetpi-elsen caddy[36993]: {"level":"info","ts":1653336224.510079,"msg":"shutdown complete","signal":"SIGTERM","exit_code":0}
May 23 22:03:44 birdnetpi-elsen systemd[1]: caddy.service: Succeeded.
May 23 22:03:44 birdnetpi-elsen systemd[1]: Stopped Caddy.
May 23 22:03:44 birdnetpi-elsen systemd[1]: caddy.service: Consumed 1min 28.558s CPU time.
May 23 22:03:44 birdnetpi-elsen systemd[1]: Starting Caddy...
May 23 22:03:44 birdnetpi-elsen caddy[37023]: caddy.HomeDir=/var/lib/caddy
May 23 22:03:44 birdnetpi-elsen caddy[37023]: caddy.AppDataDir=/var/lib/caddy/.local/share/caddy
May 23 22:03:44 birdnetpi-elsen caddy[37023]: caddy.AppConfigDir=/var/lib/caddy/.config/caddy
May 23 22:03:44 birdnetpi-elsen caddy[37023]: caddy.ConfigAutosavePath=/var/lib/caddy/.config/caddy/autosave.json
May 23 22:03:44 birdnetpi-elsen caddy[37023]: caddy.Version=v2.5.1 h1:bAWwslD1jNeCzDa+jDCNwb8M3UJ2tPa8UZFFzPVmGKs=
May 23 22:03:44 birdnetpi-elsen caddy[37023]: runtime.GOOS=linux
May 23 22:03:44 birdnetpi-elsen caddy[37023]: runtime.GOARCH=arm64
May 23 22:03:44 birdnetpi-elsen caddy[37023]: runtime.Compiler=gc
May 23 22:03:44 birdnetpi-elsen caddy[37023]: runtime.NumCPU=4
May 23 22:03:44 birdnetpi-elsen caddy[37023]: runtime.GOMAXPROCS=4
May 23 22:03:44 birdnetpi-elsen caddy[37023]: runtime.Version=go1.18.1
May 23 22:03:44 birdnetpi-elsen caddy[37023]: os.Getwd=/
May 23 22:03:44 birdnetpi-elsen caddy[37023]: LANG=en_GB.UTF-8
May 23 22:03:44 birdnetpi-elsen caddy[37023]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
May 23 22:03:44 birdnetpi-elsen caddy[37023]: NOTIFY_SOCKET=/run/systemd/notify
May 23 22:03:44 birdnetpi-elsen caddy[37023]: HOME=/var/lib/caddy
May 23 22:03:44 birdnetpi-elsen caddy[37023]: LOGNAME=caddy
May 23 22:03:44 birdnetpi-elsen caddy[37023]: USER=caddy
May 23 22:03:44 birdnetpi-elsen caddy[37023]: INVOCATION_ID=b12ea59c29c948eaae7eb08c1c466bce
May 23 22:03:44 birdnetpi-elsen caddy[37023]: JOURNAL_STREAM=8:137742
May 23 22:03:44 birdnetpi-elsen caddy[37023]: {"level":"info","ts":1653336224.6831658,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","config_adapter":""}
May 23 22:03:44 birdnetpi-elsen caddy[37023]: {"level":"warn","ts":1653336224.7003384,"msg":"Caddyfile input is not formatted; run the 'caddy fmt' command to fix inconsistencies","adapter":"caddyfile","file":"/etc/caddy/Caddyfile","line":2}
May 23 22:03:44 birdnetpi-elsen caddy[37023]: {"level":"info","ts":1653336224.7154055,"logger":"admin","msg":"admin endpoint started","address":"tcp/localhost:2019","enforce_origin":false,"origins":["//127.0.0.1:2019","//localhost:2019","//[::1]:2019"]}
May 23 22:03:44 birdnetpi-elsen caddy[37023]: {"level":"info","ts":1653336224.7171774,"logger":"http","msg":"server is listening only on the HTTPS port but has no TLS connection policies; adding one to enable TLS","serv [... cut ...]
May 23 22:03:44 birdnetpi-elsen caddy[37023]: {"level":"info","ts":1653336224.7175515,"logger":"tls.cache.maintenance","msg":"started background certificate maintenance","cache":"0x400069a5b0"}
May 23 22:03:44 birdnetpi-elsen caddy[37023]: {"level":"warn","ts":1653336224.7176347,"logger":"http","msg":"server is listening only on the HTTP port, so no automatic HTTPS will be applied to this server","server_name":"srv1","http_port":80}
^C
[2]   Done                    sudo systemctl daemon-reload && sudo systemctl restart caddy
```

mcguirepr89, 2022-05-23:
Please continue to monitor the output -- it is not nearly complete

oliverk71, 2022-05-23:
Right now the https website is reachable. Hope it will still be after a reboot. Here is the output: 

```
[4] 37088
-- Journal begins at Mon 2022-04-04 16:41:41 CEST. --
May 23 22:03:44 birdnetpi-elsen caddy[37023]: {"level":"info","ts":1653336224.7171774,"logger":"http","msg":"server is listening only on the HTTPS port but has no TLS connection policies; adding one to enable TLS","server_name":"srv0","https_port":443}
May 23 22:03:44 birdnetpi-elsen caddy[37023]: {"level":"info","ts":1653336224.7174737,"logger":"http","msg":"enabling automatic HTTP->HTTPS redirects","server_name":"srv0"}
May 23 22:03:44 birdnetpi-elsen caddy[37023]: {"level":"info","ts":1653336224.7175515,"logger":"tls.cache.maintenance","msg":"started background certificate maintenance","cache":"0x400069a5b0"}
May 23 22:03:44 birdnetpi-elsen caddy[37023]: {"level":"warn","ts":1653336224.7176347,"logger":"http","msg":"server is listening only on the HTTP port, so no automatic HTTPS will be applied to this server","server_name":"srv1","http_port":80}
May 23 22:05:13 birdnetpi-elsen caddy[37023]: {"level":"info","ts":1653336313.9371748,"logger":"http","msg":"enabling automatic TLS certificate management","domains":["grevenbroich-elsen.birdnetpi.com"]}
May 23 22:05:13 birdnetpi-elsen caddy[37023]: {"level":"info","ts":1653336313.9419954,"logger":"tls","msg":"cleaning storage unit","description":"FileStorage:/var/lib/caddy/.local/share/caddy"}
May 23 22:05:13 birdnetpi-elsen caddy[37023]: {"level":"info","ts":1653336313.942848,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}
May 23 22:05:13 birdnetpi-elsen caddy[37023]: {"level":"info","ts":1653336313.943387,"msg":"serving initial configuration"}
May 23 22:05:13 birdnetpi-elsen systemd[1]: Started Caddy.
May 23 22:05:13 birdnetpi-elsen caddy[37023]: {"level":"info","ts":1653336313.9465277,"logger":"tls","msg":"finished cleaning storage units"}
May 23 22:08:59 birdnetpi-elsen caddy[37023]: {"level":"info","ts":1653336539.9547997,"msg":"shutting down apps, then terminating","signal":"SIGTERM"}
May 23 22:08:59 birdnetpi-elsen caddy[37023]: {"level":"warn","ts":1653336539.9550087,"msg":"exiting; byeee!! 👋","signal":"SIGTERM"}
May 23 22:08:59 birdnetpi-elsen systemd[1]: Stopping Caddy...
May 23 22:09:01 birdnetpi-elsen caddy[37023]: {"level":"info","ts":1653336541.0205455,"logger":"tls.cache.maintenance","msg":"stopped background certificate maintenance","cache":"0x400069a5b0"}
May 23 22:09:01 birdnetpi-elsen caddy[37023]: {"level":"info","ts":1653336541.0221982,"logger":"admin","msg":"stopped previous server","address":"tcp/localhost:2019"}
May 23 22:09:01 birdnetpi-elsen caddy[37023]: {"level":"info","ts":1653336541.022321,"msg":"shutdown complete","signal":"SIGTERM","exit_code":0}
May 23 22:09:01 birdnetpi-elsen systemd[1]: caddy.service: Succeeded.
May 23 22:09:01 birdnetpi-elsen systemd[1]: Stopped Caddy.
May 23 22:09:01 birdnetpi-elsen systemd[1]: caddy.service: Consumed 1min 31.705s CPU time.
May 23 22:09:01 birdnetpi-elsen systemd[1]: Starting Caddy...
May 23 22:09:01 birdnetpi-elsen caddy[37094]: caddy.HomeDir=/var/lib/caddy
May 23 22:09:01 birdnetpi-elsen caddy[37094]: caddy.AppDataDir=/var/lib/caddy/.local/share/caddy
May 23 22:09:01 birdnetpi-elsen caddy[37094]: caddy.AppConfigDir=/var/lib/caddy/.config/caddy
May 23 22:09:01 birdnetpi-elsen caddy[37094]: caddy.ConfigAutosavePath=/var/lib/caddy/.config/caddy/autosave.json
May 23 22:09:01 birdnetpi-elsen caddy[37094]: caddy.Version=v2.5.1 h1:bAWwslD1jNeCzDa+jDCNwb8M3UJ2tPa8UZFFzPVmGKs=
May 23 22:09:01 birdnetpi-elsen caddy[37094]: runtime.GOOS=linux
May 23 22:09:01 birdnetpi-elsen caddy[37094]: runtime.GOARCH=arm64
May 23 22:09:01 birdnetpi-elsen caddy[37094]: runtime.Compiler=gc
May 23 22:09:01 birdnetpi-elsen caddy[37094]: runtime.NumCPU=4
May 23 22:09:01 birdnetpi-elsen caddy[37094]: runtime.GOMAXPROCS=4
May 23 22:09:01 birdnetpi-elsen caddy[37094]: runtime.Version=go1.18.1
May 23 22:09:01 birdnetpi-elsen caddy[37094]: os.Getwd=/
May 23 22:09:01 birdnetpi-elsen caddy[37094]: LANG=en_GB.UTF-8
May 23 22:09:01 birdnetpi-elsen caddy[37094]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
May 23 22:09:01 birdnetpi-elsen caddy[37094]: NOTIFY_SOCKET=/run/systemd/notify
May 23 22:09:01 birdnetpi-elsen caddy[37094]: HOME=/var/lib/caddy
May 23 22:09:01 birdnetpi-elsen caddy[37094]: LOGNAME=caddy
May 23 22:09:01 birdnetpi-elsen caddy[37094]: USER=caddy
May 23 22:09:01 birdnetpi-elsen caddy[37094]: INVOCATION_ID=ccbfc5f1945341fa87c92e9d37e12f05
May 23 22:09:01 birdnetpi-elsen caddy[37094]: JOURNAL_STREAM=8:135872
May 23 22:09:01 birdnetpi-elsen caddy[37094]: {"level":"info","ts":1653336541.2210078,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","config_adapter":""}
May 23 22:09:01 birdnetpi-elsen caddy[37094]: {"level":"warn","ts":1653336541.243013,"msg":"Caddyfile input is not formatted; run the 'caddy fmt' command to fix inconsistencies","adapter":"caddyfile","file":"/etc/caddy/Caddyfile","line":2}
May 23 22:09:01 birdnetpi-elsen caddy[37094]: {"level":"info","ts":1653336541.2646265,"logger":"admin","msg":"admin endpoint started","address":"tcp/localhost:2019","enforce_origin":false,"origins":["//localhost:2019","//[::1]:2019","//127.0.0.1:2019"]}
[... cut ...]
May 23 22:10:28 birdnetpi-elsen caddy[37094]: {"level":"info","ts":1653336628.9863558,"logger":"tls","msg":"finished cleaning storage units"}
May 23 22:10:28 birdnetpi-elsen caddy[37094]: {"level":"info","ts":1653336628.9904902,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}
May 23 22:10:28 birdnetpi-elsen caddy[37094]: {"level":"info","ts":1653336628.9909163,"msg":"serving initial configuration"}
May 23 22:10:28 birdnetpi-elsen systemd[1]: Started Caddy.
```

mcguirepr89, 2022-05-23:
[image]
:partying_face:

oliverk71, 2022-05-23:
So, it was just the timeout? I try a reboot, just to be sure.

oliverk71, 2022-05-23:
Takes its time, but it works! Thanks a lot, Patrick!

mcguirepr89, 2022-05-23:
Yes, for whatever reason, your system is taking a long time to complete the CA challenges during certificate maintenance. That is why it was working when running manually, but failing in the service. Increasing the timeout seems to allow your system the time it needs to maintain the certificates.

I'm glad your system is working for you now -- thank you for being patient as I worked through this with testing on my end.

closed 2022-05-24 by mcguirepr89