mcguirepr89/BirdNET-Pi #304: File manager: cannot calculate folder size

oliverk71, 2022-05-23:
I am so sorry, but I found another issue:

`Fatal error: Uncaught UnexpectedValueException: RecursiveDirectoryIterator::__construct(/home/oliver/.local/share): failed to open dir: Permission denied in /home/oliver/BirdNET-Pi/scripts/filemanager/filemanager.php:2518 Stack trace: #0 [internal function]: RecursiveDirectoryIterator->__construct() #1 /home/oliver/BirdNET-Pi/scripts/filemanager/filemanager.php(2518): RecursiveDirectoryIterator->getChildren() #2 /home/oliver/BirdNET-Pi/scripts/filemanager/filemanager.php(1944): fm_get_directorysize() #3 {main} thrown in /home/oliver/BirdNET-Pi/scripts/filemanager/filemanager.php on line 2518`

This happened after editing the options of the file manager. It only occurs if calculation of folder size is set to active.

mcguirepr89, 2022-05-23:
Hi, Oliver --

2 things:
1. To help me replicate the issue on my end, please provide exactly which line you edited, providing a copy of the exact line if possible so that I can use the exact same changes for troubelshooting.
2. I don't recommend the overhead that would come with calculating the folder size for every page load -- your Pi will hate you.

Those things noted, if you provide the info for the first item, I can help.

oliverk71, 2022-05-23:
I did not edit any file directly, I just went to settings in the file manager and activated calculation of folder size. After I received the error message, I went to settings again and deactivated it. Since then: no more error message.

mcguirepr89, 2022-05-23:
Hmmm I am not able to replicate that by just toggling the setting. Is there a particular page that shows you that error?

oliverk71, 2022-05-23:
I just noticed in the file manager, that the owner of the copied birds.db is caddy. The old, renamed birds.db has the owner oliver.
[image]

oliverk71, 2022-05-23:
[image]

mcguirepr89, 2022-05-23:
That is normal -- everything you do in "File Manager" is as the web server, caddy. The `caddy` user is in the `$USER` group. The file, though, looks empty.

oliverk71, 2022-05-23:
Sorry, I deleted the post because I posted it by mistake here. See screenshot above.

ehpersonal38, 2022-05-26:
I _can_ replicate this, and get the same error as in the original post. I also noticed that I can't set any of the Settings for File Manager through the web interface, for when I do they just revert. It's like file_manager.php has very limited perms to do anything.

```
pi@birdnetpi:~/BirdNET-Pi/scripts/filemanager$ ls -lh
total 312K
-rw-r--r-- 1 pi pi 182K May 26 19:02 filemanager.php
-rw-r--r-- 1 pi pi  35K Apr 17 10:21 LICENSE
-rw-r--r-- 1 pi pi 1.2K Apr 17 10:21 SECURITY.md
-rw-r--r-- 1 pi pi  85K Apr 17 10:21 translation.json
```

Patrick if you'd like to troubleshoot further then I'm happy to run some commands, but I understand if this is a little bit out of scope and maybe should be raised as an issue in the TIny File manager github.

oliverk71, 2022-05-27:
Well, I can toggle the settings. And true, the error message only shows up if error reporting and calculation of folder size is activated, no matter if set to English or German. 
Maybe it would be enough to hide the toggle switch for calculation of folder size? 

[image]

[image]

mcguirepr89, 2022-05-27:
I've replicated the issue, but again I do NOT recommend asking the BirdNET-Pi to calculate the directory size for each page load. If I had to choose, I would want to remove ALL settings for File Manager and just keep it the way it comes out of the box. That said, if folks feel they need to adjust settings, then I suppose that means we have to fix them.

closed 2022-05-29 by oliverk71