mcguirepr89/BirdNET-Pi #309: Web Updater broken after rework of update_birdnet.sh -> wrong working directory

kaeltis, 2022-05-24:
Hi,

it seems that a786e51da5fe4b48e7bb6ed091d55ae7826700fc broke the web based updater for me.

I'm now getting the following messages on updating via the Webinterface:

```+ sudo -u pi git reset --hard
fatal: not a git repository (or any of the parent directories): .git
+ set +x
+ sudo -u pi git fetch origin main
fatal: not a git repository (or any of the parent directories): .git
+ set +x
+ sudo -u pi git switch -C main --track origin/main
fatal: not a git repository (or any of the parent directories): .git
+ set +x
ExecStart=/home/pi/BirdNET-Pi/birdnet/bin/python3 /usr/local/bin/server.py
ExecStart=/home/pi/BirdNET-Pi/birdnet/bin/python3 /usr/local/bin/daily_plot.py
UNIT FILE STATE VENDOR PRESET

0 unit files listed.
```

Seems the git commands are executed in the wrong working directory for some reason, here's the output when adding `sudo -u $USER "pwd"` to the `sudo_with_user` function:

```
+ sudo -u pi pwd
/home/pi/BirdSongs/Extracted
+ sudo -u pi git reset --hard
fatal: not a git repository (or any of the parent directories): .git
+ set +x
+ sudo -u pi pwd
/home/pi/BirdSongs/Extracted
+ sudo -u pi git fetch origin main
fatal: not a git repository (or any of the parent directories): .git
+ set +x
+ sudo -u pi pwd
/home/pi/BirdSongs/Extracted
+ sudo -u pi git switch -C main --track origin/main
fatal: not a git repository (or any of the parent directories): .git
+ set +x
ExecStart=/home/pi/BirdNET-Pi/birdnet/bin/python3 /usr/local/bin/server.py
ExecStart=/home/pi/BirdNET-Pi/birdnet/bin/python3 /usr/local/bin/daily_plot.py
UNIT FILE STATE VENDOR PRESET

0 unit files listed.
```

kaeltis, 2022-05-24:
Bonus issue I've just noticed: the new update mechanism resets all unstaged changes, including the changes to `model/labels.txt` that are done automatically when changing the language in BirdNET-Pi‚ resetting the database language to english on every update.

mcguirepr89, 2022-05-24:
:unamused: bonus issues? eager to help! :unamused:

Please provide the output from this command:
```
git config -l
```

mcguirepr89, 2022-05-24:
@kaeltis `$HOME/BirdSongs/Extracted` is the webroot -- this is the `pwd` (if not its subdirectory `scripts`) for all commands run by the web server/PHP.

mcguirepr89, 2022-05-24:
@kaeltis I pushed a change to the updater.

Please run this:
```
git -C ~/BirdNET-Pi checkout -f main
git -C ~/BirdNET-Pi pull
update_birdnet.sh
```

kaeltis, 2022-05-24:
Awesome, thanks! The Web Updater is working again 🙂 

```
+ sudo -u pi git -C /home/pi/BirdNET-Pi reset --hard
HEAD is now at 4bb5256 Merge branch 'main' of github.com:mcguirepr89/BirdNET-Pi
+ set +x
+ sudo -u pi git -C /home/pi/BirdNET-Pi fetch origin main
From https://github.com/mcguirepr89/BirdNET-Pi
 * branch            main       -> FETCH_HEAD
+ set +x
+ sudo -u pi git -C /home/pi/BirdNET-Pi switch -C main --track origin/main
Reset branch 'main'
Branch 'main' set up to track remote branch 'main' from 'origin'.
Your branch is up to date with 'origin/main'.
+ set +x
ExecStart=/home/pi/BirdNET-Pi/birdnet/bin/python3 /usr/local/bin/server.py
ExecStart=/home/pi/BirdNET-Pi/birdnet/bin/python3 /usr/local/bin/daily_plot.py
UNIT FILE STATE VENDOR PRESET

0 unit files listed.
```

It's still resetting the `model/labels.txt` file though

Before the update:
```
pi@birdnetpi:~/BirdNET-Pi$ git status
On branch main
Your branch is up to date with 'origin/main'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   model/labels.txt

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        BirdDB.txt
        HUMAN.txt
        analyzing_now.txt
        apprise.txt
        birdnet/
        exclude_species_list.txt
        firstrun.ini
        include_species_list.txt
        scripts/birds.db
        scripts/disk_check_exclude.txt
        scripts/exclude_species_list.txt
        scripts/include_species_list.txt
        scripts/labels.txt
        scripts/lastrun.txt
        scripts/thisrun.txt
        templates/avahi-alias@.service
        templates/birdnet_analysis.service
        templates/birdnet_log.service
        templates/birdnet_recording.service
        templates/birdnet_server.service
        templates/birdnet_stats.service
        templates/chart_viewer.service
        templates/custom_recording.service
        templates/extraction.service
        templates/livestream.service
        templates/spectrogram_viewer.service
        templates/web_terminal.service

no changes added to commit (use "git add" and/or "git commit -a")
```

After the update:
```
pi@birdnetpi:~/BirdNET-Pi$ git status
On branch main
Your branch is up to date with 'origin/main'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        BirdDB.txt
        HUMAN.txt
        analyzing_now.txt
        apprise.txt
        birdnet/
        exclude_species_list.txt
        firstrun.ini
        include_species_list.txt
        scripts/birds.db
        scripts/disk_check_exclude.txt
        scripts/exclude_species_list.txt
        scripts/include_species_list.txt
        scripts/labels.txt
        scripts/lastrun.txt
        scripts/thisrun.txt
        templates/avahi-alias@.service
        templates/birdnet_analysis.service
        templates/birdnet_log.service
        templates/birdnet_recording.service
        templates/birdnet_server.service
        templates/birdnet_stats.service
        templates/chart_viewer.service
        templates/custom_recording.service
        templates/extraction.service
        templates/livestream.service
        templates/spectrogram_viewer.service
        templates/web_terminal.service

nothing added to commit but untracked files present (use "git add" to track)
```

`git config -l`
```
pi@birdnetpi:~/BirdNET-Pi$ git config -l
core.repositoryformatversion=0
core.filemode=true
core.bare=false
core.logallrefupdates=true
remote.origin.url=https://github.com/mcguirepr89/BirdNET-Pi.git
remote.origin.fetch=+refs/heads/*:refs/remotes/origin/*
branch.main.remote=origin
branch.main.merge=refs/heads/main
```

mcguirepr89, 2022-05-24:
I'll add `model/labels.txt` to `.gitignore` and that should suffice (I think?)

jmherbst, 2022-05-24:
You'll want to `git rm --cached model/labels.txt` @mcguirepr89 🎱

kaeltis, 2022-05-24:
Maybe `git stash` before the `git reset —hard` and `git stash apply` after the update is finished could help?

kaeltis, 2022-05-24:
Maybe alternatively just leave the file removed from the repo and just create it on install.

mcguirepr89, 2022-05-24:
>Maybe alternatively just leave the file removed from the repo and just create it on install.

Yes this is the new plan -- I put it back for now -- I will add a small bit that unzips the `labels.txt`

mcguirepr89, 2022-05-24:
this c7bac0fe59afdcba70889998e15e9dfa18a7c27e
this 364dd8d683186bc3127857d5cf2e93c7a359de59
and this 74a0e432d629eca7205cc5c531cea7b2be0c634d
should have done the trick

closed 2022-05-24 by mcguirepr89