mcguirepr89/BirdNET-Pi #312: Syntax error in server.py

oliverk71, 2022-05-24:
See screenshot:
[image]

mcguirepr89, 2022-05-24:
Please update again and provide the output here

oliverk71, 2022-05-24:
+ sudo -u oliver git reset --hard
fatal: not a git repository (or any of the parent directories): .git
+ set +x
+ sudo -u oliver git fetch origin main
fatal: not a git repository (or any of the parent directories): .git
+ set +x
+ sudo -u oliver git switch -C main --track origin/main
fatal: not a git repository (or any of the parent directories): .git
+ set +x
ExecStart=/home/oliver/BirdNET-Pi/birdnet/bin/python3 /usr/local/bin/server.py
ExecStart=/home/oliver/BirdNET-Pi/birdnet/bin/python3 /usr/local/bin/daily_plot.py
UNIT FILE STATE VENDOR PRESET

0 unit files listed.

mcguirepr89, 2022-05-24:
Ok please issue the following to fix your system:
```
git -C ~/BirdNET-Pi checkout -f main
git -C ~/BirdNET-Pi pull
update_birdnet.sh
unzip ~/BirdNET-Pi/model/labels_l18n.zip labels_de.txt -d ~/BirdNET-Pi/model
mv ~/BirdNET-Pi/model/labels_de.txt ~/BirdNET-Pi/model/labels.txt
```

oliverk71, 2022-05-24:
Here is the output: 
```
birdnetpi-elsen:~/BirdNET-Pi$ git -C ~/BirdNET-Pi checkout -f main
Already on 'main'
Your branch is up to date with 'origin/main'.
birdnetpi-elsen:~/BirdNET-Pi$ git -C ~/BirdNET-Pi pull
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
remote: Enumerating objects: 58, done.
remote: Counting objects: 100% (53/53), done.
remote: Compressing objects: 100% (23/23), done.
remote: Total 38 (delta 27), reused 26 (delta 15), pack-reused 0
Unpacking objects: 100% (38/38), 5.71 KiB | 52.00 KiB/s, done.
From https://github.com/mcguirepr89/BirdNET-Pi
   f9e6cd8..426aba1  main       -> origin/main
 * [new branch]      api        -> origin/api
Updating f9e6cd8..426aba1
Fast-forward
 README.md                    |    2 +-
 homepage/style.css           |   24 +
 model/labels.txt             | 6362 ---------------------------------------------------------------------------------------------------------------------------------------------------------------
 scripts/install_birdnet.sh   |    9 +-
 scripts/service_controls.php |    2 +-
 scripts/spectrogram.php      |   23 +-
 scripts/update_birdnet.sh    |    8 +-
 7 files changed, 59 insertions(+), 6371 deletions(-)
 delete mode 100755 model/labels.txt
birdnetpi-elsen:~/BirdNET-Pi$ update_birdnet.sh
+ sudo -u oliver git -C /home/oliver/BirdNET-Pi reset --hard
HEAD is now at 426aba1 adding time expectations for Clear All Data
+ set +x
+ sudo -u oliver git -C /home/oliver/BirdNET-Pi fetch origin main
From https://github.com/mcguirepr89/BirdNET-Pi
 * branch            main       -> FETCH_HEAD
+ set +x
+ sudo -u oliver git -C /home/oliver/BirdNET-Pi switch -C main --track origin/main
Branch 'main' set up to track remote branch 'main' from 'origin'.
Reset branch 'main'
Your branch is up to date with 'origin/main'.
+ set +x
ExecStart=/home/oliver/BirdNET-Pi/birdnet/bin/python3 /usr/local/bin/server.py
ExecStart=/home/oliver/BirdNET-Pi/birdnet/bin/python3 /usr/local/bin/daily_plot.py
UNIT FILE STATE VENDOR PRESET

0 unit files listed.
birdnetpi-elsen:~/BirdNET-Pi$ unzip ~/BirdNET-Pi/model/labels_l18n.zip labels_de.txt -d ~/BirdNET-Pi/model
Archive:  /home/oliver/BirdNET-Pi/model/labels_l18n.zip
  inflating: /home/oliver/BirdNET-Pi/model/labels_de.txt
birdnetpi-elsen:~/BirdNET-Pi$ mv -f ~/BirdNET-Pi/model/labels_de.txt ~/BirdNET-Pi/model/labels.txt
```

oliverk71, 2022-05-24:
Looks much better, now! 

[image]

closed 2022-05-25 by mcguirepr89