mcguirepr89/BirdNET-Pi #315: Analysis hangs if WAV file is incomplete

pjsg, 2022-05-25:
I've had a few hangs when the system doesn't appear to make any progress -- it sticks on analyzing the same file. It seems that if the file is incomplete, then the analysis never completes. It is easy enough to fix -- just ssh in and delete the file and kill the analysis process. I get incomplete files when the power fails to the RPi (either deliberately when I'm trying to get the right location, or accidentally when the dog yanked the USB cable out of the charger!)

mcguirepr89, 2022-05-26:
Hi, @pjsg 

We fixed this bug a few weeks ago or so -- please run an update via "Tools" > "System Controls" > "Update"

Note that any local changes will be lost, as the update will do a hard reset on the repo to get you up to date.

Let me know if you continue to see the issue after updating and we'll go from there.

My very best regards,
Patrick

pjsg, 2022-05-26:
It still happens with the update to main (commit `4ec734890`). The issue appears to be that the `analyze.py` script sends the request to the `server.py` and then hangs waiting for a response.

closed 2022-05-26 by mcguirepr89