mcguirepr89/BirdNET-Pi #316: The BirdSongs/Extracted/By_Date/ does not seem to be filled correctly

pjsg, 2022-05-25:
In particular, if I look at today's detections, then most of the detections just display the generic player. This is because the link points to a file that is not found. Also, since the mp3 isn't there either, you can't press play.

[image]

```
birdnetpi:~/BirdSongs/Extracted/By_Date$ ls 2022-05-25
House_Finch
birdnetpi:~/BirdSongs/Extracted/By_Date$ 
```

[image]

mcguirepr89, 2022-05-26:
Hello again -- let me know if this issue persists after the update and we'll take a closer look

pjsg, 2022-05-26:
I'm updating -- but I only built this system within the last week, so I assume that I would have latest code as of then.

I don't think that the automated update worked. This is my `git status` output:

```
birdnetpi:~/BirdNET-Pi$ git status
On branch main
Your branch is up to date with 'origin/main'.

Changes to be committed:
  (use "git restore --staged <file>..." to unstage)
	modified:   scripts/analyze.py
	modified:   scripts/daily_plot.py
	modified:   scripts/plotly_streamlit.py

Unmerged paths:
  (use "git restore --staged <file>..." to unstage)
  (use "git add <file>..." to mark resolution)
	both modified:   scripts/server.py
```

The git diff is 
```
birdnetpi:~/BirdNET-Pi$ git diff
diff --cc scripts/server.py
index c73b0fa,c3e9ca2..0000000
--- a/scripts/server.py
+++ b/scripts/server.py
@@@ -1,16 -1,5 +1,21 @@@
++<<<<<<< Updated upstream
 +import apprise
 +from pathlib import Path
 +from tzlocal import get_localzone
 +import datetime
 +import sqlite3
 +import requests
 +import json
 +import time
 +import math
 +import numpy as np
 +import librosa
 +import operator
 +import socket
++=======
+ #!/home/philip/BirdNET-Pi/birdnet/bin/python3
+ import socket 
++>>>>>>> Stashed changes
  import threading
  import os
  os.environ['TF_CPP_MIN_LOG_LEVEL'] = '3'
```

which shows a merge conflict. I *never* edited any of these files by hand -- I just followed the instructions (and I used the Lite distribution for the OS).

```
diff --git a/scripts/analyze.py b/scripts/analyze.py
index 21fe905..e96edd0 100755
--- a/scripts/analyze.py
+++ b/scripts/analyze.py
@@ -1,3 +1,4 @@
+#!/home/philip/BirdNET-Pi/birdnet/bin/python3
 import argparse
 import socket
 
diff --git a/scripts/daily_plot.py b/scripts/daily_plot.py
index 59747f9..aff4d42 100755
--- a/scripts/daily_plot.py
+++ b/scripts/daily_plot.py
@@ -1,3 +1,4 @@
+#!/home/philip/BirdNET-Pi/birdnet/bin/python3
 import sqlite3
 import os
 import pandas as pd
diff --git a/scripts/plotly_streamlit.py b/scripts/plotly_streamlit.py
index 2df1550..6a92124 100755
--- a/scripts/plotly_streamlit.py
+++ b/scripts/plotly_streamlit.py
@@ -1,3 +1,4 @@
+#!/home/philip/BirdNET-Pi/birdnet/bin/python3
 import os
 import streamlit as st
 import pandas as pd
```

This seems to show that (probably) the install process added the `#!` header line -- but then that causes problems for updates......

mcguirepr89, 2022-05-26:
Hi, @pjsg 
>I'm updating -- but I only built this system within the last week, so I assume that I would have latest code as of then.

There have been 101 commits in the last week :)

```
birdnetpi1:~/BirdNET-Pi$  git rev-list --count main --since=1.week
101
```

>I don't think that the automated update worked. This is my git status output:

I agree -- among the 101 commits has been an overhaul to the update process -- there will be a reset during the updates going forward. 

Please instead run `update_birdnet.sh` in the terminal and I believe that will suffice.

If it does not, then please run:
```
git checkout -f main
git pull
git stash drop
update_birdnet.sh
```

and let me know if that works for you~~

pjsg, 2022-05-26:
I ran the `update_birdnet.sh` and it did "stuff". I then rebooted the PI, but I'm not seeing the new detections appearing in the `By_Date` directory.

This is what my log shows:

```
11:02:57---ANALYZING AUDIO... DONE! Time 1.7 SECONDS
11:02:57---WRITING RESULTS TO BirdSongs/May-2022/26-Thursday/2022-05-26-birdnet-11:02:26.wav.csv ... DONE! WROTE 0 RESULTS.
11:02:57---READING AUDIO DATA... DONE! READ 5 CHUNKS.
11:02:57---Date-time: 2022-05-26 11:02:41
11:02:58---NEWFILE=House_Finch-73-2022-05-23-birdnet-18:26:51.mp3
11:02:58---Extraction exists. Moving on
11:02:58---0.0;3.0-('Protonotaria citrea_Prothonotary Warbler', 0.06125659)
11:02:58---3.0;6.0-('Setophaga pensylvanica_Chestnut-sided Warbler', 0.19413063)
11:02:58---6.0;9.0-('Setophaga cerulea_Cerulean Warbler', 0.08376353)
11:02:58---9.0;12.0-('Scolopax minor_American Woodcock', 0.0458138)
11:02:58---12.0;15.0-('Spizella pusilla_Field Sparrow', 0.046661537)
11:02:59---(standard_in) 1: syntax error
11:02:59---NEWFILE=-0-2022-05-23-birdnet-18:29:36.mp3
11:02:59---(standard_in) 1: syntax error
11:02:59---(standard_in) 1: syntax error
11:02:59---sox FAIL trim: Error parsing position 2
11:02:59---sox FAIL trim: usage: {position}
11:03:02---Files loaded
11:03:02---RECORDING_LENGTH set to 15
11:03:04---NEWFILE=House_Finch-73-2022-05-23-birdnet-18:26:51.mp3
11:03:04---Extraction exists. Moving on
11:03:04---(standard_in) 1: syntax error
11:03:04---NEWFILE=-0-2022-05-23-birdnet-18:29:36.mp3
11:03:04---(standard_in) 1: syntax error
11:03:04---(standard_in) 1: syntax error
11:03:04---sox FAIL trim: Error parsing position 2
11:03:04---sox FAIL trim: usage: {position}
```

There are a number of odd things that I see -- one of which is that it keeps trying to do something with the House Finch file from 3 days ago. Also, the syntax errors are slightly worrying....

pjsg, 2022-05-26:
Did that, and it didn't seem to help.

```
12:52:00---(standard_in) 1: syntax error
12:52:00---sox FAIL trim: Error parsing position 2
12:52:00---sox FAIL trim: usage: {position}
12:52:06---NEWFILE=House_Finch-73-2022-05-23-birdnet-18:26:51.mp3
12:52:06---Extraction exists. Moving on
12:52:06---(standard_in) 1: syntax error
12:52:06---NEWFILE=-0-2022-05-23-birdnet-18:29:36.mp3
12:52:06---(standard_in) 1: syntax error
12:52:06---(standard_in) 1: syntax error
12:52:06---sox FAIL trim: Error parsing position 2
12:52:06---sox FAIL trim: usage: {position}
12:52:06---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts\analyze.py --i BirdSongs/May-2022/26-Thursday/2022-05-26-birdnet-12:51:50.wav --o BirdSongs/May-2022/26-Thursday/2022-05-26-birdnet-12:51:50.wav.csv --lat 42.5442 --lon -71.3733 --week 20 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7 --birdweather_id "IN_USE"
12:52:07---ANALYZING AUDIO... DONE! Time 1.6 SECONDS
12:52:07---WRITING RESULTS TO BirdSongs/May-2022/26-Thursday/2022-05-26-birdnet-12:51:35.wav.csv ... DONE! WROTE 0 RESULTS.
12:52:07---READING AUDIO DATA... DONE! READ 5 CHUNKS.
12:52:07---Date-time: 2022-05-26 12:51:50
12:52:08---0.0;3.0-('Scolopax minor_American Woodcock', 0.040735528)
12:52:08---3.0;6.0-('Scolopax minor_American Woodcock', 0.043734893)
12:52:08---6.0;9.0-('Scolopax minor_American Woodcock', 0.042719517)
12:52:08---9.0;12.0-('Scolopax minor_American Woodcock', 0.0392825)
12:52:08---12.0;15.0-('Spizella pusilla_Field Sparrow', 0.052776225)
12:52:11---Files loaded
12:52:12---RECORDING_LENGTH set to 15
12:52:12---NEWFILE=House_Finch-73-2022-05-23-birdnet-18:26:51.mp3
12:52:12---Extraction exists. Moving on
12:52:12---(standard_in) 1: syntax error
12:52:12---NEWFILE=-0-2022-05-23-birdnet-18:29:36.mp3
12:52:12---(standard_in) 1: syntax error
12:52:12---(standard_in) 1: syntax error
12:52:12---sox FAIL trim: Error parsing position 2
12:52:12---sox FAIL trim: usage: {position}
12:52:19---NEWFILE=House_Finch-73-2022-05-23-birdnet-18:26:51.mp3
12:52:19---Extraction exists. Moving on
12:52:19---(standard_in) 1: syntax error
12:52:19---NEWFILE=-0-2022-05-23-birdnet-18:29:36.mp3
12:52:19---(standard_in) 1: syntax error
12:52:19---(standard_in) 1: syntax error
12:52:19---sox FAIL trim: Error parsing position 2
12:52:19---sox FAIL trim: usage: {position}
```

I added an echo of the `sox` command line (in `extract_new_birdsounds.sh`):

```
12:59:58---sox -V1 BirdSongs/May-2022/23-Monday-Analyzed/2022-05-23-birdnet-18:29:36.wav BirdSongs/Extracted/By_Date/2022-05-26//-0-2022-05-23-birdnet-18:29:36.mp3 trim =0 =
```

I think that this command fails, so it never deletes the extracted csvs, and then the next iteration runs into exactly the same problem. I suspect that either the `sox` command should not have been called, or that the error return should trigger a `continue`.

I added:
```
    if [ -z "${END}" ]; then
      continue
    fi
``` 
above the sox command, and now it appears to be processing much better

mcguirepr89, 2022-05-26:
Then you very likely not only had corrupted audio files but corrupted CSV text files, which is why the variables could not be created. I suspect that removing ALL corrupted files will help a great deal, though your solution certainly prevents things from failing so miserabily.

If you are unable to locate the corrupted csv, you can try this to get rid of it:
```
sudo rm -drfv ~/BirdSongs/*/*-Analyzed
```
and that ought to clear out any bad files (though it won't prevent them form being created again on power-cuts and other accidents).

closed 2022-05-29 by mcguirepr89