mcguirepr89/BirdNET-Pi #319: Services flapping

lloydbayley, 2022-05-26:
G'day! 

I'm not sure if this is by design or not but if I turn the Pi off to move it or re-run a power cord etc etc - let's just say after a shutdown, the arecord and ffmpeg services seem to be up and down. Whilst they are down, no recording takes place, which is understandable. Sometimes they don't start again and I need to either reboot the Pi or restart all services via the web menu.
After a few goes at this, it seems to settle down and is fine again until the Pi is rebooted.

Is it normal for the service to stop and re-launch?

My diagnostic mind says it could be a mic issue but it happens even if I change mics.

Apart from the "Well don't reboot it" solution. Any ideas? :)

lloydbayley, 2022-05-27:
It's bloody amazing. It's been annoying me for ages and I've done it twice this afternoon and each time it just gets up and boogies without a problem. It's playing with my mind!!! Will try it again later on. Or did you sneak in to my system and fix it? :) *Humour Disclaimer: Patrick does NOT gain access to systems and fix it unless invited.

lloydbayley, 2022-05-27:
Ok. I had to wait some time for it to happen this time but it has started flapping. Log attached.
[lloyd_log_since_boot.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/8784818/lloyd_log_since_boot.txt)

mcguirepr89, 2022-05-27:
howdy, Lloyd -- unfortunately, the link to your log file give me a "Not Found" response -- if you'd prefer emailing the log, that is fine, too.

lloydbayley, 2022-05-27:
Well that's just not cricket is it?! Tsk! Will email as you suggested.

phrozen77, 2022-05-27:
I think i'm seeing a similar issue, unless that is if it is by design:

The "Spectogram Viewer" service is getting stopped and started all the time, watching `journalctl -xef` almost makes you go dizzy...

```
Subject: Automatic restarting of a unit has been scheduled
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ Automatic restarting of the unit spectrogram_viewer.service has been scheduled, as the result for
░░ the configured Restart= setting for the unit.
May 27 19:55:34 birdnetpi systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
░░ Subject: A stop job for unit spectrogram_viewer.service has finished
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A stop job for unit spectrogram_viewer.service has finished.
░░ 
░░ The job identifier is 113952 and the job result is done.
May 27 19:55:34 birdnetpi systemd[1]: spectrogram_viewer.service: Consumed 1.084s CPU time.
░░ Subject: Resources consumed by unit runtime
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit spectrogram_viewer.service completed and consumed the indicated resources.
May 27 19:55:34 birdnetpi systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
░░ Subject: A start job for unit spectrogram_viewer.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit spectrogram_viewer.service has finished successfully.
░░ 
░░ The job identifier is 113952.
May 27 19:55:35 birdnetpi systemd[1]: spectrogram_viewer.service: Succeeded.
░░ Subject: Unit succeeded
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit spectrogram_viewer.service has successfully entered the 'dead' state.

```

Also, it kinda seems a flock of Sturnus vulgaris made my poor Pi 3b+ getting backlogged in the analysis queue by around 2h20m :(

Is there a way to get an overview of the backlog, i.e. does BirdNET-Pi know how many items are backlogged? Any chance to speed that up?

mcguirepr89, 2022-05-27:
The spectrogram_viewer.service ought to restart often as it just updates the "Currently Analyzing" image on the Overview page. So those can all be safely ignored.

Please forgive me if this is presumptuous and wrong, but am I correct in assuming that you're `POST`ing to BirdWeather? I assume this because this tends only to happen when the system has to `POST` a ton of data, which introduces a good bit of latency.

But, to answer your question, there is not a way to temporarily make it run any faster to catch up, other than removing your BirdWeather ID for now and then going to "Tools" > "Services" > "Restart BirdNET Analysis Client." You can add it again after the system has caught up, or wait for a fix on that (it's in the works).

Again, sorry if you're not using a BirdWeather ID, as all of that would not apply.

If you would like to see how many files are left to analyze, you can look using the command below, but the timestamp you see in the logs for the currently analyzing file might be just as helpful but easier to obtain indication of the backlog:
```
source /etc/birdnet/birdnet.conf
find $RECS_DIR -name '*Processed*' -prune -o ! -name '*Analyzed' -a -name '*wav' -print
```

phrozen77, 2022-05-27:
> Please forgive me if this is presumptuous and wrong, but am I correct in assuming that you're POSTing to BirdWeather? I assume this because this tends only to happen when the system has to POST a ton of data, which introduces a good bit of latency.

No, i don't have BirdWeather configured - never had, so that can't be the issue here - sorry for not providing more details!

`find $RECS_DIR -name '*Processed*' -prune -o ! -name '*Analyzed' -a -name '*wav' -print | wc -l
858
`
Threw in a `wc -l` for good measure - ugh... 

The 16GB SD card is pretty filled (around 1GB left) and iowait is pretty high... (around 20% average i'd say)...

Load is at around 6.6 - not sure how that poor fella got into that state - looking for recommendations on how to save my Pi(e) ;)

mcguirepr89, 2022-05-27:
>looking for recommendations on how to save my Pi(e)

Super hard to say what is going on that would cause all that latency _and_ not be able to catch up? I'm available for the next hour if you'd like to do a screen session for me to take a look, otherwise it might help for me to take a look at some general log output.

Can I ask when(ish) you installed and when (if ever) was your most recent update (BirdNET-Pi Update, not packages/apt updates)?

phrozen77, 2022-05-27:
I think the installation is barely 2 weeks old now, i did update once in between - seems `git log --oneline` thinks i'm at commit 2739d59....

I appreciate your dedication and willingness to take a look but right now i don't have the time to set something up :(

Which logfile would help you analysing this further?

mcguirepr89, 2022-05-27:
For now, it could be helpful to see the output of:

```
journalctl --no-hostname --no-pager -n300 -ubirdnet_analysis -ubirdnet_server -uextraction
```

and

```
top -cbn1 -o %CPU | head -n15
```

phrozen77, 2022-05-27:
`birdnetpi:~/BirdNET-Pi$ journalctl --no-hostname --no-pager -n300 -ubirdnet_analysis -ubirdnet_server -uextraction | sed "s|$HOME||g;/CADDY_PWD/d"`

```
-- Journal begins at Thu 2022-05-19 16:20:01 CEST, ends at Fri 2022-05-27 21:33:24 CEST. --
May 27 21:28:39 birdnet_analysis.sh[99949]: 8.0;11.0-('Delichon urbicum_Mehlschwalbe', 0.20944326)
May 27 21:28:39 birdnet_analysis.sh[99949]: 10.0;13.0-('Passer montanus_Feldsperling', 0.14899124)
May 27 21:28:39 birdnet_analysis.sh[99949]: 12.0;15.0-('Acrocephalus scirpaceus_Teichrohrsänger', 0.14699988)
May 27 21:28:40 birdnet_analysis.sh[97144]: RECORDING_LENGTH set to 15
May 27 21:28:40 birdnet_analysis.sh[97144]: analyze.py --i /BirdSongs/May-2022/27-Friday/2022-05-27-birdnet-19:06:04.wav --o /BirdSongs/May-2022/27-Friday/2022-05-27-birdnet-19:06:04.wav.csv --lat 49.5705 --lon 11.7747 --week 20 --overlap 1 --sensitivity 1.25 --min_conf 0.7 --exclude_list /BirdNET-Pi/exclude_species_list.txt
May 27 21:28:41 server.py[14985]: ANALYZING AUDIO... DONE! Time 8.7 SECONDS
May 27 21:28:41 server.py[14985]: WRITING RESULTS TO /BirdSongs/May-2022/27-Friday/2022-05-27-birdnet-19:05:49.wav.csv ... DONE! WROTE 0 RESULTS.
May 27 21:28:41 server.py[14985]: [NEW CONNECTION] ('127.0.0.1', 34968) connected.
May 27 21:28:41 server.py[14985]: [ACTIVE CONNECTIONS] 1
May 27 21:28:41 server.py[14985]: READING AUDIO DATA... DONE! READ 7 CHUNKS.
May 27 21:28:41 server.py[14985]: FULL FILENAME: -/BirdSongs/May-2022/27-Friday/2022-05-27-birdnet-19:06:04.wav-
May 27 21:28:41 server.py[14985]: Date-time: 2022-05-27 19:06:04
May 27 21:28:50 birdnet_analysis.sh[99981]: 0.0;3.0-('Corvus frugilegus_Saatkrähe', 0.050609007)
May 27 21:28:50 birdnet_analysis.sh[99981]: 2.0;5.0-('Acrocephalus scirpaceus_Teichrohrsänger', 0.06273122)
May 27 21:28:50 birdnet_analysis.sh[99981]: 4.0;7.0-('Acrocephalus scirpaceus_Teichrohrsänger', 0.09662563)
May 27 21:28:50 birdnet_analysis.sh[99981]: 6.0;9.0-('Tetrao urogallus_Auerhuhn', 0.0538763)
May 27 21:28:50 birdnet_analysis.sh[99981]: 8.0;11.0-('Tetrao urogallus_Auerhuhn', 0.0487484)
May 27 21:28:50 birdnet_analysis.sh[99981]: 10.0;13.0-('Dryocopus martius_Schwarzspecht', 0.04879905)
May 27 21:28:50 birdnet_analysis.sh[99981]: 12.0;15.0-('Passer montanus_Feldsperling', 0.22729622)
May 27 21:28:50 birdnet_analysis.sh[97144]: RECORDING_LENGTH set to 15
May 27 21:28:51 birdnet_analysis.sh[97144]: analyze.py --i /BirdSongs/May-2022/27-Friday/2022-05-27-birdnet-19:06:19.wav --o /BirdSongs/May-2022/27-Friday/2022-05-27-birdnet-19:06:19.wav.csv --lat 49.5705 --lon 11.7747 --week 20 --overlap 1 --sensitivity 1.25 --min_conf 0.7 --exclude_list /BirdNET-Pi/exclude_species_list.txt
May 27 21:28:51 server.py[14985]: ANALYZING AUDIO... DONE! Time 8.9 SECONDS
May 27 21:28:51 server.py[14985]: WRITING RESULTS TO /BirdSongs/May-2022/27-Friday/2022-05-27-birdnet-19:06:04.wav.csv ... DONE! WROTE 0 RESULTS.
May 27 21:28:51 server.py[14985]: [NEW CONNECTION] ('127.0.0.1', 34970) connected.
May 27 21:28:51 server.py[14985]: [ACTIVE CONNECTIONS] 1
May 27 21:28:52 server.py[14985]: READING AUDIO DATA... DONE! READ 7 CHUNKS.
May 27 21:28:52 server.py[14985]: FULL FILENAME: -/BirdSongs/May-2022/27-Friday/2022-05-27-birdnet-19:06:19.wav-
May 27 21:28:52 server.py[14985]: Date-time: 2022-05-27 19:06:19
May 27 21:29:01 birdnet_analysis.sh[99991]: 0.0;3.0-('Delichon urbicum_Mehlschwalbe', 0.14704272)
May 27 21:29:01 birdnet_analysis.sh[99991]: 2.0;5.0-('Delichon urbicum_Mehlschwalbe', 0.13724996)
May 27 21:29:01 birdnet_analysis.sh[99991]: 4.0;7.0-('Cettia cetti_Seidensänger', 0.11886762)
May 27 21:29:01 birdnet_analysis.sh[99991]: 6.0;9.0-('Passer montanus_Feldsperling', 0.14519536)
May 27 21:29:01 birdnet_analysis.sh[99991]: 8.0;11.0-('Fulmarus glacialis_Eissturmvogel', 0.07172574)
May 27 21:29:01 birdnet_analysis.sh[99991]: 10.0;13.0-('Fulmarus glacialis_Eissturmvogel', 0.14992985)
May 27 21:29:01 birdnet_analysis.sh[99991]: 12.0;15.0-('Phasianus colchicus_Jagdfasan', 0.06050477)
May 27 21:29:01 birdnet_analysis.sh[97144]: RECORDING_LENGTH set to 15
May 27 21:29:02 birdnet_analysis.sh[97144]: analyze.py --i /BirdSongs/May-2022/27-Friday/2022-05-27-birdnet-19:06:48.wav --o /BirdSongs/May-2022/27-Friday/2022-05-27-birdnet-19:06:48.wav.csv --lat 49.5705 --lon 11.7747 --week 20 --overlap 1 --sensitivity 1.25 --min_conf 0.7 --exclude_list /BirdNET-Pi/exclude_species_list.txt
May 27 21:29:03 server.py[14985]: ANALYZING AUDIO... DONE! Time 8.8 SECONDS
May 27 21:29:03 server.py[14985]: WRITING RESULTS TO /BirdSongs/May-2022/27-Friday/2022-05-27-birdnet-19:06:19.wav.csv ... DONE! WROTE 0 RESULTS.
May 27 21:29:03 server.py[14985]: [NEW CONNECTION] ('127.0.0.1', 34972) connected.
May 27 21:29:03 server.py[14985]: [ACTIVE CONNECTIONS] 1
May 27 21:29:03 server.py[14985]: READING AUDIO DATA... DONE! READ 7 CHUNKS.
May 27 21:29:03 server.py[14985]: FULL FILENAME: -/BirdSongs/May-2022/27-Friday/2022-05-27-birdnet-19:06:48.wav-
May 27 21:29:03 server.py[14985]: Date-time: 2022-05-27 19:06:48
May 27 21:29:30 birdnet_analysis.sh[100001]: 0.0;3.0-('Vanellus vanellus_Kiebitz', 0.05914241)
May 27 21:29:30 birdnet_analysis.sh[100001]: 2.0;5.0-('Botaurus stellaris_Rohrdommel', 0.049387187)
May 27 21:29:30 birdnet_analysis.sh[100001]: 4.0;7.0-('Fratercula arctica_Papageitaucher', 0.06835793)
May 27 21:29:30 birdnet_analysis.sh[100001]: 6.0;9.0-('Fratercula arctica_Papageitaucher', 0.059564903)
May 27 21:29:30 birdnet_analysis.sh[100001]: 8.0;11.0-('Fratercula arctica_Papageitaucher', 0.19705647)
[... cut ...]
  14985         20   0 1680072 551540  41860 S   9.5  59.2 396:56.76 server.py
     13 root      20   0       0      0      0 I   4.8   0.0   4:03.99 rcu_preempt
  68723 root       0 -20       0      0      0 D   4.8   0.0   1:39.85 kworker/2:2H+kblockd
  99927 root      20   0       0      0      0 I   4.8   0.0   0:00.98 kworker/0:0-events
 101592         20   0  549036 148316  15428 D   4.8  15.9   1:10.92 daily_plot.py
      1 root      20   0  165940   4888   2476 S   0.0   0.5   2:04.74 systemd

```

mcguirepr89, 2022-05-27:
That all looked normal _except_ that it looks like your swap might be full. 

Can you share the output for:
```
df -h
```
and
```
free -h
```

phrozen77, 2022-05-27:
Sure:

```
birdnetpi:~/BirdNET-Pi$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G   13G  1.4G  91% /
devtmpfs        326M     0  326M   0% /dev
tmpfs           455M     0  455M   0% /dev/shm
tmpfs           182M  816K  182M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  253M   32M  221M  13% /boot
tmpfs            91M  4.0K   91M   1% /run/user/1000

birdnetpi:~/BirdNET-Pi$ free -h
               total        used        free      shared  buff/cache   available
Mem:           909Mi       765Mi        26Mi        14Mi       118Mi        73Mi
Swap:           99Mi        99Mi          0B
```

mcguirepr89, 2022-05-27:
AHA!! The RAM is the culprit -- let's fix (mitigate) that with some swap space. It looks like you have 1 free Gigabyte available, let's use it.

Please run this and when the system comes back up, I suspect it will be able to catch up:
```
sudo sed -i 's/CONF_SWAPSIZE=100/CONF_SWAPSIZE=1024/g' /etc/dphys-swapfile &&
sudo sed -i 's/#CONF_MAXSWAP=2048/CONF_MAXSWAP=1024/g' /etc/dphys-swapfile &&
sudo reboot
```

mcguirepr89, 2022-05-27:
After your system is able to catch up (:crossed_fingers:) I would also recommend running an update via "Tools" > "System Controls" > "Update". If you encounter any issues with the first update, try updating once more and copy the output it provides to supply in another issue (or the second update will fix everything).

phrozen77, 2022-05-28:
Swap has been expanded, right now it's still lagging behind by 1:53:

`May 28 06:59:02 birdnetpi server.py[396]: Date-time: 2022-05-28 05:06:33`

But the number of items in the queue has dropped significantly:

```
birdnetpi:~/BirdNET-Pi$ find $RECS_DIR -name '*Processed*' -prune -o ! -name '*Analyzed' -a -name '*wav' -print | wc -l
472
```

> I would also recommend running an update via "Tools" > "System Controls" > "Update"

Will do :)

Thank you *so* much for your support!

lloydbayley, 2022-05-28:
I'm pleased to report that it was my blind following of an article about BirdNet-Pi and installing the GUI version of the OS that seems to have been vying for PulseAudio's attention. My fault and I did wonder at the time why I needed to install the GUI version but I followed along. Tip for young players, DON'T install the gui version unless you would like it for some other reason. This project functions perfectly in CLI mode.

Patrick very kindly suggested the fix.
I had to make sure that the multiuser target was the systemd default and then force a tty0 login through getty.

Not sure if you want to elaborate Patrick, in case it happens to someone else, I will leave it to you. Don't want to step on toes here. :)

mcguirepr89, 2022-05-28:
@phrozen77 I'm glad to hear that you're seeing some improvement. How is it looking 9 hours later?

@lloydbayley Thanks for the comment -- I'll add a few notes:

For anyone whose installation went atop the full Desktop version of RaspiOS (64-bit of course) and is seeing issues, do the following to ensure that PulseAudio (the `pulse` user) does not have to compete for access to a user runtime directory:

1. Ensure that the system boots into the multi-user.target
   ```
   sudo systemctl set-defaut multi-user.target
   ```
1. Ensure the system is configured to automatically login the default user to TTY0 via getty:
   ```
   sudo ln -fs /lib/systemd/system/getty@.service /etc/systemd/system/getty.target.wants/getty@tty1.service
   cat | sudo tee /etc/systemd/system/getty@tty1.service.d/autologin.conf << EOF
   [Service]
   ExecStart=
   ExecStart=-/sbin/agetty --autologin $USER --noclear %I \$TERM
   EOF
   ```
1. Reboot: ```sudo reboot```

closed 2022-05-28 by mcguirepr89

phrozen77, 2022-05-28:
Now it has caught up - it's back to processing in near-realtime :)

Thanks!