mcguirepr89/BirdNET-Pi #324: installation failed: ./install_language_label.sh: No such file or directory

moffat, 2022-05-27:
**Describe the bug**

Installation of BirdNET-Pi fails

**To Reproduce**

run installer. 

**Expected behavior**

installation should have finished successfully, instead it fails with:

```bash
      Successfully uninstalled pip-20.3.4
Successfully installed MarkupSafe-2.1.1 PyYAML-6.0 Send2Trash-1.8.0 altair-4.2.0 appdirs-1.4.4 apprise-0.9.8.3 argon2-cffi-21.3.0 argon2-cffi-bindings-21.2.0 asttokens-2.0.5 attrs-21.4.0 audioread-2.1.9 backcall-0.2.0 beautifulsoup4-4.11.1 bleach-5.0.0 blinker-1.4 cachetools-5.1.0 certifi-2022.5.18.1 cffi-1.15.0 charset-normalizer-2.0.12 click-8.0.4 colorama-0.4.4 cycler-0.11.0 debugpy-1.6.0 decorator-5.1.1 defusedxml-0.7.1 entrypoints-0.4 executing-0.8.3 fastjsonschema-2.15.3 fonttools-4.33.3 gitdb-4.0.9 gitpython-3.1.27 idna-3.3 importlib-metadata-4.11.4 ipykernel-6.13.0 ipython-8.3.0 ipython-genutils-0.2.0 ipywidgets-7.7.0 jedi-0.18.1 jinja2-3.1.2 joblib-1.1.0 jsonschema-4.5.1 jupyter-client-7.3.1 jupyter-core-4.10.0 jupyterlab-pygments-0.2.2 jupyterlab-widgets-1.1.0 kiwisolver-1.4.2 librosa-0.9.1 llvmlite-0.38.1 markdown-3.3.7 matplotlib-3.5.2 matplotlib-inline-0.1.3 mistune-0.8.4 nbclient-0.6.3 nbconvert-6.5.0 nbformat-5.4.0 nest-asyncio-1.5.5 notebook-6.4.11 numba-0.55.2 numpy-1.19.5 oauthlib-3.2.0 packaging-21.3 pandas-1.4.2 pandocfilters-1.5.0 parso-0.8.3 pexpect-4.8.0 pickleshare-0.7.5 pillow-9.1.1 pip-22.1.1 plotly-5.8.0 pooch-1.6.0 prometheus-client-0.14.1 prompt-toolkit-3.0.29 protobuf-3.20.1 psutil-5.9.1 ptyprocess-0.7.0 pure-eval-0.2.2 pyarrow-8.0.0 pycparser-2.21 pydeck-0.7.1 pygments-2.12.0 pympler-1.0.1 pyparsing-3.0.9 pyrsistent-0.18.1 python-dateutil-2.8.2 pytz-2022.1 pytz-deprecation-shim-0.1.0.post0 pyzmq-23.0.0 requests-2.27.1 requests-oauthlib-1.3.1 resampy-0.2.2 scikit-learn-1.1.1 scipy-1.8.1 seaborn-0.11.2 semver-2.13.0 setuptools-62.3.2 six-1.16.0 smmap-5.0.0 soundfile-0.10.3.post1 soupsieve-2.3.2.post1 stack-data-0.2.0 streamlit-1.9.2 tenacity-8.0.1 terminado-0.15.0 tflite-runtime-2.6.0 threadpoolctl-3.1.0 tinycss2-1.1.1 toml-0.10.2 toolz-0.11.2 tornado-6.1 traitlets-5.2.1.post0 typing-extensions-4.2.0 tzdata-2022.1 tzlocal-4.2 urllib3-1.26.9 validators-0.19.0 watchdog-2.1.8 wcwidth-0.2.5 webencodings-0.5.1 wheel-0.37.1 widgetsnbextension-3.6.0 zipp-3.8.0
+ ./install_language_label.sh -l en
/home/birder/BirdNET-Pi/scripts/install_birdnet.sh: line 38: ./install_language_label.sh: No such file or directory
+ exit 1
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
[installation-2022-05-27.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/8790078/installation-2022-05-27.txt)

ehpersonal38, 2022-05-27:
I think `cd $my_dir/scripts || exit 1` needs to be added right above where install_language_label.sh is called in install_birdnet.sh

moffat, 2022-05-28:
I managed to run the last failing command manually, which seems to be the last command of the sequence...

mcguirepr89, 2022-05-28:
I'm just now seeing this -- I'll hopefully have this fixed within the next couple of hours or less

mcguirepr89, 2022-05-28:
Hi @moffat I made an adjustment and am running a test installation now. If it succeeds I will let you know at which point you should be all set to bird with us.

mcguirepr89, 2022-05-28:
Hello again -- the installer is fixed -- please give it another go on a newly flashed SD card and you should have success.

Sorry for the inconvenience!!!
and please accept my very best regards and gratitude for your intent to install BirdNET-Pi,
Patrick McGuire

closed 2022-05-28 by mcguirepr89

moffat, 2022-05-28:
Hi @mcguirepr89 thanks for the speedy response. At the end I ran that past command manually and the installation seemed to have finished fine. 

It's working great now, such a great project!

mcguirepr89, 2022-05-28:
That's great to hear and thank you for the kind praise -- you're just now getting your beak wet (whet?) (I can never resist a bird pun on this repo), but as you start collecting great data, I hope you will feel invited to [share some of your favorites](https://github.com/mcguirepr89/BirdNET-Pi/discussions/69) with the group and will consider [making your installation public](https://github.com/mcguirepr89/BirdNET-Pi/wiki/Sharing-Your-BirdNET-Pi) :)

My best regards,
Patrick