mcguirepr89/BirdNET-Pi #327: Static noise every second after about 2 days of uptime

phrozen77, 2022-05-28:
Setup: Pi 3b+, Ugreen 2-in-1 Soundcard, AOM-5024L-HD-R mic, USB extension cord for soundcard, laboratory power supply (as each other PS i had around introduced static noise, this one doesn't but is only a stopgap measure for now...)

So, after running for a while (usually 2 to 3 days, sometime also after a half day) i get static noise on the soundcard in one second intervals - in the spectrogram viewer it looks like this:

[image]

Soundfile of the issue:
[2022-05-28-birdnet-16:37:46.wav.zip](https://github.com/mcguirepr89/BirdNET-Pi/files/8791324/2022-05-28-birdnet-16.37.46.wav.zip)

I wonder what that problem is, as i usually solve that (reliably) by rebooting - once it comes up again, the noise isn't there anymore...

mcguirepr89, 2022-05-28:
@DD4WH I'm hoping you might have some insight here since I've not seen this elsewhere.

oliverk71, 2022-05-29:
@phrozen77 Did you make any changes to the original installation? Cronjobs maybe?

phrozen77, 2022-05-29:
No, no changes at all - basically i did a vanilla RaspiOS install and then BirdNET-Pi via the "pipe-to-shell" installer...

oliverk71, 2022-05-29:
Maybe you would like to try the following: Go to tools > settings, click on advanced and enter the device number of your USB audio card. As written there, you can find out the device number of your card using the command `aplay -l` in a shell (web terminal or SSH).

mcguirepr89, 2022-05-29:
>Maybe you would like to try the following: Go to tools > settings, click on advanced and enter the device number of your USB audio card. As written there, you can find out the device number of your card using the command aplay -l in a shell (web terminal or SSH).

Please don't use the device number as those are inconsistent across boots -- the kernel will allocate the device number based on when the driver is loaded, which is means that you cannot guarantee that the device will be given the same number next boot.

phrozen77, 2022-05-29:
This is currently set to "default" (as suggested or even recommended on the settings page) - i wonder what the drawbacks are if i set it to one of these:

```
hw:CARD=Audio,DEV=0
    KT USB Audio, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Audio,DEV=0
    KT USB Audio, USB Audio
    Hardware device with all software conversions
sysdefault:CARD=Audio
    KT USB Audio, USB Audio
    Default Audio Device
front:CARD=Audio,DEV=0
    KT USB Audio, USB Audio
    Front output / input
surround21:CARD=Audio,DEV=0
    KT USB Audio, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Audio,DEV=0
    KT USB Audio, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Audio,DEV=0
    KT USB Audio, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Audio,DEV=0
    KT USB Audio, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Audio,DEV=0
    KT USB Audio, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Audio,DEV=0
    KT USB Audio, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Audio,DEV=0
    KT USB Audio, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Audio,DEV=0
    KT USB Audio, USB Audio
    Direct sample mixing device
usbstream:CARD=Audio
    KT USB Audio
    USB Stream Output
```
Tbqh, i wonder why it "just works" until it doesn't :(

mcguirepr89, 2022-05-29:
The proper ALSA information that _can_ be supplied in your "Advanced Settings" would be from the device _name_:
```
arecord -L
```
gives you that output.

phrozen77, 2022-05-29:
That brings the list down to

```
hw:CARD=Audio,DEV=0
    KT USB Audio, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Audio,DEV=0
    KT USB Audio, USB Audio
    Hardware device with all software conversions
sysdefault:CARD=Audio
    KT USB Audio, USB Audio
    Default Audio Device
front:CARD=Audio,DEV=0
    KT USB Audio, USB Audio
    Front output / input
dsnoop:CARD=Audio,DEV=0
    KT USB Audio, USB Audio
    Direct sample snooping device
usbstream:CARD=Audio
    KT USB Audio
    USB Stream Output

```

@mcguirepr89 do you reckon that'd be worth giving it a shot? Which device should be the right one? usbstream?

mcguirepr89, 2022-05-29:
@phrozen77 

If I had to **_guess_**, I would think that the noise is coming from your "laboratory power supply" -- that is a guess simply because no other installations that I've encountered that _do not_ feed power into their sound card never see this issue.

As an experiment, the next time you see the issue, please power-cycle the "laboratory power supply" and see if that resolves the rogue audio signals you're seeing.

If you wanted to TRY to bypass Pulse and go straight to ALSA (which I wouldn't expect to yield better results, since Pulse is an ALSA wrapper), you can try these, in this order:
```
dsnoop:CARD=Audio,DEV=0
```
```
usbstream:CARD=Audio
```
or
```
hw:CARD=Audio,DEV=0
```

They will likely all work, but the last option will prevent you from also using the Live Audio Stream as it will monopolize that card for 1 process only (recording).

phrozen77, 2022-05-29:
> If I had to guess, I would think that the noise is coming from your "laboratory power supply"

That very well might be the case..

> As an experiment, the next time you see the issue, please power-cycle the "laboratory power supply" and see if that resolves the rogue audio signals you're seeing.

Well, since the Pi is going to get rebooted as well i'd say "yeah, it will" - however, note, that the issue "resolves itself" by just rebooting the Pi, power supply is untouched and on all the time...

mcguirepr89, 2022-05-29:
@phrozen77 

>Well, since the Pi is going to get rebooted as well i'd say "yeah, it will" - however, note, that the issue "resolves itself" by just rebooting the Pi, power supply is untouched and on all the time...

Is the "laboratory power supply" just for the audio card, or is that the Raspberry Pi's power-supply? I think I may have misunderstood the original comment

phrozen77, 2022-05-29:
No, that's the power supply for the Pi - the soundcard is this one: https://www.aliexpress.com/item/4001299124074.html and doesn't have a separate power supply.

FWIW, the laboratory power supply is capable of 30V/3A and is this one: https://www.pollin.de/p/quatpower-regelbares-labornetzgeraet-ln-3003-0-30-v-0-3-a-351425

mcguirepr89, 2022-05-29:
And just so you know my logic on suspecting the power supply:

I have built several Bluetooth Speakers sets from scratch using all kinds of different parts -- purchased, found, modified, built from scratch -- and noticed that cheap power supplies have capacitors inside that will fill and discharge and fill and discharge and fill and discharge after sustained use. Eventually the bluetooth would be able to pick up these discharges as little ticks in the audio.

So!!! That's the root of my logic -- I could be SOOOOOOOOOOOO far off from anything that is actually happening, but I just wanted you to know my thoughts so that you can decide for yourself what makes sense

mcguirepr89, 2022-05-29:
>No, that's the power supply for the Pi - the soundcard is this one: https://www.aliexpress.com/item/4001299124074.html and doesn't have a separate power supply.

Ok, I understand now -- most of my suppositions were based off thinking your sound card had a separate power supply :thinking: hmmmmm

oliverk71, 2022-05-29:
> Please don't use the device number as those are inconsistent across boots -- the kernel will allocate the device number based on when the driver is loaded, which means that you cannot guarantee that the device will be given the same number next boot.

I am sorry, I meant the card number NOT the device number! 
Here is a part of my output of `aplay -l` citing my USB audio card:
```
card 1: Audio [USB Audio], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
```
So I used 1 and not 0 in my audio card settings which at least helped me getting rid of the crackling noise.

mcguirepr89, 2022-05-29:
Again, please don't use the card number, as that number can change on the next boot.

oliverk71, 2022-05-29:
I wonder about the correct syntax, now
This is my output from arecord -l:
```
null
    Discard all samples (playback) or generate zero samples (capture)
default
    Playback/recording through the PulseAudio sound server
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
usbstream:CARD=Headphones
    bcm2835 Headphones
    USB Stream Output
hw:CARD=Audio,DEV=0
    USB Audio, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Audio,DEV=0
    USB Audio, USB Audio
    Hardware device with all software conversions
sysdefault:CARD=Audio
    USB Audio, USB Audio
    Default Audio Device
front:CARD=Audio,DEV=0
    USB Audio, USB Audio
    Front output / input
dsnoop:CARD=Audio,DEV=0
    USB Audio, USB Audio
    Direct sample snooping device
usbstream:CARD=Audio
    USB Audio
    USB Stream Output
usbstream:CARD=vc4hdmi
    vc4-hdmi
    USB Stream Output
```
So if I enter `hw:CARD=Audio,DEV=0` in the settings, BirdNET-Pi will use the USB soundcard?
Then I will change it, now.

mcguirepr89, 2022-05-29:
Again, the `alsa` command to view the device NAMEs that _can_ be used in "Advanced Settings" is:
```
arecord -L
```
notice that it is NOT a lower-case "l", but a Capital "L" for the flag
```
arecord -L
```

oliverk71, 2022-05-29:
> notice that it is NOT a lower-case "l", but a Capital "L" for the flag

Yes, true. I used capital L, but wrote lower-case l in above post. Sorry for that.

mcguirepr89, 2022-05-29:
@phrozen77 to verify that the kernel has not had anything bad to say about the power supply, could you go to "Tools" > "System Info" and scroll down to the "Viewer" section and ensure that "throttled=0x0" is indicated?
[image]

lloydbayley, 2022-05-29:
If you're getting recordings, I can't see how it would be the wrong device. I listened to your sample file and you can hear the birds in the background. It's more likely to be interference and could be anything up to capacitive noise from the air. Microphones are shocking noise picker-uppers from anywhere. I have a few issues with ground hum here and there too.
In your case, there are a few options. I would first try temporarily getting rid of the extension usb cable just to see if it's in there. Also, you could try powering the pi from a usb powerbank so it's completely isolated but that is tricky to run it for a few days as yours only starts after a while. Question. Did you build the mic yourself using the AOM-5024L-HD-R component or is it a pre-made module?

phrozen77, 2022-05-29:
>If you're getting recordings, I can't see how it would be the wrong device.

It is not, it'd just for having another method to record the sound from the mic/soundcard - however, since it is currently at the recommended setting i don't think it should be changed.

>It's more likely to be interference and could be anything up to capacitive noise from the air.

Interference? Probably, but why and from what? Capacitive noise "from the air"? I don't think so, as it goes away once rebooted.

>ground hum

Yeah, had that with each of the other 5+ power supplies, one even with a cable wound around a ferrite core - which is negligible for the other Pis i have around but not for this application...

Hence why i opted for the laboratory power supply...

> I would first try temporarily getting rid of the extension usb cable

I'll try that once the issue is present the next time - and maybe just un- and replug the USB soundcard beforehand...

The reason the extension cord is there it is _explicitly_ recommended in having it, to extend the range between the Pi and the soundcard as a safeguard against noise introduced by the Pi...

> powering the pi from a usb powerbank 

I don't have one at hand, sorry...

> Did you build the mic yourself

Yes - it works flawlessly when this very issue is not present.

> and ensure that "throttled=0x0" is indicated?

Sadly, that's not the case - 0x5005 is indicated (so "Undervolted" now and "Throttled" yes), however that status is flapping - not sure why as the "current" knob is cranked all the way up, so the Pi should be able to get up to 3A, the display says it's pulling around 0.7A at 5.2V (so around 3.6W)...

lloydbayley, 2022-05-29:
ok. i was wrong. will fix

mcguirepr89, 2022-05-29:
>Sadly, that's not the case - 0x50005 is indicated (so "Undervolted" now and "Throttled" yes), however that status is flapping - not sure why as the "current" knob is cranked all the way up, so the Pi should be able to get up to 3A, the display says it's pulling around 0.7A at 5.2V (so around 3.6W)...

I hate to be the poopoo guy, but you'll want to get an official Raspberry Pi power supply to prevent issues like this :/

I have used REALLY great power supplies before, but have found they still won't meet the Raspberry Pi's needs as they're _not_ in keeping with other 5v power supplies.

While these kernel messages don't prove anything, they would certainly need to be addressed before we could rule out the power supply as the root cause, since the Pi is obviously having trouble with it.

phrozen77, 2022-05-29:
No worries, i recall having one somewhere - guess i gotta hunt it down *scratcheshead* if only i knew where it is...

phrozen77, 2022-05-29:
```
birdnetpi:~/BirdNET-Pi$ ../status.sh
Status: 0x20000
Undervolted:
   Now: NO
   Run: NO
Throttled:
   Now: NO
   Run: NO
Frequency Capped:
   Now: NO
   Run: YES
Softlimit:
   Now: NO
   Run: NO
birdnetpi:~/BirdNET-Pi$ uptime 
 16:12:37 up 3 min,  2 users,  load average: 4.00, 2.92, 1.26
```

phrozen77, 2022-05-29:
Great ☹️ now the ground hum is back..

Dendrocopos major from today morning:
[image]
[Buntspecht-87-2022-05-29-birdnet-15_24_33.mp3.zip](https://github.com/mcguirepr89/BirdNET-Pi/files/8793515/Buntspecht-87-2022-05-29-birdnet-15_24_33.mp3.zip)

...and from now (after the original RaspberryPi power supply has been plugged in):

[image]
[Buntspecht-95-2022-05-29-birdnet-16_20_02.mp3.zip](https://github.com/mcguirepr89/BirdNET-Pi/files/8793518/Buntspecht-95-2022-05-29-birdnet-16_20_02.mp3.zip)

mcguirepr89, 2022-05-29:
:/ sounds like another bad power supply to me

mcguirepr89, 2022-05-29:
Or you have a bad Pi

phrozen77, 2022-05-29:
>:/ sounds like another bad power supply to me

Well - we're talking about 5+ power supplies i've tested now, each and every one introduces hum as can be heard above - besides the laboratory one i've linked upthread...

I mean, each of the 5V power supplies work just fine for the other Pis i'm using them for, each one is rated 2.5A (or higher) - so... what are the chances _each_ of them is bad?

I guess i'll solder the Mean Well RS-25-5 i've ordered as a permanent solution (instead of the laboratory ps) to the Pis diagnostic points next - lets see what that does...

>Or you have a bad Pi

I've a few Pi 3b+ around, although each one is in operation... but i guess that very well may be the next wild goose chase then....

mcguirepr89, 2022-05-29:
>Well - we're talking about 5+ power supplies i've tested now, each and every one introduces hum as can be heard above - besides the laboratory one i've linked upthread...

>I mean, each of the 5V power supplies work just fine for the other Pis i'm using them for, each one is rated 2.5A (or higher) - so... what are the chances each of them being bad?

These great points you make all point to the Pi being bad, now that you put it that way.

Good luck!

closed 2022-05-29 by mcguirepr89

phrozen77, 2022-05-29:
I'm sorry if i offended you somehow - if that's the case, this was not my intention!

mcguirepr89, 2022-05-29:
I'm not personally offended, though your response style has stymied any reasonable troubleshooting. It should be obvious that hardware issues need to be addressed before we can think that there is a software issue. To that point, software support is within the scope of a project like this -- hardware support is a courtesy towards which I would hope you would show more patience. 

It is a courtesy also to extend the "benefit of the doubt" to others when they try to help, and if you feel their input is not helping, you can ask for more information/clarification since dismissal tends to beget more dismissal.

DD4WH, 2022-05-29:
@phrozen77 
Sorry fpr being late to reply!

There are two very different issues here:

1.) The noise that you call "ground hum" and which is clearly audible in your Woodpecker recording, which is present with original RPi power supply and goes away with the lab power supply:
* my suspicion is that you did not solder the mic according to my suggestion ;-). Could you give us a photo or a diagram showing the exact wiring and soldering of yourt AOM mic ? I think the wiring/shielding of the cable/mic is the reason for that noise.

2.) crackling with RPi3B+: Yes, I also experience this kind of crackling noise with the RPi3B+. It stays with every mic I own, so it is not related to the mic. It also is more frequent with longer ON-time of the RPi. It immediately goes away when I connect the very same soundcard and the very same mic to a Raspberry Pi 4B with the very same micro-SD-card. So, it is related to the hardware and not the software, because the software is exactly the same. 
However, I have no clue how to solve the crackling noise, except to change the RPi3B+ to a RPi4B.

Let us know about your mic soldering to solve problem 1.)

phrozen77, 2022-05-29:
> I think the wiring/shielding of the cable/mic is the reason for that noise.

Probably, yes - the cable i bought had 2 TRRS connectors of which i clipped one and soldered the 2 wires to the microphone capsule - there simply was no shielding i could've connected on one end and left unconnected on the other...

Not being an EE, i wonder why it works _perfectly_ with the laboratory ps (that was my last-ditch attempt after i've been basically through each available power supply in the house, i think i read it somewhere in a discussion that one user solved their otherwise unsolvable noise-issues with one...) but not with any of the wall-warts i've tried with...

Thanks for replying & helping (@all).

mcguirepr89, 2022-05-29:
Thanks, @DD4WH I really appreciate your hardware troubleshooting input and insight from experience