mcguirepr89/BirdNET-Pi #333: Caddy not serving webpage

drdougwelch, 2022-05-29:
First - this is an amazing creation and I have already enjoyed the fruits of it working! Thanks to everyone contributing.

I had BirdNET-Pi working properly until a recent complete re-install. It is clear that audio is being analyzed and MP3 and PNG files are being created. But nothing is being served up by caddy. I can reach port 80 by telnet and browser on my LAN but nothing is being returned.

Thanks in advance!!!
Doug

/etc/caddy/Caddyfile:
http://localhost http://birdnetpi.local {
  root * /home/hambirder/BirdSongs/Extracted
  file_server browse
  handle /By_Date/* {
    file_server browse
  }
  handle /Charts/* {
    file_server browse
  }
  reverse_proxy /stream localhost:8000
  php_fastcgi unix//run/php/php7.4-fpm.sock
  reverse_proxy /log* localhost:8080
  reverse_proxy /stats* localhost:8501
  reverse_proxy /terminal* localhost:8888
}

journalctl --no-hostname --no-pager -n100 -ucaddy:

birdnetpi:~/BirdNET-Pi$ journalctl --no-hostname --no-pager -n100 -ucaddy
-- Journal begins at Mon 2022-04-04 10:41:41 EDT, ends at Sun 2022-05-29 09:08:41 EDT. --
May 28 17:03:54 caddy[6799]: runtime.NumCPU=4
May 28 17:03:54 caddy[6799]: runtime.GOMAXPROCS=4
May 28 17:03:54 caddy[6799]: runtime.Version=go1.18.1
May 28 17:03:54 caddy[6799]: os.Getwd=/
May 28 17:03:54 caddy[6799]: LANG=en_GB.UTF-8
May 28 17:03:54 caddy[6799]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
May 28 17:03:54 caddy[6799]: NOTIFY_SOCKET=/run/systemd/notify
May 28 17:03:54 caddy[6799]: HOME=/var/lib/caddy
May 28 17:03:54 caddy[6799]: LOGNAME=caddy
May 28 17:03:54 caddy[6799]: USER=caddy
May 28 17:03:54 caddy[6799]: INVOCATION_ID=51e7f4ef13da495a8664ce0a6402e751
May 28 17:03:54 caddy[6799]: JOURNAL_STREAM=8:39259
May 28 17:03:54 caddy[6799]: {"level":"info","ts":1653771834.0780385,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","config_adapter":""}
May 28 17:03:54 caddy[6799]: {"level":"warn","ts":1653771834.0822742,"msg":"Caddyfile input is not formatted; run the 'caddy fmt' command to fix inconsistencies","adapter":"caddyfile","file":"/etc/caddy/Caddyfile","line":2}
May 28 17:03:54 caddy[6799]: {"level":"info","ts":1653771834.084975,"logger":"admin","msg":"admin endpoint started","address":"tcp/localhost:2019","enforce_origin":false,"origins":["//localhost:2019","//[::1]:2019","//127.0.0.1:2019"]}
May 28 17:03:54 caddy[6799]: {"level":"warn","ts":1653771834.0854454,"logger":"http","msg":"server is listening only on the HTTP port, so no automatic HTTPS will be applied to this server","server_name":"srv0","http_port":80}
May 28 17:03:54 caddy[6799]: {"level":"info","ts":1653771834.0862844,"logger":"tls.cache.maintenance","msg":"started background certificate maintenance","cache":"0x4000311180"}
May 28 17:03:54 caddy[6799]: {"level":"info","ts":1653771834.0880654,"logger":"tls","msg":"cleaning storage unit","description":"FileStorage:/var/lib/caddy/.local/share/caddy"}
May 28 17:03:54 caddy[6799]: {"level":"info","ts":1653771834.088147,"logger":"tls","msg":"finished cleaning storage units"}
May 28 17:03:54 caddy[6799]: {"level":"info","ts":1653771834.0884244,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}
May 28 17:03:54 caddy[6799]: {"level":"info","ts":1653771834.0885873,"msg":"serving initial configuration"}
May 28 17:03:54 systemd[1]: Started Caddy.
May 28 17:58:13 caddy[6799]: {"level":"info","ts":1653775093.276996,"msg":"shutting down apps, then terminating","signal":"SIGTERM"}
May 28 17:58:13 caddy[6799]: {"level":"warn","ts":1653775093.277129,"msg":"exiting; byeee!! 👋","signal":"SIGTERM"}
May 28 17:58:13 caddy[6799]: {"level":"info","ts":1653775093.278924,"logger":"tls.cache.maintenance","msg":"stopped background certificate maintenance","cache":"0x4000311180"}
May 28 17:58:13 caddy[6799]: {"level":"info","ts":1653775093.2814817,"logger":"admin","msg":"stopped previous server","address":"tcp/localhost:2019"}
May 28 17:58:13 caddy[6799]: {"level":"info","ts":1653775093.2815666,"msg":"shutdown complete","signal":"SIGTERM","exit_code":0}
May 28 17:58:13 systemd[1]: Stopping Caddy...
May 28 17:58:13 systemd[1]: caddy.service: Succeeded.
May 28 17:58:13 systemd[1]: Stopped Caddy.
-- Boot 2d5c1eeac61e4e8d9ee38f6c12fc0359 --
May 29 08:56:21 systemd[1]: Starting Caddy...
May 29 08:56:23 caddy[810]: caddy.HomeDir=/var/lib/caddy
May 29 08:56:23 caddy[810]: caddy.AppDataDir=/var/lib/caddy/.local/share/caddy
May 29 08:56:23 caddy[810]: caddy.AppConfigDir=/var/lib/caddy/.config/caddy
May 29 08:56:23 caddy[810]: caddy.ConfigAutosavePath=/var/lib/caddy/.config/caddy/autosave.json
May 29 08:56:23 caddy[810]: caddy.Version=v2.5.1 h1:bAWwslD1jNeCzDa+jDCNwb8M3UJ2tPa8UZFFzPVmGKs=
May 29 08:56:23 caddy[810]: runtime.GOOS=linux
May 29 08:56:23 caddy[810]: runtime.GOARCH=arm64
May 29 08:56:23 caddy[810]: runtime.Compiler=gc
May 29 08:56:23 caddy[810]: runtime.NumCPU=4
May 29 08:56:23 caddy[810]: runtime.GOMAXPROCS=4
[... 88 lines cut ...]
░░ Subject: Automatic restarting of a unit has been scheduled
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ Automatic restarting of the unit birdnet_analysis.service has been scheduled, as the result for
░░ the configured Restart= setting for the unit.
May 29 09:11:24 birdnetpi systemd[1]: Stopped BirdNET Analysis.
░░ Subject: A stop job for unit birdnet_analysis.service has finished
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A stop job for unit birdnet_analysis.service has finished.
░░ 
░░ The job identifier is 7463 and the job result is done.
May 29 09:11:24 birdnetpi systemd[1]: Started BirdNET Analysis.
░░ Subject: A start job for unit birdnet_analysis.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit birdnet_analysis.service has finished successfully.
░░ 
░░ The job identifier is 7463.
May 29 09:11:25 birdnetpi birdnet_analysis.sh[7572]: Files loaded
May 29 09:11:26 birdnetpi birdnet_analysis.sh[7572]: RECORDING_LENGTH set to 15
May 29 09:11:26 birdnetpi env[7619]: Line = 2022-05-29-birdnet-09:11:04.wav.csv

mcguirepr89, 2022-05-29:
Hello, @drdougwelch -- thanks for you kind words about BirdNET-Pi and for the great troubleshooting information you've provided! I'm sorry your new installation is giving you trouble. Here's a few things to try out to get a better idea of what might be going on.

1. When you do go to http://birdnetpi.local, what do you see? A white page? A message from your web browser? Whatever you see will help me know whether you reached the webserver or not
1. Is anything served when you go to http://birdnetpi.local/log?
1. When you `ping` `birdnetpi.local` from your host machine (Windows via the Command Prompt, or Linux/macOS from the Terminal), does it return the IP address that you know the installation to be using?

Those things are a good starting point for me to know best where to look next.

My best regards,
Patrick

drdougwelch, 2022-05-29:
>  2. Is anything served when you go to http://birdnetpi.local/log?
>  3. When you |ping| |birdnetpi.local| from your host machine (Windows

Hi Patrick,

Thanks for your quick reply!

1) Yes - http://birdnetpi.local does return the hoped-for page! (The LAN 
address used to do this, but does not anymore, despite being the correct 
interface from ifconfig -a.)

2) Yes - that does provide the expected log output.

3) Yes - ping does produce good turnaround replies and the "from" is 
from the correct LAN address.

So it looks like my problem was only that I was browsing with the LAN IP 
address! I'm not sure why that should fail but as long as I know not to 
do that, I'm good!!!

Best regards,
Doug

mcguirepr89, 2022-05-29:
>1) Yes - http://birdnetpi.local/ does return the hoped-for page! (The LAN
address used to do this, but does not anymore, despite being the correct
interface from ifconfig -a.)

Great, so to confirm, your installation **is** reachable at http://birdnetpi.local now?

If you'd like to make the installation respond to its IP address, you can go to "Tools" > "Settings" > "Advanced Settings" and input the installation's IP address (along with the http:// protocol) and "Update Settings" to have the installation also be reachable via its LAN IP address. There is an example IP address URL near those settings on the "Advanced Settings" page that you should be able to use as reference.

DougWelch, 2022-05-29:
Hi Patrick,

Yes it is reachable now. When I first started working with the package, 
I thought that you had to put the IP in the birdnet.conf to access the 
webpage, so that is how I got used to that incorrect assumption, FWIW!!!

So please go ahead and close this issue!

Thanks,
Doug

closed 2022-05-29 by mcguirepr89