mcguirepr89/BirdNET-Pi #341: Overview page doesnt reset top 10 at midnight

b0rnwr0ng, 2022-06-01:
The page automatically refreshes when new detections are made, along with the top 10 graph, but once the day is filled at misnight you need to refresh the page so that graph goes back and shows the earlier times.

b0rnwr0ng, 2022-06-01:
Everything updates except for the top 10 table.  Once you refresh the page the table is refresh correctly.

I'll take a photo tomorrow morning

ehpersonal38, 2022-06-01:
Okay, that's very strange. If you could also leave the developer console open ( CTRL + SHIFT + I ) so we can see if any errors pop up when it hits midnight.

b0rnwr0ng, 2022-06-02:
I didn't see this message, I'll try that tonight, screen shots below

b0rnwr0ng, 2022-06-02:
[image]
Before refresh (top 10 didn't update but everything else does)

b0rnwr0ng, 2022-06-02:
[image]
Everything up to date once i refresh the page

ehpersonal38, 2022-06-02:
@mcguirepr89 Does anything happen at 11:57 PM that would be applicable to the daily plot?

mcguirepr89, 2022-06-02:
Hi -- I think this is because `overview.php` sets its date when it is loaded, and that is the Chart image for which it looks.

https://github.com/mcguirepr89/BirdNET-Pi/blob/c876c837e5d1add95740f9d6711ffb0c9bd4a9bf/scripts/overview.php#L7

b0rnwr0ng, 2022-06-02:
Not that I'm aware of, but i can't confirm either way.

ehpersonal38, 2022-06-02:
Right you are! Thanks for that.

Fixed in 0e00b55ab02848bf27e5f071dcd918b3ecd9c53c.

closed 2022-06-02 by ehpersonal38

ehpersonal38, 2022-06-03:
Okay, this one should work. It gets its date for the chart that it's supposed to load directly from the server, same as a refresh would. Let me know how it works for you 🐤