mcguirepr89/BirdNET-Pi #347: Fresh 0W2 install, unable to reach site or SSH

csuper, 2022-06-03:
Unable to reach http://birdnetpi1.local/ or SSH into Pi after Step 20 of the 0W2 installation guide.

I was able to SSH and run the install.  The install said it was successful, then Putty disconnected me.  During the install I used all of the defaults/examples from the guide.

I can see the device connected to my router.  And that IP address responds to pings, however the hostname does not.

Putty says unable to connect host does not exist.
 
Chrome says birdnetpi1.local server IP address could not be found.
 
Trying to connect to via putty using the IP I just get a cursor, but otherwise seems unresponsive.  I have been patient waited up to 30 minutes for it to fully boot up.

mcguirepr89, 2022-06-03:
Hi, @csuper -- thanks for your interest in installing BirdNET-Pi on the RPi0W2

I've noticed that on 0W2 installs, the reboot after the installation doesn't fully bring the system down (only on 0W2) --

Assuming it is still within the same power cycle from the initial installation, try pulling the power and putting it back in and let me know if you can reach the web interface when the system is powered back up.

csuper, 2022-06-03:
@mcguirepr89 -- still having the same issues.  Before opening this issue I did an install from scratch.  Both times same outcome.

mcguirepr89, 2022-06-03:
If pulling the power and then plugging it back in didn't see the system come back up two separate times, I cannot say for any reason that I know why that is not working for you. 
I will run a new clean installation now on a Raspberry Pi 0W2 and see if I can replicate the issue.

In the meantime, if you can confirm that you have pulled the power and plugged the device back in and that hasn't worked, you can mount the SD card on another device and grab the `installation-2022-06-03.txt` file and upload that here to see if there were any failures during the installation, though you mention "The install said it was successful" which does mean there was a successful exit code for the pipe :thinking:

csuper, 2022-06-03:
Sorry, yes I did pull the power and plugged it back in which didn't change the outcome.  I don't see that file on the  SD card.

mcguirepr89, 2022-06-03:
I'm very sorry to have been unintentionally vague -- the location for the installation log will be `mountpoint:/rootfs/home/birder/BirdNET-Pi/installation-2022-06-03.txt` -- I'm not sure what your mountpoint will be, but from there, you'll find the file in that directory

csuper, 2022-06-03:
+ set -e
+ my_dir=/home/birder/BirdNET-Pi
+ export my_dir=/home/birder/BirdNET-Pi
+ my_dir=/home/birder/BirdNET-Pi
+ cd /home/birder/BirdNET-Pi/scripts
++ uname -m
+ '[' aarch64 '!=' aarch64 ']'
+ ./install_config.sh
+ set -e
+ trap 'exit 1' SIGINT SIGHUP
+ echo 'Beginning ./install_config.sh'
Beginning ./install_config.sh
+ birdnet_conf=/home/birder/BirdNET-Pi/birdnet.conf
+ '[' -f /home/birder/BirdNET-Pi/birdnet.conf ']'
+ install_config
+ cat
++ curl -s4 ifconfig.co/json
++ jq .latitude
++ curl -s4 ifconfig.co/json
++ jq .longitude
+ chmod g+w /home/birder/BirdNET-Pi/birdnet.conf
+ '[' -d /etc/birdnet ']'
+ sudo mkdir /etc/birdnet
+ sudo ln -sf /home/birder/BirdNET-Pi/birdnet.conf /etc/birdnet/birdnet.conf
+ grep -ve '^#' -e '^$' /etc/birdnet/birdnet.conf
+ sudo -E HOME=/home/birder USER=birder ./install_services.sh
+ trap 'rm -f ${tmpfile}' EXIT
+ trap 'exit 1' SIGINT SIGHUP
++ mktemp
+ tmpfile=/tmp/tmp.ddnHurQbFw
+ config_file=/home/birder/BirdNET-Pi/birdnet.conf
+ export USER=birder
+ USER=birder
+ export HOME=/home/birder
+ HOME=/home/birder
+ export PYTHON_VIRTUAL_ENV=/home/birder/BirdNET-Pi/birdnet/bin/python3
+ PYTHON_VIRTUAL_ENV=/home/birder/BirdNET-Pi/birdnet/bin/python3
+ '[' -f /home/birder/BirdNET-Pi/birdnet.conf ']'
+ source /home/birder/BirdNET-Pi/birdnet.conf
++ LATITUDE=43.0371
++ LONGITUDE=-89.3932
++ BIRDWEATHER_ID=
++ CADDY_PWD=
++ ICE_PWD=birdnetpi
++ BIRDNETPI_URL=
++ RTSP_STREAM=
++ APPRISE_NOTIFICATION_TITLE='New BirdNET-Pi Detection'
++ APPRISE_NOTIFICATION_BODY='A   was just detected with a confidence of '
++ APPRISE_NOTIFY_EACH_DETECTION=0
++ APPRISE_NOTIFY_NEW_SPECIES=0
++ FLICKR_API_KEY=
++ RECS_DIR=/home/birder/BirdSongs
++ REC_CARD=default
++ PROCESSED=/home/birder/BirdSongs/Processed
++ EXTRACTED=/home/birder/BirdSongs/Extracted
++ OVERLAP=0.0
++ CONFIDENCE=0.7
++ SENSITIVITY=1.25
++ CHANNELS=2
++ FULL_DISK=purge
++ PRIVACY_THRESHOLD=0
++ RECORDING_LENGTH=15
++ EXTRACTION_LENGTH=
++ AUDIOFMT=mp3
++ DATABASE_LANG=en
++ LAST_RUN=
++ THIS_RUN=
++ IDFILE=/home/birder/BirdNET-Pi/IdentifiedSoFar.txt
+ install_services
+ set_hostname
[... 1923 lines cut ...]
Building wheels for collected packages: pyrsistent, psutil
  Building wheel for pyrsistent (PEP 517): started
  Building wheel for pyrsistent (PEP 517): finished with status 'done'
  Created wheel for pyrsistent: filename=pyrsistent-0.18.1-cp39-cp39-linux_aarch64.whl size=106109 sha256=94276cdfe16cfb8051e8e567160ccc8befd3a444531c522528d52da9a1b14493
  Stored in directory: /home/birder/.cache/pip/wheels/87/fe/e6/fc8deeb581a41e462eafaf19fee96f51cdc8391e0be1c8088a
  Building wheel for psutil (PEP 517): started
  Building wheel for psutil (PEP 517): finished with status 'done'
  Created wheel for psutil: filename=psutil-5.9.1-cp39-cp39-linux_aarch64.whl size=281516 sha256=b711cc390701bd99f42a5edcb74eda218efb02486e9458d4b8c8b5c9846f968c
  Stored in directory: /home/birder/.cache/pip/wheels/65/66/89/7ecefa8b8cfa845352eab83061e5f0f93946a306ae25661e03
Successfully built pyrsistent psutil
Installing collected packages: traitlets, six, pyrsistent, attrs, wcwidth, tornado, pyzmq, python-dateutil, pycparser, pure-eval, ptyprocess, parso, nest-asyncio, jupyter-core, jsonschema, fastjsonschema, executing, entrypoints, asttokens, webencodings, stack-data, soupsieve, setuptools, pyparsing, pygments, prompt-toolkit, pickleshare, pexpect, nbformat, matplotlib-inline, MarkupSafe, jupyter-client, jedi, decorator, cffi, backcall, tinycss2, psutil, pandocfilters, packaging, nbclient, mistune, jupyterlab-pygments, jinja2, ipython, defusedxml, debugpy, bleach, beautifulsoup4, argon2-cffi-bindings, terminado, Send2Trash, prometheus-client, nbconvert, ipython-genutils, ipykernel, argon2-cffi, notebook, zipp, widgetsnbextension, urllib3, tzdata, smmap, pytz, numpy, llvmlite, jupyterlab-widgets, idna, charset-normalizer, certifi, toolz, threadpoolctl, scipy, requests, pytz-deprecation-shim, pillow, pandas, oauthlib, numba, kiwisolver, joblib, ipywidgets, importlib-metadata, gitdb, fonttools, cycler, commonmark, appdirs, watchdog, validators, tzlocal, typing-extensions, toml, tenacity, soundfile, semver, scikit-learn, rich, resampy, requests-oauthlib, PyYAML, pympler, pydeck, pyarrow, protobuf, pooch, matplotlib, markdown, gitpython, click, cachetools, blinker, audioread, altair, wheel, tflite-runtime, streamlit, seaborn, plotly, pip, librosa, colorama, apprise
  Attempting uninstall: setuptools
    Found existing installation: setuptools 44.1.1
    Uninstalling setuptools-44.1.1:
      Successfully uninstalled setuptools-44.1.1
  Attempting uninstall: pip
    Found existing installation: pip 20.3.4
    Uninstalling pip-20.3.4:
      Successfully uninstalled pip-20.3.4
Successfully installed MarkupSafe-2.1.1 PyYAML-6.0 Send2Trash-1.8.0 altair-4.2.0 appdirs-1.4.4 apprise-0.9.9 argon2-cffi-21.3.0 argon2-cffi-bindings-21.2.0 asttokens-2.0.5 attrs-21.4.0 audioread-2.1.9 backcall-0.2.0 beautifulsoup4-4.11.1 bleach-5.0.0 blinker-1.4 cachetools-5.2.0 certifi-2022.5.18.1 cffi-1.15.0 charset-normalizer-2.0.12 click-8.1.3 colorama-0.4.4 commonmark-0.9.1 cycler-0.11.0 debugpy-1.6.0 decorator-5.1.1 defusedxml-0.7.1 entrypoints-0.4 executing-0.8.3 fastjsonschema-2.15.3 fonttools-4.33.3 gitdb-4.0.9 gitpython-3.1.27 idna-3.3 importlib-metadata-4.11.4 ipykernel-6.13.0 ipython-8.4.0 ipython-genutils-0.2.0 ipywidgets-7.7.0 jedi-0.18.1 jinja2-3.1.2 joblib-1.1.0 jsonschema-4.6.0 jupyter-client-7.3.1 jupyter-core-4.10.0 jupyterlab-pygments-0.2.2 jupyterlab-widgets-1 [... cut ...] o-0.15.0 tflite-runtime-2.6.0 threadpoolctl-3.1.0 tinycss2-1.1.1 toml-0.10.2 toolz-0.11.2 tornado-6.1 traitlets-5.2.2.post1 typing-extensions-4.2.0 tzdata-2022.1 tzlocal-4.2 urllib3-1.26.9 validators-0.19.0 watchdog-2.1.8 wcwidth-0.2.5 webencodings-0.5.1 wheel-0.37.1 widgetsnbextension-3.6.0 zipp-3.8.0
+ cd /home/birder/BirdNET-Pi/scripts
+ ./install_language_label.sh -l en
Archive:  /home/birder/BirdNET-Pi/model/labels_l18n.zip
  inflating: /home/birder/BirdNET-Pi/model/labels_en.txt  
+ exit 0

mcguirepr89, 2022-06-03:
That installation certainly looks successful to me -- does the LED seem to do anything when you power it on? And if so, what does it do and does it change over time?

mcguirepr89, 2022-06-03:
Pulsing is an indication that something is going on with the system, and my installation has finished and is functioning properly. You said you are not able to SSH into the system either?

csuper, 2022-06-03:
Correct. Putty "Unable to open connection to birdnetpi1.local Host does not exist"

mcguirepr89, 2022-06-03:
Are you able to SSH into the machine using the IP address??

csuper, 2022-06-03:
I can't SSH with the IP.  Though it does behave differently.  Rather than the error I just get a cursor.  I think I will try it from the top again.  Since my last message it has continued to blink every second.  Do you recommend I wait awhile after the install completes to unplug it?

mcguirepr89, 2022-06-03:
If you are going to do a new attempt, try this as it may help make things more responsive:

Before running the installer, copy and paste the commands below. (If the system doesn't actually reboot, unplug it, but I don't think you'll have trouble with this one)
```
sudo sed -i 's/CONF_SWAPSIZE=100/CONF_SWAPSIZE=2048/g' /etc/dphys-swapfile &&
sudo sed -i 's/#CONF_MAXSWAP=2048/CONF_MAXSWAP=4096/g' /etc/dphys-swapfile &&
sudo -i sed '/^exit 0/i sudo iw wlan0 set power_save off' /etc/rc.local &&
sudo reboot
```

That will create a larger swap space on the system and will disable the WiFi power_save settings.

When it does come back up, then you can run the `curl` installation

Let me know if you have more success this time around and I will amend the documentation to reflect what works.

Thank you!

csuper, 2022-06-03:
Success! Those extra commands did the trick.  I didn't even have to unplug it after the installation completed. Thanks!

mcguirepr89, 2022-06-04:
Hey that's great! I'll amend the documentation to guide folks to perform that step before installation. Don't forget to set your Recording Length to 30 seconds to give the RPi0W2 enough time to process each audio file and . . . 

Happy birding!!!

closed 2022-06-04 by mcguirepr89