mcguirepr89/BirdNET-Pi #348: Excluded species list Alberts Lyrebird.

Jerrajerra, 2022-06-04:
**Describe the bug**
Alberts Lyrebird is still being detected after being added to the excluded species list

Noticed that whole name does not move across when added

[image]
[image]

Thankyou

mcguirepr89, 2022-06-04:
Thanks for letting me know! [I put a fix in place](https://github.com/mcguirepr89/BirdNET-Pi/commit/045afcdd11b15359ed24aab8ccef8c5123a9211c) for that and it should work now -- if you could please update and then test, I would appreciate it greatly!

"Tools" > "System Controls" > "Update"

mcguirepr89, 2022-06-04:
Also, unrelated:
@ehpersonal38 let's squash this -- I'm better you'll know how to address the menu bar letting text escape as in the "Tools" button in this screenshot
[image]

If you have initial ideas, I'll gladly take a look

ehpersonal38, 2022-06-04:
There you are - b24d8d9796dd32df196294cb8b8fc2d31cb47f6f. I didn't know you could do that with CSS, so thanks for letting me learn that! 🙂

mcguirepr89, 2022-06-04:
Thank you!!!!! And likewise!! I always learn something from your commits ~~

Jerrajerra, 2022-06-04:
updated  and monitoring. Will load the dishwasher straight after a Bower bird calls to apparently replicate a Lyrebird. 
Thankyou again for a great project

closed 2022-06-04 by mcguirepr89