mcguirepr89/BirdNET-Pi #350: Not recording on Pi 3B

phillipredshirt, 2022-06-05:
On new install on raspberry pi 3b

I'm using a UGREEN usb sound card which seems to be recognized by the system as "Realtek Semiconductor Corp. USB Audio"

I am playing recorded American Goldfinch sounds straight from iBird on my phone to the sound card using a cable.  It appears that sound is not being recorded.

Here is what the log looks like when I go to 
"View Log"

19:05:17---find: ‘BirdSongs/StreamData’: No such file or directory
19:05:17---Files loaded
19:05:17---RECORDING_LENGTH set to 60
19:05:52---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts\analyze.py --i BirdSongs/June-2022/04-Saturday/2022-06-04-birdnet-19:04:51.wav --o BirdSongs/June-2022/04-Saturday/2022-06-04-birdnet-19:04:51.wav.csv --lat43.0696 --lon -89.4239 --week 21 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
19:05:53---ANALYZING AUDIO... DONE! Time 19.8 SECONDS
19:05:53---WRITING RESULTS TO BirdSongs/June-2022/04-Saturday/2022-06-04-birdnet-19:03:51.wav.csv ... DONE! WROTE 0 RESULTS.
19:05:53---READING AUDIO DATA... DONE! READ 20 CHUNKS.
19:05:53---Date-time: 2022-06-04 19:04:51
19:06:09---0.0;3.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---3.0;6.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---6.0;9.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---9.0;12.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---12.0;15.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---15.0;18.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---18.0;21.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---21.0;24.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---24.0;27.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---27.0;30.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---30.0;33.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---33.0;36.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---36.0;39.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---39.0;42.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---42.0;45.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---45.0;48.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---48.0;51.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---51.0;54.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---54.0;57.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:09---57.0;60.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
19:06:12---find: ‘BirdSongs/StreamData’: No such file or directory
19:06:12---Files loaded
19:06:13---RECORDING_LENGTH set to 60

Recordings by Date is empty too

Is there an update I can run?

mcguirepr89, 2022-06-05:
Hi, @phillipredshirt I'm sorry you're having issues with your installation.

>Is there an update I can run?

You can check for updates via "Tools" > "System Controls" > "Update" -- if there are any available updates, your "Update" button will indicate as much within a small red circle.

Regarding your recording, have you performed any tests at all to determine whether your microphone is working outside of BirdNET-Pi?

phillipredshirt, 2022-06-05:
I've solved the issue with nothing recording.  I failed to heed this documented warning.

Be careful that the 3.5mm plug type of your USB audio sound card matches the one at your microphone [TRS or TRRS]: https://en.wikipedia.org/wiki/Phone_connector_(audio)

Unfortunately, the pi is struggling to serve up web pages now that it has data to work with.

closed 2022-06-05 by mcguirepr89