mcguirepr89/BirdNET-Pi #351: Sound issue - pulsing sonar sound

EdLaverne, 2022-06-05:
Hi - I'm a newb at this - trying to find a solution to the pulsing "sonar"-like sound that is the background to all audio coming from my RP. I have tried different usb mics, no change. I have even tried two different RPis, both a model 3 and model 4, to no avail. When I listen to other nodes on Birdweather.com I don't hear the pulsing, but on my node I do, although it is diminished compared to what I hear locally. Is there a separate reference for troubleshooting this? Thanks in advance.

mcguirepr89, 2022-06-05:
:thinking:
Have you tried relocating the BirdNET-Pi to eliminate the possibility that there is something nearby introducing the noise?

ehpersonal38, 2022-06-05:
Could you upload a snippet of the sound?

EdLaverne, 2022-06-05:
Hi - my suspicion is that it is power quality related. I will see if
changing to a different circuit helps. I guess I could also try a low-noise
or noise-cancelling power supply. Here is a sample file.

EdLaverne, 2022-06-05:
Thanks Patrick - it's definitely not a local ambient noise.

ehpersonal38, 2022-06-05:
Unfortunately no file was attached - try directly through github.com instead of email?

EdLaverne, 2022-06-05:
Here is a sample sound file - I had to embed it into LibreOffice Impress as it appears audio files won't attach to these comments:
[birdnet-pi-soundtest.odp](https://github.com/mcguirepr89/BirdNET-Pi/files/8840656/birdnet-pi-soundtest.odp)

EdLaverne, 2022-06-05:
I just tried plugging the RBpi into a battery backup I have, and the sound is the same. This battery was not plugged into the wall, so I guess it isn't the house power that is at fault. Are there any known issues with Pulse audio on the Raspberry Pi using this build?

mcguirepr89, 2022-06-05:
>Are there any known issues with Pulse audio on the Raspberry Pi using this build?

Nope -- there are far too many installations that don't see this issue for the issue to have to do with PulseAudio or BirdNET-Pi itself. Sorry :/

mcguirepr89, 2022-06-05:
Have you tried with any other microphones or sound cards?

EdLaverne, 2022-06-05:
Yes, tried it with another usb mic I had. I will continue to work on it and let you know if I am able to resolve. Thanks for your help!
- Ed

ehpersonal38, 2022-06-05:
So you've ruled out:
- the pi
- the mic
- the power circuit

This leaves:
- the 5v power supply
- an external source of electronic interference (use aluminum foil? 😄 )

EdLaverne, 2022-06-05:
Same issue with two different 5v power supplies on two different RPis, so probably not that. The shielding idea is a good one. I'll give it a try.

kk4pqr, 2022-06-06:
Have you tried to connect the Rpi to a hotspot and take it to a different location?

phrozen77, 2022-06-06:
Does it sound like in #327 ? Currently on mobile, can’t open the LibreOffice doc…

EdLaverne, 2022-06-06:
So today I tried fussing with alsamixer settings, to no avail, and then upon reboot the mic died. So not having another usb mic I hooked the RPi up to a usb audio interface and a XLR dynamic mic and voila! Noise gone! The alternate usb mic I used previously was borrowed, so maybe that one was of poor quality as well. Anyway, consider this issue closed. Thanks all for your help and suggestions. Great experience!

mcguirepr89, 2022-06-06:
Thanks for keeping us updated on your troubleshooting -- I'm glad that you were able to find the root of your issue and am selfishly glad it had nothing to do with BirdNET-Pi :)

Please accept my very best regards and please don't hesitate to let us know if you encounter any other issues. Please also feel invited to share your data with the community over on our "[Share your BirdNET-Pi Stuff](https://github.com/mcguirepr89/BirdNET-Pi/discussions/69)" page

Happy birding!

closed 2022-06-06 by mcguirepr89

EdLaverne, 2022-06-06:
Will do thanks!

EdLaverne, 2022-10-11:
ok will do.