mcguirepr89/BirdNET-Pi #352: Only use one channel from multiple-channel soundcard

rototom, 2022-06-07:
I use a soundcard with 4 inputs, but I want birdnetpi to use only the input from channel 1 (which is the connected microphone).
How can I do this?

mcguirepr89, 2022-06-08:
Hello -- have you tried setting "Channels" to 1 in "Tools" > "Settings" > "Advanced Settings"?

DD4WH, 2022-06-08:
Setting Channels to 1 works perfectly --> go to the Settings

Its a bit strange to put a `bug` label on this issue and you just ask a question about a feature that already exists  . . .

mcguirepr89, 2022-06-09:
Yeah I've not figured out the best set of issue templates -- right now there is one for a "Bug Report" and another for  "Feature Request", nothing yet for questions like this (well, I won't put a template together for questions like _this_, but you know what I mean)

closed 2022-06-09 by mcguirepr89