mcguirepr89/BirdNET-Pi #354: Time and Date setting does not work in Image v14.2

DD4WH, 2022-06-08:
In the latest version (Image 14.2) the setting of time does not work:

* if I set the time manually (and untick the `retrieve time from internet` box), the Pi still uses time stamps from some US time zone
* when I click the box to retrieve the time automatically, the time is not retrieved
* this has some consequences on spectrogram labelling (which does not work) and potentially on other things

ehpersonal38, 2022-06-08:
I'll need to add a timezone selector too I think - thanks for the report. Do you concur that's the right way to handle this @mcguirepr89?

mcguirepr89, 2022-06-09:
@ehpersonal38 why yes! I do believe it is the best option. I apologize that I've not been addressing the feedback from the new image just yet -- I have a lot of thoughts, though -- for this, let's utilize the filesystem's tz options:

```
/usr/share/zoneinfo/
├── Africa
│   ├── Abidjan
│   ├── Accra
│   ├── Addis_Ababa -> Nairobi
│   ├── Algiers
│   ├── Asmara -> Nairobi
│   ├── Asmera -> Nairobi
│   ├── Bamako -> Abidjan
│   ├── Bangui -> Lagos
│   ├── Banjul -> Abidjan
│   ├── Bissau
│   ├── Blantyre -> Maputo
│   ├── Brazzaville -> Lagos
│   ├── Bujumbura -> Maputo
│   ├── Cairo
│   ├── Casablanca
│   ├── Ceuta
│   ├── Conakry -> Abidjan
│   ├── Dakar -> Abidjan
│   ├── Dar_es_Salaam -> Nairobi
│   ├── Djibouti -> Nairobi
│   ├── Douala -> Lagos
│   ├── El_Aaiun
│   ├── Freetown -> Abidjan
│   ├── Gaborone -> Maputo
│   ├── Harare -> Maputo
│   ├── Johannesburg
│   ├── Juba
│   ├── Kampala -> Nairobi
│   ├── Khartoum
│   ├── Kigali -> Maputo
│   ├── Kinshasa -> Lagos
│   ├── Lagos
│   ├── Libreville -> Lagos
│   ├── Lome -> Abidjan
│   ├── Luanda -> Lagos
│   ├── Lubumbashi -> Maputo
│   ├── Lusaka -> Maputo
│   ├── Malabo -> Lagos
│   ├── Maputo
│   ├── Maseru -> Johannesburg
│   ├── Mbabane -> Johannesburg
│   ├── Mogadishu -> Nairobi
│   ├── Monrovia
│   ├── Nairobi
│   ├── Ndjamena
│   ├── Niamey -> Lagos
│   ├── Nouakchott -> Abidjan
│   ├── Ouagadougou -> Abidjan
│   ├── Porto-Novo -> Lagos
│   ├── Sao_Tome
│   ├── Timbuktu -> Abidjan
│   ├── Tripoli
│   ├── Tunis
│   └── Windhoek
├── America
│   ├── Adak
│   ├── Anchorage
│   ├── Anguilla -> Port_of_Spain
│   ├── Antigua -> Port_of_Spain
│   ├── Araguaina
│   ├── Argentina
│   │   ├── Buenos_Aires
│   │   ├── Catamarca
│   │   ├── ComodRivadavia -> Catamarca
│   │   ├── Cordoba
[... 1771 lines cut ...]
│   ├── Central -> ../America/Chicago
│   ├── Eastern -> ../America/New_York
│   ├── East-Indiana -> ../America/Indiana/Indianapolis
│   ├── Hawaii -> ../Pacific/Honolulu
│   ├── Indiana-Starke -> ../America/Indiana/Knox
│   ├── Michigan -> ../America/Detroit
│   ├── Mountain -> ../America/Denver
│   ├── Pacific -> ../America/Los_Angeles
│   └── Samoa -> ../Pacific/Pago_Pago
├── UTC -> Etc/UTC
├── WET
├── W-SU -> Europe/Moscow
├── zone1970.tab
├── zone.tab
└── Zulu -> Etc/UTC
```
symlinking any of those to `/etc/localtime` ought to work

```
ln -sf /usr/share/zoneinfo/blah_blah /etc/localtime
```

I will follow up on this and the other feedback tomorrow for certain!

Thanks so much!

DD4WH, 2022-06-09:
Sounds good ! Thanks for the quick response!

For me, a `UTC +- x ` selection box in the settings would be best and easiest !
How to deal with daylight saving ? Maybe a tick box ?

DD4WH, 2022-06-09:
I  have to be more precise:
* setting the time manually in the Settings DOES work properly (box unticked)
* ticking the box to retrieve time from the internet leads to a totally different time zone

mcguirepr89, 2022-06-09:
Thanks for the clarification ~~

ehpersonal38, 2022-06-10:
@mcguirepr89 Do you know any neat ways (like with grep, sed, awk, etc) to format the above list of timezones into something like:

```
Africa/Abidjan
Africa/Accra
Africa/Addis_Ababa
Africa/Algiers
... 
```

If that's not easily achievable, I will make the suggestion that we just use PHP's inbuilt list of timezones that already come as an array, and I believe are mostly identical to the ones above. Let me know how you'd like to proceed!

mcguirepr89, 2022-06-10:
I'm tinkering with that now and will let you know -- how might I go about seeing the exhaustive list of PHP inbuilt TZ options? I'd like to compare the list

ehpersonal38, 2022-06-10:
Okay I've pushed what I have so far to timezone-testy, feel free to make changes as you see fit

mcguirepr89, 2022-06-10:
this looks reallllllly good to me -- I'm going to test it out a bunch and will let you know, but this looks like what I had in mind. Folks interested in setting the TZ to anything outside of those options will be directed to do so using the appropriate tools, i.e., handling the symlinks themselves or manually setting things using `raspi-config`

:heart_eyes:

mcguirepr89, 2022-06-10:
All my tests are working just as expected :+1:

ehpersonal38, 2022-06-10:
Woohoo! One thing I noticed is that after you change the timezone, the time right above doesn't change to the correct time until I forcefully `service php7.4-fpm restart` - think I could slap that in a shell_exec()? Seems a bit hacky to me but what's new I guess?

mcguirepr89, 2022-06-10:
I think that php will be sending the request to restart PHP, so it may not work they way you have in mind? Worth a shot, though?

ehpersonal38, 2022-06-10:
Okay that took wayyy too long to do, but I'm pretty sure it's working now

mcguirepr89, 2022-06-10:
Awesome! I'll be adding this to the next image v14.3 as a bug fix along with a few other small changes. Thank you for the great work as always!

mcguirepr89, 2022-06-10:
[Pulled into `main`](https://github.com/mcguirepr89/BirdNET-Pi/commit/96efed452bde2bb096a8c5f83964f933cf6846c6)!! thanks again, @ehpersonal38 

@DD4WH if you want to update the image and test it out, let me know if things are smoother. I'll put together a v14.3 for the image next week for another round of testing!

I'll close this as an issue since the fix is in `main` now

My best!

closed 2022-06-10 by mcguirepr89