mcguirepr89/BirdNET-Pi #355: Update failed.. No webpage now

b0rnwr0ng, 2022-06-09:
I did an update and it failed, something regarding permissions... Webpage never came back, not sure how to kick that process off again

lloydbayley, 2022-06-09:
I just did the update, had a permissions message but everything is still working. What results does a restart yield?

@mcguirepr89 The message was: 
```
+ sudo -u birdpiadmin git -C /home/birdpiadmin/BirdNET-Pi reset --hard
HEAD is now at b796ec2 analysis uses new StreamData location
+ set +x
+ sudo -u birdpiadmin git -C /home/birdpiadmin/BirdNET-Pi fetch origin main
error: insufficient permission for adding an object to repository database .git/objects
fatal: failed to write object
fatal: unpack-objects failed
+ set +x
+ sudo -u birdpiadmin git -C /home/birdpiadmin/BirdNET-Pi switch -C main --track origin/main
Reset branch 'main'
Branch 'main' set up to track remote branch 'main' from 'origin'.
Your branch is up to date with 'origin/main'.
+ set +x
+ my_dir=/home/birdpiadmin/BirdNET-Pi/scripts
+ sudo systemctl stop birdnet_server.service
+ sudo systemctl stop birdnet_recording.service
+ services=(icecast2.service extraction.service birdnet_recording.service birdnet_log.service)
+ for i in "${services[@]}"
+ sudo systemctl restart icecast2.service
+ for i in "${services[@]}"
+ sudo systemctl restart extraction.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_recording.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_log.service
+ sudo systemctl start birdnet_server.service
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LISTEN      282386/python3      
+ logger '[/usr/local/bin/restart_services.sh] birdnet_server.service is running'
+ break
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LISTEN      282386/python3      
+ status=0
+ ((  status != 0  ))
+ sudo systemctl restart birdnet_analysis.service
```

ehpersonal38, 2022-06-09:
Sorry about this little bug, it's been fixed in newer installations. You should be able to fix it with
`sudo -E chown -R $USER:$USER $HOME/Bird*`

lloydbayley, 2022-06-09:
We all appreciate the conscientiousness for both you and Patrick. Well, I do... :)
Hope it helps the other person to fix their install.

mcguirepr89, 2022-06-09:
@lloydbayley was the `sudo -E chown` fix things for you?

If not, you can fix things with this:
```
cd ~/BirdNET-Pi
git checkout -f
git pull
update_birdnet.sh
```

lloydbayley, 2022-06-09:
I didn't have any problem with operation after the update, so I didn't mess with the system. I was going to wait until next update to see if it was still the same message.
Do you think I action one of those commands or let sleeping dogs lie?

mcguirepr89, 2022-06-09:
Thanks for letting us know --

@lloydbayley it's your call -- if it were me, I would run the commands and then re-run the update, but that's just me :)

ehpersonal38, 2022-06-09:
LOL, that was a close one. One wrong update and you'll be running LizardNET-lite. 😄 

Happy to see things worked out here, closing now!

closed 2022-06-09 by ehpersonal38