mcguirepr89/BirdNET-Pi #357: No more recordings / USB issue

mcguirepr89, 2022-06-09:
### Discussed in https://github.com/mcguirepr89/BirdNET-Pi/discussions/356

<div type='discussions-op-text'>

<sup>Originally posted by **oliverk71** June  9, 2022</sup>
Since yesterday, after update (birdnet and OS) no more recordings. I have a Pi 3B. Maybe anyone else facing this issue (3B/3B+ users)? 

Output of `dmesg`:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.32-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] percpu: Embedded 28 pages/cpu s77144 r8192 d29352 u114688
[    0.000000] pcpu-alloc: s77144 r8192 d29352 u114688 alloc=28*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=d28a8906-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 665668K/970752K available (11456K kernel code, 1962K rwdata, 4144K rodata, 3776K init, 964K bss, 42940K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 37889 entries in 149 pages
[    0.000000] ftrace: allocated 149 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: bcm2836_arm_irqchip_handle_irq
[    0.000000] random: get_random_bytes called from start_kernel+0x510/0x6f8 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000314] Console: colour dummy device 80x25
[    0.001205] printk: console [tty1] enabled
[    0.001289] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=76800)
[    0.001353] pid_max: default: 32768 minimum: 301
[    0.001554] LSM: Security Framework initializing
[    0.001768] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001838] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.003535] cgroup: Disabling memory control group subsystem
[    0.007112] rcu: Hierarchical SRCU implementation.
[    0.008643] EFI services will not be available.
[    0.009459] smp: Bringing up secondary CPUs ...
[    0.010998] Detected VIPT I-cache on CPU1
[    0.011090] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.012915] Detected VIPT I-cache on CPU2
[    0.012978] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.014653] Detected VIPT I-cache on CPU3
[... 382 lines cut ...]
[  575.805267] usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
[  575.885102] usb 1-1.2: new low-speed USB device number 13 using dwc_otg
[  576.737293] usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
[  576.737412] usb 1-1-port2: attempt power cycle
[  577.905374] usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
[  578.193174] usb 1-1.2: new low-speed USB device number 15 using dwc_otg
[  578.609185] usb 1-1.2: device not accepting address 15, error -32
[  578.609341] usb 1-1-port2: unable to enumerate USB device
[  603.702011] usb 1-1.2: new low-speed USB device number 16 using dwc_otg
[  604.766391] usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
[  627.714866] usb 1-1.2: new low-speed USB device number 18 using dwc_otg
[  630.278979] usb 1-1.2: new low-speed USB device number 19 using dwc_otg
[  631.547203] usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
[  762.913930] usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
[  762.993747] usb 1-1.2: new low-speed USB device number 24 using dwc_otg
[  763.697714] usb 1-1.2: device descriptor read/64, error -32
```
I did not check if it is a hardware or a software issue, yet. 

Output of `lsusb`:
```
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```</div>

mcguirepr89, 2022-06-09:
Hi, Oliver -- this is duplicate issue to the one regarding updates

Please try running these commands to address the bug that is affecting your installation:
```
sudo -E chown -R $USER:$USER ~/Bird*
update_birdnet.sh
```

and if that does not work, issue these commands:
```
cd ~/BirdNET-Pi
git checkout -f
git pull
update_birdnet.sh
```

closed 2022-06-09 by mcguirepr89

oliverk71, 2022-06-09:
Hi Patrick. Unfortunately no change. Tried both. 

Output of `update_birdnet.sh`
```
update_birdnet.sh
+ sudo -u oliver git -C /home/oliver/BirdNET-Pi reset --hard
HEAD is now at feca568 Update todays_detections.php
+ set +x
+ sudo -u oliver git -C /home/oliver/BirdNET-Pi fetch origin main
fatal: unable to access 'https://github.com/mcguirepr89/BirdNET-Pi.git/': Could not resolve host: github.com
+ set +x
+ sudo -u oliver git -C /home/oliver/BirdNET-Pi switch -C main --track origin/main
Branch 'main' set up to track remote branch 'main' from 'origin'.
Reset branch 'main'
Your branch is up to date with 'origin/main'.
+ set +x
+ my_dir=/home/oliver/BirdNET-Pi/scripts
+ sudo systemctl stop birdnet_server.service
+ sudo systemctl stop birdnet_recording.service
+ services=(icecast2.service extraction.service birdnet_recording.service birdnet_log.service)
+ for i in "${services[@]}"
+ sudo systemctl restart icecast2.service
+ for i in "${services[@]}"
+ sudo systemctl restart extraction.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_recording.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_log.service
+ sudo systemctl start birdnet_server.service
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LISTEN      95252/python3       
+ logger '[/usr/local/bin/restart_services.sh] birdnet_server.service is running'
+ break
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LISTEN      95252/python3       
+ status=0
+ ((  status != 0  ))
+ sudo systemctl restart birdnet_analysis.service

```
but no change. Output of `dmesg`

```
#[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.32-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[... 405 lines cut ...]
[   31.478293]  MAI: soc_pcm_open() failed (-19)
[   31.480197] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   31.480333]  MAI: soc_pcm_open() failed (-19)
[   31.481788] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   31.482249]  MAI: soc_pcm_open() failed (-19)
[   31.497987] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   31.498120]  MAI: soc_pcm_open() failed (-19)
[   31.499060] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   31.499205]  MAI: soc_pcm_open() failed (-19)
[   31.500132] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   31.500286]  MAI: soc_pcm_open() failed (-19)
[   33.757664] cam-dummy-reg: disabling
[  149.907573] usb 1-1.2: new low-speed USB device number 4 using dwc_otg
[  149.987577] usb 1-1.2: device descriptor read/64, error -32
[  150.179933] usb 1-1.2: device descriptor read/64, error -32
[  150.367599] usb 1-1.2: new low-speed USB device number 5 using dwc_otg
[  150.447607] usb 1-1.2: device descriptor read/64, error -32
[  150.635712] usb 1-1.2: device descriptor read/64, error -32
[  150.743749] usb 1-1-port2: attempt power cycle
[  151.351635] usb 1-1.2: new low-speed USB device number 6 using dwc_otg
[  151.767721] usb 1-1.2: device not accepting address 6, error -32
[  151.847854] usb 1-1.2: new low-speed USB device number 7 using dwc_otg
[  152.263712] usb 1-1.2: device not accepting address 7, error -32
[  152.264219] usb 1-1-port2: unable to enumerate USB device
```

oliverk71, 2022-06-09:
Well, sometimes the wifi connection to my router is weak. I tried a second time, but maybe I copied the first output. However, I did it again and here is the output. No change. 

```
+ sudo -u oliver git -C /home/oliver/BirdNET-Pi reset --hard
Updating files: 100% (97/97), done.
HEAD is now at d33ce41 Spectrogram now uses RECS_DIR
+ set +x
+ sudo -u oliver git -C /home/oliver/BirdNET-Pi fetch origin main
remote: Enumerating objects: 7, done.
remote: Counting objects: 100% (7/7), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 4 (delta 3), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (4/4), 746 bytes | 82.00 KiB/s, done.
From https://github.com/mcguirepr89/BirdNET-Pi
 * branch            main       -> FETCH_HEAD
   d33ce41..9392fbe  main       -> origin/main
+ set +x
+ sudo -u oliver git -C /home/oliver/BirdNET-Pi switch -C main --track origin/main
Branch 'main' set up to track remote branch 'main' from 'origin'.
Reset branch 'main'
Your branch is up to date with 'origin/main'.
+ set +x
+ my_dir=/home/oliver/BirdNET-Pi/scripts
+ sudo systemctl stop birdnet_server.service
+ sudo systemctl stop birdnet_recording.service
+ services=(icecast2.service extraction.service birdnet_recording.service birdnet_log.service)
+ for i in "${services[@]}"
+ sudo systemctl restart icecast2.service
+ for i in "${services[@]}"
+ sudo systemctl restart extraction.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_recording.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_log.service
+ sudo systemctl start birdnet_server.service
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
++ netstat -tulpn
+ grep 5050 /dev/fd/63
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LISTEN      13599/python3
+ status=0
+ ((  status != 0  ))
+ sudo systemctl restart birdnet_analysis.service
```
And here is the output of the second suggestion: 

```
birdnetpi-elsen:~/BirdNET-Pi$ git pull
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Already up to date.
birdnetpi-elsen:~/BirdNET-Pi$ update_birdnet.sh
+ sudo -u oliver git -C /home/oliver/BirdNET-Pi reset --hard
HEAD is now at 9392fbe Update birdnet_log.sh
+ set +x
+ sudo -u oliver git -C /home/oliver/BirdNET-Pi fetch origin main
From https://github.com/mcguirepr89/BirdNET-Pi
 * branch            main       -> FETCH_HEAD
+ set +x
+ sudo -u oliver git -C /home/oliver/BirdNET-Pi switch -C main --track origin/main
Branch 'main' set up to track remote branch 'main' from 'origin'.
Reset branch 'main'
Your branch is up to date with 'origin/main'.
+ set +x
+ my_dir=/home/oliver/BirdNET-Pi/scripts
+ sudo systemctl stop birdnet_server.service
+ sudo systemctl stop birdnet_recording.service
+ services=(icecast2.service extraction.service birdnet_recording.service birdnet_log.service)
+ for i in "${services[@]}"
+ sudo systemctl restart icecast2.service
+ for i in "${services[@]}"
+ sudo systemctl restart extraction.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_recording.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_log.service
+ sudo systemctl start birdnet_server.service
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LISTEN      16779/python3
+ logger '[/usr/local/bin/restart_services.sh] birdnet_server.service is running'
+ break
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LISTEN      16779/python3
+ status=0
+ ((  status != 0  ))
+ sudo systemctl restart birdnet_analysis.service
```

mcguirepr89, 2022-06-09:
Could you please provide some log output from "View Log"?

mcguirepr89, 2022-06-09:
That indicates that the system is not receiving any input -- please check your microphone hardware and try restarting the "BirdNET Recording" service in "Tools" > "Services"

A reboot is also recommended if neither of the steps above work.

oliverk71, 2022-06-09:
Guess, I should check the hardware. Unfortunately, I do not have another soundcard around.

oliverk71, 2022-06-09:
I was pretty sure that I removed the external USB soundcard. But I will double check, tomorrow. This is dmesg output after reboot:

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.32-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] percpu: Embedded 28 pages/cpu s77144 r8192 d29352 u114688
[    0.000000] pcpu-alloc: s77144 r8192 d29352 u114688 alloc=28*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=d28a8906-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 665668K/970752K available (11456K kernel code, 1962K rwdata, 4144K rodata, 3776K init, 964K bss, 42940K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 37889 entries in 149 pages
[    0.000000] ftrace: allocated 149 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: bcm2836_arm_irqchip_handle_irq
[    0.000000] random: get_random_bytes called from start_kernel+0x510/0x6f8 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000311] Console: colour dummy device 80x25
[    0.001205] printk: console [tty1] enabled
[    0.001290] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=76800)
[    0.001356] pid_max: default: 32768 minimum: 301
[    0.001557] LSM: Security Framework initializing
[    0.001774] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001842] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.003536] cgroup: Disabling memory control group subsystem
[    0.007107] rcu: Hierarchical SRCU implementation.
[    0.008635] EFI services will not be available.
[    0.009449] smp: Bringing up secondary CPUs ...
[    0.010982] Detected VIPT I-cache on CPU1
[    0.011077] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.012896] Detected VIPT I-cache on CPU2
[    0.012961] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.014616] Detected VIPT I-cache on CPU3
[    0.014672] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.014880] smp: Brought up 1 node, 4 CPUs
[    0.015038] SMP: Total of 4 processors activated.
[    0.015072] CPU features: detected: 32-bit EL0 Support
[    0.015104] CPU features: detected: 32-bit EL1 Support
[    0.015139] CPU features: detected: CRC32 instructions
[... 329 lines cut ...]
[   17.055881] Bluetooth: L2CAP socket layer initialized
[   17.055913] Bluetooth: SCO socket layer initialized
[   17.080483] Bluetooth: HCI UART driver ver 2.3
[   17.080513] Bluetooth: HCI UART protocol H4 registered
[   17.080615] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   17.080912] Bluetooth: HCI UART protocol Broadcom registered
[   17.680448] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   17.680477] Bluetooth: BNEP filters: protocol multicast
[   17.680501] Bluetooth: BNEP socket layer initialized
[   17.708878] NET: Registered PF_ALG protocol family
[... cut ...]
[   29.871491]  MAI: soc_pcm_open() failed (-19)
[   29.898094] hdmi-audio-codec hdmi-audio-codec.1.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   29.898247]  MAI: soc_pcm_open() failed (-19)
[   29.899211] hdmi-audio-codec hdmi-audio-codec.1.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   29.899388]  MAI: soc_pcm_open() failed (-19)
[   29.900502] hdmi-audio-codec hdmi-audio-codec.1.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   29.900692]  MAI: soc_pcm_open() failed (-19)
```

oliverk71, 2022-06-11:
USB soundcard is not even connected, device was rebooted, but still this errors.

oliverk71, 2022-06-11:
```
Linux birdnetpi-elsen 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Jun 11 18:44:21 2022
oliver@birdnetpi-elsen:~/BirdNET-Pi$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.32-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] percpu: Embedded 28 pages/cpu s77144 r8192 d29352 u114688
[    0.000000] pcpu-alloc: s77144 r8192 d29352 u114688 alloc=28*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=d28a8906-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 665668K/970752K available (11456K kernel code, 1962K rwdata, 4144K rodata, 3776K init, 964K bss, 42940K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 37889 entries in 149 pages
[    0.000000] ftrace: allocated 149 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: bcm2836_arm_irqchip_handle_irq
[    0.000000] random: get_random_bytes called from start_kernel+0x510/0x6f8 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000003] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000313] Console: colour dummy device 80x25
[    0.001208] printk: console [tty1] enabled
[    0.001294] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=76800)
[    0.001358] pid_max: default: 32768 minimum: 301
[    0.001559] LSM: Security Framework initializing
[    0.001774] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001843] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.003535] cgroup: Disabling memory control group subsystem
[    0.007109] rcu: Hierarchical SRCU implementation.
[    0.008620] EFI services will not be available.
[    0.009433] smp: Bringing up secondary CPUs ...
[    0.010970] Detected VIPT I-cache on CPU1
[    0.011064] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.012880] Detected VIPT I-cache on CPU2
[... 340 lines cut ...]
[   16.622550] Bluetooth: L2CAP socket layer initialized
[   16.622594] Bluetooth: SCO socket layer initialized
[   16.651290] Bluetooth: HCI UART driver ver 2.3
[   16.651325] Bluetooth: HCI UART protocol H4 registered
[   16.651977] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   16.652392] Bluetooth: HCI UART protocol Broadcom registered
[   17.372351] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   17.372395] Bluetooth: BNEP filters: protocol multicast
[   17.372423] Bluetooth: BNEP socket layer initialized[   17.400484] NET: Registered PF_ALG protocol family
[   20.446796] vc4-drm soc:gpu: [drm] Cannot find any crtc or sizes
[... cut ...]
[   31.401794] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   31.401944]  MAI: soc_pcm_open() failed (-19)
[   31.402950] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   31.403105]  MAI: soc_pcm_open() failed (-19)
[   31.404442] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   31.404571]  MAI: soc_pcm_open() failed (-19)
[   33.758679] cam-dummy-reg: disabling
oliver@birdnetpi-elsen:~/BirdNET-Pi$
```

oliverk71, 2022-06-12:
Surprise: I pulled out the SD card and put it into my Raspberry Pi 4B with 8 GB. No USB device connected to it, powered by original Raspberry Pi power supply. And voila: Still the same errors! So, it must be a software issue. I do not consider the SD card itself as error source. 

Regarding the software side: I did not change much, but made updates of OS and birdnetpi. The only thing I installed in addition is the 2FA, which works perfectly fine for SSH connection. 

Output of dmesg again: 

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.15.32-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] percpu: Embedded 28 pages/cpu s77144 r8192 d29352 u114688
[    0.000000] pcpu-alloc: s77144 r8192 d29352 u114688 alloc=28*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2028848
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:B7:D4:71 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=d28a8906-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 7668516K/8245248K available (11456K kernel code, 1962K rwdata, 4144K rodata, 3776K init, 964K bss, 249052K reserved, 327680K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 37889 entries in 149 pages
[    0.000000] ftrace: allocated 149 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 10)
[    0.000000] random: get_random_bytes called from start_kernel+0x510/0x6f8 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000001] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000250] Console: colour dummy device 80x25
[    0.000806] printk: console [tty1] enabled
[    0.000888] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000929] pid_max: default: 32768 minimum: 301
[    0.001074] LSM: Security Framework initializing
[    0.001299] Mount-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.001424] Mountpoint-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.002666] cgroup: Disabling memory control group subsystem
[    0.005151] rcu: Hierarchical SRCU implementation.
[    0.006277] EFI services will not be available.
[    0.006830] smp: Bringing up secondary CPUs ...
[... 378 lines cut ...]
[   29.151326] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   29.151395]  MAI: soc_pcm_open() failed (-19)
[   29.151967] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[... cut ...]
[   29.211665]  MAI: soc_pcm_open() failed (-19)
[   29.217607] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   29.217672]  MAI: soc_pcm_open() failed (-19)
[   29.217970] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   29.218008]  MAI: soc_pcm_open() failed (-19)
[   29.218205] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   29.218229]  MAI: soc_pcm_open() failed (-19)
```

ehpersonal38, 2022-06-12:
Unfortunately this isn't in the scope of BirdNET-Pi. Your system updates seem to have broken something (from a brief Google search maybe something with kernels?). So I would either seek help on forums like Stack Exchange or Raspberry Pi forums where they can probably help you best. Or save yourself some time and do a fresh install (make sure to backup your birds.db), if that's an option for you.

oliverk71, 2022-06-12:
Well, I don't want to reinstall whole system again and again (I did it several times), but if nothing helps... I will have to do that. 

I reconnected the USB sound card and microphone and now it is recording again and shows the spectrogram again on the raspberry pi 4B (which I do not want to use because it is the 8 GB version and I use it for something else).