mcguirepr89/BirdNET-Pi #365: Labels.txt not found

TheBigBottle, 2022-06-10:
Hey there, 
thanks for your work! 

But the Installation dosen't work for me. 

After the Installation the service is online but the page is blank. 
The /log/ output is AS follows:

Jun 10 19:13:36 birdnetpi python3[3594]: INFO: Created TensorFlow Lite XNNPACK delegate for CPU.
Jun 10 19:13:36 birdnetpi python3[3594]: INFO: Created TensorFlow Lite delegate for select TF ops.
Jun 10 19:13:36 birdnetpi python3[3594]: INFO: TfLiteFlexDelegate delegate: 1 nodes delegated out of 182 nodes with 1 partitions.
Jun 10 19:13:36 birdnetpi python3[3594]: LOADING TF LITE MODEL... Traceback (most recent call last):
Jun 10 19:13:36 birdnetpi python3[3594]:   File "/usr/local/bin/server.py", line 545, in <module>
Jun 10 19:13:36 birdnetpi python3[3594]:     start()
Jun 10 19:13:36 birdnetpi python3[3594]:   File "/usr/local/bin/server.py", line 534, in start
Jun 10 19:13:36 birdnetpi python3[3594]:     INTERPRETER = loadModel()
Jun 10 19:13:36 birdnetpi python3[3594]:   File "/usr/local/bin/server.py", line 76, in loadModel
Jun 10 19:13:36 birdnetpi python3[3594]:     with open(labelspath, 'r') as lfile:
Jun 10 19:13:36 birdnetpi birdnet_recording.sh[397]: + grep 5050 /dev/fd/63
Jun 10 19:13:36 birdnetpi birdnet_recording.sh[3714]: ++ netstat -tulpn
Jun 10 19:13:36 birdnetpi python3[3594]: FileNotFoundError: [Errno 2] No such file or directory: '/root/BirdNET-Pi/model/labels.txt'
Jun 10 19:13:36 birdnetpi birdnet_recording.sh[397]: + sleep 1

So tensorflow doesn't find the Labels.txt. 

Do you have any suggestions ? 

Kind regards and stay safe

mcguirepr89, 2022-06-10:
Dang! and that was a recent installation???

This should fix your problem, but if you could please send me your installation log, I would love to know why your labels.txt was not extracted. The log is located in your `$HOME` directory and would (assuming your installations just failed) be named `$HOME/installation-2022-06-10.txt`

But running
```
install_language_label.sh -l en
```
should create the labels.txt for your and get things on their way (if that was the only thing that failed during installation)

mcguirepr89, 2022-06-10:
Oh, now I see what is going on -- the installation has been performed as the `root` user -- this is not supported -- please try running the installer again and [follow the installation guide](https://github.com/mcguirepr89/BirdNET-Pi/wiki/Installation-Guide)

If after performing the installation while following that guide fails, please let me know and we will take things from there. 

My best!

ehpersonal38, 2022-06-10:
@mcguirepr89 Think we should tack this onto the beginning of the installer?
```
if [ "$EUID" == 0 ]
  then echo "Please run as a non-root user"
  exit
fi
```
Test:
```
pi@birdnetpi:~/$ bash test_install.sh
pi@birdnetpi:~/$ sudo bash test_install.sh
Please run as a non-root user
```

mcguirepr89, 2022-06-10:
`exit 1`

TheBigBottle, 2022-06-11:
Hey,
thanks for the fast response !!!
Thst fixed the isse, but i also think it would be a good idea to catch that in the install script.

Kind regards and stay safe!

closed 2022-06-11 by TheBigBottle