mcguirepr89/BirdNET-Pi #366: cannot access web terminal due to unrecognized credentials.

johnocampbell99, 2022-06-11:
Whenever I try to connect to the web terminal, it prompts me with:

birdnetpi login:

I use the default credentials I used when setting up birdnetpi but they are not recognized and I cannot proceed using the terminal.  It just provides: Login incorrect

ehpersonal38, 2022-06-11:
Does this happen when you ssh in (either with Putty if you're on windows, or terminal on Linux), or is it just the web terminal?

mcguirepr89, 2022-06-11:
Please check your capslock -- unfortunately, their won't be much we can do to help you recall your system credentials.

closed 2022-06-11 by mcguirepr89

johnocampbell99, 2022-06-11:
my apologies, you can remove this.  Or I will try to.  It's very early and my head must not be on straight.  I resolved the issue