mcguirepr89/BirdNET-Pi #371: Webpage no longer Accessible

abartow91, 2022-06-12:
**Describe the bug**
I no longer have access to the birdnetpi webpage created by the instance.  I can tell birdnet is still running because I can ssh into my raspberry pi and I still get emails about new species detections.

**Expected behavior**
Going to http or https://birdnetpi.local able to see birdnet webpage while within my home network.

**Additional context**
Today (6/12) I had noticed that it had not had any detections since 11am the day before (6/11).  At this point I could still access the web page.

I cycled power on my raspberry pi.  When it came back online, I could no longer access the webpage.
When it came back online, I got new species detection emails with the bird species name being in German (I live in the U.S. and they were in English/default before).

I can also SSH into the raspberry pi.

I've also tried using the IP address that the raspberry pi is at.

**Troubleshooting Items**
1. Rebooted/cycled power multiple times
2. Disabled DHCP static IP address I had previously setup
3. Using SSH, removed the custom url i had setup for remote access to the birdnetpi
4. Updated to the latest birdnet version.
5.  Performed the sudo -E chown command as described in Closed Issue #355 

I rebooted after each of these changes.

Thanks for your help

mcguirepr89, 2022-06-13:
Hi, @abartow91 -- thanks for installing BirdNET-Pi, though I'm sorry you're having issues reaching the web interface.

I have a few questions for you I'm hoping you'll answer, and I will request some log output to look closer at possible causes.

First,

1. ```
   I cycled power on my raspberry pi. When it came back online, I could no longer access the webpage.
   ```
   For this, did you use the web interface's "Tools" > "System Controls" > "Shutdown", or did you shutdown the system another way prior to the power cycle?

1. When you do try visiting http://birdnetpi.local, do you see anything in your web browser? If so, are there errors reported? If so, what are they? If you don't see any words, does anything load, or is it just a white screen?

1. Same series of question as above, but for the portion where you mention attempting to reach the web interface using the IP address: Do you see anything in your web browser? If so, are there errors reported? If so, what are they? If you don't see any words, does anything load, or is it just a white screen?

For additional information, could you please provide the output from this command,

```
journalctl --no-hostname --no-pager -n100 -ucaddy
```
formatting the output as code by first placing three backticks (\`\`\`) together on one line by themselves, followed by the log output, followed by another line after the code block closing the code block with three more backticks (\`\`\`)

For example:
\`\`\`
The three back ticks 
    in the line 
       above and below this 
          sentence will allow the log 
       output to stay formatted 
    correctly and will be 
much easier to read
\`\`\`
```
The three back ticks 
    in the line 
       above and below this 
          sentence will allow the log 
       output to stay formatted 
    correctly and will be 
much easier to read
```

My best regards,
Patrick

fossie08, 2022-06-13:
I've very recently 13/06 installed BirdNet on my RPi 4B and have not been able to access the webclient page, all it comes back with is a completly blank html page.

Here is the output of `journalctl --no-hostname --no-pager -n100 -ucaddy`

```
-- Journal begins at Mon 2022-04-04 15:41:41 BST, ends at Mon 2022-06-13 19:44:38 BST. --                               Jun 13 18:28:43 caddy[860]: runtime.NumCPU=4                                                                            Jun 13 18:28:43 caddy[860]: runtime.GOMAXPROCS=4                                                                        Jun 13 18:28:43 caddy[860]: runtime.Version=go1.18.1                                                                    Jun 13 18:28:43 caddy[860]: os.Getwd=/                                                                                  Jun 13 18:28:43 caddy[860]: LANG=en_GB.UTF-8                                                                            Jun 13 18:28:43 caddy[860]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin                           Jun 13 18:28:43 caddy[860]: NOTIFY_SOCKET=/run/systemd/notify                                                           Jun 13 18:28:43 caddy[860]: HOME=/var/lib/caddy                                                                         Jun 13 18:28:43 caddy[860]: LOGNAME=caddy                                                                               Jun 13 18:28:43 caddy[860]: USER=caddy                                                                                  Jun 13 18:28:43 caddy[860]: INVOCATION_ID=0030e8ed906442a0b0c9042800b2abfd                                              Jun 13 18:28:43 caddy[860]: JOURNAL_STREAM=8:14678                                                                      Jun 13 18:28:43 caddy[860]: {"level":"info","ts":1655141323.4044342,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","config_adapter":""}                                                                              Jun 13 18:28:43 caddy[860]: {"level":"warn","ts":1655141323.4097688,"msg":"Caddyfile input is not formatted; run the 'caddy fmt' command to fix inconsistencies","adapter":"caddyfile","file":"/etc/caddy/Caddyfile","line":1}                  Jun 13 18:28:43 caddy[860]: {"level":"info","ts":1655141323.428991,"logger":"admin","msg":"admin endpoint started","address":"tcp/localhost:2019","enforce_origin":false,"origins":["//localhost:2019","//[::1]:2019","//127.0.0.1:2019"]}      Jun 13 18:28:43 caddy[860]: {"level":"info","ts":1655141323.4298708,"logger":"tls.cache.maintenance","msg":"started background certificate maintenance","cache":"0x400042cb60"}                                                                 Jun 13 18:28:43 caddy[860]: {"level":"warn","ts":1655141323.42985,"logger":"http","msg":"server is listening only on the HTTP port, so no automatic HTTPS will be applied to this server","server_name":"srv0","http_port":80}                  Jun 13 18:28:43 caddy[860]: {"level":"info","ts":1655141323.4584606,"logger":"tls","msg":"cleaning storage unit","description":"FileStorage:/var/lib/caddy/.local/share/caddy"}                                                                 Jun 13 18:28:43 caddy[860]: {"level":"info","ts":1655141323.4586017,"logger":"tls","msg":"finished cleaning storage units"}                                                                                                                     Jun 13 18:28:43 caddy[860]: {"level":"info","ts":1655141323.4676597,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}                                                                     Jun 13 18:28:43 caddy[860]: {"level":"info","ts":1655141323.468107,"msg":"serving initial configuration"}               Jun 13 18:28:43 systemd[1]: Started Caddy.                                                                              Jun 13 19:11:18 caddy[860]: {"level":"info","ts":1655143877.472711,"msg":"shutting down apps, then terminating","signal":"SIGTERM"}                                                                                                             Jun 13 19:11:19 caddy[860]: {"level":"warn","ts":1655143878.881556,"msg":"exiting; byeee!! 👋","signal":"SIGTERM"}      Jun 13 19:11:19 caddy[860]: {"level":"info","ts":1655143879.425376,"logger":"tls.cache.maintenance","msg":"stopped background certificate maintenance","cache":"0x400042cb60"}                                                                  Jun 13 19:11:19 caddy[860]: {"level":"info","ts":1655143879.6197064,"logger":"admin","msg":"stopped previous server","address":"tcp/localhost:2019"}                                                                                            Jun 13 19:11:19 caddy[860]: {"level":"info","ts":1655143879.6647122,"msg":"shutdown complete","signal":"SIGTERM","exit_code":0}                                                                                                                 Jun 13 19:11:15 systemd[1]: Stopping Caddy...                                                                           Jun 13 19:11:19 systemd[1]: caddy.service: Succeeded.                                                                   Jun 13 19:11:19 systemd[1]: Stopped Caddy.                                                                              -- Boot 8b6fb6b0c626407da50532605d83b473 --                                                       [... cut ...]                                                         Jun 13 19:16:47 caddy[900]: {"level":"info","ts":1655144207.0584414,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}                                                                     Jun 13 19:16:47 caddy[900]: {"level":"info","ts":1655144207.0587156,"msg":"serving initial configuration"}              Jun 13 19:16:47 systemd[1]: Started Caddy.   
```

Hope that you can help.

fossie08

mcguirepr89, 2022-06-13:
I'm trying desperately to format your log so that it is readable and I am unable -- can you please try again and use a different method of copy and paste? This method did not capture any of the new line characters that are needed to make the log readable

fossie08, 2022-06-13:
I am so sorry for the fornatting issue!
I have uploaded the log as screenshots in a pdf.
Hopefully this works
[log.pdf](https://github.com/mcguirepr89/BirdNET-Pi/files/8893885/log.pdf)
!

mcguirepr89, 2022-06-13:
Thank you that was very helpful and confirms that there are no issues with the web server.

I suspect the issue is with the mDNS announcement on the network. I just pushed a change that should take care of this for users during installation and should also provide a tool for users to fix things on their installations once now so to avoid this going forward.

If you could please update your system via SSH, we will see if you can then reach your installation via its IP address. To do this, please follow the steps below:

1. ```
   update_birdnet.sh
   ```
1. ```
   sudo update_caddyfile.sh
   ```
   
and then try to visit the installation at the IP address it reports when you issue the command
```
hostname -I | cut -d' ' -f1
```

Please let me know if that helps --
Patrick

mcguirepr89, 2022-06-13:
I'm happy to have helped and am glad things are working for you. Hopefully this will also resolve things for @abartow91 :crossed_fingers:

abartow91, 2022-06-13:
Unfortunately that did not work, though it may have provided a clue.

When I updated, it had to loop multiple times to restart the birdnet_server.service.  But eventually did restart that service

```
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ status=1
+ ((  status != 0  ))
+ logger &apos;[/usr/local/bin/restart_services.sh] Unable to start birdnet_server.service... Looping until it start properly&apos;
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sudo systemctl restart birdnet_server.service
+ sleep 45
+ grep 5050 /dev/fd/63
++ netstat -tulpn
tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LISTEN      958553/python3      
+ sudo systemctl restart birdnet_analysis.service
```

The update caddy.service did not work.  It had a "caddy.service is not active error".
I think this is the issue

```
+ '[' -d /etc/caddy ']'
+ '[' -f /etc/caddy/Caddyfile ']'
+ cp /etc/caddy/Caddyfile /etc/caddy/Caddyfile.original
+ '[' -z REMOVED ']'
++ caddy hash-password -plaintext REMOVED
+ HASHWORD=REMOVED
+ cat
+ sudo caddy fmt --overwrite /etc/caddy/Caddyfile
+ sudo systemctl reload caddy
caddy.service is not active, cannot reload.
```

But to answer your original questions
1.  I cycled power by unplugging it and plugging it back in.  I'm sometimes mean to my stuff
2. I get a web browser message about "This site can't be reached: birdnetpi.local refusing to connect", ERR_CONNECTION_REFUSED
3. I get the same web browser message except replaced with my IP address

Output of journalctl command:

```
-- Journal begins at Sat 2022-06-11 14:56:41 EDT, ends at Mon 2022-06-13 16:30:05 EDT. --
Jun 12 09:43:28 caddy[816]: caddy.Version=v2.5.1 h1:bAWwslD1jNeCzDa+jDCNwb8M3UJ2tPa8UZFFzPVmGKs=
Jun 12 09:43:28 caddy[816]: runtime.GOOS=linux
Jun 12 09:43:28 caddy[816]: runtime.GOARCH=arm64
Jun 12 09:43:28 caddy[816]: runtime.Compiler=gc
Jun 12 09:43:28 caddy[816]: runtime.NumCPU=4
Jun 12 09:43:28 caddy[816]: runtime.GOMAXPROCS=4
Jun 12 09:43:28 caddy[816]: runtime.Version=go1.18.1
Jun 12 09:43:28 caddy[816]: os.Getwd=/
Jun 12 09:43:28 caddy[816]: LANG=en_GB.UTF-8
Jun 12 09:43:28 caddy[816]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
Jun 12 09:43:28 caddy[816]: NOTIFY_SOCKET=/run/systemd/notify
Jun 12 09:43:28 caddy[816]: HOME=/var/lib/caddy
Jun 12 09:43:28 caddy[816]: LOGNAME=caddy
Jun 12 09:43:28 caddy[816]: USER=caddy
Jun 12 09:43:28 caddy[816]: INVOCATION_ID=85cbd0c34c814873aab0dda8937da0ee
Jun 12 09:43:28 caddy[816]: JOURNAL_STREAM=8:15050
Jun 12 09:43:28 caddy[816]: {"level":"info","ts":1655041408.3507304,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","config_adapter":""}
Jun 12 09:43:28 caddy[816]: run: adapting config using caddyfile: parsing caddyfile tokens for 'basicauth': /etc/caddy/Caddyfile:10 - Error during parsing: unrecognized hash algorithm: views.php?view=File*
[... 63 lines cut ...]
Jun 12 16:55:07 caddy[854]: caddy.Version=v2.5.1 h1:bAWwslD1jNeCzDa+jDCNwb8M3UJ2tPa8UZFFzPVmGKs=
Jun 12 16:55:07 caddy[854]: runtime.GOOS=linux
Jun 12 16:55:07 caddy[854]: runtime.GOARCH=arm64
Jun 12 16:55:07 caddy[854]: runtime.Compiler=gc
Jun 12 16:55:07 caddy[854]: runtime.NumCPU=4
Jun 12 16:55:07 caddy[854]: runtime.GOMAXPROCS=4
Jun 12 16:55:07 caddy[854]: runtime.Version=go1.18.1
Jun 12 16:55:07 caddy[854]: os.Getwd=/
Jun 12 16:55:07 caddy[854]: LANG=en_GB.UTF-8
Jun 12 16:55:07 caddy[854]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
Jun 12 16:55:07 caddy[854]: NOTIFY_SOCKET=/run/systemd/notify
Jun 12 16:55:07 caddy[854]: HOME=/var/lib/caddy
Jun 12 16:55:07 caddy[854]: LOGNAME=caddy
Jun 12 16:55:07 caddy[854]: USER=caddy
Jun 12 16:55:07 caddy[854]: INVOCATION_ID=2ac58a922b30448598627df083b46e7f
Jun 12 16:55:07 caddy[854]: JOURNAL_STREAM=8:14991
Jun 12 16:55:07 caddy[854]: {"level":"info","ts":1655067307.692989,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","config_adapter":""}
Jun 12 16:55:07 caddy[854]: run: adapting config using caddyfile: parsing caddyfile tokens for 'basicauth': /etc/caddy/Caddyfile:10 - Error during parsing: unrecognized hash algorithm: views.php?view=File*
Jun 12 16:55:07 systemd[1]: caddy.service: Main process exited, code=exited, status=1/FAILURE
Jun 12 16:55:07 systemd[1]: caddy.service: Failed with result 'exit-code'.
Jun 12 16:55:07 systemd[1]: Failed to start Caddy.
Jun 13 15:59:53 systemd[1]: caddy.service: Unit cannot be reloaded because it is inactive
```

Thanks again for your help

mcguirepr89, 2022-06-13:
Hello -- here is what I'm seeing:

```
Jun 12 16:55:07 caddy[854]: run: adapting config using caddyfile: parsing caddyfile tokens for 'basicauth': /etc/caddy/Caddyfile:10 - Error during parsing: unrecognized hash algorithm: views.php?view=File*
```

This means that your Caddyfile is not configured properly.

Try re-running the `sudo update_caddyfile.sh`, but this time, we'll have it `restart` instead of `reload` the service since the service is not loaded.

```
sed 's/reload/restart/' ~/BirdNET-Pi/scripts/update_caddyfile.sh | sudo bash
```

Then if you could please rerun the output for 
```
journalctl --no-hostname --no-pager --boot -ucaddy
```

:crossed_fingers:

If that doesn't work, you'll likely still have a malformed Caddyfile for some reason -- if so, if you don't mind put its contents here, but just remove the hashed password if are using a web password.
```
cat /etc/caddy/Caddyfile
```

abartow91, 2022-06-14:
Restarting the caddy service seems to have worked.  Now I just need to go and fix all of the german bird identifications for the past few days.

Thanks for your help.  Here is the output for the journal entry after restarting the caddy service.

```
Jun 12 16:54:49 systemd[1]: Starting Caddy...
Jun 12 16:55:07 caddy[854]: caddy.HomeDir=/var/lib/caddy
Jun 12 16:55:07 caddy[854]: caddy.AppDataDir=/var/lib/caddy/.local/share/caddy
Jun 12 16:55:07 caddy[854]: caddy.AppConfigDir=/var/lib/caddy/.config/caddy
Jun 12 16:55:07 caddy[854]: caddy.ConfigAutosavePath=/var/lib/caddy/.config/caddy/autosave.json
Jun 12 16:55:07 caddy[854]: caddy.Version=v2.5.1 h1:bAWwslD1jNeCzDa+jDCNwb8M3UJ2tPa8UZFFzPVmGKs=
Jun 12 16:55:07 caddy[854]: runtime.GOOS=linux
Jun 12 16:55:07 caddy[854]: runtime.GOARCH=arm64
Jun 12 16:55:07 caddy[854]: runtime.Compiler=gc
Jun 12 16:55:07 caddy[854]: runtime.NumCPU=4
Jun 12 16:55:07 caddy[854]: runtime.GOMAXPROCS=4
Jun 12 16:55:07 caddy[854]: runtime.Version=go1.18.1
Jun 12 16:55:07 caddy[854]: os.Getwd=/
Jun 12 16:55:07 caddy[854]: LANG=en_GB.UTF-8
Jun 12 16:55:07 caddy[854]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
Jun 12 16:55:07 caddy[854]: NOTIFY_SOCKET=/run/systemd/notify
Jun 12 16:55:07 caddy[854]: HOME=/var/lib/caddy
Jun 12 16:55:07 caddy[854]: LOGNAME=caddy
Jun 12 16:55:07 caddy[854]: USER=caddy
Jun 12 16:55:07 caddy[854]: INVOCATION_ID=2ac58a922b30448598627df083b46e7f
Jun 12 16:55:07 caddy[854]: JOURNAL_STREAM=8:14991
Jun 12 16:55:07 caddy[854]: {"level":"info","ts":1655067307.692989,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","config_adapter":""}
Jun 12 16:55:07 caddy[854]: run: adapting config using caddyfile: parsing caddyfile tokens for 'basicauth': /etc/caddy/Caddyfile:10 - Error during parsing: unrecognized hash algorithm: views.php?view=File*
Jun 12 16:55:07 systemd[1]: caddy.service: Main process exited, code=exited, status=1/FAILURE
Jun 12 16:55:07 systemd[1]: caddy.service: Failed with result 'exit-code'.
Jun 12 16:55:07 systemd[1]: Failed to start Caddy.
Jun 13 15:59:53 systemd[1]: caddy.service: Unit cannot be reloaded because it is inactive.
Jun 13 18:30:44 systemd[1]: Starting Caddy...
Jun 13 18:30:44 caddy[1068973]: caddy.HomeDir=/var/lib/caddy
Jun 13 18:30:44 caddy[1068973]: caddy.AppDataDir=/var/lib/caddy/.local/share/caddy
Jun 13 18:30:44 caddy[1068973]: caddy.AppConfigDir=/var/lib/caddy/.config/caddy
Jun 13 18:30:44 caddy[1068973]: caddy.ConfigAutosavePath=/var/lib/caddy/.config/caddy/autosave.json
Jun 13 18:30:44 caddy[1068973]: caddy.Version=v2.5.1 h1:bAWwslD1jNeCzDa+jDCNwb8M3UJ2tPa8UZFFzPVmGKs=
Jun 13 18:30:44 caddy[1068973]: runtime.GOOS=linux
Jun 13 18:30:44 caddy[1068973]: runtime.GOARCH=arm64
Jun 13 18:30:44 caddy[1068973]: runtime.Compiler=gc
Jun 13 18:30:44 caddy[1068973]: runtime.NumCPU=4
Jun 13 18:30:44 caddy[1068973]: runtime.GOMAXPROCS=4
Jun 13 18:30:44 caddy[1068973]: runtime.Version=go1.18.1
Jun 13 18:30:44 caddy[1068973]: os.Getwd=/
Jun 13 18:30:44 caddy[1068973]: LANG=en_GB.UTF-8
Jun 13 18:30:44 caddy[1068973]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
Jun 13 18:30:44 caddy[1068973]: NOTIFY_SOCKET=/run/systemd/notify
Jun 13 18:30:44 caddy[1068973]: HOME=/var/lib/caddy
Jun 13 18:30:44 caddy[1068973]: LOGNAME=caddy
Jun 13 18:30:44 caddy[1068973]: USER=caddy
Jun 13 18:30:44 caddy[1068973]: INVOCATION_ID=1296105bca0b4becac870f88532bdf92
Jun 13 18:30:44 caddy[1068973]: JOURNAL_STREAM=8:3308900
Jun 13 18:30:44 caddy[1068973]: {"level":"info","ts":1655159444.451173,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","config_adapter":""}
Jun 13 18:30:44 caddy[1068973]: {"level":"info","ts":1655159444.466432,"logger":"admin","msg":"admin endpoint started","address":"tcp/localhost:2019","enforce_origin":false,"origins":["//localhost:2019","//[::1]:2019","//127.0.0.1:2019"]}
Jun 13 18:30:44 caddy[1068973]: {"level":"info","ts":1655159444.471696,"logger":"tls.cache.maintenance","msg":"started background certificate maintenance","cache":"0x40005cd110"}
Jun 13 18:30:44 caddy[1068973]: {"level":"warn","ts":1655159444.4727676,"logger":"http","msg":"server is listening only on the HTTP port, so no automatic HTTPS will be applied to this server","server_name":"srv0","http_port":80}
Jun 13 18:30:56 caddy[1068973]: {"level":"info","ts":1655159456.1352913,"logger":"tls","msg":"cleaning storage unit","description":"FileStorage:/var/lib/caddy/.local/share/caddy"}
Jun 13 18:30:56 caddy[1068973]: {"level":"info","ts":1655159456.1390643,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}
Jun 13 18:30:56 caddy[1068973]: {"level":"info","ts":1655159456.1395257,"msg":"serving initial configuration"}
Jun 13 18:30:56 systemd[1]: Started Caddy.
Jun 13 18:30:56 caddy[1068973]: {"level":"info","ts":1655159456.1525998,"logger":"tls","msg":"finished cleaning storage units"}
Jun 13 18:34:57 systemd[1]: Reloading Caddy.
Jun 13 18:34:57 caddy[1072738]: {"level":"info","ts":1655159697.0839634,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","config_adapter":""}
Jun 13 18:34:57 caddy[1072738]: {"level":"info","ts":1655159697.0917904,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","config_adapter":""}
[... cut ...]
Jun 13 18:35:21 caddy[1068973]: {"level":"info","ts":1655159721.110405,"logger":"tls.cache.maintenance","msg":"stopped background certificate maintenance","cache":"0x40005cd110"}
Jun 13 18:35:21 caddy[1068973]: {"level":"info","ts":1655159721.1108859,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}
Jun 13 18:35:21 caddy[1068973]: {"level":"info","ts":1655159721.1110647,"logger":"admin.api","msg":"load complete"}
Jun 13 18:35:21 systemd[1]: Reloaded Caddy.
```