mcguirepr89/BirdNET-Pi #387: No Birds Being Detected

AvnerCohen, 2022-06-17:
I have now completed the setup.

I got all the equipment as described in the various forums, I am able to access the local caddy server, see the records and overall the experience from start to end, given that I am all new to Rasperry-Pi was near perfect, thanks for making it !
[image]

However... it failed for me on the climax..  No birds are being detected, I actually tried to play a sound from xano canto next to the microphone, nothing is coming out.

The must common birds in my area (Israel) is a House Sparrow, I expect quite a high precision rate on HS calls, isn't it?

Could it be that there is an additional config I may have missed?

ehpersonal38, 2022-06-17:
Hi, could you send a screenshot of your "View Logs" page?

AvnerCohen, 2022-06-17:
Sure thing:

```17:34:59---3.0;6.0-('Passer domesticus_House Sparrow', 0.30181926)
17:34:59---6.0;9.0-('Passer domesticus_House Sparrow', 0.34740007)
17:34:59---9.0;12.0-('Passer domesticus_House Sparrow', 0.45575374)
17:34:59---12.0;15.0-('Passer domesticus_House Sparrow', 0.3442137)
17:35:02---Files loaded
17:35:03---RECORDING_LENGTH set to 15
17:35:11---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts\analyze.py --i BirdSongs/June-2022/17-Friday/2022-06-17-birdnet-17:34:56.wav --o BirdSongs/June-2022/17-Friday/2022-06-17-birdnet-17:34:56.wav.csv --lat 31.7703 --lon 34.8187 --week 23 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
17:35:12---ANALYZING AUDIO... DONE! Time 1.5 SECONDS
17:35:12---WRITING RESULTS TO BirdSongs/June-2022/17-Friday/2022-06-17-birdnet-17:34:41.wav.csv ... DONE! WROTE 0 RESULTS.
17:35:12---READING AUDIO DATA... DONE! READ 5 CHUNKS.
17:35:12---Date-time: 2022-06-17 17:34:56
17:35:13---0.0;3.0-('Passer domesticus_House Sparrow', 0.47037524)
17:35:13---3.0;6.0-('Passer domesticus_House Sparrow', 0.40673617)
17:35:13---6.0;9.0-('Passer domesticus_House Sparrow', 0.37072855)
17:35:13---9.0;12.0-('Passer domesticus_House Sparrow', 0.48044917)
17:35:13---12.0;15.0-('Passer domesticus_House Sparrow', 0.38414073)
17:35:16---Files loaded
17:35:17---RECORDING_LENGTH set to 15
17:35:27---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts\analyze.py --i BirdSongs/June-2022/17-Friday/2022-06-17-birdnet-17:35:11.wav --o BirdSongs/June-2022/17-Friday/2022-06-17-birdnet-17:35:11.wav.csv --lat 31.7703 --lon 34.8187 --week 23 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
17:35:27---ANALYZING AUDIO... DONE! Time 1.5 SECONDS
17:35:27---WRITING RESULTS TO BirdSongs/June-2022/17-Friday/2022-06-17-birdnet-17:34:56.wav.csv ... DONE! WROTE 0 RESULTS.
17:35:28---READING AUDIO DATA... DONE! READ 5 CHUNKS.
17:35:28---Date-time: 2022-06-17 17:35:11
17:35:29---0.0;3.0-('Passer domesticus_House Sparrow', 0.39043808)
17:35:29---3.0;6.0-('Passer domesticus_House Sparrow', 0.31553558)
17:35:29---6.0;9.0-('Passer domesticus_House Sparrow', 0.34832576)
17:35:29---9.0;12.0-('Passer domesticus_House Sparrow', 0.2889458)
17:35:29---12.0;15.0-('Passer domesticus_House Sparrow', 0.2838155)
17:35:32---Files loaded
17:35:33---RECORDING_LENGTH set to 15
17:35:41---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts\analyze.py --i BirdSongs/June-2022/17-Friday/2022-06-17-birdnet-17:35:26.wav --o BirdSongs/June-2022/17-Friday/2022-06-17-birdnet-17:35:26.wav.csv --lat 31.7703 --lon 34.8187 --week 23 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
17:35:41---ANALYZING AUDIO... DONE! Time 1.5 SECONDS
17:35:41---WRITING RESULTS TO BirdSongs/June-2022/17-Friday/2022-06-17-birdnet-17:35:11.wav.csv ... DONE! WROTE 0 RESULTS.
17:35:41---READING AUDIO DATA... DONE! READ 5 CHUNKS.
17:35:41---Date-time: 2022-06-17 17:35:26
17:35:43---0.0;3.0-('Passer domesticus_House Sparrow', 0.38008958)
17:35:43---3.0;6.0-('Passer domesticus_House Sparrow', 0.34855074)
17:35:43---6.0;9.0-('Passer domesticus_House Sparrow', 0.20421153)
17:35:43---9.0;12.0-('Passer domesticus_House Sparrow', 0.26688677)
17:35:43---12.0;15.0-('Passer domesticus_House Sparrow', 0.19725546)
17:35:46---Files loaded
17:35:47---RECORDING_LENGTH set to 15
```

ehpersonal38, 2022-06-17:
Okay it looks like everything is working as intended, it's just that the model is returning a confidence of around 0.35 for the House Sparrows, whereas the minimum confidence you're using (the default) is 0.7. So I would recommend going to Tools -> Settings -> Advanced Settings -> Minimum Confidence, and changing it to around 0.3. You should then see a plethora of detections!

Unfortunately this has the side effect of raising false positives pretty significantly, but what can you do? We'll have to wait for the new model to be trained and implemented.

AvnerCohen, 2022-06-17:
Thanks for that !
Yeah, I'll soon know how hard the impact on precision of the model is.

I totally wonder what would cause this to require change from the default, the mic is literally 3 meters from a `Passer domesticus` nest, and I am sure there are loads of data points to learn from on this species.

Anyway, hoping to be able to share the install publicly soon.

Thanks!

closed 2022-06-17 by AvnerCohen