mcguirepr89/BirdNET-Pi #389: Unable to access Basic Settings

Phaeton, 2022-06-18:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
If the initial installation does not auto-populate latitude, a user is unable to access the Basic Settings via web interface.

**To Reproduce**
Steps to reproduce the behavior:
1. Create new Install using installation script.
2. Review birdnet.conf (clear Latitude if populated by initial script)
3. Navigate to Tools and enter default username `birdnet` and empty password
4. Navigate to Settings and enter username and password again. User is unable to get pass the prompt.

**Expected behavior**
The user should be able to get past the login prompt to update basic settings (which includes the latitude/longitude inputs). By updating `birdnet.conf` via ssh, I was able to then access basic settings via website. If possible, populate both Latitude and Longitude with a default if the script is unable to determine the correct location.

mcguirepr89, 2022-06-21:
Hello -- thank you for bringing this to my attention! I will take a look at this more closely and will make any necessary adjustments.

My best,
Patrick

mcguirepr89, 2022-06-28:
@ehpersonal38 I’ve not been able to reproduce this problem— any chance that you can reproduce this?? Otherwise I will need to close it as an issue I can’t reproduce.

ehpersonal38, 2022-06-28:
I emptied latitude and longitude from birdnet.conf and thisrun.txt, rebooted, and I'm still able to access and update settings. @mcguirepr89

mcguirepr89, 2022-06-28:
Thanks for checking, @ehpersonal38  -- I haven't been able to replicate this either, no matter what I've tried. I'm afraid I will need to close this as an issue that we are unable to reproduce.

Sorry @Phaeton :/ Hopefully if anyone else encounters this issue, they'll find this issue and will reopen.

apawloski, 2022-08-09:
I was able to reproduce this on a fresh install (screen capture attached). The only thing I believe I've done differently from the installation guide was use an SSH key in the Raspberry Pi Image Creator settings (whereas instructions suggest password). I don't _think_ that's related, but that's the only deviation I'm aware of.

Is there other information that would be helpful to debug this?

https://user-images.githubusercontent.com/1554414/183665273-38d088d7-7e9e-4a5f-8563-2e3ae219fe03.mov

mcguirepr89, 2022-08-09:
Hi, @apawloski -- sorry to hear that you're having trouble on your installation. Thank you for the video that shows your issue and for the offer to gather more information.

- Would you mind showing me what the system shows when you visit the "View Logs" page?
- Do you have the same results in an "incognito" browser window?

My best,
Patrick

apawloski, 2022-08-09:
I can confirm same behavior in incognito (that's the firefox equivalent above but I tried it in a regular window as well). 

Here is the log, but looks unrelated:

```
09:53:38---Traceback (most recent call last):
09:53:38---  File "/usr/local/bin/server.py", line 52, in <module>
09:53:38---    audiofmt = "." + str(str(str([i for i in this_run if i.startswith('AUDIOFMT')]).split('=')[1]).split('\\')[0])
09:53:38---IndexError: list index out of range
09:53:57---Traceback (most recent call last):
09:53:57---  File "/usr/local/bin/server.py", line 52, in <module>
09:53:57---    audiofmt = "." + str(str(str([i for i in this_run if i.startswith('AUDIOFMT')]).split('=')[1]).split('\\')[0])
09:53:57---IndexError: list index out of range
09:54:17---Traceback (most recent call last):
```

mcguirepr89, 2022-08-09:
Would you mind posting your installation log here? It may have some sensitive information (location data and your system's user), as a heads up. You can either throw caution to the wind on that, [email me instead](mailto:mcguirepr89@gmail.com), or redact the portions you'd like to keep private before posting.

That log will show if anything failed during installation (looks like there maybe maybe maybe were issues with installing the python virtual environment, but hard to say without more context).

Before we get too far, would you mind opening a terminal and issuing:
```
restart_services.sh
```
and let me know the output it produces?

Thanks!

apawloski, 2022-08-09:
Thanks

restart_services reports the following:

```
+ my_dir=/home/pi/BirdNET-Pi/scripts
+ sudo systemctl stop birdnet_server.service
+ sudo systemctl stop birdnet_recording.service
+ services=(chart_viewer.service spectrogram_viewer.service icecast2.service extraction.service birdnet_recording.service birdnet_log.service)
+ for i in "${services[@]}"
+ sudo systemctl restart chart_viewer.service
+ for i in "${services[@]}"
+ sudo systemctl restart spectrogram_viewer.service
+ for i in "${services[@]}"
+ sudo systemctl restart icecast2.service
+ for i in "${services[@]}"
+ sudo systemctl restart extraction.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_recording.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_log.service
+ sudo systemctl start birdnet_server.service
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ systemctl is-active --quiet birdnet_server.service
+ status=3
+ ((  status != 0  ))
+ logger '[/usr/local/bin/restart_services.sh] Unable to start birdnet_server.service... Looping until it start properly'
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sudo systemctl restart birdnet_server.service
+ sleep 45
```

FWIW, `systemctl status` reports birdnet_server as active.

I will upload the log in a moment. I appreciate the responsiveness (very fast!) but also this is not urgent. Can pick up in a few days if you'd prefer.

apawloski, 2022-08-09:
[installation-2022-08-08.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/9291572/installation-2022-08-08.txt)

mcguirepr89, 2022-08-09:
Thanks, @apawloski -- the installation log looked good :thinking: but your system is still hanging when trying to restart services?

Would you provide the output from these commands?:

- ```
  tail -n100 /var/log/syslog | sed "/LATITUDE/d;/LONGIT/d;/BIRDWEATHER/d;/URL/d;s|$HOME||g;/EMAIL/d;/KEY/d;/STREAM/d;/PWD/d;/bitrate/d"
  ```
- ```
  grep Model /proc/cpuinfo
  ```
- ```
  df -h && free -h
  ```

apawloski, 2022-08-09:
Seeing some more errors which might be useful?

```
birdnet:~/BirdNET-Pi/scripts$ tail -n100 /var/log/syslog | sed "/LATITUDE/d;/LONGIT/d;/BIRDWEATHER/d;/URL/d;s|$HOME||g;/EMAIL/d;/KEY/d;/STREAM/d;/PWD/d;/bitrate/d;s|$USER||g"
Aug  9 10:45:43 birdnet birdnet_recording.sh[507870]: + grep 5050 /dev/fd/63
Aug  9 10:45:43 birdnet birdnet_recording.sh[520400]: ++ netstat -tulpn
Aug  9 10:45:43 birdnet birdnet_recording.sh[507870]: + sleep 1
Aug  9 10:45:44 birdnet birdnet_recording.sh[507870]: + grep 5050 /dev/fd/63
Aug  9 10:45:44 birdnet birdnet_recording.sh[520404]: ++ netstat -tulpn
Aug  9 10:45:44 birdnet birdnet_recording.sh[507870]: + sleep 1
Aug  9 10:45:45 birdnet systemd[1]: birdnet_server.service: Main process exited, code=exited, status=1/FAILURE
Aug  9 10:45:45 birdnet systemd[1]: birdnet_server.service: Failed with result 'exit-code'.
Aug  9 10:45:45 birdnet systemd[1]: birdnet_server.service: Consumed 10.339s CPU time.
Aug  9 10:45:45 birdnet birdnet_recording.sh[507870]: + grep 5050 /dev/fd/63
Aug  9 10:45:45 birdnet birdnet_recording.sh[520412]: ++ netstat -tulpn
Aug  9 10:45:45 birdnet birdnet_recording.sh[507870]: + sleep 1
Aug  9 10:45:46 birdnet birdnet_recording.sh[507870]: + grep 5050 /dev/fd/63
Aug  9 10:45:46 birdnet birdnet_recording.sh[520423]: ++ netstat -tulpn
Aug  9 10:45:46 birdnet birdnet_recording.sh[507870]: + sleep 1
Aug  9 10:45:47 birdnet birdnet_recording.sh[507870]: + grep 5050 /dev/fd/63
Aug  9 10:45:47 birdnet birdnet_recording.sh[520427]: ++ netstat -tulpn
Aug  9 10:45:47 birdnet birdnet_recording.sh[507870]: + sleep 1
Aug  9 10:45:48 birdnet birdnet_recording.sh[507870]: + grep 5050 /dev/fd/63
Aug  9 10:45:48 birdnet birdnet_recording.sh[520431]: ++ netstat -tulpn
Aug  9 10:45:48 birdnet birdnet_recording.sh[507870]: + sleep 1
Aug  9 10:45:49 birdnet birdnet_recording.sh[507870]: + grep 5050 /dev/fd/63
Aug  9 10:45:49 birdnet birdnet_recording.sh[520442]: ++ netstat -tulpn
Aug  9 10:45:49 birdnet birdnet_recording.sh[507870]: + sleep 1
Aug  9 10:45:50 birdnet systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 6665.
Aug  9 10:45:50 birdnet systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
Aug  9 10:45:50 birdnet systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
Aug  9 10:45:50 birdnet spectrogram.sh[520447]: cat: /BirdNET-Pi/analyzing_now.txt: No such file or directory
Aug  9 10:45:50 birdnet spectrogram.sh[520448]: sox FAIL formats: can't open input file `': No such file or directory
Aug  9 10:45:50 birdnet systemd[1]: spectrogram_viewer.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Aug  9 10:45:50 birdnet systemd[1]: spectrogram_viewer.service: Failed with result 'exit-code'.
Aug  9 10:45:50 birdnet birdnet_recording.sh[507870]: + grep 5050 /dev/fd/63
Aug  9 10:45:50 birdnet birdnet_recording.sh[520449]: ++ netstat -tulpn
Aug  9 10:45:50 birdnet birdnet_recording.sh[507870]: + sleep 1
Aug  9 10:45:52 birdnet birdnet_recording.sh[507870]: + grep 5050 /dev/fd/63
Aug  9 10:45:52 birdnet birdnet_recording.sh[520453]: ++ netstat -tulpn
Aug  9 10:45:52 birdnet birdnet_recording.sh[507870]: + sleep 1
Aug  9 10:45:53 birdnet birdnet_recording.sh[507870]: + grep 5050 /dev/fd/63
Aug  9 10:45:53 birdnet birdnet_recording.sh[520464]: ++ netstat -tulpn
Aug  9 10:45:53 birdnet birdnet_recording.sh[507870]: + sleep 1
Aug  9 10:45:53 birdnet python3[520365]: Traceback (most recent call last):
Aug  9 10:45:53 birdnet python3[520365]:   File "/usr/local/bin/daily_plot.py", line 67, in <module>
Aug  9 10:45:53 birdnet python3[520365]:     norm = plt.Normalize(confmax.values.min(), confmax.values.max())
Aug  9 10:45:53 birdnet python3[520365]:   File "/BirdNET-Pi/birdnet/lib/python3.9/site-packages/numpy/core/_methods.py", line 43, in _amin
Aug  9 10:45:53 birdnet python3[520365]:     return umr_minimum(a, axis, None, out, keepdims, initial, where)
Aug  9 10:45:53 birdnet python3[520365]: ValueError: zero-size array to reduction operation minimum which has no identity
Aug  9 10:45:54 birdnet birdnet_recording.sh[507870]: + grep 5050 /dev/fd/63
Aug  9 10:45:54 birdnet birdnet_recording.sh[520468]: ++ netstat -tulpn
Aug  9 10:45:54 birdnet birdnet_recording.sh[507870]: + sleep 1
Aug  9 10:45:55 birdnet birdnet_recording.sh[507870]: + grep 5050 /dev/fd/63
Aug  9 10:45:55 birdnet birdnet_recording.sh[520472]: ++ netstat -tulpn
Aug  9 10:45:55 birdnet birdnet_recording.sh[507870]: + sleep 1
Aug  9 10:45:55 birdnet systemd[1]: birdnet_server.service: Scheduled restart job, restart counter is at 119.
Aug  9 10:45:55 birdnet systemd[1]: Stopped BirdNET Analysis Server.
Aug  9 10:45:55 birdnet systemd[1]: birdnet_server.service: Consumed 10.339s CPU time.
Aug  9 10:45:55 birdnet systemd[1]: Started BirdNET Analysis Server.
Aug  9 10:45:55 birdnet systemd[1]: chart_viewer.service: Main process exited, code=exited, status=1/FAILURE
Aug  9 10:45:55 birdnet systemd[1]: chart_viewer.service: Failed with result 'exit-code'.
Aug  9 10:45:55 birdnet systemd[1]: chart_viewer.service: Consumed 15.665s CPU time.
Aug  9 10:45:56 birdnet birdnet_recording.sh[507870]: + grep 5050 /dev/fd/63
Aug  9 10:45:56 birdnet birdnet_recording.sh[520484]: ++ netstat -tulpn
Aug  9 10:45:56 birdnet birdnet_recording.sh[507870]: + sleep 1
Aug  9 10:45:57 birdnet birdnet_recording.sh[507870]: + grep 5050 /dev/fd/63
Aug  9 10:45:57 birdnet birdnet_recording.sh[520491]: ++ netstat -tulpn
Aug  9 10:45:57 birdnet birdnet_recording.sh[507870]: + sleep 1
[... cut ...]
/dev/root        59G  4.7G   52G   9% /
devtmpfs        326M     0  326M   0% /dev
tmpfs           455M     0  455M   0% /dev/shm
tmpfs           182M  752K  182M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  253M   32M  221M  13% /boot
tmpfs            91M  4.0K   91M   1% /run/user/1000
               total        used        free      shared  buff/cache   available
Mem:           909Mi       119Mi       259Mi       1.0Mi       530Mi       725Mi
Swap:           99Mi        99Mi       0.0Ki
```

mcguirepr89, 2022-08-09:
Let's try this:

- Increase your swap space (it is all used up) and reboot:
  ```
  sudo sed -i 's/CONF_SWAPSIZE=100/CONF_SWAPSIZE=2048/g' /etc/dphys-swapfile &&
  sudo sed -i 's/#CONF_MAXSWAP=2048/CONF_MAXSWAP=4096/g' /etc/dphys-swapfile && sudo reboot
  ```

We'll see if that allows the services to return.

apawloski, 2022-08-09:
Wow that did work - interesting!

ehpersonal38, 2022-08-09:
That's some pretty good troubleshooting!

mcguirepr89, 2022-08-09:
Yeah, I think I will need to instruct all RPi3B+ users to the "RPi0W2" installation instructions (which I guess makes them non-RPi4B installation instructions now)

closed 2022-08-10 by mcguirepr89