mcguirepr89/BirdNET-Pi #393: Action when analyzes are stopped

Svardsten53, 2022-06-21:
Background: I helped a bird observatory install BirdNET-Pi. The installation is local and connects via radio link to a private network which is then connected to the internet via a dynamic IP address. I can not monitor the installation remotely and it takes several hours to go out there. Daily charts are sent regularly via FTP to a web hotel.

Problem: Twice in two weeks, the analysis of the sounds has stopped working. I can see that the daily graph has not been updated and that data has not been sent to BirdWeather. When the error first occurred, the whole thing was solved with power off / power on by a visitor out there. I think, but am not sure, BirdNET-Pi somehow hangs when there is a problem with the network connection and doesn't start working again when connection is back again. The OS itself is working. I can see that FTP works with mirroring the charts directory every hour to my web hotel.

Solution: Would it be possible to build a function that automatically detects if, for example, analyzes are not performed or detections are not sent to BirdWeather and that action is taken, such as restarting services or rebooting?
I think more people can benefit from a similar feature when the program is spread and perhaps installed in places where monitoring is not possible from the outside.

mellery, 2022-06-21:
I'm seeing something similar. my 3b+ pi is running really hot so I'm waiting to see if adding a fan fixes my issue.

mellery, 2022-06-21:
@mcguirepr89 I have a normal network setup, but I see analysis stopping too, fixed by a hard reset.  Symptoms sound the same as @Svardsten53 , cause could be different.  I think my pi is overheating (it's running at 60C+) and cpu is auto throttling to the point where the analysis won't run.  I can report back later this when I get a fan installed, or open a separate ticket.  Are the running temperatures on @Svardsten53 's pi lower?

Svardsten53, 2022-06-22:
The installation is the same as the one I have at home, the same Rpi and the same software version. The network, on the other hand, is different - a subnet with .birds as a suffix instead of .local and I do not know what the connection of the main network to the internet looks like.

Unfortunately I have no opportunities to catch logs as it takes me two hours to go there by car and boat, otherwise it is unmanned most of the time. The only thing anyone else can do is to power off / power on and it will hopefully be done this weekend.

I understand that more facts are needed to make a solution. Next time I go out I will try to make a quick and dirty solution, just to avoid downtime. I'm thinking of putting restart of services in crontab every few hours and on top of that maybe even a reboot every night.

**@mellery**: My installation at home does not have an ideal air exchange and the temperature is around 70C, but I have never experienced that it has caused any problems. I assume the temperature on the other installation is lower.

JohnButcher, 2022-06-26:
This may constitute a different thread but in my case I have a pi-400 with an SSD that just can't keep up with the ingestion of new recordings (I have a plague of Goldfinches in my garden :). The effect of this is that the size of the "files" array built from the RECS_DIR listing becomes too much to process and it will hang on analyzing. Another effect is that the disk will fill up - not always due to number of WAV files but sometimes from the ffmpeg logging filling up the syslog.
My fix for this so far has been to:

- send stdout and stderr to their own service logs and have a more bespoke and aggressive logrotate
- reduce the amount of ffmpeg logging by sending stderr (in birdnet_recording.sh) to a tmp file that is overwritten each time
- use a small throttling script that monitors the number of WAV files in RECS_DIR and stops the birdnet_recording service until it drops below the threshold (I set it to 100). This runs from crontab on @reboot.

I'm still finding my way around this great project so there may well be facilities already for doing this.

Here's the throttle script - the same logic could just go into birdnet_recording.sh:

#!/bin/bash
#
# Stop the recordings while the analysis catches up
#
source /etc/birdnet/birdnet.conf
srv="birdnet_recording"
while [ 1 ];do
     sleep 61
     wavs=$(ls $RECS_DIR/*.wav | wc -l)
     if [ $(systemctl --state=active | grep $srv | wc -l) -eq 0 ];then
        state="inactive"
     else
        state="active"
     fi
     echo "$(date)    INFO ${wavs} wav files waiting, $srv state is $state"
     if [ $wavs -gt 100 -a $state = "active" ];then
        sudo systemctl stop $srv
        echo "$(date) WARNING stopped $srv service"
     elif [ $wavs -le 100 -a $state = "inactive" ];then
        sudo systemctl start $srv
        echo "$(date)    INFO started $srv service"
     fi
done

# End

Svardsten53, 2022-06-28:
Today I went out to the failing installation and found the following:

- Spectrogram working.
- Wav files were created.
- Analysis had stopped and the log showed the following:

[image]

Seven days ago the analysis stopped after some old directories were removed.
After "Restart Core Services" it all started again with analyzing all old files. I deleted the wav files and directories for all days except for today in order to see that everything worked in real time. Everything looked fine and reports are sent to BirdWeather again.  https://app.birdweather.com/stations/236 

I also found a message in the log that a subdirectory was missing in BirdSongs. Unfortunately I don't remember the name of it, but after adding the directory manually the message disappeared. I don't have that directory in my installations at home.

To be on the safe side (I hope) for that unmanned local installation I have now scheduled restart of services in crontab every six hours and also a reboot every night.

ehpersonal38, 2022-07-06:
I've had this happen, usually after I've been changing branches or updating. A restart of all services fixes things without fail. 

Today it was happening very frequently. I would see "Stopping BirdNET Analysis" as the last thing in `journalctl --no-pager`.

It turns out my problem is a hardware one. My microphone finally succomed to a rainstorm we had, and while it still works most of the time, it causes the birdnet_recording service to have some kind of fault, and subsequently all services are told to stop. Maybe that can provide some clues to anyone else having this issue.

ehpersonal38, 2022-07-06:
Ahh, it was you and not my mic?!

cam-el, 2022-07-23:
Hi I have this same issue daily. The birdnet_server.service stops processing new audio files, they build up and system gets hours behind. Only option was to reboot as not long after it appears drops the wifi connection (PiOW2).
I have enabled remote log output so was able to capture some errors:

> 2022-07-23T11:19:28+00:00 birdnetpi1 python3[79649]: Date-time: 2022-07-23 10:13:33
2022-07-23T11:19:47+00:00 birdnetpi1 python3[79649]: ANALYZING AUDIO... Exception in thread Thread-569:
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]: Traceback (most recent call last):
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:   File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:     self.run()
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:   File "/usr/lib/python3.9/threading.py", line 892, in run
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:     self._target(*self._args, **self._kwargs)
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:   File "/usr/local/bin/server.py", line 408, in handle_client
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:     sendAppriseNotifications(species,
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:   File "/home/pi/BirdNET-Pi/scripts/utils/notifications.py", line 95, in sendAppriseNotifications
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:     notify(notify_body, notify_title, image_url)
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:   File "/home/pi/BirdNET-Pi/scripts/utils/notifications.py", line 25, in notify
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:     apobj.notify(
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/apprise/Apprise.py", line 374, in notify
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:     return py3compat.asyncio.tosync(
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/apprise/py3compat/asyncio.py", line 66, in tosync
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:     return asyncio.run(cor, debug=debug)
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:   File "/usr/lib/python3.9/asyncio/runners.py", line 49, in run
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:     loop.run_until_complete(loop.shutdown_default_executor())
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:   File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
2022-07-23T11:19:48+00:00 birdnetpi1 python3[79649]:     return future.result()
2022-07-23T11:19:49+00:00 birdnetpi1 python3[79649]:   File "/usr/lib/python3.9/asyncio/base_events.py", line 563, in shutdown_default_executor
2022-07-23T11:19:49+00:00 birdnetpi1 python3[79649]:     thread.start()
2022-07-23T11:19:49+00:00 birdnetpi1 python3[79649]:   File "/usr/lib/python3.9/threading.py", line 874, in start
2022-07-23T11:19:49+00:00 birdnetpi1 python3[79649]:     _start_new_thread(self._bootstrap, ())
2022-07-23T11:19:49+00:00 birdnetpi1 python3[79649]: RuntimeError: can't start new thread
2022-07-23T11:38:56+00:00 birdnetpi1 python3[136833]: INFO: Created TensorFlow Lite XNNPACK delegate for CPU.
2022-07-23T11:38:56+00:00 birdnetpi1 python3[136833]: INFO: Created TensorFlow Lite delegate for select TF ops.

Key seems to be **RuntimeError: can't start new thread**
With this occurrence I had terminal access:

> birdnetpi1:~/BirdNET-Pi$ sudo systemctl status birdnet_server.service
● birdnet_server.service - BirdNET Analysis Server
     Loaded: loaded (/home/pi/BirdNET-Pi/templates/birdnet_server.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2022-07-23 09:03:05 AEST; 2h 34min ago
   Main PID: 79649 (python3)
      Tasks: 189 (limit: 191)
        CPU: 1h 6min 32.916s
     CGroup: /system.slice/birdnet_server.service
             └─79649 /home/pi/BirdNET-Pi/birdnet/bin/python3 /usr/local/bin/server.py

Jul 23 11:19:48 birdnetpi1 python3[79649]:     return asyncio.run(cor, debug=debug)
Jul 23 11:19:48 birdnetpi1 python3[79649]:   File "/usr/lib/python3.9/asyncio/runners.py", line 49, in run
Jul 23 11:19:48 birdnetpi1 python3[79649]:     loop.run_until_complete(loop.shutdown_default_executor())
Jul 23 11:19:48 birdnetpi1 python3[79649]:   File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
Jul 23 11:19:48 birdnetpi1 python3[79649]:     return future.result()
Jul 23 11:19:48 birdnetpi1 python3[79649]:   File "/usr/lib/python3.9/asyncio/base_events.py", line 563, in shutdown_default_executor
Jul 23 11:19:48 birdnetpi1 python3[79649]:     thread.start()
Jul 23 11:19:48 birdnetpi1 python3[79649]:   File "/usr/lib/python3.9/threading.py", line 874, in start
Jul 23 11:19:48 birdnetpi1 python3[79649]:     _start_new_thread(self._bootstrap, ())
Jul 23 11:19:48 birdnetpi1 python3[79649]: RuntimeError: can't start new thread

Here is another one from today however I could no longer access the device as it would not accept the password copy and pasted from keepass :|

> 2022-07-24T07:56:50+00:00 birdnetpi1 python3[329]: ANALYZING AUDIO... DONE! Time 9.5 SECONDS
2022-07-24T07:56:50+00:00 birdnetpi1 python3[329]: WRITING RESULTS TO /home/pi/BirdSongs/July-2022/24-Sunday/2022-07-24-birdnet-07:41:00.wav.csv ... DONE! WROTE 1 RESULTS.
[... cut ...]
2022-07-24T08:09:01+00:00 birdnetpi1 CRON[463328]: pam_unix(cron:session): session closed for user root
2022-07-24T08:09:02+00:00 birdnetpi1 systemd[1]: Starting Clean php session files...
2022-07-24T08:09:12+00:00 birdnetpi1 systemd[1]: phpsessionclean.service: Succeeded.
2022-07-24T08:09:12+00:00 birdnetpi1 systemd[1]: Finished Clean php session files.

The Pi0W2 is severely RAM limited. Is it not cleaning up after itself? **Tasks: 189 (limit: 191)**

I am happy to assist  if possible. I will look at https://github.com/JohnButcher script :+1:

cam-el, 2022-07-24:
A bit more information from the syslog. I can see errors, I just don't know what they mean :) 
The entry I have bolded below may have a solution found [here](https://www.suse.com/support/kb/doc/?id=000019044)

> 2022-07-24T10:41:12+00:00 birdnetpi1 birdnet_analysis.sh[27933]: RECORDING_LENGTH set to 30
2022-07-24T10:41:12+00:00 birdnetpi1 birdnet_analysis.sh[27933]: /home/pi/BirdNET-Pi/birdnet/bin/python3 /home/pi/BirdNET-Pi/scripts/analyze.py --i /home/pi/BirdSongs/July-2022/24-Sunday/2022-07-24-birdnet-09:27:22.wav --o /home/pi/BirdSongs/July-2022/24-Sunday/2022-07-24-birdnet-09:27:22.wav.csv --lat -34.XX --lon 149.XX --week 28 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7 --exclude_list /home/pi/BirdNET-Pi/exclude_species_list.txt --birdweather_id "IN_USE"
2022-07-24T10:41:12+00:00 birdnetpi1 python3[334]: ANALYZING AUDIO... DONE! Time 9.6 SECONDS
2022-07-24T10:41:12+00:00 birdnetpi1 python3[334]: WRITING RESULTS TO /home/pi/BirdSongs/July-2022/24-Sunday/2022-07-24-birdnet-09:26:52.wav.csv ... DONE! WROTE 0 RESULTS.
2022-07-24T10:41:15+00:00 birdnetpi1 python3[334]: READING AUDIO DATA... DONE! READ 10 CHUNKS.
2022-07-24T10:41:15+00:00 birdnetpi1 python3[334]: Date-time: 2022-07-24 09:27:22
**2022-07-24T10:41:46+00:00 birdnetpi1 kernel: [ 6449.052673] cgroup: fork rejected by pids controller in /system.slice/birdnet_server.service**
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]: ANALYZING AUDIO... Exception in thread Thread-565:
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]: Traceback (most recent call last):
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:   File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:     self.run()
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:   File "/usr/lib/python3.9/threading.py", line 892, in run
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:     self._target(*self._args, **self._kwargs)
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:   File "/usr/local/bin/server.py", line 408, in handle_client
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:     sendAppriseNotifications(species,
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:   File "/home/pi/BirdNET-Pi/scripts/utils/notifications.py", line 95, in sendAppriseNotifications
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:     notify(notify_body, notify_title, image_url)
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:   File "/home/pi/BirdNET-Pi/scripts/utils/notifications.py", line 25, in notify
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:     apobj.notify(
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/apprise/Apprise.py", line 374, in notify
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:     return py3compat.asyncio.tosync(
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/apprise/py3compat/asyncio.py", line 66, in tosync
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:     return asyncio.run(cor, debug=debug)
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:   File "/usr/lib/python3.9/asyncio/runners.py", line 49, in run
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:     loop.run_until_complete(loop.shutdown_default_executor())
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:   File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:     return future.result()
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:   File "/usr/lib/python3.9/asyncio/base_events.py", line 563, in shutdown_default_executor
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:     thread.start()
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:   File "/usr/lib/python3.9/threading.py", line 874, in start
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]:     _start_new_thread(self._bootstrap, ())
2022-07-24T10:41:46+00:00 birdnetpi1 python3[334]: RuntimeError: can't start new thread
2022-07-24T10:42:01+00:00 birdnetpi1 CRON[44855]: pam_unix(cron:session): session opened for user pi(uid=1000) by (uid=0)
2022-07-24T10:42:01+00:00 birdnetpi1 CRON[44856]: (pi) CMD (/usr/local/bin/cleanup.sh >/dev/null 2>&1)
2022-07-24T10:42:01+00:00 birdnetpi1 CRON[44855]: pam_unix(cron:session): session closed for user pi
2022-07-24T10:45:01+00:00 birdnetpi1 CRON[46017]: pam_unix(cron:session): session opened for user pi(uid=1000) by (uid=0)
2022-07-24T10:45:04+00:00 birdnetpi1 CRON[46016]: pam_unix(cron:session): session opened for user pi(uid=1000) by (uid=0)
2022-07-24T10:45:04+00:00 birdnetpi1 CRON[46018]: (pi) CMD (/usr/local/bin/cleanup.sh >/dev/null 2>&1)
2022-07-24T10:45:04+00:00 birdnetpi1 CRON[46019]: (pi) CMD (/usr/local/bin/disk_check.sh >/dev/null 2>&1)
2022-07-24T10:45:04+00:00 birdnetpi1 CRON[46016]: pam_unix(cron:session): session closed for user pi
2022-07-24T10:45:05+00:00 birdnetpi1 CRON[46017]: pam_unix(cron:session): session closed for user pi
2022-07-24T10:48:01+00:00 birdnetpi1 CRON[47109]: pam_unix(cron:session): session opened for user pi(uid=1000) by (uid=0)
2022-07-24T10:48:01+00:00 birdnetpi1 CRON[47110]: (pi) CMD (/usr/local/bin/cleanup.sh >/dev/null 2>&1)
2022-07-24T10:48:01+00:00 birdnetpi1 CRON[47109]: pam_unix(cron:session): session closed for user pi
2022-07-24T10:50:01+00:00 birdnetpi1 CRON[47856]: pam_unix(cron:session): session opened for user pi(uid=1000) by (uid=0)
[... cut ...]
2022-07-24T11:00:02+00:00 birdnetpi1 CRON[51762]: pam_unix(cron:session): session opened for user pi(uid=1000) by (uid=0)
2022-07-24T11:00:02+00:00 birdnetpi1 CRON[51764]: (pi) CMD (/usr/local/bin/cleanup.sh >/dev/null 2>&1)
2022-07-24T11:00:02+00:00 birdnetpi1 CRON[51765]: (pi) CMD (/usr/local/bin/disk_check.sh >/dev/null 2>&1)
2022-07-24T11:00:02+00:00 birdnetpi1 CRON[51762]: pam_unix(cron:session): session closed for user pi
2022-07-24T11:00:03+00:00 birdnetpi1 CRON[51763]: pam_unix(cron:session): session closed for user pi

closed 2022-09-30 by ehpersonal38