mcguirepr89/BirdNET-Pi #397: Test Notification in Tools -> Setting

sjg, 2022-06-22:
**Is your feature request related to a problem? Please describe.**
Testing Notifications to see if the settings are correct

**Describe the solution you'd like**
A button or link to send a test notification to the provided strings so that we can test that notifications are being sent correctly

**Describe alternatives you've considered**
Nothing yet as there isn't any

**Additional context**
Feedback from UCL CE Workshop

mcguirepr89, 2022-06-22:
Howdy!

Thanks! Wonderful to meet you today! I'm opening an issue for the JSON notification body now :)

ehpersonal38, 2022-06-22:
I've pushed a commit hopefully adding this feature. Do test it out and report if there are any bugs or if I've missed something. 🐤

mcguirepr89, 2022-06-25:
This is also in `main` now :)

THANK YOU @ehpersonal38 !

closed 2022-06-25 by mcguirepr89