mcguirepr89/BirdNET-Pi #399: JSON Object in Notification being Stripped in Notification Body

sjg, 2022-06-22:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
Interface strips quotes in the Notification Body when trying to create a JSON payload

**To Reproduce**
Steps to reproduce the behavior:
1. Go to  Tools -> Settings
2. In Notification Body ....
3. Enter {"sciname": "$sciname"}  (strips double quotes)
4. Enter {\"sciname\": \"$sciname\"}   (strips everything after first \)

**Expected behavior**
String either escapes the double quotes or urlencodes it  or have a button to create a JSON payload

**Additional context**
Discovered in UCL CE Workshop

mcguirepr89, 2022-06-22:
Hi, Steven!!!

I think I have this working now -- would you mind testing for me? It seems like you're comfortable with `git` so this shouldn't be tricky:

You can just
```
git pull
git checkout json_notification_body
```
then try setting the JSON formatted stuff in the notification body and let me know if that works well for you.

Feel free to correspond with me using the Zoom if you want to troubleshoot this in realtime

sjg, 2022-06-22:
just looked at the logs and getting -- database busy printed a few times

```
23:31:00---RECORDING_LENGTH set to 15
23:31:00---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts\analyze.py --i BirdSongs/June-2022/22-Wednesday/2022-06-22-birdnet-23:30:31.wav --o BirdSongs/June-2022/22-Wednesday/2022-06-22-birdnet-23:30:31.wav.csv --lat 51.6010 --lon -0.3972 --week 24 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7 --birdweather_id "IN_USE"
23:31:00---ANALYZING AUDIO... DONE! Time 1.5 SECONDS
23:31:00---WRITING RESULTS TO BirdSongs/June-2022/22-Wednesday/2022-06-22-birdnet-23:30:16.wav.csv ... DONE! WROTE 3 RESULTS.
23:31:00---Database busy
23:31:00---Database busy
23:31:00---Database busy
23:31:00---Notification Body {"sciname":"$sciname","comname":"$comname","confidence":$confidence,"lat":"$latitude","lng":"$longitude","dt":"$date $time","week":"$week","model":[{"cutoff":$cutoff,"sens":$sens,"overlap":$overlap}]}
23:31:00---2022-06-22;23:30:16;Actitis hypoleucos;Common Sandpiper;0.990927;51.601;-0.3972;0.7;25;1.25;0.0;Common_Sandpiper-99-2022-06-22-birdnet-23:30:16.mp3
23:31:00---Soundscape POST Response Status -  201
23:31:00---{ "timestamp": "2022-06-22T23:30:19+01:00","lat": 51.601,"lon": -0.3972,"soundscapeId": 12905415,"soundscapeStartTime": 3.0,"soundscapeEndTime": 6.0,"commonName": "Common Sandpiper","scientificName": "Actitis hypoleucos","algorithm": "alpha","confidence": 0.990927 }
23:31:00---Detection POST Response Status -  201
23:31:00---Database busy
23:31:00---Database busy
23:31:00---Database busy
23:31:00---Notification Body {"sciname":"$sciname","comname":"$comname","confidence":$confidence,"lat":"$latitude","lng":"$longitude","dt":"$date $time","week":"$week","model":[{"cutoff":$cutoff,"sens":$sens,"overlap":$overlap}]}
23:31:00---2022-06-22;23:30:16;Actitis hypoleucos;Common Sandpiper;0.9266642;51.601;-0.3972;0.7;25;1.25;0.0;Common_Sandpiper-93-2022-06-22-birdnet-23:30:16.mp3
23:31:00---{ "timestamp": "2022-06-22T23:30:22+01:00","lat": 51.601,"lon": -0.3972,"soundscapeId": 12905415,"soundscapeStartTime": 6.0,"soundscapeEndTime": 9.0,"commonName": "Common Sandpiper","scientificName": "Actitis hypoleucos","algorithm": "alpha","confidence": 0.9266642 }
23:31:00---Detection POST Response Status -  201
23:31:00---Database busy
23:31:00---Database busy
23:31:00---Database busy
23:31:00---Notification Body {"sciname":"$sciname","comname":"$comname","confidence":$confidence,"lat":"$latitude","lng":"$longitude","dt":"$date $time","week":"$week","model":[{"cutoff":$cutoff,"sens":$sens,"overlap":$overlap}]}
```

mcguirepr89, 2022-06-23:
Thanks for the catch!!! I accidentally pushed a change that was specific to my test system (I have an extra field in my database)

I removed that portion and your detections should get logged into the database now -- this isn't resolved, however, as @ehpersonal38 discovered -- I've been able to make JSON work at the expense of breaking the normal notifications!!!

I'll keep you _apprised_ (terribly pun intended) on this as I'm able to get it working for all types of notification bodies.

mcguirepr89, 2022-06-25:
Hello, again, @sjg! I hope this finds you enjoying your weekend ~~

I think I have the JSON body working as needed while also allowing normal notification to function as usual.

@ehpersonal38, would you also mind testing?

```
update_birdnet.sh -b daily_new_species_notifier
```

Thanks!!!!!!!!!

mcguirepr89, 2022-06-25:
@sjg Things move fast!

This is back in the `main` branch now, so you can switch back over and get that just with "Tools" > "System Controls" > "Update" -- that will reset you to the HEAD on main, which now accepts JSON objects in the notification body.

Definitely re-open this issue if it doesn't work for you!

My very best regards,
Patrick

closed 2022-06-25 by mcguirepr89