mcguirepr89/BirdNET-Pi #401: Add the ability to notify on species' first detection of the day

joeweiss, 2022-06-23:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
I would like to receive a notification on the first detection of the day for each species. The alert should be configurable by setting `APPRISE_NOTIFY_NEW_SPECIES_EACH_DAY` to `1` in the .conf file. 

**Describe alternatives you've considered**
N/A

**Additional context**
I've implemented this and I'll open a PR with the code shortly.

ehpersonal38, 2022-06-23:
@joeweiss Those commits were done in a separate branch from main, so I believe your PR can still be accepted without much conflict. Just put the burden on Patrick to reconcile _his_ code with yours when the time comes, hehehe

joeweiss, 2022-06-23:
@ehpersonal38 Cool. PR in.

mcguirepr89, 2022-06-25:
Howdy, @joeweiss!! I merged the branch that was created from the PR for this into `main` just now -- thanks again for the awesome refactoring!

closed 2022-06-25 by mcguirepr89