mcguirepr89/BirdNET-Pi #407: Recording 24/7 & privacy questions

BjornW, 2022-06-26:
I've just stumbled upon your brilliant project which I'd like to participate in, but I don't quite understand how the recording 24/7 relates to privacy. Especially if I'd open up my installation to the public Internet. 

Given that this a feature request template ;) 

My request is: Could you elaborate a bit more on the recording part in the documentation & website? 

For example: 
- Are recordings sliced automatically? What if my neighbors are having a private conversation would this become part of the recordings if there's a bird call in there as well?
- Can the audio stream & recordings stay private, but can I still share the analyzed data e.g. which birds, frequency time & such? 

Thank you in advance!

mcguirepr89, 2022-06-26:
Hello! Thanks for your interest in BirdNET-Pi.

I will do my best to address your "feature request" :)

While the onus to respect and address any and all privacy concerns lies solely with the user, there are a some things built into BirdNET-Pi to help users with those concerns and responsibilities.

The main concerns of which users must be mindful:
1) The system indiscriminately collects ambient audio data 24/7 wherever it is recording. This means the users **_MUST_** be legally allowed to collect that data as it is often not legal to record someone without both their knowledge and consent.
2) The BirdNET model expects most of the data it encounters to be of avian origin -- this will not always be the case, and will sometimes result in the system collecting and storing non-avian data that is actually of human origin. Ultimately, this means the system has the potential to collect human sounds and store that data.
3) Public installations and those that share their data with BirdWeather.com need to be extremely mindful that the system will occasionally collect human sounds (mislabeled as bird sounds, naturally). Public installations present the user with the additional onus of protecting administrative access to the system, including access to the live audio stream, the web terminal, and all system settings.

To mitigate some of these concerns, we've added a privacy feature that will force the model to consider human sounds with a (user-defined) threshold of sensitivity. Users are able to increase this privacy threshold to force the model to cull data that _could_ be of human origin, based on the threshold the user sets for this behavior.

Example:
For this example, we will assume that the example sound _is_ of human origin.
So, for instance, in an installation with no mitigating privacy settings, a sound may receive label predictions for which the `HUMAN` label is the **60th** most likely species to have made the sound in our example. Without a configured privacy threshold, the sound will get a `BIRD` label, but as we know, it actually contains some human speech. 
If we set the threshold to the top 1%, however, the model will be more sensitive to these `HUMAN` predictions, and will assume that if `HUMAN` is predicted _anywhere_ within the top 63 predictions (1% of the 6,300 species labels), the sound sample will get labeled as `HUMAN` and thus the data will _not_ be collected.

Additionally, the installation uses Basic Authentication over TLS to protect the administrative settings and live audio stream.

I hope those things begin answering your question about the privacy concerns, though the bulk of that discussion thread has a great deal of other useful information and considerations.

Let me know what follow up questions you may have~~

My best regards,
Patrick

lloydbayley, 2022-06-27:
If I may weigh in here. Mine occasionally catches me singing/whistling/talking/farting etc and usually logs it as HUMAN and throws the recording away. There has only been one occasion when it thought I was a bird (I was whistling some classical music with lots of little twiddly bits). I can't remember what it thought I was (hopefully not a "Great Tit") but I happened to notice it and just removed it as it wasn't a real bird. It's really a matter of personal responsibility as Patrick said.
 
If it did capture the bird with a voice 'underneath' it, then it would have to be a very clear recording of both to have both distinguishable. Not an impossible scenario but hopefully reasonably seldom.

Having said that, if you're a member of ASIO or the FBI, perhaps it would be unwise to make the BirdNet system public. :)

BjornW, 2022-06-29:
@mcguirepr89 & @lloydbayley thank you for sharing your thoughts and information on this topic. I'm going to have a look and see what would work for me.