mcguirepr89/BirdNET-Pi #410: Use picture from Flickr in notifications

lexologe, 2022-06-28:
**Is your feature request related to a problem? Please describe.**
Not a problem. Nice to have.

**Describe the solution you'd like**
I am using Telegram and I would like to see the pictures of detected bird (e.g. Flickr image) to be send along with the notification.

**Describe alternatives you've considered**
May be an alternative would be to send an URL in the notification text. I hope that Telegram shows a preview of this URL containing an image of the detected bird.

**Additional context**
No.

mcguirepr89, 2022-06-28:
Cool idea -- I know that Apprise can attach files, so that may be one way to get this working if the link won't suffice (I'm curious if a link to the image will show the image or just a link) :thinking:

Thanks for the feature request!

lexologe, 2022-06-28:
I tried it by adding an URL [1] to an image at the end of the telegram message. This seems not to work as I supposed. It shows the URL in text form.

In [2] there is a parameter called 'preview' that defaults to 'no'. Maybe it helps if it is set to 'on'. Even if this works, I or any other user needs to convert the detected bird name to an URL that directs to the image of the bird. I did not find a service for that for now.

Maybe the better solution would be to send the image that is already obtained from Flickr is sent as a attachment. There is a parameter called 'image' in [2] for that.

[1] https://upload.wikimedia.org/wikipedia/commons/4/4a/Buteo_buteo_%285571928172%29.jpg
[2] https://github.com/caronc/apprise/wiki/Notify_telegram#parameter-breakdown

ehpersonal38, 2022-06-28:
I think the next step is determining whether Apprise's default file attaching system will work for this kind of thing. 

https://github.com/caronc/apprise/blob/master/README.md#attaching-files

> Even if this works, I or any other user needs to convert the detected bird name to an URL that directs to the image of the bird. I did not find a service for that for now.

We'd need to recreate the Flickr api logic from `overview.php` in Python, I don't mind helping with that

ehpersonal38, 2022-07-14:
Okay, that should do it. Update your BirdNET-Pi, test it out, and let me know if that works for you!

Sp4rkR4t, 2022-07-15:
[image]

Updated to latest and enabled telegram notifications but this is the only image it sends when I append ?image=yes to the end of the line.

ehpersonal38, 2022-07-15:
Sorry, should've been more clear. You need to add "$flickrimage" in your Apprise notification body. You can remove ?image=yes

Sp4rkR4t, 2022-07-15:
[image]

Ok so I removed the ?image=yes and added $flickrimage in the message body but that doesn't display an image.

I thought there was a telegram API limitation that meant you could send images or text but not both?

ehpersonal38, 2022-07-15:
Are you updated to the latest version (Tools -> System Controls -> Update) and do you have your Flickr API key set?

Apprise sends images as a separate message, I think.

Sp4rkR4t, 2022-07-15:
Working now, thanks and excellent work.

ehpersonal38, 2022-07-18:
@phillprice Sure, just pushed a commit that should allow this.

closed 2022-07-18 by ehpersonal38