mcguirepr89/BirdNET-Pi #414: Apprise Notifications Config box clears when saving

cornerkilling, 2022-06-30:
The Apprise notification config box is clearing when updating the settings. Unable to save the config info to send notifications. 

**To Reproduce**
Steps to reproduce the behavior:
1. Input MQTT notification config to the Apprise config box
2. Send a test notification and it is successful
3. Click Update settings 
4. Page reloads and the update settings button says "Success"
5. Apprise config box is then blank 

**Expected behavior**
Apprise Config URL should be present after updating settings

mcguirepr89, 2022-06-30:
Hello — there is a permissions issue that should be fixed with an update. Please let me know if you still see the issue after an update via “Tools” > “System Controls” > “Update”

my best

ehpersonal38, 2022-06-30:
I think you'll need to `rm $HOME/BirdNET-Pi/apprise.txt` and then try again. I fixed this bug for future installations

mcguirepr89, 2022-06-30:
`update_birdnet.sh` now runs `sudo -E chown -R $USER:$USER $HOME/*` and `sudo -E chmod g+wr $HOME/*` which ought to fix that without needing to remove it (the issue is permissions so far as I could tell)

ehpersonal38, 2022-06-30:
`$HOME/BirdNET-Pi/apprise.txt` needs to be owned by `caddy` though not `pi`. So he can either execute a chown to do that, or just remove the file since config.php will recreate it anyway with proper perms once he goes to save his config.

mcguirepr89, 2022-06-30:
`caddy` is in the $USER group, which is why `sudo chmod g+wr` -- this way it is all owned by $USER:$USER, _and_ caddy can do with it what it needs :)

ehpersonal38, 2022-06-30:
Ohh, that's what groups are for. Sorry, I've been solely using "root" to login to all my servers for the past decade. *whistles*

Thanks for the fix! So yeah @cornerkilling, Patrick's first comment should sort you out!