mcguirepr89/BirdNET-Pi #417: AttributeError: module 'librosa' has no attribute 'load'

koehntopp, 2022-07-01:
Fresh installation, this is the log:

:31:50---Traceback (most recent call last):
17:31:50---  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
17:31:50---    self.run()
17:31:50---  File "/usr/lib/python3.9/threading.py", line 892, in run
17:31:50---    self._target(*self._args, **self._kwargs)
17:31:50---  File "/usr/local/bin/server.py", line 321, in handle_client
17:31:50---    audioData = readAudioData(args.i, args.overlap)
17:31:50---  File "/usr/local/bin/server.py", line 129, in readAudioData
17:31:50---    sig, rate = librosa.load(path, sr=sample_rate, mono=True, res_type='kaiser_fast')
17:31:50---AttributeError: module 'librosa' has no attribute 'load'

mcguirepr89, 2022-07-01:
Hello, @koehntopp 

I'm sorry your installation wasn't as smooth as I would have liked. There is an installation log available in your `$HOME` directory -- if you would be willing to share that information here, I would be happy to take a look more closely at potential issues during installation that may be causing these `librosa` errors.

If you are concerned with sharing your installation information (location and user info), you can email the log [to me](mailto:mcguirepr89@gmail.com) instead.

koehntopp, 2022-07-01:
I have fixed it by re-installing, the first installation may not have finished completely.

Also, I expanded the file system before installing, a step that is not in the documentation - maybe that was the problem in the first place.

closed 2022-07-01 by koehntopp