mcguirepr89/BirdNET-Pi #426: BirdnetPi webserver not reachable error 403 (forbidden)

kinteki, 2022-07-07:
Hi,
After new installation:
when i try my IP http://192.168.1.157/ this is the answer:

[image]

Read Permission problem?

kinteki, 2022-07-08:
> Hi, @kinteki
> 
> I'm sorry you're having trouble with your installation. In order to figure out what is going on, I will need some sort of log info.
> 
> You can get this by either posting here the `installation-date-.txt` log that can be found in the Raspberry Pi's `$HOME` directory, or we try troubleshooting each service. I recommend the former, as we have nothing to go on for the later.
> 
> My very best, Patrick

Hi, Patrick!
Here is the _installation-date-.txt log_

------------------------
+ set -e
+ my_dir=/home/zoli/BirdNET-Pi
+ export my_dir=/home/zoli/BirdNET-Pi
+ my_dir=/home/zoli/BirdNET-Pi
+ cd /home/zoli/BirdNET-Pi/scripts
++ uname -m
+ '[' aarch64 '!=' aarch64 ']'
+ ./install_config.sh
+ set -e
+ trap 'exit 1' SIGINT SIGHUP
+ echo 'Beginning ./install_config.sh'
Beginning ./install_config.sh
+ birdnet_conf=/home/zoli/BirdNET-Pi/birdnet.conf
+ '[' -f /home/zoli/BirdNET-Pi/birdnet.conf ']'
+ install_config
+ cat
++ curl -s4 ifconfig.co/json
++ jq .latitude
++ curl -s4 ifconfig.co/json
++ jq .longitude
+ chmod g+w /home/zoli/BirdNET-Pi/birdnet.conf
+ '[' -d /etc/birdnet ']'
+ sudo mkdir /etc/birdnet
+ sudo ln -sf /home/zoli/BirdNET-Pi/birdnet.conf /etc/birdnet/birdnet.conf
+ grep -ve '^#' -e '^$' /etc/birdnet/birdnet.conf
+ sudo -E HOME=/home/zoli USER=zoli ./install_services.sh
+ trap 'rm -f ${tmpfile}' EXIT
+ trap 'exit 1' SIGINT SIGHUP
++ mktemp
+ tmpfile=/tmp/tmp.DmQkR23QzC
+ config_file=/home/zoli/BirdNET-Pi/birdnet.conf
+ export USER=zoli
+ USER=zoli
+ export HOME=/home/zoli
+ HOME=/home/zoli
+ export PYTHON_VIRTUAL_ENV=/home/zoli/BirdNET-Pi/birdnet/bin/python3
+ PYTHON_VIRTUAL_ENV=/home/zoli/BirdNET-Pi/birdnet/bin/python3
+ '[' -f /home/zoli/BirdNET-Pi/birdnet.conf ']'
+ source /home/zoli/BirdNET-Pi/birdnet.conf
++ LATITUDE=47.949
++ LONGITUDE=21.705
++ BIRDWEATHER_ID=
++ CADDY_PWD=
++ ICE_PWD=birdnetpi
++ BIRDNETPI_URL=
++ RTSP_STREAM=
++ APPRISE_NOTIFICATION_TITLE='New BirdNET-Pi Detection'
++ APPRISE_NOTIFICATION_BODY='A   was just detected with a confidence of '
++ APPRISE_NOTIFY_EACH_DETECTION=0
++ APPRISE_NOTIFY_NEW_SPECIES=0
++ FLICKR_API_KEY=
++ FLICKR_FILTER_EMAIL=
++ RECS_DIR=/home/zoli/BirdSongs
++ REC_CARD=default
++ PROCESSED=/home/zoli/BirdSongs/Processed
++ EXTRACTED=/home/zoli/BirdSongs/Extracted
++ OVERLAP=0.0
++ CONFIDENCE=0.7
[... 1127 lines cut ...]
Building wheels for collected packages: pyrsistent, psutil
  Building wheel for pyrsistent (PEP 517): started
  Building wheel for pyrsistent (PEP 517): finished with status 'done'
  Created wheel for pyrsistent: filename=pyrsistent-0.18.1-cp39-cp39-linux_aarch64.whl size=106108 sha256=f72e3cd47753354b17b737ff2b6376fa89d40a152b892f96f8c79221e6094526
  Stored in directory: /home/zoli/.cache/pip/wheels/87/fe/e6/fc8deeb581a41e462eafaf19fee96f51cdc8391e0be1c8088a
  Building wheel for psutil (PEP 517): started
  Building wheel for psutil (PEP 517): finished with status 'done'
  Created wheel for psutil: filename=psutil-5.9.1-cp39-cp39-linux_aarch64.whl size=281510 sha256=ec3799d3596ec0e6d1aeef249ed890b626347e4513283f7e6679d9218de48fa8
  Stored in directory: /home/zoli/.cache/pip/wheels/65/66/89/7ecefa8b8cfa845352eab83061e5f0f93946a306ae25661e03
Successfully built pyrsistent psutil
Installing collected packages: traitlets, six, pyrsistent, attrs, wcwidth, tornado, pyzmq, python-dateutil, pycparser, pure-eval, ptyprocess, parso, nest-asyncio, jupyter-core, jsonschema, fastjsonschema, executing, entrypoints, asttokens, webencodings, stack-data, soupsieve, setuptools, pyparsing, pygments, prompt-toolkit, pickleshare, pexpect, nbformat, matplotlib-inline, MarkupSafe, jupyter-client, jedi, decorator, cffi, backcall, tinycss2, psutil, pandocfilters, packaging, nbclient, mistune, jupyterlab-pygments, jinja2, ipython, defusedxml, debugpy, bleach, beautifulsoup4, argon2-cffi-bindings, terminado, Send2Trash, prometheus-client, nbconvert, ipython-genutils, ipykernel, argon2-cffi, notebook, zipp, widgetsnbextension, urllib3, tzdata, smmap, pytz, numpy, llvmlite, jupyterlab-widgets, idna, charset-normalizer, certifi, toolz, tomli, threadpoolctl, scipy, requests, pytz-deprecation-shim, py, pluggy, pillow, pandas, oauthlib, numba, kiwisolver, joblib, ipywidgets, iniconfig, importlib-metadata, gitdb, fonttools, cycler, commonmark, appdirs, watchdog, validators, tzlocal, typing-extensions, toml, tenacity, soundfile, semver, scikit-learn, rich, resampy, requests-oauthlib, PyYAML, pytest, pympler, pydeck, pyarrow, protobuf, pooch, matplotlib, markdown, gitpython, click, cachetools, blinker, audioread, altair, wheel, tflite-runtime, streamlit, seaborn, pytest-mock, plotly, pip, paho-mqtt, librosa, colorama, apprise
  Attempting uninstall: setuptools
    Found existing installation: setuptools 44.1.1
    Uninstalling setuptools-44.1.1:
      Successfully uninstalled setuptools-44.1.1
  Attempting uninstall: pip
    Found existing installation: pip 20.3.4
    Uninstalling pip-20.3.4:
      Successfully uninstalled pip-20.3.4
Successfully installed MarkupSafe-2.1.1 PyYAML-6.0 Send2Trash-1.8.0 altair-4.2.0 appdirs-1.4.4 apprise-0.9.9 argon2-cffi-21.3.0 argon2-cffi-bindings-21.2.0 asttokens-2.0.5 attrs-21.4.0 audioread-2.1.9 backcall-0.2.0 beautifulsoup4-4.11.1 bleach-5.0.1 blinker-1.4 cachetools-5.2.0 certifi-2022.6.15 cffi-1.15.1 charset-normalizer-2.1.0 click-8.1.3 colorama-0.4.4 commonmark-0.9.1 cycler-0.11.0 debugpy-1.6.0 decorator-5.1.1 defusedxml-0.7.1 entrypoints-0.4 executing-0.8.3 fastjsonschema-2.15.3 fonttools-4.33.3 gitdb-4.0.9  [... cut ...] o-0.15.0 tflite-runtime-2.6.0 threadpoolctl-3.1.0 tinycss2-1.1.1 toml-0.10.2 tomli-2.0.1 toolz-0.11.2 tornado-6.1 traitlets-5.3.0 typing-extensions-4.2.0 tzdata-2022.1 tzlocal-4.2 urllib3-1.26.9 validators-0.20.0 watchdog-2.1.9 wcwidth-0.2.5 webencodings-0.5.1 wheel-0.37.1 widgetsnbextension-3.6.1 zipp-3.8.0
+ cd /home/zoli/BirdNET-Pi/scripts
+ ./install_language_label.sh -l en
Archive:  /home/zoli/BirdNET-Pi/model/labels_l18n.zip
  inflating: /home/zoli/BirdNET-Pi/model/labels_en.txt  
+ exit 0

mcguirepr89, 2022-07-08:
That installation looks successful, though I noticed this line:
```
The following packages will be upgraded:
  python3-sense-hat
```

Which implies that this installation is going atop some other software somewhere (as that package is not installed at any point, though your system _is_ using it)

What else is installed other than BirdNET-Pi?

kinteki, 2022-07-08:
This one installed : wiedehopf's readsb (fork of readsb / dump1090-fa)

https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb

2022. 07. 08. 19:56 keltezéssel, Patrick McGuire írta:

> That installation looks successful, though I noticed this line:
>
> |The following packages will be upgraded: python3-sense-hat |
>
> Which implies that this installation is going atop some other [...]

mcguirepr89, 2022-07-08:
Hi, @kinteki -- that program is not compatible with BirdNET-Pi -- it hosts a web server https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb#web-interface

if you can disable that somehow, things should work -- but as a rule, only one process can bind to a port at a time, and it is obvious that `Automatic-installation-for-readsb` wants to bind to one of the ports that BirdNET-Pi uses.

In the future, things like this are really helpful to include in the issue. A quick look at that repo would have allowed me to troubleshoot this much earlier.

My best regards,
Patrick

closed 2022-07-08 by mcguirepr89

derektgardner, 2023-02-25:
@kinteki I was able to get this working. I just changed the lighttpd port to something BirdNET-PI doesn't use (I use 8081).

Change the `server.port` in `/etc/lighttpd/lighttpd.conf` restart the service and you should be all set. You may have to flush your browser's cache to get birdnet to show up.