mcguirepr89/BirdNET-Pi #428: Error with audio?

b0rnwr0ng, 2022-07-07:
[image]

Sorry i dont have an easy way to post the error, but doesnt seem to restart, no observations today.  Only thing other than updating was enabled the reporting using discord that i can recall.

lloydbayley, 2022-07-07:
Just off to bed and thought I'd chime in quickly.
To me, it looks like ffmpeg isn't installed. You could try installing it with 
```sudo apt install ffmpeg```
but it smacks of something that went horribly wrong with your initial install. The others will put you on the right track.

b0rnwr0ng, 2022-07-07:
> Hi, @b0rnwr0ng<https://github.com/b0rnwr0ng>
>
> Could you please access the installation's shell, either via SSH or through "Tools" > "Web Terminal" and run the following commands sending me the output for each?
>
>   1.
>
> restart_services.sh
>
>   2.
>
> journalctl --no-hostname --no-pager -n100 -u birdnet_analysis -u birdnet_server -u birdnet_recording | sed "s|$HOME||g"
>
>   3.
>
> df -h && free -h
>
> Hopefully there will be some helpful information somewhere within those logs and command outputs to move forward with troubleshooting or resolve your issue.
>
> My best,
> Patrick

birdnetpi:~/BirdNET-Pi$ journalctl --no-hostname --no-pager -n100 -u birdnet_analysis -u birdnet_server -u birdnet_recording | sed "s|$HOME||g"
-- Journal begins at Thu 2022-06-30 06:51:39 EDT, ends at Thu 2022-07-07 17:14:19
EDT. --
Jul 07 16:06:35 birdnet_recording.sh[538]: W: [autospawn] lock-autospawn.c: Cannot access autospawn lock.
Jul 07 16:06:35 birdnet_recording.sh[538]: E: [pulseaudio] main.c: Failed to acquire autospawn lock
Jul 07 16:06:35 birdnet_recording.sh[402]: + pgrep arecord
Jul 07 16:06:35 birdnet_recording.sh[402]: + grep 5050 /dev/fd/63
Jul 07 16:06:35 birdnet_recording.sh[569]: ++ netstat -tulpn
Jul 07 16:06:35 birdnet_recording.sh[402]: + sleep 1
Jul 07 16:06:35 systemd[1]: Stopping BirdNET Recording...
Jul 07 16:06:35 systemd[1]: birdnet_recording.service: Succeeded.
Jul 07 16:06:35 systemd[1]: Stopped BirdNET Recording.
Jul 07 16:06:35 systemd[1]: Stopping BirdNET Analysis...
Jul 07 16:06:35 systemd[1]: birdnet_analysis.service: Succeeded.
Jul 07 16:06:35 systemd[1]: Stopped BirdNET Analysis.
Jul 07 16:06:35 systemd[1]: Stopping BirdNET Analysis Server...
Jul 07 16:06:35 systemd[1]: birdnet_server.service: Succeeded.
Jul 07 16:06:35 systemd[1]: Stopped BirdNET Analysis Server.
Jul 07 17:14:03 systemd[1]: Started BirdNET Recording.
Jul 07 17:14:03 birdnet_recording.sh[6240]: + source /etc/birdnet/birdnet.conf
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ LATITUDE=
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ LONGITUDE=
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ CADDY_PWD
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ ICE_PWD=
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ BIRDNETPI_URL=http://192.168.1.62
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ PUSHED_APP_KEY=
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ PUSHED_APP_SECRET=
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ RECS_DIR=/BirdSongs
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ REC_CARD=default
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ PROCESSED=/BirdSongs/Processed
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ EXTRACTED=/BirdSongs/Extracted
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ OVERLAP=0.0
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ CONFIDENCE=0.65
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ SENSITIVITY=1.25
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ CHANNELS=2
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ FULL_DISK=purge
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ PRIVACY_MODE=on
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ RECORDING_LENGTH=15
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ EXTRACTION_LENGTH=
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ AUDIOFMT=mp3
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ LAST_RUN=
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ THIS_RUN=
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ IDFILE=/BirdNET-Pi/IdentifiedSoFar.txt
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ PRIVACY_THRESHOLD=3
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ APPRISE_NOTIFICATION_TITLE='New BirdNET-Pi Detection'
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ APPRISE_NOTIFICATION_BODY='A $sciname $comname was just detected with a confidence of $confidence'
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ APPRISE_NOTIFY_EACH_DETECTION=0
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ RTSP_STREAM=
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ APPRISE_NOTIFY_NEW_SPECIES=1
Jul 07 17:14:03 birdnet_recording.sh[6240
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ DATABASE_LANG=en
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ FLICKR_FILTER_EMAIL=
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ APPRISE_NOTIFY_NEW_SPECIES_EACH_DAY=1
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ APPRISE_WEEKLY_REPORT=1
Jul 07 17:14:03 birdnet_recording.sh[6240]: ++ SILENCE_UPDATE_INDICATOR=0
Jul 07 17:14:03 birdnet_recording.sh[6240]: + '[' -z 15 ']'
Jul 07 17:14:03 birdnet_recording.sh[6240]: + '[' '!' -z ']'
Jul 07 17:14:03 birdnet_recording.sh[6240]: + pulseaudio --check
Jul 07 17:14:03 birdnet_recording.sh[6240]: + pgrep arecord
Jul 07 17:14:03 birdnet_recording.sh[6240]: + grep 5050 /dev/fd/63
Jul 07 17:14:03 birdnet_recording.sh[6254]: ++ netstat -tulpn
Jul 07 17:14:03 birdnet_recording.sh[6240]: + sleep 1
Jul 07 17:14:03 systemd[1]: Started BirdNET Analysis Server.
Jul 07 17:14:04 birdnet_recording.sh[6240]: + grep 5050 /dev/fd/63
Jul 07 17:14:04 birdnet_recording.sh[6281]: ++ netstat -tulpn
Jul 07 17:14:04 birdnet_recording.sh[6240]: + sleep 1
Jul 07 17:14:05 birdnet_recording.sh[6240]: + grep 5050 /dev/fd/63
Jul 07 17:14:05 birdnet_recording.sh[6291]: ++ netstat -tulpn
Jul 07 17:14:05 birdnet_recording.sh[6240]: + sleep 1
[... 24 lines cut ...]
Jul 07 17:14:13 python3[6260]: INFO: TfLiteFlexDelegate delegate: 1 nodes delegated out of 182 nodes with 1 partitions.
Jul 07 17:14:13 birdnet_recording.sh[6240]: + grep 5050 /dev/fd/63
Jul 07 17:14:13 birdnet_recording.sh[6399]: ++ netstat -tulpn
Jul 07 17:14:14 birdnet_recording.sh[6400]: tcp        0      0 127.0.0.1:5050
      0.0.0.0:*               LISTEN      6260/python3
Jul 07 17:14:14 birdnet_recording.sh[6240]: + '[' -z default ']'
Jul 07 17:14:14 birdnet_recording.sh[6240]: + arecord -f S16_LE -c2 -r48000 -t wav --max-file-time 15 -D default --use-strftime /BirdSongs/%B-%Y/%d-%A/%F-birdnet-%H:%M:%S.wav
Jul 07 17:14:14 systemd[1]: Started BirdNET Analysis.
[... cut ...]
devtmpfs        1.7G     0  1.7G   0% /dev
tmpfs           1.9G     0  1.9G   0% /dev/shm
tmpfs           759M  912K  759M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  253M   31M  222M  12% /boot
tmpfs           380M  4.0K  380M   1% /run/user/1000
               total        used        free      shared  buff/cache   available
Mem:           3.7Gi       318Mi       2.7Gi       3.0Mi       737Mi       3.3Gi
Swap:           99Mi          0B        99Mi
birdnetpi:~/BirdNET-

mcguirepr89, 2022-07-08:
Unfortunately, I don't see any unexpected errors from those logs. :thinking:

Does the "Currently Analyzing" spectrogram at the very bottom of the "Overview" page show audio content (be sure to check the timestamp on the image to see that it is current)?

mcguirepr89, 2022-07-08:
Try "Tools" > "Services" > Restart Spectrogram Viewer and let me know if it begins updating

b0rnwr0ng, 2022-07-08:
I see, sorry didn't have a great way to post this on the phone.

spectrogram_viewer.service - BirdNET-Pi Spectrogram Viewer
     Loaded: loaded (/home/pi/BirdNET-Pi/templates/spectrogram_viewer.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Fri 2022-07-08 12:55:04 EDT; 3s ago
    Process: 3510 ExecStart=/usr/local/bin/spectrogram.sh (code=exited, status=2)
   Main PID: 3510 (code=exited, status=2)
        CPU: 16ms`

`

mcguirepr89, 2022-07-08:
Hmm that also looks normal.

At this point, I have no idea what is broken. To me, it appears that everything is working, but that the audio files are empty. Can you test the microphone on another device to see that it is working properly?

If it is working on another device, let's try rebooting so that I can only view the log information from a clean boot:

1. ```
   sudo reboot
   ```
1. After the reboot, wait for the system to exhibit the issue, then we will collect the logs
1. ```
   journalctl --no-hostname --no-pager --boot -ubirdnet_analysis -ubirdnet_server -ubirdnet_recording -uextraction | sed "s|$HOME||g;/LATITUDE=/d;/LONGITUDE=/d;/BIRDWEATHER/d;/PWD/d;/URL/d"
   ```
   That command should scrub your personal info, but please double-check that there is no sensitive information within those logs --

b0rnwr0ng, 2022-07-08:
mic works on my laptop, swapped with a different mic, same thing...should I just reimage at this point?

b0rnwr0ng, 2022-07-08:
`
-- Journal begins at Fri 2022-07-01 23:31:05 EDT, ends at Fri 2022-07-08 14:19:00 EDT. --
Jul 08 14:18:03 systemd[1]: Started BirdNET Recording.
Jul 08 14:18:03 systemd[1]: Started BirdNET Analysis Server.
Jul 08 14:18:03 systemd[1]: Started BirdNET Analysis.
Jul 08 14:18:03 systemd[1]: Started BirdNET BirdSound Extraction.
Jul 08 14:18:03 birdnet_recording.sh[400]: + source /etc/birdnet/birdnet.conf
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ PUSHED_APP_SECRET=
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ RECS_DIR=/BirdSongs
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ REC_CARD=default
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ PROCESSED=/BirdSongs/Processed
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ EXTRACTED=/BirdSongs/Extracted
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ OVERLAP=0.0
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ CONFIDENCE=0.65
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ SENSITIVITY=1.25
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ CHANNELS=2
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ FULL_DISK=purge
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ PRIVACY_MODE=on
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ RECORDING_LENGTH=15
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ EXTRACTION_LENGTH=
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ AUDIOFMT=mp3
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ LAST_RUN=
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ THIS_RUN=
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ IDFILE=/BirdNET-Pi/IdentifiedSoFar.txt
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ PRIVACY_THRESHOLD=3
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ APPRISE_NOTIFICATION_TITLE='New BirdNET-Pi Detection'
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ APPRISE_NOTIFICATION_BODY='A $sciname $comname was just detected with a confidence of $confidence'
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ APPRISE_NOTIFY_EACH_DETECTION=0
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ APPRISE_NOTIFY_NEW_SPECIES=1
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ DATABASE_LANG=en
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ FLICKR_FILTER_EMAIL=
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ APPRISE_NOTIFY_NEW_SPECIES_EACH_DAY=1
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ APPRISE_WEEKLY_REPORT=1
Jul 08 14:18:05 birdnet_recording.sh[400]: ++ SILENCE_UPDATE_INDICATOR=0
Jul 08 14:18:05 birdnet_recording.sh[400]: + '[' -z 15 ']'
Jul 08 14:18:05 birdnet_recording.sh[400]: + '[' '!' -z ']'
Jul 08 14:18:05 birdnet_recording.sh[400]: + pulseaudio --check
Jul 08 14:18:05 birdnet_recording.sh[467]: E: [pulseaudio] core-util.c: Failed to create secure directory (/run/user/1000/pulse): No such file or directory
Jul 08 14:18:05 birdnet_recording.sh[400]: + pulseaudio --start
Jul 08 14:18:05 birdnet_recording.sh[544]: E: [autospawn] core-util.c: Failed to create secure directory (/run/user/1000/pulse): No such file or directory
Jul 08 14:18:05 birdnet_recording.sh[544]: W: [autospawn] lock-autospawn.c: Cannot access autospawn lock.
Jul 08 14:18:05 birdnet_recording.sh[544]: E: [pulseaudio] main.c: Failed to acquire autospawn lock
Jul 08 14:18:05 birdnet_recording.sh[400]: + pgrep arecord
Jul 08 14:18:05 birdnet_recording.sh[400]: + grep 5050 /dev/fd/63
Jul 08 14:18:05 birdnet_recording.sh[568]: ++ netstat -tulpn
Jul 08 14:18:05 birdnet_recording.sh[400]: + sleep 1
Jul 08 14:18:05 systemd[1]: Stopping BirdNET Recording...
Jul 08 14:18:05 systemd[1]: birdnet_recording.service: Succeeded.
Jul 08 14:18:05 systemd[1]: Stopped BirdNET Recording.
Jul 08 14:18:05 systemd[1]: Stopping BirdNET Analysis...
Jul 08 14:18:05 systemd[1]: birdnet_analysis.service: Succeeded.
Jul 08 14:18:05 systemd[1]: Stopped BirdNET Analysis.
Jul 08 14:18:05 systemd[1]: Stopping BirdNET Analysis Server...
Jul 08 14:18:05 systemd[1]: birdnet_server.service: Succeeded.
Jul 08 14:18:05 systemd[1]: Stopped BirdNET Analysis Server.
Jul 08 14:18:05 systemd[1]: Stopping BirdNET BirdSound Extraction...
Jul 08 14:18:06 env[434]: Terminated
Jul 08 14:18:06 systemd[1]: extraction.service: Succeeded.
Jul 08 14:18:06 systemd[1]: Stopped BirdNET BirdSound Extraction.

`

b0rnwr0ng, 2022-07-08:
I deleted Thursdays folders as it seems I have wav files from today (Friday-08) at 12:23 and on..  I removed all the July folders, wondering if its hung up on that since the error still states for a Thursday file..

b0rnwr0ng, 2022-07-08:
rebooted and that seems to work now..

b0rnwr0ng, 2022-07-08:
`-- Journal begins at Fri 2022-07-01 23:31:05 EDT, ends at Fri 2022-07-08 14:28:09 EDT. --
Jul 08 14:25:17 systemd[1]: Started BirdNET Recording.
Jul 08 14:25:17 systemd[1]: Started BirdNET Analysis Server.
Jul 08 14:25:17 systemd[1]: Started BirdNET Analysis.
Jul 08 14:25:17 systemd[1]: Started BirdNET BirdSound Extraction.
Jul 08 14:25:17 birdnet_recording.sh[393]: + source /etc/birdnet/birdnet.conf
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ PUSHED_APP_SECRET=
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ RECS_DIR=/BirdSongs
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ REC_CARD=default
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ PROCESSED=/BirdSongs/Processed
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ EXTRACTED=/BirdSongs/Extracted
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ OVERLAP=0.0
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ CONFIDENCE=0.65
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ SENSITIVITY=1.25
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ CHANNELS=2
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ FULL_DISK=purge
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ PRIVACY_MODE=on
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ RECORDING_LENGTH=15
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ EXTRACTION_LENGTH=
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ AUDIOFMT=mp3
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ LAST_RUN=
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ THIS_RUN=
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ IDFILE=/BirdNET-Pi/IdentifiedSoFar.txt
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ PRIVACY_THRESHOLD=3
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ APPRISE_NOTIFICATION_TITLE='New BirdNET-Pi Detection'
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ APPRISE_NOTIFICATION_BODY='A $sciname $comname was just detected with a confidence of $confidence'
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ APPRISE_NOTIFY_EACH_DETECTION=0
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ APPRISE_NOTIFY_NEW_SPECIES=1
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ DATABASE_LANG=en
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ FLICKR_FILTER_EMAIL=
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ APPRISE_NOTIFY_NEW_SPECIES_EACH_DAY=1
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ APPRISE_WEEKLY_REPORT=1
Jul 08 14:25:18 birdnet_recording.sh[393]: ++ SILENCE_UPDATE_INDICATOR=0
Jul 08 14:25:18 birdnet_recording.sh[393]: + '[' -z 15 ']'
Jul 08 14:25:18 birdnet_recording.sh[393]: + '[' '!' -z ']'
Jul 08 14:25:18 birdnet_recording.sh[393]: + pulseaudio --check
Jul 08 14:25:19 birdnet_recording.sh[442]: E: [pulseaudio] core-util.c: Failed to create secure directory (/run/user/1000/pulse): No such file or directory
Jul 08 14:25:19 birdnet_recording.sh[393]: + pulseaudio --start
Jul 08 14:25:19 birdnet_recording.sh[528]: E: [autospawn] core-util.c: Failed to create secure directory (/run/user/1000/pulse): No such file or directory
Jul 08 14:25:19 birdnet_recording.sh[528]: W: [autospawn] lock-autospawn.c: Cannot access autospawn lock.
Jul 08 14:25:19 birdnet_recording.sh[528]: E: [pulseaudio] main.c: Failed to acquire autospawn lock
Jul 08 14:25:19 birdnet_recording.sh[393]: + pgrep arecord
Jul 08 14:25:19 birdnet_recording.sh[393]: + grep 5050 /dev/fd/63
Jul 08 14:25:19 birdnet_recording.sh[553]: ++ netstat -tulpn
Jul 08 14:25:19 birdnet_recording.sh[393]: + sleep 1
Jul 08 14:25:19 birdnet_recording.sh[393]: + grep 5050 /dev/fd/63
Jul 08 14:25:19 birdnet_recording.sh[623]: ++ netstat -tulpn
Jul 08 14:25:19 birdnet_recording.sh[393]: + sleep 1
Jul 08 14:25:20 birdnet_recording.sh[393]: + grep 5050 /dev/fd/63
Jul 08 14:25:20 birdnet_recording.sh[642]: ++ netstat -tulpn
Jul 08 14:25:20 birdnet_recording.sh[393]: + sleep 1
Jul 08 14:25:21 birdnet_recording.sh[393]: + grep 5050 /dev/fd/63
Jul 08 14:25:21 birdnet_recording.sh[653]: ++ netstat -tulpn
Jul 08 14:25:21 birdnet_recording.sh[393]: + sleep 1
Jul 08 14:25:22 birdnet_recording.sh[393]: + grep 5050 /dev/fd/63
Jul 08 14:25:22 birdnet_recording.sh[665]: ++ netstat -tulpn
Jul 08 14:25:22 birdnet_recording.sh[393]: + sleep 1
Jul 08 14:25:23 birdnet_recording.sh[393]: + grep 5050 /dev/fd/63
Jul 08 14:25:23 birdnet_recording.sh[677]: ++ netstat -tulpn
Jul 08 14:25:23 birdnet_recording.sh[393]: + sleep 1
Jul 08 14:25:24 birdnet_recording.sh[393]: + grep 5050 /dev/fd/63
Jul 08 14:25:24 birdnet_recording.sh[685]: ++ netstat -tulpn
Jul 08 14:25:24 birdnet_recording.sh[393]: + sleep 1
Jul 08 14:25:25 birdnet_recording.sh[393]: + grep 5050 /dev/fd/63
Jul 08 14:25:25 birdnet_recording.sh[740]: ++ netstat -tulpn
Jul 08 14:25:25 birdnet_recording.sh[393]: + sleep 1
Jul 08 14:25:26 birdnet_recording.sh[393]: + grep 5050 /dev/fd/63
Jul 08 14:25:26 birdnet_recording.sh[748]: ++ netstat -tulpn
Jul 08 14:25:26 birdnet_recording.sh[393]: + sleep 1
Jul 08 14:25:27 birdnet_recording.sh[393]: + grep 5050 /dev/fd/63
[... 256 lines cut ...]
Jul 08 14:27:45 systemd[1]: Stopped BirdNET Analysis.
Jul 08 14:27:45 systemd[1]: Started BirdNET Analysis.
Jul 08 14:27:46 birdnet_analysis.sh[2294]: find: ‘/BirdSongs/StreamData’: No such file or directory
Jul 08 14:27:46 birdnet_analysis.sh[2276]: Files loaded
Jul 08 14:27:47 birdnet_analysis.sh[2276]: RECORDING_LENGTH set to 15
Jul 08 14:27:48 env[2323]: Line = 2022-07-08-birdnet-14:27:26.wav.csv
Jul 08 14:27:57 birdnet_analysis.sh[2276]: /BirdNET-Pi/birdnet/bin/python3 /BirdNET-Pi/scripts\analyze.py --i /BirdSongs/July-2022/08-Friday/2022-07-08-birdnet-14:27:41.wav --o /BirdSongs/July-2022/08-Friday/2022-07-08-birdnet-14:27:41.wa [... cut ...]
Jul 08 14:28:01 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 11.
Jul 08 14:28:01 systemd[1]: Stopped BirdNET Analysis.
Jul 08 14:28:01 systemd[1]: Started BirdNET Analysis.
Jul 08 14:28:02 birdnet_analysis.sh[2494]: find: ‘/BirdSongs/StreamData’: No such file or directory
Jul 08 14:28:02 birdnet_analysis.sh[2473]: Files loaded
Jul 08 14:28:03 birdnet_analysis.sh[2473]: RECORDING_LENGTH set to 15
Jul 08 14:28:03 env[2523]: Line = 2022-07-08-birdnet-14:27:41.wav.csv
`

b0rnwr0ng, 2022-07-08:
I must've rebooted after I made the discord notification changes and did apt get upgrade, and I'm assuming the wav file got corrupted during the reboot and stuck there for processing.

b0rnwr0ng, 2022-07-08:
Eh, things are a bit wacky still.  Live audio not working.  Lot of noise in my area tofay so figured thats why it wasn't picking anything up

mcguirepr89, 2022-07-08:
I'm afraid I will need a lot more information than its wackyness :)

```
journalctl --no-hostname --no-pager -n300 -ubirdnet_analysis -ubirdnet_server -ubirdnet_recording -uextraction | sed "s|$HOME||g;/LATITUDE=/d;/LONGITUDE=/d;/BIRDWEATHER/d;/PWD/d;/URL/d;/KEY/d;/STREAM/d"
```

b0rnwr0ng, 2022-07-08:
Lol, yeah sorry lots going on atm.  Seems normal right now, but nothing showing, but there is a lot of work going on in my area which may have something to do with it.  Saw errors earlier but not so much now.. May need to wait until tomorrow

b0rnwr0ng, 2022-07-08:
Created streamdata folder:
Jul 08 18:06:35 systemd[1]: Stopped BirdNET Analysis.
Jul 08 18:06:35 systemd[1]: Started BirdNET Analysis.
Jul 08 18:06:36 birdnet_analysis.sh[124853]: find: ‘/BirdSongs/StreamData’: No such file or directory
Jul 08 18:06:36 birdnet_analysis.sh[124835]: Files loaded
Jul 08 18:06:37 birdnet_analysis.sh[124835]: RECORDING_LENGTH set to 15
Jul 08 18:06:38 env[124889]: Line = 2022-07-08-birdnet-18:06:16.wav.csv
Jul 08 18:06:47 birdnet_analysis.sh[124835]: /BirdNET-Pi/birdnet/bin/python3 /BirdNET-Pi/scripts\analyze.py --i /BirdSongs/July-2022/08-Friday/2022-07-08-birdnet-18:06:31.wav --o /BirdSongs/July-2022/08-Friday/2022-07-08-birdnet-18:06:31.wav.csv
--lat 40.XX --lon -74.XX --week 26 --overlap 0.0 --sensitivity 1.25 --min_conf 0.65 --birdweather_id "IN_USE"
Jul 08 18:06:48 python3[407]: ANALYZING AUDIO... DONE! Time 1.6 SECONDS
Jul 08 18:06:48 python3[407]: WRITING RESULTS TO /BirdSongs/July-2022/08-Friday/2022-07-08-birdnet-18:06:16.wav.csv ... DONE! WROTE 0 RESULTS.
Jul 08 18:06:48 python3[407]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
Jul 08 18:06:48 python3[407]: Date-time: 2022-07-08 18:06:31
Jul 08 18:06:49 birdnet_analysis.sh[124984]: 0.0;3.0-('Scolopax minor_American Woodcock', 0.046568252)
Jul 08 18:06:49 birdnet_analysis.sh[124984]: 3.0;6.0-('Scolopax minor_American Woodcock', 0.03760868)
Jul 08 18:06:49 birdnet_analysis.sh[124984]: 6.0;9.0-('Dryocopus pileatus_Pileated Woodpecker', 0.038295817)
Jul 08 18:06:49 birdnet_analysis.sh[124984]: 9.0;12.0-('Scolopax minor_American Woodcock', 0.041059177)
Jul 08 18:06:49 birdnet_analysis.sh[124984]: 12.0;15.0-('Vireo solitarius_Blue-headed Vireo', 0.036714423)
Jul 08 18:06:49 systemd[1]: birdnet_analysis.service: Succeeded.
Jul 08 18:06:51 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 677.
Jul 08 18:06:51 systemd[1]: Stopped BirdNET Analysis.
Jul 08 18:06:51 systemd[1]: Started BirdNET Analysis.
Jul 08 18:06:52 birdnet_analysis.sh[125023]: Files loaded
Jul 08 18:06:53 birdnet_analysis.sh[125023]: RECORDING_LENGTH set to 15
Jul 08 18:06:54 env[125080]: Line = 2022-07-08-birdnet-18:06:31.wav.csv

b0rnwr0ng, 2022-07-08:
This appear to show resulta not written:
`

-- Boot 897bc0f7d6854cbc8da81ff2a294a998 --
Jul 08 18:10:25 systemd[1]: Started BirdNET Recording.
Jul 08 18:10:25 systemd[1]: Started BirdNET Analysis Server.
Jul 08 18:10:25 systemd[1]: Started BirdNET Analysis.
Jul 08 18:10:25 systemd[1]: Started BirdNET BirdSound Extraction.
Jul 08 18:10:25 birdnet_recording.sh[395]: + source /etc/birdnet/birdnet.conf
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ PUSHED_APP_SECRET=
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ RECS_DIR=/BirdSongs
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ REC_CARD=default
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ PROCESSED=/BirdSongs/Processed
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ EXTRACTED=/BirdSongs/Extracted
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ OVERLAP=0.0
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ CONFIDENCE=0.65
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ SENSITIVITY=1.25
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ CHANNELS=2
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ FULL_DISK=purge
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ PRIVACY_MODE=on
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ RECORDING_LENGTH=15
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ EXTRACTION_LENGTH=
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ AUDIOFMT=mp3
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ LAST_RUN=
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ THIS_RUN=
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ IDFILE=/BirdNET-Pi/IdentifiedSoFar.txt
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ PRIVACY_THRESHOLD=3
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ APPRISE_NOTIFICATION_TITLE='New BirdNET-Pi Detection'
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ APPRISE_NOTIFICATION_BODY='A $sciname $comname was just detected with a confidence of $confidence'
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ APPRISE_NOTIFY_EACH_DETECTION=0
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ APPRISE_NOTIFY_NEW_SPECIES=1
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ DATABASE_LANG=en
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ FLICKR_FILTER_EMAIL=
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ APPRISE_NOTIFY_NEW_SPECIES_EACH_DAY=1
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ APPRISE_WEEKLY_REPORT=1
Jul 08 18:10:27 birdnet_recording.sh[395]: ++ SILENCE_UPDATE_INDICATOR=0
Jul 08 18:10:27 birdnet_recording.sh[395]: + '[' -z 15 ']'
Jul 08 18:10:27 birdnet_recording.sh[395]: + '[' '!' -z ']'
Jul 08 18:10:27 birdnet_recording.sh[395]: + pulseaudio --check
Jul 08 18:10:27 birdnet_recording.sh[450]: E: [pulseaudio] core-util.c: Failed to
create secure directory (/run/user/1000/pulse): No such file or directory
Jul 08 18:10:27 birdnet_recording.sh[395]: + pulseaudio --start
Jul 08 18:10:27 birdnet_recording.sh[536]: E: [autospawn] core-util.c: Failed to create secure directory (/run/user/1000/pulse): No such file or directory
Jul 08 18:10:27 birdnet_recording.sh[536]: W: [autospawn] lock-autospawn.c: Cannot access autospawn lock.
Jul 08 18:10:27 birdnet_recording.sh[536]: E: [pulseaudio] main.c: Failed to acquire autospawn lock
Jul 08 18:10:27 birdnet_recording.sh[395]: + pgrep arecord
Jul 08 18:10:27 birdnet_recording.sh[395]: + grep 5050 /dev/fd/63
Jul 08 18:10:27 birdnet_recording.sh[552]: ++ netstat -tulpn
Jul 08 18:10:27 birdnet_recording.sh[395]: + sleep 1
Jul 08 18:10:28 birdnet_recording.sh[395]: + grep 5050 /dev/fd/63
Jul 08 18:10:28 birdnet_recording.sh[645]: ++ netstat -tulpn
Jul 08 18:10:28 birdnet_recording.sh[395]: + sleep 1
Jul 08 18:10:29 birdnet_recording.sh[395]: + grep 5050 /dev/fd/63
Jul 08 18:10:29 birdnet_recording.sh[661]: ++ netstat -tulpn
Jul 08 18:10:29 birdnet_recording.sh[395]: + sleep 1
Jul 08 18:10:30 birdnet_recording.sh[395]: + grep 5050 /dev/fd/63
Jul 08 18:10:30 birdnet_recording.sh[675]: ++ netstat -tulpn
Jul 08 18:10:30 birdnet_recording.sh[395]: + sleep 1
Jul 08 18:10:31 birdnet_recording.sh[395]: + grep 5050 /dev/fd/63
Jul 08 18:10:31 birdnet_recording.sh[694]: ++ netstat -tulpn
Jul 08 18:10:31 birdnet_recording.sh[395]: + sleep 1
Jul 08 18:10:32 birdnet_recording.sh[395]: + grep 5050 /dev/fd/63
Jul 08 18:10:32 birdnet_recording.sh[706]: ++ netstat -tulpn
Jul 08 18:10:32 birdnet_recording.sh[395]: + sleep 1
Jul 08 18:10:33 birdnet_recording.sh[395]: + grep 5050 /dev/fd/63
Jul 08 18:10:33 birdnet_recording.sh[714]: ++ netstat -tulpn
Jul 08 18:10:33 birdnet_recording.sh[395]: + sleep 1
Jul 08 18:10:34 birdnet_recording.sh[395]: + grep 5050 /dev/fd/63
Jul 08 18:10:34 birdnet_recording.sh[768]: ++ netstat -tulpn
Jul 08 18:10:34 birdnet_recording.sh[395]: + sleep 1
[... 159 lines cut ...]
Jul 08 18:11:34 birdnet_analysis.sh[1339]: /BirdNET-Pi/birdnet/bin/python3 /BirdNET-Pi/scripts\analyze.py --i /BirdSongs/July-2022/08-Friday/2022-07-08-birdnet-18:11:19.wav --o /BirdSongs/July-2022/08-Friday/2022-07-08-birdnet-18:11:19.wav.csv --lat 40.XX --lon -74.XX --week 26 --overlap 0.0 --sensitivity 1.25 --min_conf 0.65
--birdweather_id "IN_USE"
Jul 08 18:11:34 python3[396]: ANALYZING AUDIO... DONE! Time 1.6 SECONDS
Jul 08 18:11:34 python3[396]: WRITING RESULTS TO /BirdSongs/July-2022/08-Friday/2022-07-08-birdnet-18:11:04.wav.csv ... DONE! WROTE 0 RESULTS.
Jul 08 18:11:34 python3[396]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
Jul 08 18:11:34 python3[396]: Date-time: 2022-07-08 18:11:19
Jul 08 18:11:36 birdnet_analysis.sh[1508]: 0.0;3.0-('Human_Human', 0.0)
Jul 08 18:11:36 birdnet_analysis.sh[1508]: 3.0;6.0-('Human_Human', 0.0)
[... cut ...] -18:11:34.wav.csv --lat 40.XX --lon -74.XX --week 26 --overlap 0.0 --sensitivity 1.25 --min_conf 0.65
--birdweather_id "IN_USE"
Jul 08 18:11:50 python3[396]: ANALYZING AUDIO... DONE! Time 1.7 SECONDS
Jul 08 18:11:50 python3[396]: WRITING RESULTS TO /BirdSongs/July-2022/08-Friday/2022-07-08-birdnet-18:11:19.wav.csv ... DONE! WROTE 0 RESULTS.
Jul 08 18:11:50 python3[396]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
Jul 08 18:11:50 python3[396]: Date-time: 2022-07-08 18:11:34
birdnetpi:~/BirdNET-Pi$

b0rnwr0ng, 2022-07-09:
Everything seems ok but nondetections which is highly unusual.  Restarted live and got:

`
Restarting icecast2 (via systemctl): icecast2.service.
Unit restart.service could not be found.

`

ehpersonal38, 2022-07-09:
Having this same exact problem since commit ce017a9baea92169f44a9ed5d277dd777ada9901. The problem is that `scripts/cleanup.sh` is detecting a backlog 10 or more files in `/home/pi/BirdSongs/Processed`, and subsequently stopping all services. But a few lines above that script, it only ever deletes files from Processed if theres 100 or more? Can you shed any light @mcguirepr89 ?

For now I'm commenting out the last 3 lines in `scripts/cleanup.sh` and then running `bash ~/BirdNET-Pi/scripts/restart_services.sh`

b0rnwr0ng, 2022-07-09:
I wound up doing the inatall_birdnet script but also did an update after this comment above, seems ok now?  Ill keep an eye out

closed 2022-07-10 by b0rnwr0ng