mcguirepr89/BirdNET-Pi #430: Create a restorable data backup from the web ui.

Frono, 2022-07-09:
It would be nice to have a function for potential failed platform hardware (given that the default pi is on an SD Card) that would create a data backup that could be restored on another instance. Depending on the stability of the underlying data structure it the backup may need to be version aware...

mcguirepr89, 2022-07-09:
Hi, @Frono!! Great idea!!

I'm actually working on this and have most of it built, except the last part, which is the user-configured crontab -- I want to make that less esoteric and maybe create a that I can create those crontabs automatically.

The backend will be `rclone` and I'm bringing its web-gui into BirdNET-Pi as well -- it will all be configured and setup from the "Tools" > "Backups" page. It will make for super easy and regular backups and restoration. Currently, I have my entire database and all extracted sound clips from a 16GB SD card backed up to a free Gmail account and I LOVE it!

If you have any experience developing front-end stuff, I'd be happy to work with you on the `rclone` branch that is still development and will eventually be the recommended backup method. Otherwise, I will keep you updated on its progress!

My very best regards,
Patrick

mcguirepr89, 2022-07-11:
@ehpersonal38 I am curious what your GUI thoughts are on this? Part of me is inclined to try to make a GUI tool that will help users configure an appropriate `cron` entry -- another part of me wants to create a `systemd` service for logging and troubleshooting and easy toggling to disable/enable on demand.

GUI wise, this is going to be so tricky!!! Or, at least, I'm scratching my head on the most user-friendly way to actually _use_ rclone. I _think_ I've figured the most user-friendly way to _configure_ rclone . . . :thinking:, but sending the rclone request to the rclone remote control isn't user-intuitive and can be accomplished in so many ways (automatically through cron or systemd, or as a manual button, or as a reaction to a trigger, etc)

Let me know what you think as I bet your approach is a lot prettier and cooler than mine :)

Frono, 2022-07-11:
Is the intent of the software, that someone with little technical ability can run it on a Pi? If would be much easier for the end user to have a config/database backup and just document that in the case of disk loss, the recordings directory would not be restored. 
If so I can think of some options for allowing a user to just download their configs and database from the front end:
a) maybe a cron job to make a nightly backup automatically.
b) maybe allow the user to kick the job off manually
c) It would be nice if they system automatically backed up prior to a software update.
Give the user a screen that shows 5 or so most recent backups (# could be in configuration file) along with the date/time they were made.
It would be nice if the system automatically deleted backups older than #. 

The documented recovery procedure would involve the two steps they already know.
1) reload the OS
2) run the curl to get the repository
3) upload the backup file

My challenge with this is to understand what would have to be backed up to keep their audio settings from the pi, or even if it should be kept...

These are just thoughts... They may not be coherent...  :)

I really appreciate all that you have done with this software! My wife and I watch the front-end all the time for new birds and to analyze the patterns of current birds...

mcguirepr89, 2022-08-22:
Sorry for the tremendous delay on this feature request -- I got it working and use it myself, but it has not been easy to integrate as an update (and I don't have the time/energy to put together documentation for the manual process).

I wish I had more to provide as an update, but this note will have to do until I get rclone into the installer.

mcguirepr89, 2022-09-15:
I wanted to add these steps as potentially valuable notes:

@Frono's mention:
```
. . .
The documented recovery procedure would involve the two steps they already know.

    1. reload the OS
    2. run the curl to get the repository
    3. upload the backup file
. . .
```

To get the stuff needed for step #&#x2060;3 would be the `~/BirdNET-Pi/birdnet.conf` file, and the database, the `~/BirdNET-Pi/scripts/birds.db` file. You can download a copy of both using the "Tools" > "File Manager"

After steps #&#x2060;1 and #&#x2060;2, you would need to put those two files back in place, making sure they have the proper permissions (that is, their user and group match that of the user under whom the installation happened and permission to **r**ead, **w**rite, and e**x**ecute them `sudo -E chown $USER:$USER ~/BirdNET-Pi/birdnet.conf ~/BirdNET-Pi/scripts/birds.db && sudo chmod ugo+wrx ~/BirdNET-Pi/birdnet.conf ~/BirdNET-Pi/scripts/birds.db`)
Once those two items are back in place, that would restore most things (_except_ the actual audio files).

To backup the audio as well, grab the `~/BirdSongs/Extracted` directory and all of its contents. After replacing the directory on a restored system, ensure again that the ownership and permissions are appropriate:`sudo -E chown -R $USER:$USER ~/BirdSongs/Extracted && sudo chmod -R ugo+rw ~/BirdSongs/Extracted`

closed 2022-12-13 by mcguirepr89