mcguirepr89/BirdNET-Pi #431: Service broken after config is updated via web UI

Bobblybook, 2022-07-10:
Hi! I've recently set this BirdNET-Pi up on a Pi 3B+ rev.1.3 with a completely fresh install of x64 lite pi.

Installation went well, I'm setting everything up in preparation for testing. However when making any changes on the web UI's "tools" page, after clicking "Update Settings" it hangs forever on "Updating... please wait.".
Initially I thought the website hard crashed, but reloading the front page does produce html output, the contents are just blank. All the headers are still there, the web server is still running, and I can ssh into the Pi just fine, it hasn't crashed: https://imgur.com/nVV9aUm

I checked the syslog and it seems like python is crashing/hanging when trying to re-graph the charts.
I don't currently have a microphone connected and therefore have zero data points, maybe this is causing issues? Interestingly after a reset, the blank empty graphs do generate correctly. It's only after clicking update here that it seems to break.

The modified settings do appear to save, but I need to restart the pi to get the service back up and running properly.

I also noticed a similar python crashing error encountered by @pjsg in https://github.com/mcguirepr89/BirdNET-Pi/issues/429#issuecomment-1178388182_

Syslog snippet from when the service was operational to not:
```
Jul 10 20:22:59 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:22:59 blueberrypi birdnet_recording.sh[10062]: ++ netstat -tulpn
Jul 10 20:22:59 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:23:00 blueberrypi systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 80.
Jul 10 20:23:00 blueberrypi systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
Jul 10 20:23:00 blueberrypi systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
Jul 10 20:23:00 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:23:00 blueberrypi birdnet_recording.sh[10069]: ++ netstat -tulpn
Jul 10 20:23:00 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:23:00 blueberrypi systemd[1]: spectrogram_viewer.service: Succeeded.
Jul 10 20:23:01 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:23:01 blueberrypi birdnet_recording.sh[10087]: ++ netstat -tulpn
Jul 10 20:23:01 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:23:02 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:23:02 blueberrypi birdnet_recording.sh[10096]: ++ netstat -tulpn
Jul 10 20:23:02 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:23:03 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:23:03 blueberrypi birdnet_recording.sh[10101]: ++ netstat -tulpn
Jul 10 20:23:03 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:23:04 blueberrypi python3[9907]: Traceback (most recent call last):
Jul 10 20:23:04 blueberrypi python3[9907]:   File "/usr/local/bin/daily_plot.py", line 67, in <module>
Jul 10 20:23:04 blueberrypi python3[9907]:     norm = plt.Normalize(confmax.values.min(), confmax.values.max())
Jul 10 20:23:04 blueberrypi python3[9907]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/numpy/core/_methods.py", line 43, in _amin
Jul 10 20:23:04 blueberrypi python3[9907]:     return umr_minimum(a, axis, None, out, keepdims, initial, where)
Jul 10 20:23:04 blueberrypi python3[9907]: ValueError: zero-size array to reduction operation minimum which has no identity
Jul 10 20:23:04 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:23:04 blueberrypi birdnet_recording.sh[10119]: ++ netstat -tulpn
Jul 10 20:23:04 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:23:05 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:23:05 blueberrypi birdnet_recording.sh[10123]: ++ netstat -tulpn
Jul 10 20:23:05 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:23:06 blueberrypi systemd[1]: chart_viewer.service: Main process exited, code=exited, status=1/FAILURE
Jul 10 20:23:06 blueberrypi systemd[1]: chart_viewer.service: Failed with result 'exit-code'.
Jul 10 20:23:06 blueberrypi systemd[1]: chart_viewer.service: Consumed 16.347s CPU time.
Jul 10 20:23:06 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:23:06 blueberrypi birdnet_recording.sh[10127]: ++ netstat -tulpn
Jul 10 20:23:06 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:23:07 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:23:07 blueberrypi birdnet_recording.sh[10150]: ++ netstat -tulpn
Jul 10 20:23:07 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:23:08 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:23:08 blueberrypi birdnet_recording.sh[10154]: ++ netstat -tulpn
Jul 10 20:23:08 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:23:09 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:23:09 blueberrypi birdnet_recording.sh[10158]: ++ netstat -tulpn
Jul 10 20:23:09 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:23:10 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:23:10 blueberrypi birdnet_recording.sh[10177]: ++ netstat -tulpn
Jul 10 20:23:10 blueberrypi systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 81.
Jul 10 20:23:10 blueberrypi systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
Jul 10 20:23:10 blueberrypi systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
Jul 10 20:23:10 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:23:11 blueberrypi systemd[1]: spectrogram_viewer.service: Succeeded.
Jul 10 20:23:11 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:23:11 blueberrypi birdnet_recording.sh[10184]: ++ netstat -tulpn
Jul 10 20:23:11 blueberrypi birdnet_recording.sh[9959]: + sleep 1
[... 161 lines cut ...]
Jul 10 20:23:59 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:23:59 blueberrypi birdnet_recording.sh[10625]: ++ netstat -tulpn
Jul 10 20:23:59 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:24:00 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:24:00 blueberrypi birdnet_recording.sh[10643]: ++ netstat -tulpn
Jul 10 20:24:00 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:24:01 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:24:01 blueberrypi birdnet_recording.sh[10647]: ++ netstat -tulpn
Jul 10 20:24:01 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:24:01 blueberrypi CRON[10652]: (pi) CMD (/usr/local/bin/cleanup.sh >/dev/null 2>&1)
Jul  10 08:24:01 Stopped Core Services -- It looks like analysis stopped. Check raw recordings in /home/pi/BirdSongs and check the birdnet_analysis.service and birdnet_server.service "journalctl -eu birdnet_analysis -u birdnet_server"
Jul 10 20:24:02 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:24:02 blueberrypi birdnet_recording.sh[10679]: ++ netstat -tulpn
Jul 10 20:24:02 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:24:03 blueberrypi systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 86.
Jul 10 20:24:03 blueberrypi systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
Jul 10 20:24:03 blueberrypi systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
Jul 10 20:24:03 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:24:03 blueberrypi birdnet_recording.sh[10700]: ++ netstat -tulpn
Jul 10 20:24:03 blueberrypi birdnet_recording.sh[9959]: + sleep 1
Jul 10 20:24:03 blueberrypi systemd[1]: spectrogram_viewer.service: Succeeded.
Jul 10 20:24:04 blueberrypi birdnet_recording.sh[9959]: + grep 5050 /dev/fd/63
Jul 10 20:24:04 blueberrypi birdnet_recording.sh[10705]: ++ netstat -tulpn
Jul 10 20:24:04 blueberrypi birdnet_recording.sh[9959]: + sleep 1
```

Bobblybook, 2022-07-11:
> I can't say for sure why this is happening. The main thing that links both updating and saving the config is that `~/BirdNET-Pi/scripts/restart_services.sh` is ran at the end of both of their operations. Maybe you could try running that command by itself and pasting the output here.

I have played around with this script yesterday. It makes no difference in both scenarios: When the server is running, running the restart script causes it to continue running fine. Once I break it, the restart script doesn't fix anything. Output is below, same for both cases:
```
bash restart_services.sh
+ my_dir=/home/pi/BirdNET-Pi/scripts
+ sudo systemctl stop birdnet_server.service
+ sudo systemctl stop birdnet_recording.service
+ services=(chart_viewer.service spectrogram_viewer.service icecast2.service extraction.service birdnet_recording.service birdnet_log.service)
+ for i in "${services[@]}"
+ sudo systemctl restart chart_viewer.service
+ for i in "${services[@]}"
+ sudo systemctl restart spectrogram_viewer.service
+ for i in "${services[@]}"
+ sudo systemctl restart icecast2.service
+ for i in "${services[@]}"
+ sudo systemctl restart extraction.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_recording.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_log.service
+ sudo systemctl start birdnet_server.service
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
```
> 
> Also this line is interesting to me (and looks quite similar to an issue I'm having [#429 (comment)](https://github.com/mcguirepr89/BirdNET-Pi/issues/429#issuecomment-1179761552)): `Jul 10 08:24:01 Stopped Core Services -- It looks like analysis stopped. Check raw recordings in /home/pi/BirdSongs and check the birdnet_analysis.service and birdnet_server.service "journalctl -eu birdnet_analysis -u birdnet_server"`
> 
> Could you paste the output of `ls ~/BirdSongs/Processed/ | wc -l` ?

The output of this was a single '0'. I've also tried deleting all raw and processed recordings incase there was something corrupted, but the same thing still happens after a reboot.

pjsg, 2022-07-11:
After clearing out the backlog, things seem to be working for me....

ehpersonal38, 2022-07-11:
So it's interesting that the top bar of your webpage still shows... could you try adding this to the very beginning of `~/BirdNET-Pi/homepage/views.php`
```
ini_set('display_errors', 1);
ini_set('display_startup_errors', 1);
error_reporting(E_ALL);
```

and then running
`curl localhost/views.php` ?

ehpersonal38, 2022-07-11:
So going to your http://{BIRDNETPI_URL} still shows the same as https://imgur.com/nVV9aUm ?

How about if you go to http://{BIRDNETPI_URL}/views.php?view=Overview in your browser? And then try curling that link too?

mcguirepr89, 2022-07-11:
Hello -- sorry for the delay while I've been out of town.

You can update now to remove the cleanup.sh scripts backlog check and it will not stop reacting to a backlog of recordings. I do encourage you to find out why your system is accruing a backlog of recordings.

"Tools" > "System Controls" > "Update" ought to suffice

Bobblybook, 2022-07-11:
> Hello -- sorry for the delay while I've been out of town.
> 
> You can update now to remove the cleanup.sh scripts backlog check and it will stop reacting to a backlog of recordings. I do encourage you to find out why your system is accruing a backlog of recordings.
> 
> "Tools" > "System Controls" > "Update" ought to suffice

No prob.

This doesn't work for me, as mentioned I can't actually run the update without the whole site messing up. I did a complete uninstall and cleanout and reinstall, and I'm getting the same issue still..

mcguirepr89, 2022-07-11:
Have you attempted any data collection yet? Are we still troubleshooting services being weird with a completely empty (or nonexistent) database?

Bobblybook, 2022-07-11:
> Have you attempted any data collection yet? Are we still troubleshooting services being weird with a completely empty (or nonexistent) database?

No data yet, I don't have a microphone plugged in to the pi. It is taking recordings, I assume they are just empty/static from the mic port. Any updated config settings are just breaking my setup until a hard restart.

mcguirepr89, 2022-07-11:
How about using a microphone to collect data so that the services _can_ restart -- let me know if that works

mcguirepr89, 2022-07-11:
Also, try the webGUI from an "Incognito" browser and let us know if you see more of the GUI

Bobblybook, 2022-07-11:
> Also, try the webGUI from an "Incognito" browser and let us know if you see more of the GUI

Actually from incognito mode I see nothing at all, the page never loads (blank white screen). I think the green page must be just cache. It just hangs indefinitely now.

I will try a microphone tomorrow, it's 1:30 am now and the pi is outside in the cold. Will report back on how it goes!

Bobblybook, 2022-07-12:
I've done a bit of testing today. Gave it a USB microphone, it's now picking up detections and the database is filled and the issue still occurs.

If I don't touch anything it seems to work fine, just when I try and update settings or update the software it breaks.
I've also been playing around with the recording length - I noticed the pi is running hot (61C) and is thermal throttled. Long-term I will improve cooling, but it seems to be keeping up with processing anyway (30-40% cpu load) so I don't think that's the issue. I did change recording length to 30s to test, but it made no difference.

I noticed a few things:
If I change lat/long to another value (changing one digit) it breaks
If I change the password to ANOTHER password (by adding a character) it succeeds and doesn't break the site
However if I attempt to "change" the password to the same current password, it hangs and breaks.

This makes me think it's nothing to do with audio processing or the birdnet system itself, but some interaction and checks when submitting the web form. I don't see how this ties in with the altered lat/long coordinate changes or the software update feature.. but there seems to be a pattern.
After all of these scenarios, if I sudo reboot the system via ssh, it's up and running again.

mcguirepr89, 2022-07-12:
That is all very strange. Is there any chance you would be willing to host a remote desktop session so that I might be able to take a look at the system myself? If so, please [email me](mailto:mcguirepr89@gmail.com) and we will arrange for some remote work, if not, I totally understand and will request the following from you:

1. Reboot
1. Make a breaking change
1. ```
   journalctl --no-hostname --no-pager --boot -ucaddy -uphp7.4-fpm
   ```

After looking through those logs, I'll request that you add some debugging to the PHP scripts and then will request screenshots thereafter. I'll provide those instructions at that time.

My best,
Patrick

Bobblybook, 2022-07-12:
> That is all very strange. Is there any chance you would be willing to host a remote desktop session so that I might be able to take a look at the system myself? If so, please [email me](mailto:mcguirepr89@gmail.com) and we will arrange for some remote work, if not, I totally understand and will request the following from you:
> 
> 1. Reboot
> 2. Make a breaking change
> 3. ```
>    journalctl --no-hostname --no-pager --boot -ucaddy -uphp7.4-fpm
>    ```
> 
> After looking through those logs, I'll request that you add some debugging to the PHP scripts and then will request screenshots thereafter. I'll provide those instructions at that time.
> 
> My best, Patrick

Hi Patrick,
See email response. Happy to figure something out, whatever is easier and best for you.

mcguirepr89, 2022-07-14:
I'm looking into this with @Bobblybook 

Does anyone running on a Raspberry Pi 3B+ see the system hang when updating settings in "Tools" > "Settings"? If so, can you please report what you typically observe?

Thank you!!!!!!

simogogo, 2022-07-14:
@mcguirepr89 I think my device was in the same situation after a platform update (I have a 3B+). 

I found 3 wrong things that were blocking my platform after the update:
1 - Pulseaudio was misconfigured, there were errors on my dmesg output, but I think this could be a peculiarity of my environment
2 - the filename contained in my "BirdNET-Pi/analyzing_now.txt" file referred to a non-existent / corrupt audio file
3 - Some unprocessed audio files (check your "BirdSongs/MONTHNAME/SINGLEDAY/" folders...) where corrupted or shorter than the recording lenght parameter configured in your Settings

I solved the problem in this way:
1 - stop all birdnet processes
2 - kill all python processes (server.py sometimes is hard to die! 
3 - edit BirdNET-Pi/analyzing_now.txt file and remove the filename contained
4 - remove all "BirdSongs/MONTHNAME/" folders. <<---- !!!!!!! CAUTION: this will erase your current queue!
5 - reboot

I think that on my environment the main problem was because of the many corrupted / shorter audio files that were generated  because of the many reboots. I say this because during the test phase, after a reboot, the environment worked correctly for some minutes and then it stopped again... I don't know if this could be the same issue of @Bobblybook but could be an idea..

simogogo, 2022-07-15:
@Bobblybook for me what you are describing is because of a stopped/notworking analyzing service. 
Recording services is working, but server.py does not analyze correctly your recordings (probably because of an error with the filename containet in "analyzin_now.txt" file). In this scenario you keep recordings correctly, but the recordings are not analyzed/cleaned and then this generate the full-storage issue.

Bobblybook, 2022-07-15:
> @Bobblybook for me what you are describing is because of a stopped/notworking analyzing service. Recording services is working, but server.py does not analyze correctly your recordings (probably [...]

Yeah I guessed that was probably it, it was stuck on that ~250ms recording. I didn't need to manually clear it out though, just a reboot seemed to fix it and now it's slowly catching back up.
However I'm watching it go, and after it writes the .wav.csv of the results, the .wav is still left behind. Is this normal? I figured it would free up the space once the .csv is written and any appropriate .mp3's are extracted.

simogogo, 2022-07-15:
I don't remember which process take care of moving analyzed audios to "daily folders", but it could be normal. Probably @mcguirepr89 could provide a more clear explanation for this

mcguirepr89, 2022-07-15:
Hi, Simione!!! I'm sorry to hear about the issues with updating -- I'm glad you were able to sort things out. I want to see the `currently_analyzing.txt` breakage you mention, as it seems plausible that could be an issue re: locale differences, though I haven't seen it for myself first hand.

>@Bobblybook However I just came home and noticed that the pi seems to be way behind on processing . . .

Hi, @Bobblybook -- It looks like your system throttled again from a thermal issue -- it's CPU temp is around 68℃, which I suspect is slowing things down a good bit (it's frequency capped right now at 1.2GHz)

>However I'm watching it go, and after it writes the .wav.csv of the results, the .wav is still left behind. Is this normal? I figured it would free up the space once the .csv is written and any appropriate .mp3's are extracted.

>@simogogo I don't remember which process take care of moving analyzed audios to "daily folders", but it could be normal. Probably @mcguirepr89 could provide a more clear explanation for this

The original raw recordings from which extracted sound clips are made get stored in the "`Processed`" directory after the bird sound clip is extracted, though only 50 of the most recent raw recordings are kept, and of those, only the raw recordings from which a detection lead to an extracted sound clip. When new recordings need to take one of the 50 available slots, the oldest recordings and their CSV are removed. Keeping this raw data temporarily has been requested as an ornithological review process (the greater audio context is helpful for researchers) as well as to help facilitate running the samples against other models for comparison.

The audio lifecycle (on a default installation) is:
```
RECS_DIR=$HOME/BirdSongs
EXTRACTED=$HOME/BirdSongs/Extracted
PROCESSED=$HOME/BirdSongs/Processed

         (Raw audio from mic)
                  |
                  |
                  V
       $RECS_DIR/[nested-by-date]
                  |
                  |
                  V
       `birdnet_analysis` service
       runs the audio through the
          BirdNET-Lite model
        |                   |
        |                   |
        V                   V
extracted bird        the raw audio files
sounds go to       from which the exctrations
$EXTRACTED          were made to go $PROCESSED
```

I hope that is helpful!!!

Bobblybook, 2022-07-15:
Thanks, yes that helps a lot. Kepping the last 50 recordings is no issue, however my 'Friday 15th' folder is currently 18GB in size. I just checked quickly and it currently has 5751 15-sec .wav files, most of them already processed. I'm wondering if the deletion process is somehow messed up on it..

Yes it's currently throttled and running very hot, it's been catching up on processing and I'm not able to cool it any better right now. However I was interested if it's going to work fine under throttled conditions - currently it is making progress and processing each 15-sec segment in around 6 seconds, so it should catch up. I'll keep an eye on it and see if it can cope while throttled.

mcguirepr89, 2022-07-15:
Your edit is correct -- the backlog you mention are the 5,751 files. I went through your logs and they're unfortunately totally worthless in troubleshooting since all but the `spectrogram_viewer` and `chart_viewer` services all stopped without any errors. It's super unusual that the python scripts failed without even their tracebacks, but I didn't see any indication in the logs about _why_ everything suddenly stopped on the July 14th around 17:00:00ish and didn't restart until a manual reboot, then the analysis hung but the recording persisted (creating the backlog) -- then you said the most recent reboot got the backlog processing. Strange and just hard to find the pattern here

ehpersonal38, 2022-07-20:
Ran into this exact issue today. I was editing server.py and made a typo, so when I went to "Restart Core Services", `restart_services.sh` was just looping forever trying to start up a broken server.py. You can see that's exactly what's happening in your [earlier comment](https://github.com/mcguirepr89/BirdNET-Pi/issues/431#issuecomment-1180369703):
```
+ sudo systemctl start birdnet_server.service
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
```
^ this would continue looping forever

What differs is that your server.py breaks only under certain circumstances, still don't know what that's about. Maybe after it breaks you can give the output of `sudo systemctl status birdnet_server.service`

Bobblybook, 2022-07-21:
@mcguirepr89 Ok so I've had the system running for a few days now and have some things to report back.
I did do a system update as there were a number of updates shown, which overrode the changes you made (an oversight by me..) so the config page is back to breaking. It's not a big deal and I commented out the lines again in the service restart script, and I haven't been needing to change any settings after the initial config anyway, now that you've identified the problem.

The Pi has been running thermal throttled at around 60C for a few days. It's fine and is able to keep up - each 15sec audio segment tends to finish processing after 6-8 secs total so there is still a healthy margin of time, even running at 1.2GHz.

However I checked it last night and there were issues. It was not keeping up, and was processing recordings from almost 24 hours prior. I checked the console and it looked like it had frozen, but it was successfully processing segments, just VERY slowly (probably 30 secs processing time instead of 5). Not sure what was going on. The drive was 90% full again of .wav files, though as mentioned it wasn't stuck, just going very slow.

A couple of strange findings:
- Soft thermal throttle had been triggered, which was normal in this setup as mentioned. However the CPU frequency was 1.4GHz. I think you mentioned that once throttling is triggered, a restart is needed to clear the bit, so I'm not sure why it was back up at 1.4. The php page showed that the soft throttle flag was set still.
- Hard throttling had not been triggered at any point, which is good.
- CPU temps were down to 54C or so, which is below the ~60C usually seen when it's running normally. This makes it seem like the CPU was not being loaded properly... it seemed to have the potential to process much more but was not being given the load for some reason. AKA not CPU bottlenecked, but by something else...

After a reboot, it went straight back to processing each clip in ~6 secs, jumped to 61C (which is what I've been seeing when all is well and processing at the right pace), thermal throttled to 1.2GHz,  and has been fine since for about 24 hours so far.

I have a proper case on the way and I likely won't look into this too much until I get the case on there and reduce/remove the thermal throttling to see if it's the culprit. However from this experience, the throttling doesn't seem to be an issue and there is something causing an issue after some time or event.

It seems to happen rarely, and unfortunately I'm not always around or aware when it occurs. I could always set a cron job to restart every night to try and avoid it, though that seems like a poor attempt at a solution.

Is there any way for me to be notified when there is a backlog of files to be processed? As mentioned, when running normally it processes within ~6 secs and never has more than one file to process. I'm wondering if I can be alerted when there are eg. >5 or >10 files to be processed sitting in the directory at a time. Then I could provide you with a log right after the issue/trigger has occurred in order to try and track it down.

Either way I will do more monitoring once I fix up the heat issue.

mcguirepr89, 2022-07-21:
Thank you for keeping an eye on things and for your work troubleshooting. I think your idea to be notified when the backlog reaches a certain threshold is wise and should be able to accomplish that with a tweak to you `cleanup.sh` file. Just be sure to configure an Apprise notification string in "Tools" > "Settings" for things to work.

You can copy and paste the script below so that your `cleanup.sh` looks like this (I just tested this for myself and it worked as expected so :crossed_fingers:):

```

#!/usr/bin/env bash
source /etc/birdnet/birdnet.conf
set -x

cd "${PROCESSED}" || exit 1
empties=($(find ${PROCESSED} -size 57c))
for i in "${empties[@]}";do
  rm -f "${i}"
  rm -f "${i/.csv/}"
done

if [[ "$(find ${PROCESSED} | wc -l)" -ge 100 ]];then
  ls -1t . | tail -n +100 | xargs -r rm -vv
fi

accumulated_files=$(find $RECS_DIR -path $PROCESSED -prune -o -path $EXTRACTED -prune -o -type f -print | wc -l)
if [ $accumulated_files -ge 10 ];then
  echo "$accumulated_files\!\!\!" | $HOME/BirdNET-Pi/birdnet/bin/apprise -c $HOME/BirdNET-Pi/apprise.txt
fi
```

After which you can take a look at your system and run these commands for some insight:
```
free -h
```
```
source /etc/birdnet/birdnet.conf
accumulated_files=$(find $RECS_DIR -path $PROCESSED -prune -o -path $EXTRACTED -prune -o -type f -print | wc -l)
echo $accumulated_files
```
```
journalctl -n300 --no-hostname --no-pager | sed "s|$HOME||;s|$USER||;/LATITUDE/d;/LONGITUDE/d;/PWD/d;/KEY/d;/BIRDWEATHER/d;/STREAM/d;/URL/d;"```

Bobblybook, 2022-07-22:
Great, I've set that up now and tested it and it's working. I'll leave it running and report back when I run into the problem again.

closed 2022-08-10 by mcguirepr89